openv/openv #14: Adressliste für VScotHO1 ?

WikiSpaxe, 2017-10-25:
* Subject: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: wikiwitt <None>
* Date: Tue,  7 Dec 2010 08:30:05 UT
* Message-ID: <31296741-31296741$openv.wikispaces>

Hallo,

gibt es mittlerweile auch eine Adressliste für VScotHO1?
Anlage: Vitodens 333-F WS3C

MfG
Bernd

WikiSpaxe, 2017-10-25:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Wed,  8 Dec 2010 12:11:06 UT
* Message-ID: <31296741-31369747$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Eine Liste aller VScotHO1 Adressen findest du auf meiner Vitoopen homepage
http://vitoopen.tripod.com/page002.html
(einige user haben gemeldeet, dass diese page nicht mit FireFox funktioniert)

WikiSpaxe, 2017-10-25:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: wikiwitt <None>
* Date: Thu,  9 Dec 2010 10:16:34 UT
* Message-ID: <31296741-31443389$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Hallo,

vielen Dank, habe die Liste auf Deiner HP gefunden, allerdings nur mit dem Browser IE vom MS.
Wie schon richtig bemerkt: Mit Firefox und mit dem von mir benutzten Google Chrome Browser wird nur die Seite aber ohne Inhalt angezeigt.

Gruß Bernd

WikiSpaxe, 2017-10-25:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: chris_at <None>
* Date: Fri, 17 Dec 2010 16:56:15 UT
* Message-ID: <31296741-31891647$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Ebenfalls Danke - bin jetzt auch stolzer Beistzer der VScotHO1-Adressen (und korrekt, mit Firefox gehts nicht). Trotzdem, vitograph zeigt bei allen Temperaturen nur 0 an - irgendwelche Anregungen?

Danke & Gruß
Chris

WikiSpaxe, 2017-10-25:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: shartdac <None>
* Date: Sat, 18 Dec 2010 16:54:15 UT
* Message-ID: <31296741-31925615$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

hallo,
VOIdent zeigt mir:

Device       VScotHO1
ID           20CB
ExtID        0009
Protocol     300

und meine Solardaten werden von v-control auch nicht gelesen. 
Ich habe mir die Liste mit den Adressen auch mal runtergeladen (Danke übrigens, super arbeit!). Aber mal ganz dumm: wie kriege ich die Adressen in z.B. v-control oder vitograph rein, damit ich die Solardaten lesen kann? Scheinbar klappts ja auch bei anderen nicht mit vitograph...?
Muesste man da v-control neu kompilieren?

Danke,
Shartdac

WikiSpaxe, 2017-10-25:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: chris_at <None>
* Date: Sat, 18 Dec 2010 21:48:37 UT
* Message-ID: <31296741-31930543$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Hi there,
 
@shartdac:

von v-control gibts hier keine sources, nur von v-comm.dll; das ist noch unter VB6 gestrickt, ich bin grad dabei das so herzurichten dass ich meine Anlage vernünftig auslesen kann (vorläufig eine VB6 form ähnlich v-control). Wenn ich was zusammengeschustert hab stell ich ein update zur Verfügung mit Kochrezept.

LG
Chris

WikiSpaxe, 2017-10-26:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: texomex <userid-41982818@openv.wikispaces>
* Date: Wed, 28 Aug 2013 22:47:49 UT
* Message-ID: <31296741-64548754$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Also ich habe jetzt mal von vito.rampro.de ein vito.xml für die B3HA (ID 20CB). Allerdings stimmen die Adressen dort nicht alle mit dem Google-Doc überein. BetriebsartA1M1 ist laut Google Doc z.b. auf 2301, im xml aber auf 2323. Was kann man denn da als aktueller betrachten?

WikiSpaxe, 2017-10-26:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: heher001 <userid-53544314@openv.wikispaces>
* Date: Fri, 26 Jun 2015 16:44:37 UT
* Message-ID: <31296741-82795821$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

Guten Tag, kann mir jemand mal seine Adressliste für die 20CB per Email senden?
Den o.g. Link kann ich zwar öffnen, bleibt aber leer. Sowohl mit Crome als auch Inet Explorer.
Vielen herzlichen Dank

LG
Henrik

WikiSpaxe, 2017-10-26:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: promyBE <userid-52718232@openv.wikispaces>
* Date: Sat, 27 Jun 2015 21:48:24 UT
* Message-ID: <31296741-82807481$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

I am looking for the adresses for the vitocal 300G heatpump. I found some of them, but things like het Natural Cooling box info i am missing.
The google spreadsheet seems to be dead?

WikiSpaxe, 2017-10-26:
* Subject: Re: Adressliste für VScotHO1 ?
* Wiki-Page: Adressen
* From: leifnel <userid-57973459@openv.wikispaces>
* Date: Wed, 22 Mar 2017 22:17:06 UT
* Message-ID: <31296741-93474225$openv.wikispaces>
* In-Reply-To: <31296741-31296741$openv.wikispaces>

I'm looking for adresses for the Vitocal200S WO1C-Rev.C
The spreadsheet seems very thinly populated.
I have a trial version of vitosoft 300, it shows a lot of interesting values.

Hardy74, 2017-11-07:
Hallo,
ich kann die Liste leider auch mit keinem Browser öffnen. Kann die vielleicht jemand hier veröffentlichen?
Danke!

dima-357, 2020-06-01:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses