openv/openv #20: Fernsteuerung eines Vitodens 300 (WB3C)

WikiSpaxe, 2017-10-25:
* Subject: Fernsteuerung eines Vitodens 300 (WB3C)
* Wiki-Page: home
* From: lambdapro <None>
* Date: Sat,  1 Jan 2011 16:15:26 UT
* Message-ID: <32151854-32151854$openv.wikispaces>

Hallo zusammen,

erstmal vielen Dank für die Protokollbeschreibung und die Adresslisten.

Das Auslesen von Prozesswerten funktioniert bei mir schon sehr gut.
Allerdings möchte ich den Kessel selber regeln, d.h. die Leistung (nicht die Temperatur) von aussen vorgeben. Dieses wird sicherlich auch im Kaskadenbetrieb genutzt. Hat jemand hierfür schon die entsprechenden Adressen herausgefunden?

Für jeden Hinweis wäre ich dankbar.

Gruß
lambdapro

WikiSpaxe, 2017-10-25:
* Subject: Re: Fernsteuerung eines Vitodens 300 (WB3C)
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat,  1 Jan 2011 17:29:20 UT
* Message-ID: <32151854-32152274$openv.wikispaces>
* In-Reply-To: <32151854-32151854$openv.wikispaces>

Hallo labdapro
Wenn du irgendwo an deiner Vitotronic die Leistung einstellen kannst, dann wird es auch Adressen dafür geben. Du wirst über Optolink nur das einstellen können, was die Steuerung grundsätzlich zur Verfügung stellt. Es kann noch sein, dass gewisse Einstellungen abhängig von anderen Einstellungen sichtbar sind.

Ich selber betreibe eine Kaskaden-Regelung (1 x V333MW1, 2 x V100GC1) die beiden Brenner sind aber nur 2 stufig. Die Kommunikation zwischen der V333 und den V100 geschieht über LON. Leider kenne ich keine günstige HW um auf dem LON zu lauschen, und auch wenn diese Kommunikation entschlüsselt werden könnte, ist noch nicht sicher, ob die entsprechenden Funktionen und Datenpunkte auch über OptoLink adressierbar sind.

Zum LON gibt es von Viessmann auch ein KNX/EIB Gateway. vielleicht kennt jemand hier die Funktionen, die über dieses Gateway zur Verfügung stehen und kann dir damit weiter helfen.

WikiSpaxe, 2017-10-25:
* Subject: Re: Fernsteuerung eines Vitodens 300 (WB3C)
* Wiki-Page: home
* From: ceteris_paribus <None>
* Date: Wed, 24 Aug 2011 11:26:28 UT
* Message-ID: <32151854-41623067$openv.wikispaces>
* In-Reply-To: <32151854-32151854$openv.wikispaces>

Hallo zusammen
unter http://vitoopen.tripod.com/files/VScotHO1.zip

Findet man Adressen. Die Hohen (Zeile 1030 oder so) sind m.E. die LON Kommunikation.

Die Beschreibungen die mit "niv" Anfangen.

Evtl. auch hier nochmal reinsehen, synoptische darstellung verschiedener Anlagen/Regelungen:

http://vitoopen.tripod.com/files/Optolink_Datenpunkte_Logger_v2.zip

WikiSpaxe, 2017-10-25:
* Subject: Re: Fernsteuerung eines Vitodens 300 (WB3C)
* Wiki-Page: home
* From: ramack <None>
* Date: Wed, 24 Aug 2011 21:53:26 UT
* Message-ID: <32151854-41638547$openv.wikispaces>
* In-Reply-To: <32151854-32151854$openv.wikispaces>

Hallo,

in der ersten Listen die ceteris_paribus verlinkt hat (Danke dafür) sehe ich ein paar Fragezeichen. Was ist z. B. an der Adresse 0x0810 hinterlegt? VL Temp, Kesseltemp oder Puffer oben? - Es hat ja drei Zeilen die auf der Adresse liegen... Was bedeutet das?

Grüße,
Rapha

WikiSpaxe, 2017-10-25:
* Subject: Re: Fernsteuerung eines Vitodens 300 (WB3C)
* Wiki-Page: home
* From: ceteris_paribus <None>
* Date: Thu, 25 Aug 2011 22:49:00 UT
* Message-ID: <32151854-41673401$openv.wikispaces>
* In-Reply-To: <32151854-32151854$openv.wikispaces>

In der Version V4 sind die berichtigt und heißen 810, 812, 814 - ich kann nur mutmaßen, daß dies aus der Anfangszeit stammt, und kopierte Zeilen waren. Man muß einfach mal Sinnig die Adress-Lücken ansehen

V333MW1	0x00F8	20B8	2	
Addr	"Values Factor"	Bytes

Kesseltemperatur Tiefpass KTS	
0x0810	/ 10	2
Speichertemperatur Tiefpass STS1		
0x0812	/ 10	2
Speichertemperatur 2 Tiefpass	STS2		
0x0814	/ 10	2

Mir hat es geholfen, die hier verlinkten XML-Dateien im IE anzusehen. Nicht alle Adressen sind für alle Anlagen gleich, aber es fördert mal die Herangehensweise an die Systematik.

SteffMUC, 2024-12-29:
Hallo zusammen,
Auch wenn der Beitrag etwas "älter" ist: Welche device ID verwendet Ihr für die WB3C ? Bin mir nicht sicher, was ich da einstellen soll..
Vielen lieben Dank
Stefan

SteffMUC, 2024-12-29:
Google hilft manchmal weiter, habe hier eine Indikation gefunden: [Forum](https://community.viessmann.de/t5/Gas/Vitodens-300-W-vcontrold-xml-Konfiguration/td-p/468702).
Wenn ich das richtig interpretiere, dann muss ich die WB3C als VScotHO1_70 konfigurieren.