openv/openv #25: v-control mit Vitotronic 200 KW 6 kompatibel??

WikiSpaxe, 2017-10-25:
* Subject: v-control mit Vitotronic 200 KW 6 kompatibel??
* Wiki-Page: home
* From: King-Nova <None>
* Date: Sun, 16 Jan 2011 18:43:00 UT
* Message-ID: <32750452-32750452$openv.wikispaces>

Guten Tag, ich überlege gerade ob ich mir auch ein Optolinkkabel anschaffe. 
Nun lese ich hier aber immer nur von der KW1 und KW2. Ist die Software mit der KW6 auch kompatibel? Diese hab ich nämlich an meinem Vitoladens 300 C.

WikiSpaxe, 2017-10-25:
* Subject: Re: v-control mit Vitotronic 200 KW 6 kompatibel??
* Wiki-Page: home
* From: King-Nova <None>
* Date: Sun, 30 Jan 2011 22:33:07 UT
* Message-ID: <32750452-33395938$openv.wikispaces>
* In-Reply-To: <32750452-32750452$openv.wikispaces>

Hat keiner ne KW 6 und das schonmal getestet?

Al-, 2019-06-30:
Zwar schon alt, aber falls jemand die gleiche Frage hat: KW6B funktioniert auch.

Ich nehme als device id '2098' (als ob es ein V200KW2 wäre) und es funktioniert. Vielleicht hätte KW6B zusätzliche Funktionen, welche von der Community (noch) nicht identifiziert sind; aber alle von mir benötigten Funktionen sind verfügbar, sowohl lesen als schreiben.

valirion, 2020-11-27:
@AI: Wie ist der Adapter in Ihrer Steuerung eingebaut? Die Lichtleiter für die LEDs sind ja eigentlich im Weg.

Al-, 2020-11-28:
@valirion Bei mir hatte es ein durchsichtiges Plastikteil als Lichtleiter, welches die beiden LEDs bei geschlossenem Deckel von aussen sichtbar macht. Dieses entfernen, möglichst ohne einen Teil des orangen Deckels abzubrechen. Dann den USB Stecker des Optolinkkabels durch die Öffnung neben den LEDs in die Heizung schieben (von innen ‘fischen’ und sauber verlegen bis zum Steuerungsgerät; bei mir ein Raspberry Pi), das V einstecken und fertig ist die Verkabelung.

sbidy, 2024-10-25:
@Al- zwar lange her, aber mal eine Frage: hast du bei dem KW6B die Brennerstufen herausfinden können (vorausgesetzt du hast einen Stufenbrenner)? Beibringt vor allem der `0x0849` (Brenner zweite Stufe) keine Anständigen werte.

Al-, 2024-10-26:
@sbidy Leider blieb ich vollkommen erfolglos bei meinen Versuchen, weitere Parameter zu finden. Zum Beispiel Innentemperatur(en): den ganzen Addressbereich ausgelesen bei heissem und dann wieder bei kälterem Wetter; alle unveränderten Bytes verworfen, mit den übrigbleibenden weitergesucht. Allzu lange habe ich allerdings dann nicht weitergesucht, weil meine Grundbedürfnisse (Solltemperatur einstellen und Aussentemperatur auslesen) abgedeckt sind.