openv/openv #26: Verbindung zu Vitotres 343

WikiSpaxe, 2017-10-25:
* Subject: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: Alibaba20 <None>
* Date: Fri, 21 Jan 2011 22:06:25 UT
* Message-ID: <33000304-33000304$openv.wikispaces>

Hallo,

hat von euch schon mal jemand erfolgreich "Optolink-Kontakt" mit einer Vitotres 343 aufgenommen? Ich kriege es nicht hin und mir fällt nichts mehr ein, was ich noch ausprobieren könnte. Es geht um Folgendes:

Ein Nachbar mit der gleichen Anlage hat sich einen Optolink-USB-Adapter gekauft und ihn aber nicht zum Funktionieren bekommen (unter Windows). D.h. seine Anlage wurde nicht identifiziert, weil noch nicht mal die ID ermittelt werden konnte. Da ich mich mit Elektronik ganz gut auskenne, habe ich angeboten, es bei uns mal auszuprobieren. Ergebnis war leider genau das gleiche, egal ob Windows (voIdent) oder Linux (vcontrold). Die Anlage liefert immer nur 0x05, ca. ein mal pro Sekunde. Man kann senden was man will, es ändert sich nichts. Daraufhin habe ich mir den RS232-Adapter nach der Anleitung hier im Forum gebaut. Keine Verbesserung, genau gleiches Verhalten. Fremdlicht und optischer Kurzschluss sind (meiner Ansicht nach) nicht möglich. Zu guter letzt habe ich mit dem Oszi die Spannung bzw. das Signal an der Empfängerdiode mal gemessen, sieht eigentlich super aus. Ich weiss nicht mehr weiter...

Als Beispiel hier mal ein Protokoll von vcontrold:

[3613] Fri Jan 21 21:49:08 2011 : Befehl: idgwg
[3613] Fri Jan 21 21:49:08 2011 : konfiguriere serielle Schnittstelle /dev/ttyS0
[3613] Fri Jan 21 21:49:08 2011 : >SEND: 04
[3613] Fri Jan 21 21:49:08 2011 : Warte auf 05
[3613] Fri Jan 21 21:49:08 2011 : <RECV: 05 (640.0 ms)
[3613] Fri Jan 21 21:49:08 2011 : >SEND: 01
[3613] Fri Jan 21 21:49:08 2011 : >SEND: C7
[3613] Fri Jan 21 21:49:08 2011 : >SEND: F8
[3613] Fri Jan 21 21:49:08 2011 : >SEND: 04
[3613] Fri Jan 21 21:49:08 2011 : >SEND: 04
[3613] Fri Jan 21 21:49:09 2011 : <RECV: 05 (1030.0 ms)
[3613] Fri Jan 21 21:49:10 2011 : <RECV: 05 (1020.0 ms)
[3613] Fri Jan 21 21:49:11 2011 : <RECV: 05 (1030.0 ms)
[3613] Fri Jan 21 21:49:12 2011 : <RECV: 05 (1020.0 ms)
[3613] Fri Jan 21 21:49:12 2011 : Recv Timeout: 4100 ms  > 150 ms (Retry: 2)

Die ID der Anlage ist übrigens 0x2043, das habe ich dem Menue entnommen.

Kann mir jemand weiterhelfen oder hat eine Idee, was ich falsch mache? Kann es sein, dass diese Anlage keines der bekannten Protokolle unterstützt?

Herzlichen Dank
Alex

WikiSpaxe, 2017-10-25:
* Subject: Re: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Fri, 21 Jan 2011 23:03:04 UT
* Message-ID: <33000304-33002452$openv.wikispaces>
* In-Reply-To: <33000304-33000304$openv.wikispaces>

Hallo Alibaba, willkommen im Forum!

ich befürchte mal, dass deine Steuerung über Optolink nicht kommunizieren kann. Meine Version der Vitosoft (1.20f) [Inbetriebnahme und Diagnose SW von Viessman] hat keine Einträge für eine Steuerung mit der ID 0x2043.

In der Montage und Service Anleitung zur Vitotres 343 habe ich auch keine Hinweise auf Laptop-Anbindung gefunden - einzig im Übersichtsplan ist mit (C) die Optolink-Schnittstelle bezeichnet.

Allenfalls könnte eine Anfrage bei Viessmann Klarheit bringen "Kann die Vitotres 343 mittels Vitosoft 200 LNR parametriert werden?".

Weitere Hoffnungen könnte es geben, wenn die Vitotres mit irgend einer Vitocom Anbindung fernbedient werden kann...

Dass die Vitotres ein anderes Protokoll verwendet, halte ich für unwahrscheinlich. Die oben verwendete ID-Abfrage wird von allen bekannten Protokollen unterstützt (GWG, KW, 300) GWG ist speziell, weil nur 1 Byte Adressen möglich sind, aber Steuerungen die das 300er unterstützen können auch mit dem KW Protokoll kommunizieren.

Das einzige, was du noch probieren könntest, ist auf das initiale Send 04 zu verzichten und nur auf das Empfangen von 05 zu warten.

Gruss
Hanspeter

WikiSpaxe, 2017-10-25:
* Subject: Re: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: Alibaba20 <None>
* Date: Sat, 22 Jan 2011 18:25:06 UT
* Message-ID: <33000304-33021308$openv.wikispaces>
* In-Reply-To: <33000304-33000304$openv.wikispaces>

Hallo Hanspeter,

vielen Dank für deine freundliche Begrüßung und Antwort!
Ich habe in der Zwischenzeit einen kleinen Fortschritt erzielt, d.h. eine Kommunikation scheint möglich. Ich habe nämlich das KW-Protokoll für die Identifizierung ausprobiert, und es kommt tatsächlich eine Antwort, die zwar noch nicht perfekt ist, aber was nicht ist, kann ja noch werden :-) Sieh selbst:

[21035] Sat Jan 22 18:06:04 2011 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[21035] Sat Jan 22 18:06:04 2011 : Warte auf 05
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 05 (100.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : >SEND: 01
[21035] Sat Jan 22 18:06:04 2011 : >SEND: F7
[21035] Sat Jan 22 18:06:04 2011 : >SEND: 00
[21035] Sat Jan 22 18:06:04 2011 : >SEND: F8
[21035] Sat Jan 22 18:06:04 2011 : >SEND: 04
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 20 (50.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 00 (10.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 43 (0.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 00 (0.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : Empfangen: 20 00 43 00 

[21035] Sat Jan 22 18:06:04 2011 : Befehl: idkwdirect
[21035] Sat Jan 22 18:06:04 2011 : >SEND: F7
[21035] Sat Jan 22 18:06:04 2011 : >SEND: 00
[21035] Sat Jan 22 18:06:04 2011 : >SEND: F8
[21035] Sat Jan 22 18:06:04 2011 : >SEND: 04
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 20 (30.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 00 (0.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 43 (20.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : <RECV: 00 (0.0 ms)
[21035] Sat Jan 22 18:06:04 2011 : Empfangen: 20 00 43 00 

[21035] Sat Jan 22 18:06:04 2011 : Befehl: idgwg
[21035] Sat Jan 22 18:06:04 2011 : Warte auf 05
[21035] Sat Jan 22 18:06:05 2011 : <RECV: 05 (820.0 ms)
[21035] Sat Jan 22 18:06:05 2011 : >SEND: 01
[21035] Sat Jan 22 18:06:05 2011 : >SEND: C7
[21035] Sat Jan 22 18:06:05 2011 : >SEND: F8
[21035] Sat Jan 22 18:06:05 2011 : >SEND: 04
[21035] Sat Jan 22 18:06:05 2011 : >SEND: 04
[21035] Sat Jan 22 18:06:06 2011 : <RECV: 05 (980.0 ms)
[21035] Sat Jan 22 18:06:07 2011 : <RECV: 05 (1030.0 ms)
[21035] Sat Jan 22 18:06:08 2011 : <RECV: 05 (1020.0 ms)
[21035] Sat Jan 22 18:06:09 2011 : <RECV: 05 (1030.0 ms)
[21035] Sat Jan 22 18:06:09 2011 : Recv Timeout: 4060 ms  > 150 ms (Retry: 0)
[21035] Sat Jan 22 18:06:09 2011 : Recv Timeout, Abbruch
[21035] Sat Jan 22 18:06:09 2011 : Fehler beim ausfuehren von idgwg

[21035] Sat Jan 22 18:06:09 2011 : Befehl: id1kw
[21035] Sat Jan 22 18:06:09 2011 : Warte auf 05
[21035] Sat Jan 22 18:06:10 2011 : <RECV: 05 (980.0 ms)
[21035] Sat Jan 22 18:06:10 2011 : >SEND: 01
[21035] Sat Jan 22 18:06:10 2011 : >SEND: F7
[21035] Sat Jan 22 18:06:10 2011 : >SEND: 00
[21035] Sat Jan 22 18:06:10 2011 : >SEND: F8
[21035] Sat Jan 22 18:06:10 2011 : >SEND: 02
[21035] Sat Jan 22 18:06:10 2011 : <RECV: 20 (30.0 ms)
[21035] Sat Jan 22 18:06:10 2011 : <RECV: 00 (0.0 ms)
[21035] Sat Jan 22 18:06:10 2011 : Empfangen: 20 00 

[21035] Sat Jan 22 18:06:10 2011 : Befehl: id2kw
[21035] Sat Jan 22 18:06:10 2011 : Warte auf 05
[21035] Sat Jan 22 18:06:11 2011 : <RECV: 05 (970.0 ms)
[21035] Sat Jan 22 18:06:11 2011 : >SEND: 01
[21035] Sat Jan 22 18:06:11 2011 : >SEND: F7
[21035] Sat Jan 22 18:06:11 2011 : >SEND: 00
[21035] Sat Jan 22 18:06:11 2011 : >SEND: F9
[21035] Sat Jan 22 18:06:11 2011 : >SEND: 02
[21035] Sat Jan 22 18:06:11 2011 : <RECV: 43 (50.0 ms)
[21035] Sat Jan 22 18:06:11 2011 : <RECV: 00 (0.0 ms)
[21035] Sat Jan 22 18:06:11 2011 : Empfangen: 43 00 
[21035] Sat Jan 22 18:06:11 2011 : Befehl: quit

D.h. ID-Abfrage per KW-Protokoll liefert eine Antwort, per GWG nicht. Die Steuerung der Vitotres 343 heißt CD70, vielleicht sagt dir das was?
Jedenfalls taucht ja immerhin 0x2043 in der Antwort auf, aber es scheint nach jedem Byte noch ein 0x00 gesendet (oder zumindest empfangen) zu werden. Ich hab's mit RS232- und mit USB-Adapter getestet, ist bei beiden gleich. Wenn ich jeweils zwei Bytes ab 0x00F8, 0x00F9 abfrage, bekomme ich die korrekte Antwort 0x20, 0x43, aber auch mit 0x00 dazu. Ein-Byte-Abfragen funktionieren nicht.

Wenn du oder jemand anderes weitere Ideen hat, ich freue mich über jede Anregung.

Viele Grüße
Alex

WikiSpaxe, 2017-10-25:
* Subject: Re: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat, 22 Jan 2011 19:54:00 UT
* Message-ID: <33000304-33023312$openv.wikispaces>
* In-Reply-To: <33000304-33000304$openv.wikispaces>

ist ja interessant!

Wenn du mal in voIdent mit der Option "KW Protocol" testen kannst, dann müsste auch 20 00 43 00 empfangen werden (entspricht deinem idkwdirect)

Eine Idee wäre noch, das 300er Protokoll zu testen (fehlt noch in voIdent).
1. auf den Empfang von 05 mit 16 00 00 antworten.
hört dann das periodische 05 auf?

2. ID abfragen mit
41 05 00 01 00 F8 02 00

die Antwort müsste in etwa so sein:
06 41 07 01 01 00 F8 02 20 43 66

3. nach dem Senden von 04 müsste wieder periodisch 05 empfangen werden.

WikiSpaxe, 2017-10-25:
* Subject: Re: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: Alibaba20 <None>
* Date: Thu, 28 Apr 2011 14:05:56 UT
* Message-ID: <33000304-38252572$openv.wikispaces>
* In-Reply-To: <33000304-33000304$openv.wikispaces>

Hallo,

jetzt war lange Zeit Sendepause bei mir :-( - aber kürzlich habe ich doch wieder ein bisschen weiterexperimentiert und einige Dinge herausgefunden:

1. Von der Vitotres 343 scheint nur das KW-Protokoll unterstützt zu werden. Jedenfalls reagiert die Steuerung nicht auf die 300er-Initialisierung und auf GWG genauso wenig.

2. voIdent liefert bei Auswahl des "KW Protokolls" 20 00 43 00 zurück. D.h. die ID scheint 2043 zu sein.

3. Ein Scan des Adressbereichs hat einige sinnvolle Adressen geliefert (diverse Temperaturen, Lüfterdrehzahlen, -ströme, etc.); ich werden demnächst mal die Adressen-Tabelle mit diesen Daten füttern.

4. Was ich nicht gefunden habe, sind die in der Anlage gespeicherten statistischen Daten wie Anzahl der Einschaltungen oder Betriebsstunden der Pumpen/des Verdichters etc.
Im Moment lese ich alle 10 Min. alle Adressen aus, die bisher sinnvolle Werte zurückgegeben haben. Die habe ich daraufhin untersucht, ob es irgendwo monoton ansteigende Werte gibt, aber keine gefunden. Hat da jemand von euch noch eine Idee?

Viele Grüße
Alex

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindung zu Vitotres 343
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Fri, 25 Jul 2014 08:57:58 UT
* Message-ID: <33000304-74153484$openv.wikispaces>
* In-Reply-To: <33000304-33000304$openv.wikispaces>

Hallo Alex,

sorry für das "Aufwärmen" dieses alten Threads - ich hätte großes Interesse an den Adressen / vito.xml / vcontrold.xml für die CD70-Steuerung - kannst Du weiterhelfen und das zur Verfügung stellen?

Danke!
Jens

parasender, 2025-01-21:
Any further Information ? Thanks Stephan

Fhiel, 2026-10-15:
Anyone still interested in reading Temperatures from Vitotres 343?
I see some progress with ESPhome. Currently verifiing adresses and signal names.  

Viele Grüsse
Frank