openv/openv #3: Bauanleitungen

WikiSpaxe, 2017-10-25:
* Subject: Bauanleitungen
* Wiki-Page: home
* From: eisberg <None>
* Date: Tue, 12 Jan 2010 19:53:13 UT
* Message-ID: <18814747-18814747$openv.wikispaces>

Hallo,

gibt es für die Herstellung der Platinen auch Dateien im Gerber oder Eagle Format?
Mich interessiert vor allem die USB Version.
Danke im voraus.
:-)

WikiSpaxe, 2017-10-25:
* Subject: Re: Bauanleitungen
* Wiki-Page: home
* From: stb3 <None>
* Date: Tue, 19 Jan 2010 19:42:35 UT
* Message-ID: <18814747-19054423$openv.wikispaces>
* In-Reply-To: <18814747-18814747$openv.wikispaces>

Du kannst Leerplatinen von mir bekommen. Das ist vermutlich billiger als eine selbst zu ätzen.

Grüße

STB

WikiSpaxe, 2017-10-25:
* Subject: Re: Bauanleitungen
* Wiki-Page: home
* From: BrainyRP <None>
* Date: Mon,  6 Feb 2012 12:00:18 UT
* Message-ID: <18814747-50101310$openv.wikispaces>
* In-Reply-To: <18814747-18814747$openv.wikispaces>

Hallo STB (falls Du hier noch mitliest)

Stellst Du noch Leerplatinen her?

mactoolz, 2018-10-25:
Hi

tut mir leid das ich hierüber Nachfrage. Ich würde gerne die Adapter beziehen.
Kannst du mir sagen welche Email ich für die Bestellung dich ansprechen kann.

Gruß