openv/openv #30: Vitotronic 100 FC1

WikiSpaxe, 2017-10-25:
* Subject: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <None>
* Date: Sun, 21 Aug 2011 14:25:59 UT
* Message-ID: <41543251-41543251$openv.wikispaces>

Hello. I have vitoligno 100 with Vitotronic 100 type FC1. On controllers plate where is a RJ14 connector with "PROG" label on it. Maybe it possible to connect PC here(Rj14 connector),not with OptoLink? Thank you.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: warner10 <userid-38434168@openv.wikispaces>
* Date: Thu,  3 Jan 2013 20:12:32 UT
* Message-ID: <41543251-59448704$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Hi,

Did you manage to do anything with the FC1 controller? I have the same problem as you: I want to store the data from the Vitoligno 100 into a database, but I cannot find any information on FC1 connectivity.

Any kind of help would be greatly appreciated.

Thank you,

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Fri,  4 Jan 2013 10:04:13 UT
* Message-ID: <41543251-59459832$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Hi. Actually, I postponed this (use fc1 controller) for better times. Now I made so: AVR controller (with Ethernet) has One Wire  temp. sensors which monitors temp. + in feature I will be able to manipulate with valves to transfer heat separately to different sources (radiators, boiler, heat accumulator).

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: warner10 <userid-38434168@openv.wikispaces>
* Date: Fri,  4 Jan 2013 10:56:50 UT
* Message-ID: <41543251-59459986$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Hi,

I managed to contact Viessmann because my heater provider was lost to my request. I was told that the Optolink is compatible with the FC1 controller. It just plugs in at the front into the "V", rather than at the back.

I think I will purchase the USB adapter from here, and probably a Raspberry PI to link it all up afterwards.

I will update this once I get it working.

Cheers,

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: warner10 <userid-38434168@openv.wikispaces>
* Date: Thu, 24 Jan 2013 14:50:22 UT
* Message-ID: <41543251-59901126$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Hi,

I received the Optolink adapater this morning, but when I plug it in, I just can't get any data. I phoned back Viessmann and they tell me maybe it isn't supported!

How is it possible to check if the "V" is linked to something?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Thu, 24 Jan 2013 15:06:33 UT
* Message-ID: <41543251-59901486$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Ohhhh, kinda strange...  I assume one way to check is just remove controller cover and see if there some leds infront of "V"

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Thu, 24 Jan 2013 15:07:29 UT
* Message-ID: <41543251-59901510$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

btw I can do it today, and post result tomorrow.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: diefenbecker <userid-37970446@openv.wikispaces>
* Date: Thu, 24 Jan 2013 15:30:26 UT
* Message-ID: <41543251-59902066$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Is it possible to see with in the display of digital fotocamera if there is a signal on the LEDs? Trying this with a regular remote control (e.g. for TV) this is working. You will see the LED blink.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: warner10 <userid-38434168@openv.wikispaces>
* Date: Thu, 24 Jan 2013 16:20:07 UT
* Message-ID: <41543251-59903554$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Hi,

With the remote control, I can get the leds to blink on the USB adapter. But nothing on the Vitotronic 100.
I then opened up the back, and there's nothing behind the V !!! What a disappointment.

Now I have an Optolink that I have no use for, and no way to query my Vitotronic FC1.

Maybe I have to build a 1-wire system also. Can you tell me more about your 1wire solution?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Thu, 24 Jan 2013 16:38:39 UT
* Message-ID: <41543251-59904192$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Oh, this is very sad :( ... But I'm sure that there must be solution with the PROG connector than.

One wire solution is simple. Try google it: arduino with ds18b20 or you can use 1Wire network  without arduino but then 1wire muster needed. For me solution with arduino (or st similar) is better.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Thu, 24 Jan 2013 16:41:34 UT
* Message-ID: <41543251-59904366$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

..is better than 1wire muster, because can me maid energy undependable (just +5V needed)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: warner10 <userid-38434168@openv.wikispaces>
* Date: Thu, 24 Jan 2013 17:16:36 UT
* Message-ID: <41543251-59906184$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

How many sensors did you find useful to monitor the vitoligno 100, and where did you place them?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic 100 FC1
* Wiki-Page: home
* From: edkaLv <userid-22417205@openv.wikispaces>
* Date: Fri, 25 Jan 2013 07:29:04 UT
* Message-ID: <41543251-59929186$openv.wikispaces>
* In-Reply-To: <41543251-41543251$openv.wikispaces>

Useful and most critical - suppose one sensor, now it is stays in  output pipe (but I will place it together with vitotronic sensors in near feature).
Also I have 1 sensor in IN pipe and 1 sensor per room and outdoor sensor - but at the moment it is just for statistics.

dima-357, 2020-11-22:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses