openv/openv #50: Vitodens 200 WB2 und Optolink?

WikiSpaxe, 2017-10-25:
* Subject: Vitodens 200 WB2 und Optolink?
* Wiki-Page: home
* From: ao16 <None>
* Date: Mon,  9 Jan 2012 17:32:41 UT
* Message-ID: <48944535-48944535$openv.wikispaces>

Hallo zusammen,

ist jemand bekannt, ob bei dem Modell Vitodens 200 WB2 (6-24kW, installiert im Januar 2005) 
als Wandgerät für witterungsgeführten Betrieb mit Kesselkreisregelung und Standard-Bedieneinheit
der Anschluß des Optolinkadapters und die Nutzung von V-control oder Viess-Data möglich ist?

In den vergangenen Tagen habe ich hierzu umfangreiche Tests und Untersuchungen an unserer
Heizungsanlage gemacht, leider stellte sich jedoch bisher kein Erfolg ein. 
Vielleicht habe ich jedoch auch einen wichtigen Punkt übersehen, deshalb bin ich für jegliche
Anregungen, Tips und Hinweise dankbar.

Zu den durchgeführten Tests und den erhaltenen Ergebnissen:

Auf der rechten Seite unserer Vitodens Standard-Bedieneinheit ist ebenfalls die "V"-Vertiefung 
mit den beiden Betriebs-LED's, in welche sich der Optolinkadapter einstecken läßt.
In der Explosionszeichnung der Serviceanleitung ist das Optolinkmodul standardmäsig
als "bestückt" gekennzeichnet, was mich vermuten ließ, daß auch mit diesem bereits
älteren Vitodens-Models sich die Daten elektronisch auslesen lassen sollten.

Deshalb habe ich den Adapter (Schaltplan 3 aus diesem Wiki) aufgebaut, 
jedoch erhalte ich keine Datenverbindung mit der Heizung (getestet mit V-control 2.5 und 3.0m,
VoIdent und Viess-Data).

In diesem Wiki sowie in Forenbeiträgen im "Haustechnikdialog" wurde dieses Problem bereits 
mehrfach von Usern angesprochen und mögliche Lösungsvorschläge beschrieben.

Leider hat bei unserer Heizungsanlage keine der beschriebenen Vorschläge zum Erfolg geführt,
weshalb ich nun den Verdacht habe, daß in dem o.g. Vitodens-Modell möglicherweise doch 
kein Interface für die Optolinkverbindung enthalten ist?

Den Optolinkadapter habe ich mit folgenden Rechnerkonfigurationen getestet:
- Laptop (IBM T42p mit RS232 aus der Dockingstation und alternativ per USB-RS232-Adapter) 
- "Desktop-PC" (Neoware Thinclient) mit eingebauter RS232

Auf dem Optolinkadapter ist als IR-Sender die SFH485 und als IR-Empfänger die SFH309 (Reichelt,
ohne Tageslichtfilter!) verbaut.

Der Adapter sendet IR-Signale aus (geprüft mit Digicam) und der Empfänger kann IR-Signale
empfangen (geprüft mit einer TV IR-Fernsteuerung und Scope am RXD-Ausgang des Adapters),
jedoch nur unter der Vorraussetzung, daß er von hellem Umgebungslicht isoliert ist - sonst
steuert der Fototransistor voll durch und RXD bleibt auf Dauer-High.

Tests an der Heizungsanlage führte ich deshalb durch
a) bei abgedunkeltem Raum und 
b) IR-Empfangstransistor mit schwarzem Schrumpfschlauchtubus bis zum "LED Fenster" 
   -optisch von Umlicht entkoppelt- zur Vitodens geführt.

Die Pegel an der RS232 vom Laptop mit eingestecktem Optolink-Adapter sind +/-5V.

Leider empfange ich jedoch keine "05" von der Heizungssteuerung noch reagiert diese
auf V-Control oder VoIdent oder Viess-Data.
Das Beobachten des "Sendefensters" der Vitodens mit einer Digicam ließ keine periodischen
IR-Impule erkennen, vielleicht sind diese jedoch auch zu kurz und zu schwach, als daß sie
die Digicam auflösen kann?
Bei der SFH485 Sendediode am Adapter waren sie nur knapp mit der Digicam wahrnehmbar,
wenn diese dynamisch von V-Control angesteuert wurde. 
Bei statischer Ansteuerung der TXD-Leitung war die Sende-LED in der Digicam gut sichbar.

Wenn ich den "optischen Schrumpfschlauchtubus" beim IR-Empfangstransistor entferne,
scheint dieser die Sendesignale der SFH485 im Adapter aufzunehmen. 
V-Control loggt im Debugmodus die selbst ausgesendeten Bytes, 
welche dann teilweise über RX wieder angezeigt werden.

Auszug aus Log:
08-01-2012 17:24:36 DATA TX: 1 F7 25 44 8 
08-01-2012 17:24:36 DATA RX: 1 
08-01-2012 17:24:36 DATA RX: F7 
08-01-2012 17:24:36 DATA RX: 25 
08-01-2012 17:24:37 DATA RX: 44 
08-01-2012 17:24:37 DATA RX: 8 
08-01-2012 17:24:38 DATA RX: 0 
08-01-2012 17:24:39 DATA RX: 0 
08-01-2012 17:24:41 DATA RX: 0 
08-01-2012 17:24:41 DATA TX: 1 F7 8 42 8 
08-01-2012 17:24:41 DATA RX: 1 
08-01-2012 17:24:41 DATA RX: F7 
08-01-2012 17:24:41 DATA RX: 8 
08-01-2012 17:24:41 DATA RX: 42 
08-01-2012 17:24:41 DATA RX: 8 
08-01-2012 17:24:42 DATA RX: 0 

Sporadisch erscheinen jedoch auch andere (gültige??) Empfangsdaten, s.u.:
07-01-2012 23:31:14 SYS Version: 0086
07-01-2012 23:31:14 SYS Protokoll = KW
07-01-2012 23:31:14 DATA Init - OK
07-01-2012 23:31:14 COMM Comm Port OK
07-01-2012 23:31:14 COMM Verbunden: COM1
07-01-2012 23:31:15 DATA RX: 0 
07-01-2012 23:31:16 DATA RX: 0 
07-01-2012 23:31:16 DATA TX: 1 F7 39 0 8 
07-01-2012 23:31:16 DATA RX: 0 
07-01-2012 23:31:16 DATA RX: E6 
07-01-2012 23:31:16 DATA RX: 30 
07-01-2012 23:31:16 DATA RX: 0 
07-01-2012 23:31:16 DATA RX: 0  

Über jegliche Hinweise, Tips und Anregungen zu dem geschilderten Fehlerbild und 
Lösungsvorschläge freue ich mich und bedanke mich vorab.

Beste Grüße,

ao16

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitodens 200 WB2 und Optolink?
* Wiki-Page: home
* From: V200logger <userid-55539681@openv.wikispaces>
* Date: Sun, 31 Jan 2016 11:57:43 UT
* Message-ID: <48944535-87169509$openv.wikispaces>
* In-Reply-To: <48944535-48944535$openv.wikispaces>

Hallo ao16, 
falls es vier Jahre später noch von Interesse ist: An unserer Vitodens 200 aus dem Jahre 2000 habe ich eine Datenloggerfunktion implementiert und in diesem Artikel kurz beschrieben: http://openv.wikispaces.com/Vitodens+200+aus+Baujahr+2000+auslesen
Gruß,
V200logger

hollie, 2021-10-16:
In case somebody else is looking for the original content linked to in this ticket, it is available in the project wiki [here](https://github.com/openv/openv/wiki/Vitodens-200-aus-Baujahr-2000-auslesen).

GitHubTW, 2025-12-28:
@WikiSpaxe 
Hallo, bist du der V200logger?

Ich beziehe mich auf folgenden Link: [: http://openv.wikispaces.com/Vitodens+200+aus+Baujahr+2000+auslesen](url) und habe folgendes Gerät zu Hause:
[https://up.picr.de/50417184me.png](url)

Es hat die gleiche Bedieneinheit wie in dem o.a. Link im ersten Bild sichtbar; allerdings nicht direkt im Heizgerät montiert, sondern in einem anderen Raum als "Fernbedienung".

Folgende Frage habe:
Ist mein Gerät genau das gleiche wie deins?
Und wenn ja: Welches Protokoll spricht es (GWG?) und welches Gerät (Device ID = 2053) handelt es sich?
Und wenn nein: Welches Protokoll spricht mein Gerät und welche Device ID hast es?
😉

Schon mal vorab vielen Dank für eine kurze Rückmeldung und schöne Zeit noch!

Gruß
Thomas