openv/openv #59: Operating mode (Betriebstatus)

WikiSpaxe, 2017-10-25:
* Subject: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: tomazmla <None>
* Date: Sun, 22 Jan 2012 22:34:14 UT
* Message-ID: <49477202-49477202$openv.wikispaces>

Hi

I'm using VScotHO1 (ID=20CB) and can't get current operating mode (Betriebstatus) like 0 - 0ff, 1 - reduced, 2,3 - normal. I tryed with address 2501 (and 3501) as noted in Optolink_Datenpunkte_Logger_v3 but I'm always getting value -1 (FF).
Is there some undocumented address for this?

Thanks for help,
Tomaz

WikiSpaxe, 2017-10-25:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Mon, 23 Jan 2012 19:05:31 UT
* Message-ID: <49477202-49509010$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

The current operating mode is the second byte of a 22 byte datastructure at address 2500.
Can you try to read 22 bytes at 2500?
If this gives the correct result you can also try to read 2 bytes at 2500.
May be that the control needs always even addresses.

WikiSpaxe, 2017-10-25:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: tomazmla <None>
* Date: Tue, 24 Jan 2012 00:40:01 UT
* Message-ID: <49477202-49524994$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

Another big thanks. That's it. Operating mode is as you suggested the second byte at 3500 (I'm just interested in M2). Now I get 02 02 at normal operation and 02 01 at reduced operation. I will now try to write custom unit to extract the second byte.

WikiSpaxe, 2017-10-25:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: tomazmla <None>
* Date: Tue, 24 Jan 2012 13:26:31 UT
* Message-ID: <49477202-49539426$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

I now created custom unit BS (Betriebstatus) based on  unit Bitstatus where I test the second byte. I'm checking the second bit of the second byte and if it is 01 I have Normal othewise Reduced operation.
Thanks for the help.

Tomaz

WikiSpaxe, 2017-10-26:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Tue, 20 Mar 2012 16:05:04 UT
* Message-ID: <49477202-51922176$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

I have another question. I would like to change BetriebsArt from H+WW (03) to WW (00). I can successfully read BetriebsArtM2 at address 3301 but changing is not possible (I get an OK message but change is not visible). Any help would me much appreciated. 

Regards
Tomaz

WikiSpaxe, 2017-10-26:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Wed, 21 Mar 2012 21:28:02 UT
* Message-ID: <49477202-51984416$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

I found it. I can read and change it successfully at address 2323.

Regards
Tomaz

WikiSpaxe, 2017-10-26:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: lanvezhon <userid-51768096@openv.wikispaces>
* Date: Fri, 30 Jan 2015 12:54:51 UT
* Message-ID: <49477202-78932722$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

Hello,
I'm using VScotHO1 (ID=20CB) also and try to connect it to a RASPI and translate vito and vcontrold in "french" because i don't understand German.
I have same pb with 2323 in write. Could you please give me your "special BS"
Many thanks
Regards
René

WikiSpaxe, 2017-10-26:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: mqu31 <userid-52314612@openv.wikispaces>
* Date: Fri, 24 Apr 2015 11:37:04 UT
* Message-ID: <49477202-81481764$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

salut René ; je viens de t'envoyer un message en privé.

WikiSpaxe, 2017-10-26:
* Subject: Re: Operating mode (Betriebstatus)
* Wiki-Page: Adressen
* From: lanvezhon <userid-51768096@openv.wikispaces>
* Date: Sat, 25 Apr 2015 22:20:12 UT
* Message-ID: <49477202-81513548$openv.wikispaces>
* In-Reply-To: <49477202-49477202$openv.wikispaces>

Merci pour l'info

smbunn, 2018-07-18:
Could I also get a copy of the special 'BS' unit type.  Thanks 
Simon