openv/openv #108: Wert = 128.5

WikiSpaxe, 2017-10-26:
* Subject: Wert = 128.5
* Wiki-Page: vclient
* From: NeillHog <userid-4257943@openv.wikispaces>
* Date: Tue, 25 Dec 2012 18:21:51 UT
* Message-ID: <61152434-59383822$openv.wikispaces>

Hat sonst jemand folgend Problem?
Manchmal liefert „vclient -h 127.0.0.1:1234 -c getTempA, getTempWWist“ für ein der zwei Werte 128.5.
Kommt ein paar mal am Tag vor (Abfrage jeder 5 Minuten).
Gibt's eine Lösung?
Danke
Neill

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 25 Dec 2012 18:30:52 UT
* Message-ID: <61152434-59383852$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Hi,

das hatte ich auch.

Wenn ich mich recht erinnere geht in dem Moment die Kommunikation vom vcontrold mit der Steuerung daneben. Statt eine realen wert liest die zwei Bytes 0x0505. Konvertiert in Temparatur kommt dabei genau 128,5 raus.

0x05 ist das Zeichen der Steuerung das sie bereit für Kommandos ist.

Meine ursprüngliche Lösung war exakt diesen Wert zu ignorieren, weil >100 ist für getTempA recht unwahrscheinlich.

Leider haben sich auch an anderen Stellen immer wieder ähnliche Übertragungsfehler eingeschlichen.

Das war für mich der Grund auf das 300 Protokoll umzusteigen, siehe vconnect.

Gruß
Frank

PS. Frohes Fest Euch Allen.

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 25 Dec 2012 18:43:24 UT
* Message-ID: <61152434-59383872$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

```

$ vConnect xxxxx.homeunix.net 3000 getTempA getTempWWist
getTempA.value          9.3
getTempWWist.value         48.8

```

Das läuft auf einem Rootserver im Rechenzentrum. Der Roundtrip zu mir nach Hause mit ping gemessen liegt bei 41 ms.

Gruß
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: NeillHog <userid-4257943@openv.wikispaces>
* Date: Tue, 25 Dec 2012 18:46:43 UT
* Message-ID: <61152434-59383886$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Danke Frank.
Auch ich ignoriere alle Werte über 100 Grad. Funktioniert aber elegant ist es nicht.
Und jetzt muss ich vconnect genauere anschauen. 
Danke.
Neill

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 25 Dec 2012 18:51:22 UT
* Message-ID: <61152434-59383904$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Hallo Neil,

vconnect ist noch ziemlich beta.

Ich würde Dir per E-Mail die aktuellen Sourcen zukommen lassen.

Oder ich muss es endlich hinkriegen, das mein rep. lesenden Zugriff gestattet.

Gruß
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: NeillHog <userid-4257943@openv.wikispaces>
* Date: Tue, 25 Dec 2012 19:46:40 UT
* Message-ID: <61152434-59384022$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Hallo Frank,

Ein paar (vielleicht dummer) fragen.
Läuft vconnect ohne vclient, vcontrold, usw. ? Einfach „vConnect localhost 3000 get...“ ?
Verwendet es auch vito.xml oder sind die Befehle alle definiert in den Quellcode?
GetTempA und getTempWWist sind in der Quellcode definiert aber getBrennerStunden1 nicht. Kann ich auch die Brennerstunden lesen.

Danke
Neill

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 25 Dec 2012 20:04:16 UT
* Message-ID: <61152434-59384064$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Schau mal hier http://openv.wikispaces.com/vconnect

Mein setup ist ser2net auf einem Sheevaplug mit dem Optolink Adapter. ser2net lauscht dann auf Port 3000

Von aussen oder auch localhost greift vConnect dann direkt auf die Optolinkschnittstelle zu.

Alles ohne vclient, vcontrold etc.

Die Befehle sind zur Zeit hardcoded im Quelltext. 

Gruß
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Wert = 128.5
* Wiki-Page: vclient
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 25 Dec 2012 21:46:14 UT
* Message-ID: <61152434-59384298$openv.wikispaces>
* In-Reply-To: <61152434-59383822$openv.wikispaces>

Letztes Update für heute.

Die Sourcen auf der vconnect Seite sind jetzt aktuell von heute.

Gruß
Frank

Bjoern3003, 2022-08-16:
Falls noch jemand das Problem hat, dass nach 1-2 Minuten nur noch 128.5 oder 5 Grad Werte ankommen.

Ich habe festgestellt, dass das Problem mit der Version 0.98.8 auftrat. Habe diese dann mit der Version 0.98.7 verglichen und die Änderungen aus der io.c zurückportiert.

Sprich, ich habe die function my_send https://github.com/openv/vcontrold/blob/v0.98.7/src/io.c#L132 in die io.c der aktuellsten Release Version eingespielt und mir dann wieder eine neue Version kompiliert. Seitdem bekomme ich auch wieder saubere Werte :-)