openv/openv #117: Keine Reaktion auf gesendete Daten

WikiSpaxe, 2017-10-26:
* Subject: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sun, 20 Jan 2013 23:22:53 UT
* Message-ID: <61274078-59811884$openv.wikispaces>

Ich habe laut Bedienungsanleitung eine Vitotronic 200 HO 1A. Gemäß Forum müsste das eine VDensHO1 sein und damit die Protokolle 300 und KW unterstützen.

Zur Kommunikation mit der Steuerung nutze ich ein Arduino Uno und die Library SoftSerial mit einigen Anpassungen, damit sie 4800 8E2 spricht.

An den Arduino habe ich eine Schaltung nach Bauanleitung LAN-Ethernet gebastelt. Da der Arduino 5V Pegel liefert, habe ich den Widerstand für die Sendediode angepasst.

Auf dem Arduino läuft ein Programm, das 5 Mal auf den Empfang von 0x05 wartet. Dann sendet es umgehend die Abfrage der Geräte-ID bzw. der Außentemp.

Mit diesem Aufbau kann ich die 0x05 zyklisch empfangen. Das funktioniert auch in einigen Zentimetern Abstand noch.

Auf das Senden verschiedener Abfragen in unterschiedlichen Protokollen (GWG, KW und 300) bekomme ich keine Reaktion. Die Steuerung sendet weiterhin zyklisch 0x05.

Was mache ich falsch?
- das Oszi sagt, dass an der Sendediode die vom Arduino gesendeten Daten anliegen. => SW scheint zu funktionieren.
- Die Sendediode habe ich ausgetauscht (Mehrere SFH485 und SFH487-2). Wenn man direkt von vorn auf die Diode schaut, kann man ein leichtes rotes Glimmen sehen. => Die Dioden scheinen alle zu funktionieren.
- Den Widerstand an der Sendediode habe ich verändert, da beschrieben wurde, dass die Diode nicht übersteuern sollte. Die Betriebsspannung der Dioden habe ich damit zwischen 1,1V und 1,3V verändert. => Bin mir fast sicher, dass es daran auch nicht liegt.

So, und jetzt bin ich mit meinem Latein am Ende. An was kann es noch liegen? Habe ich was übersehen? 

Kann mit jemand helfen?

Vielen Dank

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sun, 20 Jan 2013 23:29:31 UT
* Message-ID: <61274078-59811970$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hier noch die Singalverläufe vom Oszi ...

[image]

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Mon, 21 Jan 2013 21:31:02 UT
* Message-ID: <61274078-59828860$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo Ofenrohr,

ich teste meine Platinen so:
Spiegel (Alu-Papier tut es auch) ca. 1-2 cm gegenüber der beiden Dioden anbringen. Alles ganz dunkel (lichtdicht) einpacken (zB. kleine Kiste drüber, Tuch,..). Wenn Du nun ein Signal sendest müsste es auf der Empfängerseite genau so ankommen. (ich messe auch mit dem Oszi auf der Empfänger- und Sender gleichzeitig und die Signale sollten identisch sein).
Du solltest wissen, dass die Empfängerdiode bei Licheinfall vorgespannt ist. Die Vorspannung kann ausreichen das sie ständig den Pegel auf + zieht.
Also auch Deine Koppelung zum .V. und zwichen den Dioden schön lichtdicht halten (Ich ziehe über jede Diode einen schwares 5mm Röhrchen was die Dioden nach außen und gegeneinander abschottet .)

Der Widerstand an der Sende-Diode ist nicht kritisch. Der Strom durch die Diode sollte lt. Datenblatt nicht >100mA sein. Die 100Ohm bei 5V sind also kein Problen. Ich habe da 47Ohm bei 3.3V, das sind dann ca. 40mA. Zu großer Widerstand senkt die Lichleistung. (rechnen oder einfach Spannungsabfall am Widerstand mit dem Oszi messen und URI)

Gruß
Stefan

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Mon, 21 Jan 2013 21:39:19 UT
* Message-ID: <61274078-59828964$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo,
wenn Du schon ein Oszi hast ... prüf doch mal bitte, ob an der Sendediode die Spannung "genügend" wechselt um von hell auf dunkel zu schalten. Ich hatte bei meinem Aufbau das Problem, dass U(BE) am Sendetransistor immer > 0.6 V war, somit der Transistor immer durchschaltet. Ich habe das mit einem Pullup von 7k5 nach 5V behoben.
Tipp: Bau Dir doch mal einen zweiten Empfänger, Spannung, Widerstand, Empfänger und schau mal, ob Du damit das gesendete Signal am Oszi sehen kannst. Ob es ein Spiegel an der Schaltung mit Reflexion auch tut, kann ich nicht sagen. Ich war beim Aufsetzen der Schaltung etwas irritiert, was Beinchen und Polung des Senders angeht, also Prüfen. Wenn bei Dir aber schon Strom fliesst (Spannusabfall am Widerstand) sollte die Richhtung OK sein. Das Schalten der Sendediode solltest Du aber auch am Spannungsabfall am Widerstand finden. Der sollte um ca. 1.5 V schwanken - Oszi 
vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Tue, 22 Jan 2013 22:28:51 UT
* Message-ID: <61274078-59856290$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

erstmal vielen Dank für Eure Vorschläge und Tipps. Ich habe mich gleich in den Keller gesetzt und gebastelt und gemessen. Leider noch ohne Erfolg.

- Sendediode und Empfangsdiode vor einer DVD dunkel eingepackt
  => egal was ich sende, empfange ich eine 0x00. Das liegt aber am Arduino. Senden in der zyklischen Task und Empfangen im Interrupt. Nun, das zeigt, dass die Empfangsseite zumindest das Startbit mitbekommt.

- meine Sendediode mit einem ungeschrumpften Stück Schrumpfschlauch versehen und das Licht im Raum ausgemacht.
  => keine Veränderung.

- Die Schaltung an der Sendediode auf die Diode und Poti reduziert. Das Poti so eingestellt, dass ca. 40mA durch die Doide fließen und ein Spannungsabfall von ca. 1,8V entsteht. Arduino liefert ein sauberes Rechtecksignal. Das habe ich dann an der Diode auch gemessen. D.h. ein Signal zwischen 0V und 1,8V.
  => keine Veränderung

- Einen zweiten Empfänger baue ich bei Gelegenheit auch noch. Habe zur Zeit nur keinen PC mit serieller Schnittstelle und mein USB-RS232-Interface habe ich verliehen.
  => ToDo für mich

- Beim Schmökern im Wiki ist mir noch aufgefallen, dass das Senden "unmittelbar" nach dem Empfangen der 0x05 starten soll. Wie viel ist "unmittelbar"? Habe das bei mir mal gemessen. Der Sendevorgang (Flanke Startbit) beginnt 560us nach dem Empfangsvorgang (Flanke Stoppbit). Dann habe ich unmittelbar die Zeit verändert. 
  => +1ms und + 5ms Verzögerung haben keine Veränderung gezeigt.

In der SW auf dem Arduino habe ich eine Zeitmessung eingebaut. Bei der Auswertung ist mir aufgefallen, dass .V. nachdem er meine Daten empfangen hat, kurz Hicks macht. Siehe unten. Ist das normal? - Meine SW ist so gestrickt, dass sie nach jedem 5. Empfang von 0x05 Daten sendet. Die gesendeten Daten habe ich mit den Protokollen GWG, KW und 300 variiert. Das Beispiel zeigt das Protokoll 300. 

```

...
TX:	0
RX:		5		(1991ms)	Hicks beim Empfangen nach jedem Senden
RX:		5		(2238ms)
RX:		5		(2240ms)
RX:		5		(2238ms)
RX:		5		(2243ms)
TX:	16		(460us)
TX:	0
TX:	0
RX:		5		(1997ms)	Hicks
RX:		5		(2241ms)
RX:		5		(2238ms)
RX:		5		(6514ms)	Entfernen der Sendediode
RX:		5		(2237ms)
TX:	16		(464us)
TX:	0
TX:	0
RX:		5		(2236ms)	=> Hicks verschwindet
RX:		5		(2240ms)
RX:		5		(2236ms)
RX:		5		(2238ms)
RX:		5		(2241ms)
TX:	16		(464us)
TX:	0
TX:	0
RX:		5		(2235ms)	kein Hicks
RX:		5		(2238ms)
...

```

Jetzt glaube ich dann bald, dass die Empfangsdiode im .V. kaputt ist. Kann man die deaktivieren? Nein, das war eine blöde Frage. Aber nachdem was ich gemessen und probiert habe, müsste mit meinem Aufbau alles passen. Oder sehe ich das falsch? Habt Ihr noch Vorschläge und Tipps? Ich bin dankbar für alles.

So, und jetzt trinke ich ein Frustbier. Prost.

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Tue, 22 Jan 2013 22:49:35 UT
* Message-ID: <61274078-59856758$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo,
es gibt in den diversen Schaltungen Kommentare, dass der Strom durch die Sendediode nicht mehr als 10 mA sein soll, sonst Übersteuerungen (an der Empfangsdiode in der Steuerung, könnte temporär "blind" werden). Vermutlich sind die 40 mA einfach zuviel, 1.8 V an der Diode scheint mir auch viel... Poti so einstellen, dass 10 mA laufen.
vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Wed, 23 Jan 2013 02:18:25 UT
* Message-ID: <61274078-59862052$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo,

mit den 40mA sehe ich nicht als großes Problem (die Sonne schiebt deutlich mehr Licht ins .V., wenn sie offen steht), aber wir sollten auf vheat hören. Es wird wohl so sein, dass der Eingang übersteuert und dann eine Schutzschaltung in der .V. zuschlägt und sie Temp. lahm legt. (habe ich aber noch nicht festgestellt)
Werde sicherheitshalber meinen R mal auf 150Ohm (ca. 10mA bei 3.3 V Versorgungsspannung) setzen und testen.

@Ofenrohr:
Du solltest wirklich mal mit dem Oszi messen.
Bin z.Zt. an einem Java-Programm auf einem Raspberry Pi am schreiben.
Was ich dabei festgestellt habe, dass der Eingangspuffer des Seriellen Interfaces die 0x05 sammelt.
Wenn man nun ein 0x16 0x00 0x00 sendet bekommt man die Antwort 0x06 nicht mehr mit da man ewig mit den 0x05 zugeworfen wird. 

```java

void flush() throws Exception {
		// Flush input Buffer 
		input.skip(input.available());
		if (debug) System.out.println("#    Input Buffer flushed");
}

```

Ich setze vor jedem Start der Session einen flash auf den Eingangs-Puffer auf. 

```java

    void startSession() throws Exception {	
		if (session) {
			System.out.println("#    Session already opened");
			return;
		}
		if (debug) System.out.println("#    Open Session to Optolink");
		flush();                                 // Puffer leeren - Notwendig wenn tty nicht sauber geschlossen wurde 
		if (debug) System.out.print("TxD:");
		psend(0x04);                            // wenn noch offene Kommunikation vorhanden, schließen
		if (debug) System.out.println();

		for (int i=1; i<4; i++) {                              // drei mal versuchen auf Sync
		   if (debug) System.out.println("##   Try "+i+ " ");
		   if (debug) System.out.print("RxD:");
		   while (pread() != 0x05 );                          // Warten bis 0x05 kommt
		   if (debug) System.out.println ();
		   if (debug) System.out.print("TxD:");
		   psend(0x16);                                       // Init senden
		   psend(0x00);
		   psend(0x00);
		   if (debug) System.out.println();
		   if (debug) System.out.print("RxD:");
		   if (pread() == 0x06) {                            // Ack abfangen
			   session = true;
			   if (debug) System.out.println(" [ACK]");
			   if (debug) System.out.println("#    Session opened");
			   return; // Init war OK
		   }
		   if (debug) System.out.println(" [ACK failed]");
		}
		// kein init, Exception werfen
		throw new RuntimeException("Unable to start Session to Optolink");
	}
...........

	private void psend(int by) throws IOException {
		if (debug) System.out.printf(" %02X", (byte)by);
		output.write((byte)by);
		
	}
	
	private int pread() throws IOException {
		int  i;
		i=input.read();
		if (debug) System.out.printf(" %02X", i);
		return i;
		
	}

```

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Wed, 23 Jan 2013 23:32:23 UT
* Message-ID: <61274078-59889284$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

vielen Dank für Eure Hilfe. Leider hatte ich immer noch keinen Erfolg.

- Strom an der Sendediode auf 10mA getrimmt. Damit ergibt sich bei mir ein Spannungsabfall von 1,39 V an der Diode.  => Leider ohne Erfolg. Auch bei Licht aus im Raum.

- Parity bit und Bitmuster nochmal geprüft (Oszi-Messung, Review Library-Code). Scheint zu passen. Entspricht 4800 8E2. => Leider ohne Erfolg

- Auf dem Arduino polle ich den Eingang zyklisch und nach Bauchgefühl auch schnell genug. Das Polling sollte im 1 bis 2 stelligen Millisekundenbereich passieren. Die 0x05 wird im Sekundenbereich versendet. Um sicher zu gehen, werde ich das die Tage mal ausmessen.  

Als nächstes werde ich mir mein USB-RS232-Interface wieder zurück holen und damit mal ein paar Versuche machen. Das war auch die Empfehlung von Carsten, den ich über das Wiki direkt angeschrieben hatte. Er hatte mit der Library Softserial leider auch keinen Erfolg. 

Es wird wohl ein paar Tage dauern, bis ich weiter machen kann :-(

Vieln Dank nochmal für Euere Tipps und Vorschläge.

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Thu, 24 Jan 2013 00:12:18 UT
* Message-ID: <61274078-59890272$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

nochmal der Versuch Sendediode in Empfangsdiode einstrahlen lassen? Gegeneinander biegen?

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Mon, 28 Jan 2013 22:37:57 UT
* Message-ID: <61274078-59996468$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi,

habe die Sendediode auf die Empfangsdiode einstrahlen lassen. Sowohl auf Sender als auch auf Empfängerseite mit dem Oszi gemessen. Das sollte so passen.

Dann habe ich noch am Timing der Tx-Routine gespielt. Ein bisschen mehr als 4800bps und ein bisschen weniger verändern das Ergebnis nicht. :-(

Jetzt mache ich mich an einen zweiten Adapter. Vielleicht lerne ich dabei, warum die aktuelle Lösung nicht funktioniert. 

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Sun,  3 Feb 2013 16:09:18 UT
* Message-ID: <61274078-60135756$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

@Stovepipe

Hi,  I am starting a similar project, trying to monitor the optolink with an arduino. I am not on the arduino  step yet. Right now, I made a device with just the diodes and transistors parts, and  connected it  to my mac (using coolterm)  with a TTL to USB  card (FOCA, that is based on a FT232RL chip). After some tries, I manage to get the  05 05 05 sequence, but I am  unable to send data. 
In order to test the emitting diode, I use a digital camera, and do not see that the IR led (SFH406) is really on. I have hte impression that whaterver I type, the transistor (2N3906) is always switched off).

I am beginner in electronics, and will appreciate any help.  

Best Wishes 

Jacques

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sun,  3 Feb 2013 20:59:32 UT
* Message-ID: <61274078-60140534$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

habe mittlerweile einen 2. Empfägner gebaut, der brav zyklisch 0x05 empfängt.

Der 2. Empfägner versteht auch prima was der Arduino sendet. Leider reagiert .V. nicht auf das was Arduino sendet. 

Ich verstehe das einfach nicht. Hat noch jemand eine Idee?

Vielen Dank

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sun,  3 Feb 2013 21:09:19 UT
* Message-ID: <61274078-60140688$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

@Jacques,

two hints in the forum really helped me a lot, so far. Ok, I was not successful, yet. But I gained confidence.
1. place the emitting and the receiving diodes opposit of each other, in a way that the receiver will see what the transmitter was sending. => If you transmit you should see the same values on the receiver.
2. build up a second optolink. maybe use a second computer as well.

By the way, I also use a RS232 to TTL board as my second optolink.

For the Arduino there is a library called SoftSerial. This lib was not prepared for 8E2. I'm currently working on this. As soon as my changes for 8E2 will work, I'll provide the lib on this forum.

Regards

Stovepipe

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Mon,  4 Feb 2013 22:47:40 UT
* Message-ID: <61274078-60176534$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi ofenrohr,

da fällt mir nur ein, dass der UART falsch eingestellt (programmiert) sein kann. Versuche mal Dinge RAW -echo -xon -xoff. 
Ich arbeite mit remserial  und den Einstellungen: "4800 raw cs8 cstopb -parodd -echo".  Die Einstellung läuf jetzt schon Tage stabil, zusammen mit JAVA auf einem raspberry Pi.
Keine Ahnung ob du remserial (Linux i386) auf dem Arduino übersetzen kannst.

Gruß Stefan

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Tue,  5 Feb 2013 09:21:17 UT
* Message-ID: <61274078-60194428$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

ich klinke mich hier mal ein, weil ich praktisch dasselbe Vorhaben habe: Per Arduino und Ethernetshield meine Vitotronic 200 auslesen. Ich stehe allerdings noch ganz am Anfang, sprich hab noch keinen Arduino Sketch.

@Ofenrohr: Könntest du bitte nochmals kurz schildern, welche Änderungen bei dir notwendig waren? Vielleicht kommen wir dem Problem ja auf die Spur, denn ein erster Test meines Transceivers (Modell TTL) zeigt dasselbe Phänomen am Rechner (Mac): Ich sehe die 0x05 periodisch im Sekundentakt, die VT reagiert aber nicht auf gesendete Signale. Es scheint also nicht unbedingt am Arduino zu liegen.

@vstyx: der Arduino ist ein Mikrocontroller - da läuft leider keinerlei Linuxsoftware drauf, sondern lediglich die Libraries, die dafür zur Verfügung gestellt werden.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Tue,  5 Feb 2013 09:47:04 UT
* Message-ID: <61274078-60194528$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Versucht doch mal bitte eine Schleife aufzubauen, also der Sender soll den Empfänger sehen, dann ein Terminalprogramm wie minicom oder busybox microcom auf diese Schnittstelle und schauen, ob das Geschriebene korrekt zurückkommt.
Für die Verwendung an der Vitotronic... bitte checkt doppelt (... und dreifach..) das der Port wirklich auf 4800 8E2 steht. Üblicherweise gibt es kaum Parity und auch die 2 Stoppbits sind eigenartig.   
Ich hab mir jetzt die 05 nicht angeshen, aber es kann sein, dass diese auch mit falschen Einstellungen richtig detektiert wird.
Im TX Bild oben für 01 und F7 scheint mir die Parity falsch zu sein, bei der 05 stimmt es, da gibt es bei den Datenbits+Parity eine gerade Anzahl = Even = E.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Tue,  5 Feb 2013 21:49:07 UT
* Message-ID: <61274078-60213326$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

vielen Dank für Euren vielen Tipps. Ich komme vermutlich diese und nächste Woche nicht viel zum Experimentieren.

- Das Protokoll hatte ich bereits geprüft. Auch die Parität. Oben im Diagramm ist sie wirklich falsch. Werde ich nochmal prüfen. Danke für den Hinweis.

- Ich habe vom Arduino gesendet und mit dem PC empfangen. Dabei hatte ich festgestellt, dass der Arduino, wenn ich die Sendediode direkt an einen Digitalport anschließe, invers sendet. Das habe ich in der Library korrigiert. Anschließend konnte der PC die Daten vom Arduino empfangen. 

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Tue,  5 Feb 2013 22:21:40 UT
* Message-ID: <61274078-60214378$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

... und andersherum ? Also PC senden, Arduino empfangen ?

ofenrohr, 2017-10-26:
Der Sync ist Mist... Ich bitte darum die Erwähnungen mit @ zu entfernen. Das sind nicht meine Fragen hier.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed,  6 Feb 2013 10:26:26 UT
* Message-ID: <61274078-60229902$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

vheat: Ich hatte die Sendeprobleme am PC unter Linux mit vcontrold. Ich gehe schwer davon aus, dass dort die seriellen Parameter korrekt gesetzt werden. Ausserdem hat der Adapter "mal" funktioniert. Da ich aber bereits Transen und Sendediode getauscht habe - ohne Ergebnis, kann ich einen Hardwaredefekt eigentlich auch ausschliessen.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sat,  9 Feb 2013 21:07:08 UT
* Message-ID: <61274078-60320790$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo vheat,

das geht nicht. Ich habe an den Digital-Ausgang des Ardunio direkt die Sendediode mit einem Widerstand angeschlossen. Das habe ich bei meinem RS232-TTL-Konverter auch versucht. Dadurch bekomme ich aber ein invertiertes Signal. Beim Ardunio habe ich das über SW ausgeglichen. Beim PC kann ich das nicht ausgleichen. Bei der nächsten Elektronikbestellung ist ein 2N3906 dabei ...

Grüße
 Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Ofenrohr <userid-38438678@openv.wikispaces>
* Date: Sat,  9 Feb 2013 22:12:43 UT
* Message-ID: <61274078-60321468$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

ich habe die Parität nochmal überprüft. Sieht gut aus, oder habe ich das falsch interpretiert?

[image] 

Grüße

Ofenrohr

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Sat,  9 Feb 2013 23:37:29 UT
* Message-ID: <61274078-60322320$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

I have resumed my  experiments. 
I have the same problem as Stovepipe; I receive 05 sequence but I am not able to send messages and get answers. 
My setting is (for the emitter)
SFH486 IR led
2N3906 transistor
10 kohm resistor
100 ohm resistor

and I use a TTL to USB card to connect to the computer (a macbook). On the computer I use 
Coolterm with 4800 8E2.
I  have verified that the system works when inputting a square signal  and looking with the oscilloscope everything is fine, and as expected. 
When I put the emitter in front of the reciever, 
I properly get the echo of everything that is typed on the keypad so the emitter is working well, as well as the receiver.

I have verified that the led is blinking when I type a key on the computer (using a digita camera). 

Nevertheless,  I do not get anything when I type 
01 F7 55 25 02

Only the 05 sequence. 

I am not sure of the way to type the above sequence. 

Should it be 
01 F7 55 25 02
or 
01F7552502

or something else 

Thanks

Jacques

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Sat,  9 Feb 2013 23:49:25 UT
* Message-ID: <61274078-60322432$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi Jacques,

sorry if that sounds like a stupid question to you, but what exactly are you sending out?

If it is written as 01 F7 or 01F7 than in both cases the hex values 0x01 and 0xf7 are meant.

You are not sending the literal string "01F7" aka the four letters 0 1 F and 7 as typed from the keyboard, aren't you?

Regards
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sun, 10 Feb 2013 00:01:34 UT
* Message-ID: <61274078-60322532$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Salu Jacques

And if you send it with a terminal, it will be difficult to send it at the right time.
You must send the sequence just after the 05 arrived - max tolerance is about 100ms - so you should try several times - may be you are lucky to hit at the right time once.
If you can run Windows Programs on your macbook you can use RS232Test or VitoTest from "Windows Software" on this site.
Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sun, 10 Feb 2013 14:07:09 UT
* Message-ID: <61274078-60327188$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Salu Jacques

I have found a Terminal Programs in JAVA that works from within the Browser - it should also work on OS X.

1. open this url:
http://code.google.com/p/java-simple-serial-connector/wiki/jSSC_Terminal

2. accept the execution of java code
3. With "Settings" select your serial port, 4800 Baud, 8 Databits, 2 Stopbits and Even Parity and save the settings
5. Open the port and choose "View as HEX" and you should see 05 05 ...
6. Press "Send as HEX", enter "01 C7 F8 04" (without ") and "Send" 
I had to try about 20 times and once the answer was 20 98 00 02 (Vitotronic 200 KW2). I had to press "Send" a little bit before the received 05 was displayed....
Regards, Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Sun, 10 Feb 2013 17:35:07 UT
* Message-ID: <61274078-60329554$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

To Franck (fnobis) 

I am sending hexadecimal codes. 

To Hanspeter (Vitoopen) 

Thanks a lot, this was really helpfull. I did not realised that one has to answer so quickly. 

For the terminal, I have on the mac the application CoolTerm which is very adequate to send and receive serial data. 

I finally got an answer from my Vitronic. Following your suggestion, 

I send the hexa  01 C7 F8 04
and got back  20 98 00 07

With Parallel, I can also have a Windows XP emulation, and could run Vito Test, but I do not understand very well what I can get from it. 
(are there some help or manual page somewhere ? ). 
I got : 

Port Open 
Start synch
tty type 2
Online type 2
20 98 

Can I do more ? 

I tried also to start v-control 1.2.5
but it did not went very far (see log below) 

10-02-2013 17:32:15 DATA Init
10-02-2013 17:32:15 COMM Init auf COM4
10-02-2013 17:32:15 DATA TX: 16 0 0 
10-02-2013 17:32:16 DATA TX: 16 0 0 
10-02-2013 17:32:18 DATA TX: 16 0 0 
10-02-2013 17:32:20 DATA TX: 16 0 0 
10-02-2013 17:32:22 TX  01 F7 00 F8 02
10-02-2013 17:32:23 RX  20 98
10-02-2013 17:32:23 SYS  Version: V200KW2
10-02-2013 17:32:23 COMM Comm Port OK
10-02-2013 17:32:23 COMM Verbunden: COM4
10-02-2013 17:32:24 DATA  Lese Daten
10-02-2013 17:32:24 DATA TX: 41 5 0 1 0 0 4 A 
10-02-2013 17:32:25 TX  01 F7 08 00 04
10-02-2013 17:32:25 RX  21 00 B7 02
10-02-2013 17:32:27 DATA TX: 41 5 0 1 0 0 4 A 
10-02-2013 17:32:30 DATA TX: 41 5 0 1 0 0 4 A 
10-02-2013 17:32:33 DATA TX: 41 5 0 1 0 0 4 A 
10-02-2013 17:32:36 DATA Read Data - Fehler
10-02-2013 17:32:36 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:39 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:42 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:45 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:48 DATA Read Data - Fehler
10-02-2013 17:32:48 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:51 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:54 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:32:57 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:00 DATA Read Data - Fehler
10-02-2013 17:33:00 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:03 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:06 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:09 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:12 DATA Read Data - Fehler
10-02-2013 17:33:12 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:15 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:18 DATA TX: 41 5 0 1 0 0 2 8 
10-02-2013 17:33:21 DATA TX: 41 5 0 1 0 0 2 8 

Thanks again for all your help. I am sure it will finally work. 

Jacques

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Sat, 16 Feb 2013 17:11:22 UT
* Message-ID: <61274078-60489132$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

It works !! (or nearly ) 

As I have verified that with the terminal I can send and receive messages. I try again with vcontrol 1.2.5  on the mac  wiht windows on a virtual machine, and then on a PC with Windows 7. Nothing worked. 

I then try vcontrol 1.3.0
and everything  is working properly. 

Can you tell why ??

I can change some settings, as the night temp. but if I try to change the time setting it does not work. I probably missed something. 

Best WIshes 

Jacques

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat, 16 Feb 2013 20:12:33 UT
* Message-ID: <61274078-60490294$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hello Jacques

vcontrol 1.3.0 supports all three protocols (GWG,KW,300) version 1.2.5 supports only protocol "300".

Since your device V200KW2 (2098) does not support the protocol "300" you need version 1.3.0

it is that simple ;-)

the protocol is documented in the third column of the list on the page "Geräte"

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed, 20 Feb 2013 11:45:37 UT
* Message-ID: <61274078-60553078$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo zusammen,

ich habe mittlerweile einen Optolink zusammgelötet und vcontrold/vclient auf einem Pogoplug mit Debian Linux kompiliert - läuft soweit. Allerdings bekomme ich keine Verbindung zur Vito:

[1616] Wed Feb 20 11:40:49 2013 : TCP socket 1234 geoeffnet
[1616] Wed Feb 20 11:41:06 2013 : 127.0.0.1 in allowList (127.0.0.1)
[1616] Wed Feb 20 11:41:06 2013 : Client verbunden 127.0.0.1:45097 (FD:6)
[1616] Wed Feb 20 11:41:06 2013 : Befehl: getTempA
[1616] Wed Feb 20 11:41:06 2013 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[1616] Wed Feb 20 11:41:06 2013 : Process 1616 tries to aquire lock
[1616] Wed Feb 20 11:41:06 2013 : Process 1616 got lock
[1616] Wed Feb 20 11:41:06 2013 : >SEND: F7
[1616] Wed Feb 20 11:41:06 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:06 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:06 2013 : >SEND: 02
[1616] Wed Feb 20 11:41:08 2013 : <RECV: 05 (1990.0 ms)
[1616] Wed Feb 20 11:41:10 2013 : <RECV: 05 (1990.0 ms)
[1616] Wed Feb 20 11:41:10 2013 : Recv Timeout: 3980 ms  > 150 ms (Retry: 2)
[1616] Wed Feb 20 11:41:10 2013 : >SEND: F7
[1616] Wed Feb 20 11:41:10 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:10 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:10 2013 : >SEND: 02
[1616] Wed Feb 20 11:41:12 2013 : <RECV: 05 (2000.0 ms)
[1616] Wed Feb 20 11:41:14 2013 : <RECV: 05 (1990.0 ms)
[1616] Wed Feb 20 11:41:14 2013 : Recv Timeout: 3990 ms  > 150 ms (Retry: 1)
[1616] Wed Feb 20 11:41:14 2013 : >SEND: F7
[1616] Wed Feb 20 11:41:14 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:14 2013 : >SEND: 01
[1616] Wed Feb 20 11:41:14 2013 : >SEND: 02
[1616] Wed Feb 20 11:41:16 2013 : <RECV: 05 (1990.0 ms)
[1616] Wed Feb 20 11:41:18 2013 : <RECV: 05 (1990.0 ms)
[1616] Wed Feb 20 11:41:18 2013 : Recv Timeout: 3980 ms  > 150 ms (Retry: 0)
[1616] Wed Feb 20 11:41:18 2013 : Recv Timeout, Abbruch
[1616] Wed Feb 20 11:41:18 2013 : Process 1616 released lock
[1616] Wed Feb 20 11:41:18 2013 : Fehler beim ausfuehren von getTempA
[1616] Wed Feb 20 11:41:18 2013 : Befehl: quit
[1616] Wed Feb 20 11:41:19 2013 : Verbindung beendet (fd:6)

Ich hatte erst den Adapter in Verdacht, das Dumme ist aber, dass dieser 1:1 mit demselben FTDI-Wandler am selben Kabel auf meinem Macbook funktioniet. Ich kann dort in CoolTerm die periodischen 0x05 sehen. Leider weiss ich auf Anhieb nicht, wie ich die serielle Schnittstelle unter Linux mitlesen kann. Mit screen und minicom komme ich nicht weiter. Jemand Vorschläge?

Danke!
Ralf.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: ramack <userid-19832842@openv.wikispaces>
* Date: Wed, 20 Feb 2013 20:36:30 UT
* Message-ID: <61274078-60567544$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo Haggy,

also ich würde sagen, dass das empfangen tut - du siehst ja auch RECV: 05 alle 2 Sekunden.  Das Problem scheint auf der Sende-seite zu sein. Hast du schon mal versucht einfach nur die Geräte ID auszulesen? Adresse 00F8 2 bytes lesen. Du kannst auch deinem Adapter mal einen Spiegel vorhalten und schauen, ob das was du sendest auch wieder ankommt...

Als terminal-Programm unter Linux kannst du mal c-kermit versuchen. Ist von der Lizenz der nicht so hübsch, aber viele kommen damit besser zurecht als mit minicom.

Rapha

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Wed, 20 Feb 2013 20:40:36 UT
* Message-ID: <61274078-60567762$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo Ralf
Leider zeigt dein Protokoll-Mitschnitt keine Millisekunden...

Soviel kann ich sagen:
nach dem Empfang von 05 muss innerhalb kurzer Zeit 01 gesendet werden (das fehlt bei dir ganz), andernfalls reagiert die Steuerung nicht auf deine Befehle.
Weitere Probleme könnte es geben, wenn das Protokoll nicht richtig eingestellt ist (insbesondere Even-Parity)
Ich hoffe, du kannst mit diesen Vermutungen was anfangen...
Gruss, Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed, 20 Feb 2013 20:46:17 UT
* Message-ID: <61274078-60567952$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Guten Abend,

@:ramack: Ja, es kommt was an. Allerdings alle nur 2 Sekunden. Wenn ich den Adapter am Mac mitlese erhalte ich schön regelmässig 05 jede Sekunde. Ich probier das aber mal mit c-kermit, danke für den Tipp!

@:vitoopen: Der Mitschnitt ist soweit komplett - bis auf die Initialisierung am Anfang - benötigst du die? Habe vcontrold mit -n und -g gestartet. Ich denke vcontrold kümmert sich um die richtige Konfiguration der seriellen Schnittstelle, oder nicht ?

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed, 20 Feb 2013 21:45:43 UT
* Message-ID: <61274078-60569654$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

So, bin nun ein wenig schlauer:

1) Sendediode funktioniert - kann das Blinken mit einer Digitalkamera sehen.
2) Es wird augenscheinlich auch "ordentlich" gesendet, sprich Sendediode ist synchron zur Kontroll-LED auf dem FTDI-Dongle.
3) Mittels Spiegel mitlesen funktioniert NICHT. Ich empfange rein gar nichts - wenn ich aber mit der TV-Fernbedienung auf den Spiegel zeige, kann ich deren Signale mitlesen. Evtl. kann CoolTerm und/oder der Adapter nicht gleichzeitig Empfangen und Senden?
4) vcontrol in eine Linux-VM verhält sich exakt gleich wie die Installation auf dem Pogoplug
5) Zum Verständnis: Ich habe die 3,3V TTL-Variante gebaut, habe allerdings nicht wie in der LAN-Anleitung einen 100 Ohm Vorwiderstand vor der Sendediode, sondern wie in der TTL-Anleitung einen 330 Ohm Vorwiderstand. Das führt dazu, dass die Diode (minimal?) weniger Ausgangsleistung hat. Der Adapter sitzt aber nur wenige Milimeter vor den Dioden der Vitotronic - sollte also noch dicke ausreichen.

Etwas ratlos,
Ralf.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Wed, 20 Feb 2013 22:07:13 UT
* Message-ID: <61274078-60570094$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Taschenrechner: Die Diode soll ca. 10 mA Strom haben, Spannungsabfall dan ca.1.5 V
Also für 5 V bleiben 3.5 V, mit 330 Ohm macht das 10 mA,
Der 3.3V Fall: bleiben ca. 1.8 V bei 330 Ohm ungefähr 5 mA also die Hälfte, bei 100 Ohm 18 mA also zuviel. 
Schalte doch einfach 2 x 330 Ohm parallel, damit solltest Du bei etwa 10 mA landen.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed, 20 Feb 2013 22:11:30 UT
* Message-ID: <61274078-60570202$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Vielen Dank für den schnellen Tipp! Probiere ich aus... die Frage ist dann aber nur: wie kann die 3.3V TTL-Variante hier aus den Anleitungen jemals funktioniert haben?

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Wed, 20 Feb 2013 23:34:57 UT
* Message-ID: <61274078-60572280$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

So, wieder da. Habe jetzt einen 330 Ohm nochmal parallel vor die Sendediode geschaltet. Ersatzwiderstand wäre ~165 Ohm und somit rein rechnerisch ~10mA - hat aber leider nichts gebracht, selbes Verhalten wie vorher:

```

[1726] Wed Feb 20 23:31:42 2013 : Client verbunden 127.0.0.1:54623 (FD:6)
[1726] Wed Feb 20 23:31:42 2013 : Befehl: sync
[1726] Wed Feb 20 23:31:42 2013 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[1726] Wed Feb 20 23:31:42 2013 : Process 1726 tries to aquire lock
[1726] Wed Feb 20 23:31:42 2013 : Process 1726 got lock
[1726] Wed Feb 20 23:31:42 2013 : Warte auf 05
[1726] Wed Feb 20 23:31:42 2013 : <RECV: 05 (30.0 ms)
[1726] Wed Feb 20 23:31:42 2013 : >SEND: 01
[1726] Wed Feb 20 23:31:42 2013 : Process 1726 released lock
[1726] Wed Feb 20 23:31:42 2013 : Befehl: getTempWWist
[1726] Wed Feb 20 23:31:42 2013 : Process 1726 tries to aquire lock
[1726] Wed Feb 20 23:31:42 2013 : Process 1726 got lock
[1726] Wed Feb 20 23:31:42 2013 : >SEND: F7
[1726] Wed Feb 20 23:31:42 2013 : >SEND: 01
[1726] Wed Feb 20 23:31:42 2013 : >SEND: 0D
[1726] Wed Feb 20 23:31:42 2013 : >SEND: 02
[1726] Wed Feb 20 23:31:44 2013 : <RECV: 05 (1980.0 ms)
[1726] Wed Feb 20 23:31:46 2013 : <RECV: 05 (1990.0 ms)
[1726] Wed Feb 20 23:31:46 2013 : Recv Timeout: 3970 ms  > 150 ms (Retry: 2)
[1726] Wed Feb 20 23:31:46 2013 : >SEND: F7
[1726] Wed Feb 20 23:31:46 2013 : >SEND: 01
[1726] Wed Feb 20 23:31:46 2013 : >SEND: 0D
[1726] Wed Feb 20 23:31:46 2013 : >SEND: 02
[1726] Wed Feb 20 23:31:48 2013 : <RECV: 05 (1990.0 ms)
[1726] Wed Feb 20 23:31:50 2013 : <RECV: 05 (1990.0 ms)
[1726] Wed Feb 20 23:31:50 2013 : Recv Timeout: 3980 ms  > 150 ms (Retry: 1)
[1726] Wed Feb 20 23:31:50 2013 : >SEND: F7
[1726] Wed Feb 20 23:31:50 2013 : >SEND: 01
[1726] Wed Feb 20 23:31:50 2013 : >SEND: 0D
[1726] Wed Feb 20 23:31:50 2013 : >SEND: 02
[1726] Wed Feb 20 23:31:52 2013 : <RECV: 05 (1990.0 ms)
[1726] Wed Feb 20 23:31:54 2013 : <RECV: 05 (1990.0 ms)
[1726] Wed Feb 20 23:31:54 2013 : Recv Timeout: 3980 ms  > 150 ms (Retry: 0)
[1726] Wed Feb 20 23:31:54 2013 : Recv Timeout, Abbruch
[1726] Wed Feb 20 23:31:54 2013 : Process 1726 released lock
[1726] Wed Feb 20 23:31:54 2013 : Fehler beim ausfuehren von getTempWWist
[1726] Wed Feb 20 23:31:54 2013 : Befehl: quit
[1726] Wed Feb 20 23:31:55 2013 : Verbindung beendet (fd:6)

```

Ich habe jetzt noch ein sync vor die Abfrage gepackt, damit auf 05 gewartet wird. Das wird dann sofort mit 01 bestätigt und das Kommando hinterhergeschickt. Die Vito sendet aber stur ihre 05 weiter.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Thu, 21 Feb 2013 10:43:12 UT
* Message-ID: <61274078-60581670$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

OK, noch ein Update: Habe mir in einer VM Windows XP mit voIdent installiert. Ergebnis: funktioniert auch nicht. Langsam habe ich die Vermutung, dass die Stücklisten beim TTL-Bausatz evtl. nicht ganz stimmen?!

Würde mir bitte Jemand eine Stückliste für einen Adapter posten, der garantiert funktioniert? Ich benötige die ganz simple Variante ohne USB<>Seriell-Wandler oder so etwas. Mehr als 2 Transistoren, 2 Dioden und paar Widerstände sollte da nicht bei sein. Betrieben werden soll der Adapter später an einem Arduino.

Danke & Grüße!
Ralf.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Thu, 21 Feb 2013 15:21:48 UT
* Message-ID: <61274078-60584766$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Schau Dir mal meinen Schaltplan unter FriedlyARM an. Die Schaltung hängt direkt an den TXD/RXD Ports eines ARM9. Hat ein paar Änderungen wegen der elektrischen Verhältnisse beim ARM

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Sun, 24 Feb 2013 20:20:44 UT
* Message-ID: <61274078-60646188$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Nabend,

danke für den Schaltplan - ist so ziemlich derselbe Aufbau wie bei mir - habe mich eben exakt an die Bauteile aus der TTL-Anleitung gehalten. Veränderte elektrische Verhältnisse habe ich wegen dem USB<>Seriell-Wandler ja keine. Übersteuere ich evtl. die Empfangsdiode? Mal paar cm Abstand nehmen?

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Mon, 25 Feb 2013 10:23:14 UT
* Message-ID: <61274078-60660232$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi Haggy, 

I have the same project, but had no time lately to works on it.  AT least my optolink is working. 
For the emitter, which was more difficult for me to set up, I have the same design as vheat, with some differences (that do not pretend to be a better design, as I am a beginner).
I   do not have R4
D1 = SFH486 IR led 
T2 = 2N3906 transistor 
R1= 10 kohm 
R3 = 77ohm 
 I use a TTL to USB card to connect to the computer 
(a foca card from iteadstudio
http://imall.iteadstudio.com/im120525005.html)
and everything is set at 3.3V as I intend next to use an xbee arduino board for communication from the basement to main floor. 

as my leds are independent of the board, with long wires, I could easily verify that there is a good communication between emitter and receiver. 

Hope this helps. 

Jacques

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Mon, 25 Feb 2013 13:16:01 UT
* Message-ID: <61274078-60661462$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo,
habe auch mit den hier im Forum dargestellten Schaltungen so meine Probleme gehabt (insb. Empfängerseite). Irgendwie hat's nie richtig geklappt. Habe dann nochmals "auf der grünen Wiese" angefangen und nun funktionierts problemlos (an einem USB-Ser-Wandler). 
Senderseitig verwende ich (bei 5V!) einen BC547 als IR-LED-Treiber, als IR-LED eine SFH485 mit 1kOhm Vorwiderstand. Die LED sitzt unmittelbar auf der Vito-Empfängerdiode, abgeschirmt von Fremdlichteinflüssen in einem Stück schwarzem Moosgummi.

Gruß mimax

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Mon, 25 Feb 2013 22:07:47 UT
* Message-ID: <61274078-60677694$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hello Jacques,

thanks for your suggestions. It seems our circuits don't differ that much. I have 330 Ohms on R3, but lately switched to about 160 Ohms in order to drive a bit more current through the ir transmitter (as vheat suggested above). However it still does not work as expected. On the receiver side i can see periodic 0x05 "pings", but my Vitotronic does not react to any codes sent by vcontrold nor voIdent. I verified the transmitter works (by watching it through a digital camera). Still somewhat clueless about that.

My exact parts are:

- D1: SFH 487-2 IR transmitter LED
- T2: 2N 3906
- R1: 10 kOhms
- R3: was 330 Ohms, now 165 Ohms - no change
- no R4
- USB<>TTL-converter from eBay:
http://www.ebay.de/itm/USB-To-RS232-TTL-PL2303HX-Auto-Converter-Module-Converter-Adapter-5V-3-3V-Output-/350568364250?pt=LH_DefaultDomain_0&hash=item519f80b8da#ht_6470wt_1220

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Mon, 25 Feb 2013 22:10:21 UT
* Message-ID: <61274078-60677768$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Oh i forgot: i'm using the 3V3 Pin on the converter to power the circuit.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Mon, 25 Feb 2013 22:18:15 UT
* Message-ID: <61274078-60678064$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi mimax, sicher dass du einen BC547 auf der Senderseite hast? Das wäre ein NPN - kann eigentlich nicht funktionieren. Meinst sicher den 3906, den auch Jacques und alle anderen (inkl. mir) verwenden. Bei 5V Betriebsspannung hast du dann ja nur noch rund 3,5mA Strom durch die LED - vll. ist das die Lösung, dass die 10mA immer noch zu viel sind?!

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Tue, 26 Feb 2013 08:22:12 UT
* Message-ID: <61274078-60690150$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo Haggy,
ja ich bin sicher... allerdings - das hatte ich dummerweise vergessen zu schreiben - verwende ich einen  74LS04 als Inverter noch zwischen USB-Interface und Dioden. Auf der Empfängerseite invertiere ich z.B. zweimal. Weshalb das ganze? Ich hatte Anfangs immer Probleme/Zweifel wegen der Polaritäten bzw. Hi/Lo und ich habe dadurch saubere TTL-Pegel.

Aber hast gut aufgepasst!  ;-)

Wie gesagt, sie Sende-LED sitzt wirklich direkt ohne Abstand vor der Vito-Schnittstelle, abgeschirmt von Fremdlicht etc. Und tut zuverlässig.

Probiers einfach mal. Ist vielleicht ein Sensibelchen. ;-)

Gruß mimax

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Tue, 26 Feb 2013 19:55:39 UT
* Message-ID: <61274078-60705840$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hi Haggy,
The first thing to test is putting your  emitter in front of your receiver and test that when you type something on the terminal, it is echoed by the receiver. If this does not work, I do not think that you can go further. The problem may be your USB-TTL converter. I think I tried with a similar one before, and it did not work,  but I will have to try again to be sure, as I did so many tries that 
it may have been another reason. I will try tomorrow. My memories is that all the TX and RX are in 5V and only the 3.3 output is in 3.3 V. 
This may be the problem. You should verify.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: maza_survey <userid-38423536@openv.wikispaces>
* Date: Wed, 27 Feb 2013 12:34:05 UT
* Message-ID: <61274078-60724952$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Contrary to what I said, I have tested again my old TTL-Usb card. All the pins are at 3.3 V, and not 5V as what I said. Only The 5V pin is at 5V.
In fact, this board (based on CP2102) works also with the optolink, but the Rx and Tx pins need to be inverted with respect to the foca TTL-USB card (based on FT232RL).

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Sat,  2 Mar 2013 10:36:22 UT
* Message-ID: <61274078-60813650$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Sorry für Englisch, aber maza_survey soll auch was mitbekommen :):

SUCCESS!

After exchanging R3 to a ~900 Ohms resistor and fiddling a bit with the position of the optolink in front of the communication LEDs, i got a positive result from voIdent: Device ID 2049, ExtID 020B - it says "unknown", but that doesn't matter.
vcontrold was able to communicate with my Vitotronic. However, connection is not that stable - 2 out of 10 trials fail with a server error. I think i'll follow mimax' suggestion and put the LEDs separate from the circuit directly in front of the LEDs on the Vitotronic, putting them in some kind of tube or foam to minimize daylight influence. 

Thanks for all your suggestions, it helped A LOT!

/ralf.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Sat,  2 Mar 2013 13:09:28 UT
* Message-ID: <61274078-60814154$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

Hallo Haggy,

gratuliere, willkommen auf der Seite des Lichts.
Bezüglich des Anbringens der Dioden: Ja, ich habe mich recht früh dazu entschieden, die eigentliche Schaltung von den Dioden zu entkoppeln, da ich in der Experimentierphase ständig an den Bauteilen rumpfriemeln musste (und das Anfangs auf einem Breadboard geschah).
Seit anderthalb Jahren hängt daher am Bedienteil der Vito ein Provisorium, aber nichts hält länger...
[image]

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Reaktion auf gesendete Daten
* Wiki-Page: home
* From: Haggy <userid-39025898@openv.wikispaces>
* Date: Sat,  2 Mar 2013 19:19:40 UT
* Message-ID: <61274078-60816678$openv.wikispaces>
* In-Reply-To: <61274078-59811884$openv.wikispaces>

:) Tesa hält die Welt zusammen. Bei mir Malerkrepp was noch über war. Danke!