openv/openv #120: V200KW2 setParty* nicht möglich

WikiSpaxe, 2017-10-26:
* Subject: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tka11 <userid-38985478@openv.wikispaces>
* Date: Fri,  1 Feb 2013 17:04:24 UT
* Message-ID: <61359168-60104498$openv.wikispaces>

Hallo!

Ich nutze vconrold mit einer V200KW2. Werte auslesen ist kein Problem, Soll-Temperaturen können auch geschrieben werden.

Leider kann ich den Partymodus nicht einschalten. Returncode ist zwar "OK" aber die Steuerung übernimmt es nicht.

Ideen?

Danke!

vg
tka

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: probono <userid-33176236@openv.wikispaces>
* Date: Sat,  2 Feb 2013 14:32:11 UT
* Message-ID: <61359168-60123614$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Bei mir funktioniert es...

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tka11 <userid-38985478@openv.wikispaces>
* Date: Sun,  3 Feb 2013 02:09:09 UT
* Message-ID: <61359168-60130762$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Wie sieht bei dir der Befehl dazu aus? Hast du die "Originale" vito.xml?

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Thu, 21 Feb 2013 16:46:51 UT
* Message-ID: <61359168-60587786$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Hi
I don't speak German, so please let me write in English.
I have the exact problem. I can't "switch on" Party or Economic mode with "setBetriebPartyM1 1" or "setBetriebSparM1". However if I enable the economic or party mode on vitotronic 200A then I can disable them with "setBetriebPartyM1 0" or "setBetriebSparM1 0" and re-enable them ! This very interesting. It looks like something is blocking the setting. Can anyone help ?

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Thu, 21 Feb 2013 21:28:53 UT
* Message-ID: <61359168-60595838$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Hello

Same here (VScotHO1, ID=20CB, Protocol=KW2).

Regards,
Tomaz

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Thu, 21 Feb 2013 22:11:22 UT
* Message-ID: <61359168-60596880$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Yup, I have the same VScotH01 boiler.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat, 23 Feb 2013 18:24:26 UT
* Message-ID: <61359168-60633350$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I have switched on my V200KW2 test control and I have no problems to set the Party mode for A1M1 and for M2.

Perhaps we first must set the Party-Temperature (Address 2308 (A1M1), 3308 (M2))?

The commands:
set Party Temperatur = 21°C (A1M1)
01 F4 23 08 01 15
read Party Mode (A1M1)
01 F7 23 03 01
activate Party Mode (A1M1)
01 F4 23 03 01 01
deactivate Party Mode (A1M1)
01 F4 23 03 01 00

The control answers 00 for all set-commands (set party temp, activating and deactivating the Party Mode

For M2 use Addresses 3303, 3308  instead of 2303, 2308

I have uploaded Version 1.7 of VitoTest (Windows) with the option, to send commands synchronized (send after receiving 05) - I did the above testing with this version.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat, 23 Feb 2013 18:46:54 UT
* Message-ID: <61359168-60633496$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Sorry, update of previous posting!
it works only with M2 and not as described with A1M1 ;-(
If I set M2 to Party Mode I can read immediately the status 01 and on the control the Party-LED switches on within about 30 seconds.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Sat, 23 Feb 2013 22:24:08 UT
* Message-ID: <61359168-60635542$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Strange, when I used sugested sequence of commands it worked! Commands: setTempPartyM2 21, getBetriebPartyM2, setBetriebPartyM2 1. I also enabled debug mode and waited few seconds between sending commands.
I will try it again tomorrow to confirm it is working but looks good. 

Thanks!

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Sat, 23 Feb 2013 22:29:07 UT
* Message-ID: <61359168-60635582$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Nope, still not working for me. I have only M1, and setting Partytemp before enabling it doesn't help.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Sat, 23 Feb 2013 22:32:03 UT
* Message-ID: <61359168-60635610$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

As I said before it is possible to set PARTY/ECONO, but before you do it, you have to enable or disable it with vitotronic button. I have no clue why is this happening.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Sun, 24 Feb 2013 23:29:36 UT
* Message-ID: <61359168-60649632$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Sorry, it's not working any more today. I think that yesterday worked because I tried it just after (5 minutes) the heating went to reduction mode.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Mon, 25 Feb 2013 07:26:27 UT
* Message-ID: <61359168-60659126$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I think this is somehow related to "reduction mode". If i press Party button on vitotronic 200A I can enable or disable  Party mode with vcontrold. But after 12 hours, when the program sets the heating to reduction mode I can no longer disable/enable party mode.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Mon, 25 Feb 2013 22:47:59 UT
* Message-ID: <61359168-60678978$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I'm thinking of using an alternative with setTimerM2XX 00:00 24:00 where XX is the current day initial. The down side is that the heating regime must be reset sometime later. So the proper way would be to store current regime, set "party" mode and set some kind of timer to reset back to saved regime. I will try to make this work with shell script and cron job and will post it here if it works.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 26 Feb 2013 08:48:46 UT
* Message-ID: <61359168-60690300$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Last evening I tested setting party mode on my system here. It is a VDensHO1 ID=20C2 with a remote for M2.
Seting setTempPartyM2 and then setBetriebPartyM2 did not activate party mode. I need to test enabling/disabling it after pressing party mode button on the remote.
What I know about two facts.
1. The remote is hardcoded into the system, that means I can't control temp from the main system panel
2. The time for party mode is set in the main system. In my case 8 hours. After that it goes back either to reduction mode in night or normal day mode.

I assume, that the fact that a remote is in the system generally disables some settings via the optolink.

Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: goumb <userid-39384814@openv.wikispaces>
* Date: Tue, 26 Feb 2013 09:34:38 UT
* Message-ID: <61359168-60690550$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

So you are suggesting that my remote "vitotronic 200A" is somehow blocking the optolink transmission ?
Is there any way to fix it ?

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Tue, 26 Feb 2013 21:20:09 UT
* Message-ID: <61359168-60709110$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I have no remote at all!

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Tue, 26 Feb 2013 21:28:11 UT
* Message-ID: <61359168-60709490$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

@goumb
I can only say that as long as the remote is online I can't change anything for M2 on the main system, for example room temperature. 
If I activate part mode is stays on for 8 hours or until the normal day mode kicks in at 6:00 in the morning.

I'm sure there is no physical blocking of the optolink interface, but it might be a programatic one by Viessmann.

Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Wed, 27 Feb 2013 01:01:40 UT
* Message-ID: <61359168-60715746$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Just for your information:
within the Viessmann software "Vitosoft 200" it is not possible to control the party mode nor the temperatur for party mode. Only the time limit can be set.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Fri,  4 Oct 2013 10:19:40 UT
* Message-ID: <61359168-65750642$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Did anyone have any luck with this issue, i.e. by using a different address?

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: tomazmla <userid-28366149@openv.wikispaces>
* Date: Wed, 16 Oct 2013 23:43:59 UT
* Message-ID: <61359168-66155212$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Hi

As planned I've achieved an alternative method for setting party mode. I've written two python scripts for setting current day timer to 00 - 24 and resetting it back to normal after predefined period. The first script (PartySet.py) reads timer for current day and writes it to file for later resetting. The second script (PartyJob.py) must be run periodically (every 5 minutes for example) and checks if the party mode should be turned off. Both files and a template can be found here: https://app.box.com/s/bvww6e4ktvbvkoyfuw7r. Read file comments for more info.
The scripts are not perfect (version 1.0) but they work for me. Any suggestions appreciated.

Regards,
Tomaz

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Tue,  2 Sep 2014 13:38:51 UT
* Message-ID: <61359168-74646392$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Incidentally, the now available Viessmann KNX gateway datapoint doc http://www.viessmann.de/content/dam/internet-global/software-daten/vitogate_200_eib-daten2/vt300kmwb_en.pdf lists 0x2330 as the address for enabling Party Mode, but that doesn't work for my controller (0x20cb) either (error when writing)

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: Alyoscha <userid-48133008@openv.wikispaces>
* Date: Fri,  5 Sep 2014 21:15:01 UT
* Message-ID: <61359168-74740760$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

@owagner42: 0x2330 for party mode works perfectly. Thanks for the hint to the pdf. I own a VScotHO1 (device id 0x20cb). Best regards - Alyoscha.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Wed, 29 Oct 2014 20:12:25 UT
* Message-ID: <61359168-76262698$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I just retried and writing to 0x2330 doesn't work for me. I do get an error, and the mode does not change:

Received error reply:  03 02 23 00 01 01 30

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Wed, 29 Oct 2014 20:20:10 UT
* Message-ID: <61359168-76262906$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Trying to remotely set economy mode with address 0x2330 also triggers the same error. Writing the temperatures (e.g. 0x2307) works, however.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Thu, 30 Oct 2014 14:07:08 UT
* Message-ID: <61359168-76283450$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I own a 20CB as well, and I can verify 0x2330 for party and 0x2331 for economy mode work both with values 0 and 1. You might receive an error though, if your heating mode is not appropriate, e.g. I can only activate economy mode during standard heating schedule, and party and economy modes cannot be used at the same time. I am using Viessdata 2.0 to communicate to the heater.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: Martin_24 <userid-51960962@openv.wikispaces>
* Date: Sat, 22 Nov 2014 21:36:22 UT
* Message-ID: <61359168-77130982$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I have a 20CB, too. But 0x2330 does not work. 0x2303 can be read and corresponds to the party mode setting made directly on the heater. But 0x2303 cannot be written. 
If you look into http://www.viessmann.de/content/dam/internet-global/software-daten/vitogate_200_eib-daten2/vt200ho1a_en.pdf you will find that 0x2303 is marked as "read". 
I am not sure if this is the right file for my Vitotronic, but it seems there are Vitotronics out there that cannot be set to party mode via optolink. 
I have no clue why anyone would do that. If not setting party mode remotely, what else?

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: probono <userid-33176236@openv.wikispaces>
* Date: Sat,  3 Jan 2015 06:27:54 UT
* Message-ID: <61359168-78187488$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I can successfully activate and deactivate the party mode on the Vitotronic 200 KW 2 using vcontrold. I can't remember to have done anything special.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Wed, 11 Feb 2015 14:18:55 UT
* Message-ID: <61359168-79282454$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

@kehrenberger. Interesting that it works with Viessdata. 

I wonder whether my own implementation of the Viessmann protocol, based on information here, is somehow broken, or the reverse-engineered information is incomplete and the single-bit registers require a different write transaction sequence. I do use byte writes to register 0x2323 to switch operation modes between Off, WW and WW+H, which works correctly, though.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Fri, 13 Feb 2015 10:09:19 UT
* Message-ID: <61359168-79351542$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

Yes, I am using single byte writes to 0x2330, switching option on writing a 1, off with a 0. Obviously this also works with other software, so it might be a limitation of your heater, not offering Party-Mode. 
Your error reply above lets me assume you used the wrong address. Which protocol do you use to write? I am using Protocol 300, and the sequence looks like this, all hex: 41 06 00 02 23 30 01 01 5D.

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: owagner42 <userid-43519386@openv.wikispaces>
* Date: Fri, 13 Feb 2015 10:39:11 UT
* Message-ID: <61359168-79351830$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

I'm also using 300. I just double checked the data which is exchanged, and what's send is identical to what you write.

Sent 41 06 00 02 23 30 01 01 5d
Received error reply:  03 02 23 30 01 01 60

WikiSpaxe, 2017-10-26:
* Subject: Re: V200KW2 setParty* nicht möglich
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Fri, 13 Feb 2015 15:28:06 UT
* Message-ID: <61359168-79355846$openv.wikispaces>
* In-Reply-To: <61359168-60104498$openv.wikispaces>

well, you are not the only person with this problem, see Martin_24 post above. Seems that some versions of 20CB accept writing party and economy mode via Optolink, some don´t. Maybe these switches should only be enabled once a Vitotrol is connected, as usually these switches are controlled with this control device, the two functions are not available directly at the Vitotronic management panel. Maybe this restriction has been removed in certain firmware versions. I do not have Vitotrol connected, still works to set party/economy via Optolink. To be honest there is not a lot of use, because party not only increases target room temperature, it also enables water circulation pump, which is kind of stupid. Enabling economy does simply reduce room target temperature, which can also be achieved by accessing room target temperatur at 0x2306 directly.

morbusp, 2020-04-17:
Hi! This thread has grown old now but this may be of interest for someone else: yesterday I did not start party mode on the front panel of my 20C2 and I wasn’t able to activate it by 0x2303 with vcontrold (after using the front panel 0x2303 works as stated previously in this thread but not without pushing the button at least once). Setting temperature for party mode by 0x2308 before didn’t change behavior. I then implemented party mode duration (0x27F2) in my vito.xml. Setting duration to “0” and sending 0x2303 afterwards successfully started party mode! This has be proven today and has to be tested with/without setting temperature for party mode in this context.

hirni99, 2022-02-17:
setting 0x27F2 before activating 0x2303 works for me too!
I was experiencing the same problem as described. I couldn't activate PartyMode.
After enabling it by menu, it worked until the next day.

When i now set 0x27F2, i can enable it and it is really written!

Thank you very much for this hint!

jankornblum, 2023-12-27:
Works for me, too. Thanks a lot! What would be the equivalent address for M2?

sigma415, 2024-12-13:
@morbusp: Thank you very much for your ingenious hint to set `0x27F2=0`. You saved my day :)
For days I tried a lot of other solutions to enable party mode for my VScot1HO 20CB heating with my FHEM/VCONTROL300.pm, but enabling works only for some hours (obviously the 8 hours set by parameter 0x27F2).
Now `0x2303=1` and `0x2303=0` works perfect the whole day :)