openv/openv #127: Vitocal 200-S Datenpunkte bestimmen

WikiSpaxe, 2017-10-26:
* Subject: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: elasc <userid-40155666@openv.wikispaces>
* Date: Thu,  4 Apr 2013 21:56:25 UT
* Message-ID: <61813420-61608454$openv.wikispaces>

Meine Anfrage richtet sich an das Vorgehen, wie ich die Datenpunkte bzw. Adressen für die Vitocal 200-S Anlage am besten rausfinden kann. Das Windows Programm vcontrol zeigt mir die ID 204A - UNKNOWN an. Anschließend habe ich den Adapter an meinen rasPi verbunden und vcontrold installiert. Ich habe die Standard Konfigs aus den Sources kopiert und in der vcontrold.xml das Device
<device ID="204A" name="V200-S" protocol="KW2"/>
hinzugefügt.
 
Anschließend habe ich die folgenden Ausgaben bekommen:
 
root@raspberrypi:~# vclient -h 127.0.0.1:3002 -c getDevType
getDevType:
UNKNOWN
root@raspberrypi:~# vclient -h 127.0.0.1:3002 -c getTempA
getTempA:
25.600000 Grad Celsius
root@raspberrypi:~# vclient -h 127.0.0.1:3002 -c getTempVListM1
SRV ERR: command unknown
getTempVListM1:
server error
root@raspberrypi:~# vclient -h 127.0.0.1:3002 -c getTempVLsollM2
getTempVLsollM2:
-0.100000 Grad Celsius
 
Die Aussentemperatur kann nicht stimmen, wir haben ja noch leider Winter - zumindest wenn ich aufs Thermometer schaue.
 
Wie kann ich nun die verschiedenen Adressen am sinnvollsten ermitteln?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Thu,  4 Apr 2013 22:25:19 UT
* Message-ID: <61813420-61609314$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo elasc

Ich würde mich an die Adressen der anderen Vitocal Steuerungen orientieren (2046, 2047, 2049 sind unter "Adressen" zu finden und haben ähnliche Adressen.).
Z.B. ist da die Aussentemperatur an 0x0101 - wenn du dein XML nicht angepasst hasst, steht vermutlich noch 0x0800 drin.

Ab morgen bin ich in einem Land mit 22 bis 30 °C Aussentemperatur - aber ohne Internet. Also werde ich in diesem Thread nicht so schnell wieder melden können ;-( 
Gruss
Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: elasc <userid-40155666@openv.wikispaces>
* Date: Tue,  9 Apr 2013 10:07:44 UT
* Message-ID: <61813420-61704516$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo Hanspeter,
Danke für deinen Tipp, der war goldrichtig. Habe nun einige Punkte bestimmen können. Werde meine Ergebnisse später teilen.
Alle bekannten Punkte für Wärmepumpen sind read-only. Gibt es einen Tipp wie man write Punkte herausfinden kann?

Gruß
Lars

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: SliderSD <userid-46534044@openv.wikispaces>
* Date: Sat,  4 Jan 2014 13:45:42 UT
* Message-ID: <61813420-68586532$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo elasc,

ich besitze auch eine Vitocal 200-S, und deshalb würde es mich sehr interessieren, welche Datenpunkte du bereits bestimmen konntest. Ich bin bisher über ein paar wenige Punkte wie z.B. die Außentemperatur nicht hinausgekommen. Im speziellen habe ich die Vorlauftemperaturen noch nicht bekommen, sowohl Ist als auch Soll.

Grüße
Stefan

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: elasc <userid-40155666@openv.wikispaces>
* Date: Tue, 25 Feb 2014 11:48:30 UT
* Message-ID: <61813420-69974050$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo Stefan,
ich habe die Adressen die ich ermittelt (und einigermaßen verifiziert) habe in die Tabelle unter "Adressen" eingetragen. An die Soll Werte für die normale Raumtemperatur und die reduzierte Raumtemperatur komme ich auch nicht dran.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: StefanJoe <userid-52212324@openv.wikispaces>
* Date: Thu, 18 Dec 2014 01:13:49 UT
* Message-ID: <61813420-77978400$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo Lars,
welche Revision von der WO1C hast du denn ? Mit der Rev. C scheinen nciht alle zu stimmen.
Soll-Werte Raumtemperatur ist bei mir 0x2000 und reduziert 0x2001 (jeweils unit UT, len 2)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: StefanJoe <userid-52212324@openv.wikispaces>
* Date: Thu, 18 Dec 2014 01:23:48 UT
* Message-ID: <61813420-77978608$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Was ich übrigens verzweifelt suche ist Laufzeit oder Status Verdichter ... wenn ich auf die 0x5800 gehe, dann schmeisst der mich irgendwie komplett raus ("connection closed by foreign host") und ich muss den vcontrol resetten...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: der_vitti <userid-45889722@openv.wikispaces>
* Date: Sat, 20 Dec 2014 22:03:00 UT
* Message-ID: <61813420-78040436$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo Lars, 
RaumtempNormal sollte bei der WO1C für HK1 0x2000 sein, RaumtempReduziert dann 0x2001. Für HK2 0x3000 und 0x3001.
Die SOLL Leistung Verdichter(1) liegt evtl bei 0x5030, das Verdichterschütz bei 0x0480.
Gruß
der_vitti

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: PyroX <userid-52254854@openv.wikispaces>
* Date: Mon, 22 Dec 2014 16:36:19 UT
* Message-ID: <61813420-78063574$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hallo in die Runde,

die Adresse 0x0480 für den Verdichter scheint bei meiner Anlage (222-S mit WO1C) leider ebenfalls nicht zu stimmen:

vctrld>Verdichter1(Schuetz)
DEBUG:Mon Dec 22 16:31:01 2014 : Befehl: Verdichter1(Schuetz)
DEBUG:Mon Dec 22 16:31:01 2014 : Process 2541 tries to aquire lock
DEBUG:Mon Dec 22 16:31:01 2014 : Process 2541 got lock
DEBUG:Mon Dec 22 16:31:01 2014 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
DEBUG:Mon Dec 22 16:31:01 2014 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
DEBUG:Mon Dec 22 16:31:01 2014 : >SEND: 04
DEBUG:Mon Dec 22 16:31:03 2014 : <RECV: len=1 00 (1990.0 ms)
DEBUG:Mon Dec 22 16:31:03 2014 : <RECV: received 00
DEBUG:Mon Dec 22 16:31:03 2014 : >FRAMER: unexpected data 00k
DEBUG:Mon Dec 22 16:31:03 2014 : >SEND: 04
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 05 (2000.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 05
DEBUG:Mon Dec 22 16:31:05 2014 : >FRAMER: closed
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 16
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 00
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 00
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 06 (10.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 06
DEBUG:Mon Dec 22 16:31:05 2014 : >FRAMER: opened
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 41
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 05
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 00
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 01
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 04
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 80
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 01
DEBUG:Mon Dec 22 16:31:05 2014 : >SEND: 8B
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 06 (20.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 06
DEBUG:Mon Dec 22 16:31:05 2014 : >FRAMER: addr was still active FE06
DEBUG:Mon Dec 22 16:31:05 2014 : >FRAMER: Command send
DEBUG:Mon Dec 22 16:31:05 2014 : >FRAMER: no preset result
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 41 (10.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 41
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 06 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 06
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 04 (10.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 80 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 02 (0.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: len=1 8F (10.0 ms)
DEBUG:Mon Dec 22 16:31:05 2014 : <RECV: received 01 01 04 80 01 02 8F
DEBUG:Mon Dec 22 16:31:05 2014 : 02 -> NOT OK
DEBUG:Mon Dec 22 16:31:05 2014 : NOT OK
NOT OK

Laut Viessmann Doku (http://www.viessmann.de/content/dam/internet-global/software-daten/Vitogate_200_Listen/vt200wo1c.pdf) sollte das ja aber die richtige Adresse zu sein.

Gibt's sonst noch kreative Vorschläge? :)

Grüße,
Christian

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: PyroX <userid-52254854@openv.wikispaces>
* Date: Mon, 22 Dec 2014 17:13:05 UT
* Message-ID: <61813420-78064038$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Nachtrag:

Die Adresse 0x0400 (aus http://openv.wikispaces.com/Adressen) passt leider ebenfalls nicht:

vctrld>VerdichterStatus
DEBUG:Mon Dec 22 17:12:48 2014 : Befehl: VerdichterStatus
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 41
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 05
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 00
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 01
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 04
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 00
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 01
DEBUG:Mon Dec 22 17:12:48 2014 : >SEND: 0B
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 06 (20.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: received 06
DEBUG:Mon Dec 22 17:12:48 2014 : P��7
DEBUG:Mon Dec 22 17:12:48 2014 : >FRAMER: Command send
DEBUG:Mon Dec 22 17:12:48 2014 : >FRAMER: no preset result
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 41 (20.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: received 41
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 06 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: received 06
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 03 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 04 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 00 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 01 (10.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: len=1 10 (0.0 ms)
DEBUG:Mon Dec 22 17:12:48 2014 : <RECV: received 03 01 04 00 01 01 10
DEBUG:Mon Dec 22 17:12:48 2014 : >FRAMER: ERROR address 0400 code 1
DEBUG:Mon Dec 22 17:12:48 2014 : Fehler recv, Abbruch
DEBUG:Mon Dec 22 17:12:48 2014 : Fehler beim ausfuehren von VerdichterStatus
ERR: P��7
>FRAMER: Command send
>FRAMER: ERROR address 0400 code 1
Fehler recv, Abbruch
Fehler beim ausfuehren von VerdichterStatus

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: StefanJoe <userid-52212324@openv.wikispaces>
* Date: Tue, 30 Dec 2014 23:09:25 UT
* Message-ID: <61813420-78164170$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Ich hab hier mal "meine" vito.xml angehängt, würde mich mal interessieren wie das Profil über die Zeit der verschiedenen Temperaturen und Lasten bei euch aussieht ...
<vito>
<devices>
 <device ID="204D" name="V200-S" protocol="P300"/>
</devices>
<commands>

<command name="Outdoor_Fanspeed" protocmd="getaddr">
 <addr>1A52</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>Outdoor Fanspeed</description>
</command>

<command name="Compressor_Freq" protocmd="getaddr">
 <addr>1A54</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>Compressor Frequency</description>
</command>

<command name="Status_Fanspeed" protocmd="getaddr">
 <addr>1A53</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>Geschwindigkeit Luefter</description>
</command>

<command name="OAT_Temperature" protocmd="getaddr">
 <addr>1A5C</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>OAT Temperature</description>
</command>

<command name="OCT_Temperature" protocmd="getaddr">
 <addr>1A5D</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>OCT Temperature</description>
</command>

<command name="CCT_Temperature" protocmd="getaddr">
 <addr>1A5E</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>CCT Temperature</description>
</command>

<command name="HST_Temperature" protocmd="getaddr">
 <addr>1A5F</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>HST Temperature</description>
</command>

<command name="OMT_Temperature" protocmd="getaddr">
 <addr>1A60</addr>
 <len>1</len>
 <unit>CO</unit>
 <description>OMT Temperature</description>
</command>

[... 173 lines cut ...]
</command>

<command name="HeizkennlinieSteig" protocmd="getaddr">
 <addr>2007</addr>
 <len>2</len>
 <unit>UT</unit>
 <description>Heizkennlinie Niveau</description>
</command>

<command name="LZVerdichter_" protocmd="getaddr">
 <addr>5005</addr>
 <len>4</len>
 <unit>CO</unit>
 <description>WPR_Verdichter1_laufzeit_wp</description>
</command>

<command name="Type" protocmd="getaddr">
 <addr>00F8</addr>
 <len>2</len>
 <unit>UT</unit>
 <description>Type</description>
</command>

</commands>
</vito>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: PyroX <userid-52254854@openv.wikispaces>
* Date: Wed,  7 Jan 2015 21:08:50 UT
* Message-ID: <61813420-78277286$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Hi StefanJoe,

herzlichen Dank für deine vito.xml, die meisten Adressen funktionieren auch bei meiner Vitocal 222-S.
Ich hab Mal einige davon in meine Datenerfassung eingebaut und schaue mal, wie die Werte bei mir aussehen.
Ein paar Dinge sind mir momentan noch unklar:

1.) In welcher Einheit ist der Wert für "Outdoor_Fanspeed" angegeben? Ich erhalte meist einstellige Werte wie z. B. 2, scheint also mal nicht nicht in Umdrehungen pro Minute angegeben zu sein.

2.) Dito bei "Status_Fanspeed". Hier erhalte ich meist zweistellige Werte.. könnte also in Hertz angegeben sein.

3.) Die Adressen für die einzelnen Laufzeite passen auf meine Anlage entweder nicht, oder sind nicht in Stunden angegeben. Für "LZHeizstab1" erhalte ich beispielsweise einen Wert von 433994, also über 18000 Tage, was nicht wirklich sein kann (die Anlage wurde 05/2013 in Betrieb genommen) ;)

Gruß,
Christian

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: PyroX <userid-52254854@openv.wikispaces>
* Date: Wed,  7 Jan 2015 21:10:20 UT
* Message-ID: <61813420-78277318$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Nachtrag:

Wo hast du eigentlich deine Adressen samt Zuordnung her bzw. wie hast du diese ermittelt?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: Morg42 <userid-54514477@openv.wikispaces>
* Date: Thu, 22 Oct 2015 21:38:31 UT
* Message-ID: <61813420-84989797$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Nur falls hier noch jemand liest... :)

habe die folgenden Daten (aus dem PDF und mit Probieren) ermitteln können:

http://openv.wikispaces.com/file/detail/V200WO1C.xlsx
http://openv.wikispaces.com/file/detail/vito_204d_v2.xml

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-S Datenpunkte bestimmen
* Wiki-Page: home
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Mon, 17 Oct 2016 11:36:06 UT
* Message-ID: <61813420-91376458$openv.wikispaces>
* In-Reply-To: <61813420-61608454$openv.wikispaces>

Servus Zusammen, 

liest hier noch jemand?!? ;-)

Kämpfe an der gleichen Umgebung, kriege aber als Rückgabewerte immer -77.199997 Degrees?

Mach ich da grundsätzlich was falsch, oder ist der Windows Deamon zum Testen einfach sch...?!?

Vielleicht hat ja jemand einen kleinen Tipp für mich!

Danke und Grüße
Leo

besimo1, 2022-03-28:
Hallo Leo,

letzter Post ist schon fast 5 Jahre her. Bist du noch aktiv ?
Ich habe eine Vitocal 200-S D08 BJ 2018 mit Vitotronic 200 (W01C).

Kannst du mir deine neueste vito.xml und vcontrold.xml schicken ?

Wie klappt es bei dir mit den Set-Anweisungen ?

Hast du auch Adressen für den Wochentimer ?

Welches Programm nutzt du aktuell - z.B. ioBroker - Viessmann-Adapter (fuer vcontrold) Datenbank influx ?

LG
Bernd

m1k3f15h, 2022-09-27:
Hy Leute,

Ich häng mich da mal an.. Hab auch eine Vitocal 200-S.
Nutz die zur Zeit mit der API.. jedoch gefällt mir das nicht so ganz (Schlechte Auflösung, Cloud, Ausfälle,...)
Läuft das ganze bei euch stabil?
Hat jemand den Service in Docker am laufen?
Wie siehts mit den Datenpunkten aus? (Auch den SET)

mfg
Mike

msglueck, 2023-02-26:
@m1k3f15h war ebenfalls auf der Suche und habe jetzt mal mit den Konfigs von hier gestartet. Vieles scheint zu stimmen, aber nicht alles (spiele aber gerade 30min damit rum): https://www.schlocke.de/viessmann-heizungswerte-mit-openv-auslesen-und-mit-volkszaehler-darstellen/

m1k3f15h, 2023-05-23:
@msglueck 
Wie läuft dein Projekt?

mfg
Mike

m1k3f15h, 2023-05-30:
@msglueck das sieht ja echt cool aus :-) Find das echt geil.
Hast du eigentlich nur vcontrold laufen, oder das VitoConnect auch noch?

mfg
Mike

msglueck, 2023-05-30:
Vitocontrold + Shellscript / vclient + JSON Exporter -> Prometheus TSDB -> Grafana für die Visualisierung. 😄 Da es noch nichts zu Prometheus gab, hatte ich vor eine Anleitung dazu schreiben. Kann dann das komplette Setup beschreiben und bspw. das Dashboard sharen.

m1k3f15h, 2023-06-01:
Ah okay :)
Ja mir schwebt ähnliches vor.. aber ich würd dann doch gerne den Vitoconnect auch noch laufen lassen...
Grafana is echt cool.. Ich hab aktuell InfluxDB2 am laufen :)

mfg
Mike

Jumbo125, 2023-08-27:
hallo zusammen
habe ebenfalls eine vitocall 200S und ich schaffe es nicht gewisse Datenpunkte zu bekommen....
bspw. Raumtemperatur, Betriebsmodus,
wird gerade aktiv WW geheizt.
Verdichter 
usw.

habt ihr ev. etwas für mich

stehlifn, 2023-10-15:
Hallo zusammen,

ich kann mit meiner Vitocal 200-S unter anderem die folgenden Datenpunkte lesen und schreiben:

* Betriebsmodus
* WW Soll Temperatur
* WW Soll Temperatur 2
* Raumtemperatur Soll Normal
* Raumtemperatur reduziert
* Raumtemperatur Party
* WW einmal
* Heizstab Stufe 1, 2, 3
* Heizstab für WW 
* Heizstab für Heizen
* Verdichter für WW, Heizen
* weitere

vcontrold.xml und vito.xml habe ich über folgenden Link besorgt:
https://www.schlocke.de/viessmann-heizungswerte-mit-openv-auslesen-und-mit-volkszaehler-darstellen/

Anschließend noch mit den beiden Konfigurationsdateien aus dem folgenden Beitrag ergänzt:
https://forum.iobroker.net/topic/61912/viessmann-adapter-mit-vcontrol/106

Alle von mir verwendeten Commands aus dem Beitrag des iobroker Forums haben dabei auf Anhieb funktioniert.

Viele Grüße

Florian

stehlifn, 2023-10-16:
Der Durchlauferhitzer kann in 3 Stufen reguliert werden - jede fügt 3 kW hinzu. Das zugehörige Command ist das Folgende (siehe Konfigurationsfiles aus dem iobroker Forum):

`<command name="setHeater" protocmd="setaddrValue">
        <addr>7907</addr>
        <len>1</len>
        <unit>XX</unit>
        <description>Leistung Durchlauferhitzer (1=3, 2=6 oder 3=9kW)</description>
</command>
<command name="getHeater" protocmd="getaddr">
        <addr>7907</addr>
        <len>1</len>
        <unit>XX</unit>
        <description>Leistung Durchlauferhitzer (1=3, 2=6 oder 3=9kW)</description>
</command>`

"WW mit Elektro" und "Heizen mit Elektro" werden über die folgenden beiden Commands gesetzt:
`<command name="setHeaterforWW" protocmd="setaddrValue">
        <addr>6015</addr>
        <len>1</len>
        <unit>XX</unit>
        <description>Druchlauferhitzer für Warmwasser aktivieren</description>
</command>
<command name="setHeaterforRoomtemp" protocmd="setaddrValue">
        <addr>7902</addr>
        <len>1</len>
        <unit>XX</unit>
        <description>Durchlauferhitzer für Heizung aktivieren</description>
</command>`

Jumbo125, 2023-10-16:
danke dir!
bedeutet 1=3 das Stufe 1 -> 3 KW Heizleistung erzeugt? Das können ja keine 3kw Strom sein. bei stufe zwei wären das 6kw. Das geht ja nichtmal mit 16A sicherung....

Jumbo125, 2023-10-18:
> Der Durchlauferhitzer kann in 3 Stufen reguliert werden - jede fügt 3 kW hinzu. Das zugehörige Command ist das Folgende (siehe Konfigurationsfiles aus dem iobroker Forum):
>
> `<command [...]

danke dir!
bedeutet 1=3 das Stufe 1 -> 3 KW Heizleistung erzeugt? Das können ja keine 3kw Strom sein. bei stufe zwei wären das 6kw. Das geht ja nichtmal mit 16A sicherung....

msglueck, 2024-04-03:
Oh doch, das Teil (200-S D08) hat einen Anschlusswert von 11kW und ist dreiphasig angeschlossen. 3,6kW = 1ph 16A mal 3 ist man bei 10,8kW wie beim eAuto. 🤓

@stehlifn danke für den Durchlauferhitzer, der fehlte mir noch. Jetzt fehlt nur noch das Ferienprogramm 😜