openv/openv #133: Connection to Raspberry Pi for PLC-based home automation

WikiSpaxe, 2017-10-26:
* Subject: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Wed, 12 Jun 2013 17:08:35 UT
* Message-ID: <62826000-63505232$openv.wikispaces>

Hello,

Great appreciation for all that has been made available on this site! :)

I would like to connect my Vitodens 200-W to Raspberry Pi to use for my PLC-based home automation system (www.edom-plc.pl).  What soulution would you advice?

1. Using RS232 - I have no problem with building a RS232 extension for the RPi but will it work with the RS232 module described here?  Does anybody have an SMD version of RS232 optolink?  Would such a moudle require a separate power source?

2. Using USB. An SMD version of the optolink is available.  My Viessmann and the RPi are 5m apart.  Will that be a problem?  

Which module is easier to use/program on the RPi?

Thank you in advance for any hint or support!

Piotr

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: smbunn <userid-40480070@openv.wikispaces>
* Date: Fri, 14 Jun 2013 01:41:31 UT
* Message-ID: <62826000-63532898$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

USB 2.0 supports 5m cables so you should be OK.  As most people on this site will attest, the vcontrold application compiles fine on Debian LInux used by the RPi.  You could then use the Ethernet port or a WiFi dongle to get data back to your control system.
There is no "modbus" interface, you would have to write scripts to transfer data across from the RPi.

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Fri, 23 Aug 2013 09:07:18 UT
* Message-ID: <62826000-64449396$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Hello,

I am sorry for not replying immediatelly.  I have spend the time on collecting the components and on redrafting the USB board to fit into a smaller case (25x40mm).  I have soldered it all together and tested.  

I have a 5-year old Vitodens 200. The voIdent returns the following data:

Device unknown
ID 01F7
ExtID 00F8

or 

Device unknown
ID 01C7
ExtID F804

Does it meant there is a connection problem or I need to set up in the Viessman itself?

Luckly the v-controll 1.3 reads all the information, so... I feel quite successful.  Nevertheless the "Device unknown" is worrying.

Is there a way to post my board layout?  Maybe somebody finds it useful?

Best regards!

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: smbunn <userid-40480070@openv.wikispaces>
* Date: Thu, 12 Dec 2013 04:16:14 UT
* Message-ID: <62826000-68245610$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Since everything is set up in the two xml files (vito and vcontrold) you need to look in these.  In my vito.xml I have defined three types of device IDs;
devices>
	<device ID="2098" name="V200KW2" protocol="KW2"/>
	<device ID="2053" name="GWG_VBEM" protocol="GWG"/>
	<device ID="20C2" name="VDensHO1" protocol="KW2"/>
</devices>

 and I have the command "getDevType" as follows;
	<command name='getDevType' protocmd='getaddr'>
		<addr>00F8</addr>
		<len>2</len>
		<unit>DT</unit>
		<description>Get the description of the attached boiler</description>

This returns the value 2098, defining my boiler as a V200 using KW2 protocol, however i only know this because I told the software in the vcontrold.xml file that a value of 2098 meant it had to return the string "V200KW2" when a unit type DT returned a value of 2098.  Sort of circular logic, but something we understand that other sin this forum have figured out by using a data sniffer on a real Viessmann interface and seeing what it sent and received when it was talking to these boilers.
The easiest thing to do is run a telnet interface to the vconotrold daemon, just type "telnet localhost 3002" and it should probably work assuming you are on the raspberry pi and you used port 3002 in your vcontrold.xml file, look for
	<net>
		<port>3002</port>
		<allow ip='127.0.0.1'/>
		<allow ip='192.168.2.0/24'/>
	</net>

Now you are in the Telnet session you can run the following commands
pi@raspberrypi2 ~ $ telnet localhost 3002
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
vctrld>debug on
vctrld>getDevType
DEBUG:Thu Dec 12 16:12:59 2013 : Befehl: getDevType
DEBUG:Thu Dec 12 16:12:59 2013 : ClI Net: verbunden 127.0.0.1:3000 (FD:0)
DEBUG:Thu Dec 12 16:12:59 2013 : Process 11953 tries to aquire lock
DEBUG:Thu Dec 12 16:12:59 2013 : Process 11953 got lock
DEBUG:Thu Dec 12 16:12:59 2013 : >SEND: 04
DEBUG:Thu Dec 12 16:12:59 2013 : Warte auf 05
DEBUG:Thu Dec 12 16:13:01 2013 : <RECV: 05 (1460.0 ms)
DEBUG:Thu Dec 12 16:13:01 2013 : >SEND: 01
DEBUG:Thu Dec 12 16:13:01 2013 : >SEND: F7
DEBUG:Thu Dec 12 16:13:01 2013 : >SEND: 00
DEBUG:Thu Dec 12 16:13:01 2013 : >SEND: F8
DEBUG:Thu Dec 12 16:13:01 2013 : >SEND: 02
DEBUG:Thu Dec 12 16:13:01 2013 : <RECV: 20 (20.0 ms)
DEBUG:Thu Dec 12 16:13:01 2013 : <RECV: 98 (0.0 ms)
DEBUG:Thu Dec 12 16:13:01 2013 : 20 98 -> V200KW2 ID=2098 Protokoll:KW
DEBUG:Thu Dec 12 16:13:01 2013 : Process 11953 released lock
DEBUG:Thu Dec 12 16:13:01 2013 : V200KW2 ID=2098 Protokoll:KW
V200KW2 ID=2098 Protokoll:KW
vctrld>

Seeing the data come back from register 00F8 as the value '2098' confirmed to me that everything was OK.

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Tue, 31 Dec 2013 00:51:01 UT
* Message-ID: <62826000-68553796$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

@edomplc 
a really late answer ;-)

If you receive ID 01F7, ExtID 00F8 or ID 01C7, ExtID F804 then you may have an "optical short circuit" of the IR signals. 0x01 0xF7/0xC7 0x00 0xF8 are the telegrams sent to the vitotronic control.
You have to ensure that there is a Separation between Send- and Receive-IR-Diodes. Otherwise the Raspi is receiving what it was just sending before receiving any answer from the control.
Or it may be an electrical short circuit between Tx and Rx lines with the same effect...

Does your USB solution work now? 
regards from switzerland, Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Tue, 31 Dec 2013 11:49:23 UT
* Message-ID: <62826000-68555452$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Hello guys!  

Many thanks for your replies.  I have tested my board for a while.  I wanted it to be connected with RPi, which is in another room.  At the end I failed as the distance and the distortions were too high for the USB connection (the usb device was not recognized).

I want to split the board into two parts to have the USB->RS232 conversion close to the RPi and then the rest at the Viessman unit.  The boards are ready, I just need to find time for the soldering.  At the end, still, I am not sure if that approach is any better ;)

Than you very much for the details of the communication.  I will make another in-depth attempt when the boards are ready :)

All the best in the New Year!

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Mon,  3 Feb 2014 20:47:07 UT
* Message-ID: <62826000-69355288$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Hello,

I finally got my device working.  As you remember my RPi is 5m away from the Viessman and the USB connection failed.  I ended up buying a USB-RS232 converter based on FT232RL for $5 and copied a scheme from your USB adapter making a new small board looking as follows:

http://www.edom-plc.pl/wago/openV/BoardScheme.PNG
http://www.edom-plc.pl/wago/openV/Ready1.jpg

I attached the board with a double-sided tape with foam inside, which keeps the board quite firmly in place.

I followed the steps described on the Raspberry Pi page and had the following issues:

- in Point 4 "Schnittstelle prüfen", giving the vcontrold $ lsusb command returned an error.  The program said the vcontrold.xml is missing.  I found in point 5 what should be copied where and tried again.  The vconttrold $ lsusb worked... but renurned no result.  Why?

- I followed the further steps and got all the messages as described:

ls -l /dev/serial/{by-path,by-id}/* returns:
lrwxrwxrwx 1 root root 13 Jan  1  1970 /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9251FBZ-if00-port0 -> ../../vitoir0
lrwxrwxrwx 1 root root 13 Jan  1  1970 /dev/serial/by-path/platform-bcm2708_usb-usb-0:1.2:1.0-port0 -> ../../vitoir0

when I launch vcontrold -n I get "socket error:: could not open socket.  What is the problem?

- I launch telnet localhost 3002, debug on, getDevType I get the following:

DEBUG:Mon Feb  3 20:37:23 2014 : Befehl: getDevType
DEBUG:Mon Feb  3 20:37:23 2014 : Process 2336 tries to aquire lock
DEBUG:Mon Feb  3 20:37:23 2014 : Process 2336 got lock
DEBUG:Mon Feb  3 20:37:23 2014 : >FRAMER: open device /dev/vitoir0 ProtocolID 00
DEBUG:Mon Feb  3 20:37:23 2014 : konfiguriere serielle Schnittstelle /dev/vitoir0
DEBUG:Mon Feb  3 20:37:23 2014 : >SEND: 04
DEBUG:Mon Feb  3 20:37:23 2014 : >FRAMER: no preset result
DEBUG:Mon Feb  3 20:37:23 2014 : Warte auf 05
DEBUG:Mon Feb  3 20:37:25 2014 : <RECV: 05 (1200.0 ms)
DEBUG:Mon Feb  3 20:37:25 2014 : >SEND: 01
DEBUG:Mon Feb  3 20:37:25 2014 : >SEND: F7
DEBUG:Mon Feb  3 20:37:25 2014 : >SEND: 00
DEBUG:Mon Feb  3 20:37:25 2014 : >SEND: F8
DEBUG:Mon Feb  3 20:37:25 2014 : >SEND: 02
DEBUG:Mon Feb  3 20:37:25 2014 : >FRAMER: no preset result
DEBUG:Mon Feb  3 20:37:25 2014 : <RECV: len=1 20 (20.0 ms)
DEBUG:Mon Feb  3 20:37:25 2014 : <RECV: len=1 CB (0.0 ms)
DEBUG:Mon Feb  3 20:37:25 2014 : <RECV: received 20 CB
DEBUG:Mon Feb  3 20:37:25 2014 : 20 CB -> Vitodens ID=20CB Protokoll:KW
DEBUG:Mon Feb  3 20:37:25 2014 : Vitodens ID=20CB Protokoll:KW
Vitodens ID=20CB Protokoll:KW

but to get my device identified I had to add a line in the vcontrold.xml at <unit name="DeviceType">.  Putting it in vito.xml under <devices> did not work.

...so... using telnet I can read all the data step by step so the connection works.  If you are still reading and cared to look at the problems described above I have one additional question.  I would like to read/write data through a php file.  I understand php and need it to talk to my PLC.  How would you approach such a need?  Should I use the vcontrold deamon or vconnect?  Please give you your best feeling.  I will dig through the rest...

Many thanks to all of you who figured made the software and the idea public.  It is great to talk to the mysterious Viessman unit!

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: smbunn <userid-40480070@openv.wikispaces>
* Date: Tue,  4 Feb 2014 04:07:00 UT
* Message-ID: <62826000-69369660$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Try looking at 
http://www.viessmann.de/content/dam/internet-global/software-daten/vitogate_200_eib-daten2/vt200_300gwb_en.pdf 
 this is the official document from Viessmann with all of the internal addresses.  By the way this site has documents on all Viessmann devices with all addresses, let me know if you are after any other device type.

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Wed, 26 Feb 2014 21:28:51 UT
* Message-ID: <62826000-70024230$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Hi,

I have got it all working with ViTalk.  So... if anyone is interested in my device (split in half - self made only the part after FT232RL) and some scripts, you can see it at http://www.edom-plc.pl/index.php/en/1-wire-i-rpi-en/175-komunikacja-z-viessmann-em-vitodens-200.

Many thanks to all who keep openv.wikispaces running!

WikiSpaxe, 2017-10-26:
* Subject: Re: Connection to Raspberry Pi for PLC-based home automation
* Wiki-Page: home
* From: edomplc <userid-28940242@openv.wikispaces>
* Date: Wed, 26 Feb 2014 21:28:52 UT
* Message-ID: <62826000-70024234$openv.wikispaces>
* In-Reply-To: <62826000-63505232$openv.wikispaces>

Hi,

I have got it all working with ViTalk.  So... if anyone is interested in my device (split in half - self made only the part after FT232RL) and some scripts, you can see it at http://www.edom-plc.pl/index.php/en/1-wire-i-rpi-en/175-komunikacja-z-viessmann-em-vitodens-200.

Many thanks to all who keep openv.wikispaces running!

trimixdiver1, 2019-01-15:
Hi
So you made a pi interface a vitodens and give and receive modbus commands from a plc? If so I would be interested.

Thanks

edomplc, 2019-01-15:
Hi,

it is rather a php script, which uses Telnet.class.php and publishes data.
Modbus wouldn't be hard to implement.

You can find my files at:

https://edom-plc.pl/index.php/en/1-wire-i-rpi-en/178-viessman-sterowanie-przez-www

Best regards,

P.