openv/openv #152: Patch für FreeBSD

WikiSpaxe, 2017-10-26:
* Subject: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Fri, 18 Oct 2013 16:01:54 UT
* Message-ID: <64418924-66225014$openv.wikispaces>

Hallo,

zu Testzwecken habe ich vcontrold auch mal unter FreeBSD kompiliert. Dazu waren ein paar kleine Anpassungen am Code notwendig. Hier der Patch:

```

diff -urN vcontrold.orig/vcontrold/common.h vcontrold/vcontrold/common.h
--- vcontrold.orig/vcontrold/common.h	2013-10-18 15:28:54.000000000 +0200
+++ vcontrold/vcontrold/common.h	2013-10-18 15:30:30.000000000 +0200
@:@ -15,4 +15,7 @:@
         #define MAXBUF 4096 
 #endif
 
+#ifndef ETIME
+#define ETIME           101             /* STREAM ioctl timeout */
+#endif
 
diff -urN vcontrold.orig/vcontrold/semaphore.c vcontrold/vcontrold/semaphore.c
--- vcontrold.orig/vcontrold/semaphore.c	2013-10-18 15:28:54.000000000 +0200
+++ vcontrold/vcontrold/semaphore.c	2013-10-18 15:31:17.000000000 +0200
@:@ -15,7 +15,7 @:@
 
 #define MAX_RETRIES 10
 
-#if !defined(__APPLE__)
+#if !defined(__APPLE__) && !defined(__FreeBSD__)
 /* on my mac this is a redefinition */
 union semun {
   int val;
diff -urN vcontrold.orig/vcontrold/unit.c vcontrold/vcontrold/unit.c
--- vcontrold.orig/vcontrold/unit.c	2013-10-18 15:28:54.000000000 +0200
+++ vcontrold/vcontrold/unit.c	2013-10-18 15:39:38.000000000 +0200
@:@ -18,7 +18,7 @:@
 #include <asm/byteorder.h>
 #endif
 
-#ifdef __CYGWIN__ 
+#if defined (__CYGWIN__) || (__FreeBSD__)
 #include "byteorder.h"
 #endif

```

Beste Grüße,
 Kai.

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Fri, 18 Oct 2013 16:31:57 UT
* Message-ID: <64418924-66227250$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

Hallo Kai,
mich stört ein bisschen die 101 bei ETIME - magic number :-) . Hat BSD nicht auch vordefinierte Werte bzw eigene Bezeichner? Evtl ETIMEOUT - oder was ist 101 in BSD ? Aus welchem include File?
vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Fri, 18 Oct 2013 17:15:10 UT
* Message-ID: <64418924-66229638$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

Hi vheat,

scheinbar kocht da jedes OS sein eigenes Süppchen. Ein ETIME, wie bei Linux oder MacOS, ist bei FreeBSD leider nicht definiert. Nur ETIMEDOUT. Nur unterscheiden sich Linux und die BSD-Abkömmlinge bzgl. des Werts:

Linux ( /usr/include/asm-generic/errno.h ):
```

#define	ETIME		62	/* Timer expired */
#define	ETIMEDOUT	110	/* Connection timed out */

```

MacOS ( /usr/include/sys/errno.h ):
```

#define	ETIMEDOUT	60		/* Operation timed out */
#define	ETIME		101		/* STREAM ioctl timeout */

```

FreeBSD ( /usr/include/sys/errno.h ):
```

#define	ETIMEDOUT	60		/* Operation timed out */

```

Und da MacOS und FreeBSD eher miteinander verwandt sind, hab ich ETIME als 101 definiert.

Beste Grüße,
 Kai.

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Fri, 18 Oct 2013 18:26:38 UT
* Message-ID: <64418924-66232960$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

Hallo Kai,
die einzigeStelle mit ETIME ist im initsem(), und da wird das nur proforma als Wert gesetzt. Mein Vorschlag wäre, das da auf einen anderen allgemeinen Code zu setzen. Im Aufrufer wird sogar ein perror darauf gemacht, was da bei der 101 passiert weiss ich nicht. ETIMEDOUT passt auch nicht ganz, ENODATA ginge vielleicht, EPROTO wäre vielleicht auch eine Option.Ich würde aber auf jeden Fall vorschlagen, keinen eigenen Wert neu zu definieren. Gibt nur Probleme wenn der mal Wirklichkeit wird, und keiner sich erinnert :-) Was wären denn Deiner Meinung nach passende Bezeichner in der BSD Welt?
Vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Fri, 18 Oct 2013 22:37:44 UT
* Message-ID: <64418924-66242058$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Fri, 18 Oct 2013 22:54:22 UT
* Message-ID: <64418924-66242390$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

Hallo zusammen,

nur mal so generell. Wenn in einem OS ein bestimmter #define nicht vorhanden ist, dann ist der entweder vergessen worden oder wird im weiteren System nicht verwendet.

Da ich schon lange nicht mehr tiefer in den FreeBSD Sourcen drin stecke, würde ich vermuten, dass es keinen Sinn macht den Wert für ETIME einfach aus anderen OS zu importieren.

Gruß
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Fri, 18 Oct 2013 22:55:42 UT
* Message-ID: <64418924-66242420$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

hmm. Ich bin kein Programmierer (schon gar nicht systemnah). Aber so aus'm Bauch raus, wenn ich mir initsem() ansehe, hätte ich gesagt "ETIMEDOUT" (mit der Beschreibung "Operation timed out") klingt nicht so schlecht. Vor allem weil dies z.B. bei MacOS im Abschnitt "ipc/network software" definiert ist, würde also auch passen.

Bei Cygwin sind ETIME und ETIMEDOUT übrigens so definiert:
```

#define ETIME 62       /* Timer expired */
#define ETIMEDOUT 116          /* Connection timed out */

```

Grüße, Kai.

WikiSpaxe, 2017-10-26:
* Subject: Re: Patch für FreeBSD
* Wiki-Page: vcontrold
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Fri, 18 Oct 2013 23:03:08 UT
* Message-ID: <64418924-66242672$openv.wikispaces>
* In-Reply-To: <64418924-66225014$openv.wikispaces>

Ich würde mir dazu gar nicht so viele Gedanken machen. Ein system call funktioniert oder er gibt einen Fehler zurück. In dem Fall kannst Du im Prgramm sowieso nicht alles abfangen. 

An der Stelle eine Fehlermeldung ausgeben und mit exit(1) raus aus dem Programm ist in den meisten Fällen die sauberste Lösung.

Gruß
Frank

Xenolog2, 2023-01-22:
Salü,
bin bei meinem aktuellen Versuch, vcontrold auf TrueNAS Core (basierend auf FreeBSD) zu bauen, auf praktisch die gleichen Probleme gestoßen.
Für die Frage nach dem korrekten Wert für ITIME habe ich durch ein zusätzliches #include von /sys/compat/linuxkpi/common/include/linux/errno.h gelöst; siehe: https://github.com/streamich/freebsd/blob/master/sys/compat/linuxkpi/common/include/linux/errno.h
Es gab dann noch Link-Probleme, für die ich ein #include /sys/compat/linuxkpi/common/include/asm/byteorder.h in unit.c ergänzte; siehe https://github.com/streamich/freebsd/blob/master/sys/compat/linuxkpi/common/include/asm/byteorder.h

Ob es wirklich läuft, kann ich erst sagen, sobald der Rechner in den Keller in die Nähe der Viessmann-WP gewandert ist.