openv/openv #160: Brennerstunden in Gasverbrauch umrechnen

WikiSpaxe, 2017-10-26:
* Subject: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: miracoolix_VScotHO1 <userid-44177880@openv.wikispaces>
* Date: Fri,  8 Nov 2013 17:49:01 UT
* Message-ID: <64812634-67030888$openv.wikispaces>

Hallo,

ich lese nun seit Tagen die Zählerstände jeweils um 13 Uhr zum Vortagesvergleich ab. Der alte Gaszähler ist nicht (ohne weiteres) auslesbar. Parallel dazu zeichne ich via getBrennerStunden1 die Brennerstunden auf.

Was genau sind diese Brennerstunden? Wie lange der Brenner in Betrieb war? Spielt die Leistung dabei eine Rolle? Also sind 5 Stunden mit 75% Leistung genau 5 Brennerstunden wie 5 Stunden mit 10 %? 

getBrennerStunden2 liefert bei mir immer 0,0.

In folgender Tabelle sieht man, dass die Umrechnung nur etwa ungefähr 0,7 qm pro Brennerstunde ergibt:

http://energy.michaelpucher.net/images/Gasverbrauch_umrechnen.png

Hat jemand eine genauere/andere Berechnungsmethode?

Vielen Dank für Hinweise!
Michael

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: texomex <userid-41982818@openv.wikispaces>
* Date: Fri,  8 Nov 2013 20:28:22 UT
* Message-ID: <64812634-67039850$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

ist BrennerStunden2 nicht nur für zweistufige Ölbrenner?
Ich denke Du müsstest schon die Brennerleistung mitloggen. Bzw. eigentlich nur diese. Aus den Zeitintervallen und der Brennerleistung kannst Du dann die Energiemenge berechnen. Im Prinzip eine Integralbildung. Nur dass das aufgrund der endlich kleinen Zeitintervalle zur Summenbildung wird. Korrekturfaktor für Gas usw... dann stimmt das recht genau.

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: miracoolix_VScotHO1 <userid-44177880@openv.wikispaces>
* Date: Fri,  8 Nov 2013 21:41:16 UT
* Message-ID: <64812634-67042158$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

Hm, klingt kompliziert...

Viessmann ist da leider auch nicht ganz up to date. Aus dem Benutzerhandbuch:

Abfragemöglichkeiten im erweiterten Menü:

„Brennst. Verbr.“
(Brennstoffverbrauch, nur bei Heizkes-
seln mit Ölbetrieb)

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Fri,  8 Nov 2013 21:54:28 UT
* Message-ID: <64812634-67042756$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

Mathe lässt grüßen, ich habe auch nichts für den Gasverbrauch gefunden, also: schön stabil mitloggen und die Brennerleistung integrieren. Wenn das Messinterval hab so groß ist wie das Zeitinterval, in dem die Regelung den Modulationsgrad anpasst, sollte das Integral deb Verbrauch geben, indem man den Wert noch normiert, per abgelesenem Gaszähler.
Würde mich auch interessieren, ob das funktioniert :-) Ich will immer mal versuchen im rrdtool die Leistung als ...wie hiess das noch? ... aufzuzeichnen, angebelich macht dann das rrdtool genau diese Summierung der Prozente, und dann fehlt nur noch der Multiplikator, ein CDEF mit einen auf lange Zeit gemitteltem Verbrauch, durch die zuvor emittelte Summe. Also ein richtig schönen numerisches Integral :-) 
Schönen Gruss aus der Schule.
vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: miracoolix_VScotHO1 <userid-44177880@openv.wikispaces>
* Date: Fri,  8 Nov 2013 22:06:27 UT
* Message-ID: <64812634-67043332$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

Ich habe die Brennerleistung schon als AREAs in einem rrd graph. Kann rrd die Berechnung nicht? 

Finde die rrd Dokumentation sehr verwirrend und teilweise frustrierend (genau das einfache Beispiel, das man braucht, ist nie dabei)...

http://energy.michaelpucher.net/cms/?energylog&Neue_Detailauswertungen&realblogaction=view&realblogID=5&page=1

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Sat,  9 Nov 2013 02:28:14 UT
* Message-ID: <64812634-67048120$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

"If you enter 'ABSOLUTE', rrdtool will assume that the counter resets upon reading, and plots the value divided by the time since the previous measurement."
Hatte gedacht, das sei dieses ABSOLUTE, aber das sin nur Raten.
Müsste man am Ende des Tages die Leitungsintervalle zusammenrechnen, eine Summenfunktion über einen Zeitraum habe ich nicht gesehen ... grr
vheat
vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: miracoolix_VScotHO1 <userid-44177880@openv.wikispaces>
* Date: Sat,  9 Nov 2013 19:25:18 UT
* Message-ID: <64812634-67062132$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

Sers,

durch eine Fehlkonfiguration ("Problem saß wieder vor dem Rechner") fehlt mir heute wieder eine Stunde an Daten.

Genau das ist das Problem bei der Berechnung anhand der Leistungs-Istdaten. Da die Daten nicht historisch gehalten werden, wäre die Berechnung jetzt fehlerhaft. Ich glaube, ich bilde mir lieber aus manuell ermitteltem Verbrauch und den Brennerstunden (da habe ich immer den letzten Wert des Kessels) und Gasverbrauch Monatsdurchschnittswerte und errechne daraus einen "geschätzten Verbrauch"

VG
Michael

WikiSpaxe, 2017-10-26:
* Subject: Re: Brennerstunden in Gasverbrauch umrechnen
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Sun, 10 Nov 2013 00:49:06 UT
* Message-ID: <64812634-67069516$openv.wikispaces>
* In-Reply-To: <64812634-67030888$openv.wikispaces>

Ich kann mir eigentlich nicht vorstellen, dass dieses Summen-Datum nirgendwo in der Regelung gebildet wird ... aber wo? Der Ölverbrauch geht ja nicht bei Gas. Gibt es bei den Extra-Werten für Solaranlagen keine Werte für die einzelnen Verbrauchsanteile Solar/Gas?
vheat

PS: das ist bei Produktivsystemen so, dass man sie nicht stören darf :-)

mpseybold, 2018-03-03:
Supi interessant! 
An meiner 20C2 Steuerung  gibt es eine kumulierte BrennerStunden1, prozentuale Leistung als BrennerStatus und kumulierte Anzahlen an BrennerStarts.
Ich hab mal einen Tag lang diese Werte alle 2 Minuten weggespeichert.
Das kürzeste Intervall, in dem der BrennerStatus >0% ist, war ~30 Minuten.
Aber es gibt auch Plateaus mit 50 oder mit 60 Minuten Dauer.
Leider habe ich keine entsprechenden Gaszählerstände zum Formel fitten...

qwert089, 2019-06-04:
mein Vitodens 300 W mit VScotHO1 hat so ein graphisches Energiecockpit.
Da Rechnet die Vitotronic schon den Gasverbrauch in kwh um.
Weiss jemand ob man diese Energiecockpit auslesen kann.
Danke

GolterU, 2019-06-04:
Meines Wissens nutzt die Viessmann-App „ViCare“ das Energiecockpit, jedenfalls den Beiträgen im [Forum](https://www.viessmann-community.com/t5/forums/searchpage/tab/message?advanced=false&allow_punctuation=false&q=Cockpit) zufolge.

Gruß 
Golter

qwert089, 2019-06-05:
Hallo Golter,

vielen Dank. 
d.h. es müsste technisch gehen.
Jemand der so einem "Duo-Sniffer" IR Adapter hat, könnte die Parameter mitscheiden.

Vielleicht ließt das ja mal jemand und hat Lust das zu machen...

qwert089, 2019-07-19:
habe jetzt durch Zufall einen anderen issue gefunden wo genau das erklärt wird.
https://github.com/openv/openv/issues/171

16D0 kWh Strom Warmwasser
16D1 kWh Strom Heizung
16D2 kWh Wärme Warmwasser
16D3 kWh Wärme Heizung