openv/openv #164: Thermisches Problem mit Kesselkreisregelung

WikiSpaxe, 2017-10-26:
* Subject: Thermisches Problem mit Kesselkreisregelung
* Wiki-Page: home
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Sun, 10 Nov 2013 23:00:34 UT
* Message-ID: <64847564-67095122$openv.wikispaces>

Hallo zusammen,

etwas off topic, aber vielleicht kann mir ja jemand weiterhelfen. Meine Vitodens 200 (WB2A, Baujahr 2007) hat offensichtlich ein thermisches Problem mit der Kesselkreisregelung. Das scheint ein bekanntes Problem bei diversen Vitodens-Regelungen zu sein. 
Siehe z.B. http://www.haustechnikdialog.de/Forum/t/114474/Vitodens-300-WB3B-Regelung-defekt-?page=3  oder http://www.fingers-welt.de/phpBB/backup/06/39588_2.htm

Jedenfalls steigt die Regelung aus wenn die Therme ca. 1-2 Stunden eingeschaltet ist (Gebläse läuft hoch, aber Brenner zündet nicht), und funktioniert wieder, nachdem die Therme eine gewisse Zeit am Netzschalter ausgeschaltet war. Im Fehlerfall steht auch der "berühmte" Wert 33 in Codieradresse 38, was anscheinend eine Unterspannung bedeutet. Nachdem ich beide Abdeckungen der Platine entfernt habe, läuft die Therme jetzt seit 2 Tagen ohne Ausfall. Sieht also tatsächlich stark nach einem thermischen Problem aus. Die Platine hat starke Ähnlichkeit mit der Platine auf den Fotos im 2. Link.

Kennt jemand dieses Problem? Evtl. sogar eine Lösung ohne gleich die komplette Regelung für +/-350,- Eur tauschen zu müssen? An welcher Stelle der Platine können die Spannungen 5V, 32V und ~22V (nicht 24V ??) gemessen werden?

Bin für jeden Tip dankbar.

Beste Grüße,
 Kai.

WikiSpaxe, 2017-10-26:
* Subject: Re: Thermisches Problem mit Kesselkreisregelung
* Wiki-Page: home
* From: texomex <userid-41982818@openv.wikispaces>
* Date: Sun, 10 Nov 2013 23:55:15 UT
* Message-ID: <64847564-67097062$openv.wikispaces>
* In-Reply-To: <64847564-67095122$openv.wikispaces>

Also sachdienliche Hinweise kann ich leider nicht liefern, aber wenn Dir gar nichts mehr einfällt, kannst Du Dir die restliche Homepage aus dem 2. link (http://www.fingers-welt.de) ansehen. Das ist äußerst unterhaltsam. Auch dass Finger jetzt in der Heizungstechnik angekommen ist.... Bin gespannt wann er sich in die Luft sprengt.....

WikiSpaxe, 2017-10-26:
* Subject: Re: Thermisches Problem mit Kesselkreisregelung
* Wiki-Page: home
* From: vheat <userid-37986540@openv.wikispaces>
* Date: Mon, 11 Nov 2013 00:07:09 UT
* Message-ID: <64847564-67097504$openv.wikispaces>
* In-Reply-To: <64847564-67095122$openv.wikispaces>

Mal mit Fön oder Kühlspray spielen, um den Bereich zu identifizieren? Evtl. Betriebsspannungen identifizierbarer Chips raussuchen?

vheat

WikiSpaxe, 2017-10-26:
* Subject: Re: Thermisches Problem mit Kesselkreisregelung
* Wiki-Page: home
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Mon, 11 Nov 2013 23:01:52 UT
* Message-ID: <64847564-67142304$openv.wikispaces>
* In-Reply-To: <64847564-67095122$openv.wikispaces>

Ja, an Fön und Kältespray habe ich auch schon gedacht. Leider ist es mir bisher noch nicht gelungen, der Fehler durch Erwärmen zu provozieren um dann mit gezielter Kühlung die richtige Stelle zu finden. Aber ich bleibe dran ...

Anderer Ansatz derweil. Im Haustechnikdialog-Forum wird berichtet dass Viessmann scheinbar die Kesselkreisregelung überarbeitet hat und der beschriebene Fehler mit der neueren Regelung nicht mehr auftritt. Ich vermute, dass ich noch eine der "alten" Regelungen habe: Therme Vitodens 200 (WB2A; 8,8 -20kW), Regelung GG1, Art.Nr. 7187362. Eine Suche im Netz nach dieser Art.Nr. bringt bis auf 1-2 eBay Aktionen aber keine Treffer, auch nicht bei diversen Online-Shops. Eine Suche nach Kesselkreisregelung GG1 bringt statt dessen aber die Art.Nr. 7825241, u.a. geeignet für WB2A 20/26kW. Vermutlich ist dies das Nachfolgemodell (ich habe den Eindruck, es gibt für die aktuelle Vitodens-Reihe eh nur eine einzige Platine welche dann über den Kesselcodierstecker für das  spezifische Modell parametrisiert wird??)

Hat jemand eine Regelung GG

WikiSpaxe, 2017-10-26:
* Subject: Re: Thermisches Problem mit Kesselkreisregelung
* Wiki-Page: home
* From: khambrecht <userid-43832634@openv.wikispaces>
* Date: Mon, 11 Nov 2013 23:04:50 UT
* Message-ID: <64847564-67142436$openv.wikispaces>
* In-Reply-To: <64847564-67095122$openv.wikispaces>

(... Fortsetzung ;-) )

Hat jemand eine Regelung GG1 mit Art.Nr. 7825241 mit Fotos von der Platine, so dass man vergleichen kann, ob und wie sich Layout und/oder Bestückung geändert hat?

Beste Grüße,
 Kai.

CSZS-HU, 2021-09-24:
Hallo!
Hast du mit diesem Fehler Ergebnisse erzielt?
Dieses Problem habe ich auch beim Vitodens200 Heizgerät
Jetzt werde ich die Regelung GG1 7825241 zum zweiten Mal ersetzen ....