openv/openv #169: vitotronic V200K01B Ölverbrauch abfragen Probleme

WikiSpaxe, 2017-10-26:
* Subject: vitotronic V200K01B Ölverbrauch abfragen Probleme
* Wiki-Page: home
* From: Schwammerl <userid-44718776@openv.wikispaces>
* Date: Mon, 16 Dec 2013 15:33:21 UT
* Message-ID: <65452150-68388892$openv.wikispaces>

Hallo,

ich habe am 5.12. begonnen meinen Ölverbrauch mit aufzunehmen. Bis gestern war auch alles ok. Als ich heute den Ölverbrauch anschauen wollte hat er wieder bei Null angefangen. Noch ein paar Daten. Bin mit Optolink verbunden Anlagenkennung ist 209F.
Die Adresse für den Ölverbrauch ist 0x7574 die ich abfrage. Ich hatte bis gestern, also 15.12. einen Verbrauch von 65 Liter. Heute hat er bei Null angefangen. Wie lässt sich das erklären? Er hat zuvor den Verbrauch immer schön dazu gezählt.

Gruß
Alex

WikiSpaxe, 2017-10-26:
* Subject: Re: vitotronic V200K01B Ölverbrauch abfragen Probleme
* Wiki-Page: home
* From: Schwammerl <userid-44718776@openv.wikispaces>
* Date: Wed, 18 Dec 2013 10:38:03 UT
* Message-ID: <65452150-68443514$openv.wikispaces>
* In-Reply-To: <65452150-68388892$openv.wikispaces>

Hallo,

habe nach längerem probieren den Fehler gefunden. An statt 4 Byte aus zu lesen habe ich nur 2 Byte gelesen. Dies hatte den Effekt, das er nach 10 Tagen neu angefangen hatt.

Gruß
Schwammerl

helmar74, 2023-07-26:
Hallo

mal eine Frage. Bei mir wird immer Ölverbrauch 0 angezeigt. alle anderen Werte kann ich auslesen.
Ich habe folgendes Kommando in vito.xml angelegt:
```
 <command name="getOelverbrauch" protocmd="getaddr">
      <addr>7574</addr>
      <len>4</len>
      <unit>COL</unit>
      <description>Ermittle den Oelverbrauch in Liter kumuliert</description>
      <device ID="2053"/>
    </command>
```

vcontrold.xml:
```
<unit name="Counter liter">
      <abbrev>COL</abbrev>
      <calc get="V/1000" set="V*1000"/>
      <type>int</type>
      <entity/>
    </unit>
```

Muss ich da sonst noch was tun, damit ich den Ölverbrauch bekomme?