openv/openv #170: Vitotronic V200KO1B / V200KO2B Protokoll

WikiSpaxe, 2017-10-26:
* Subject: Vitotronic V200KO1B / V200KO2B Protokoll
* Wiki-Page: home
* From: Schwammerl <userid-44718776@openv.wikispaces>
* Date: Mon, 16 Dec 2013 15:40:54 UT
* Message-ID: <65452488-68389444$openv.wikispaces>

Hallo,

bin dringend auf der suche nach den Adressen der oben aufgeführten Steuerung. Hat jemand so eine Steuerung und schon die Adressen herausgefunden. Mit der Liste über das 300 Protokoll bin ich langsam an meine Grenzen gestoßen. Mir fehlen da einige Adressen, wie z. B. Raumsolltemperaturen von der Anlage und den Beiden Heizkreisen HK2 und HK3 genau so die richtigen Adressen über die Abfrage der Mischerstellungen.
Hat irgend jemand schon eine komplette Lister der Oben aufgeführten Steuerung?

Gruß
Alex

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200KO1B / V200KO2B Protokoll
* Wiki-Page: home
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat, 11 Apr 2015 23:21:17 UT
* Message-ID: <65452488-81067728$openv.wikispaces>
* In-Reply-To: <65452488-68389444$openv.wikispaces>

Hallo Alex,

bin Neuling hier und suche die Adressliste für meine Vitotronic 200KO1B mit Kennung
209F. Soweit ich das sehen kann habt Ihr auch so eine Steuerung. Bitte seid so nett und teilt mit mir Eure Erfahrungen und Adressen.

Habe selbst openhab seit 2 Jahren im Einsatz mit einem KNX System und möchte jetzt auch die Heizung einbinden. Erster Schritt war erfolgreich (heute). Ich habe den Optolink in Betrieb genommen und meiner Anlagenversion ausgelesen:

ID 209F
ExtID 0002
Protocol 300
Description Vitotronic 200 KO2B: Witterungsgeführte Kessel- u. Heizkreisregelung für gleitend abgesenkte Kesselwassertemperatur mit Regelung für 2 Heizkreise mit Mischer

Lieben Dank
Michael

mactoolz, 2018-11-04:
Hi,

ich habe auch eine KO1B Steuerung. Bin auch gerade am rätseln in wei weit die HErsteller Anleitung da hilfreich ist. Ich habe noch keinen Adapter, kommt aber die kommende Woche.

Vielleicht kann man sich hier nochmal zusammen raufen.

Gruß

MacToolz

mactoolz, 2018-11-06:
Hallo,

falls es jemanden interessiert ich habe Informationen was die KO1B angeht.

Gruß

Cibuc, 2020-12-01:
> Hallo,
> 
> falls es jemanden interessiert ich habe Informationen was die KO1B angeht.
> 
> Gruß

Moin Mactoolz,

suche die xml Datei für eine KO1B. Kannst du mir diese zur Verfügung stellen?

Liebe Grüße!

mactoolz, 2020-12-01:
Hi,

sorry das ich deine Nachricht jetzt erst sehe.

Wie soll ich dir meine Infos zukommen lassen?

Gruß

Cibuc, 2020-12-01:
Moin, alles gut. Cool das du dich meldest. 

Kannst du mir alles auf meine Mail schicken? 
Schneider.preist@gmx.de

Grüße!
Stefan Schneider

Cibuc, 2020-12-02:
Hast du meine Nachricht erhalten?

mactoolz, 2020-12-02:
Hi,Ja habe ich. Ich melde mich im Laufe des Tages. Will da nicht lange schreiben. Ich habe ein PDF von Viessmann. Das wollte ich erklären etc.Gruß -- Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

abrits56, 2024-10-30:
Hallo Stefan,

habe frisch mit vcontrold  mit einem Vitoladens 300C und einer K02B begonnen, läuft soweit aber mangels Info nur mit einigen values.
Ist Dein PDF noch verfügbar?
Grüße Johannes