openv/openv #171: Vitotronic V200 WO1B - Energiebilanz

WikiSpaxe, 2017-10-26:
* Subject: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: cchiro <userid-46021696@openv.wikispaces>
* Date: Wed, 18 Dec 2013 17:34:07 UT
* Message-ID: <65482626-68453028$openv.wikispaces>

Hallo,
ich bin auf der suche nach die Adressen um Energiebilanz Werte abzufragen. Soweit ich weiss, gibt es Heizung und Warmwasser Werte aber die Adressen hierfür nie gesehen. Any ideas?
Thanks!

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Sun, 23 Feb 2014 18:38:03 UT
* Message-ID: <65482626-69918096$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

hi,
falls noch aktuell, hier sind wie adressen von Vitotronic mit ID "2049":
WPR_Heizwaerme01~0x1640
WPR_WO1H_Energie_Verdichter1~0x1650
WPR_WO1H_Energie_Verdichter1_1660~0x1660
WPR_WO1H_ElektroenergieWW_Verdichter1~0x1670
WPR_WO1H_JAZ_0x1680
WPR_WO1H_JAZ_Heizen~0x1681
WPR_WO1H_JAZ_WW_Betrieb~0x1682
energie und wärme: 4 bytes, JAZ: 1 byte.
lg
steve

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: cchiro <userid-46021696@openv.wikispaces>
* Date: Mon, 24 Feb 2014 12:38:29 UT
* Message-ID: <65482626-69936454$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

1000 Dank! Es ist noch aktuell.
Ich wollte nachfragen wegen Adresse 0x1640, 0x1650, 0x1660 und 0x1670.
Sind das Elektro- Input und Wärme- Output Energie oder nur Input?
Alles Watt Einheiten?
Gibt es nur Energie Werte für die Verdichter (WW u. HZg)? was ist mit die Pumpen Energie?
und zum Schluss, woher bekommt man die Info? JAZ Adressen habe ich schon im Doku gesehen, Verdichter Energie usw. noch nie!
Noch mal vielen vielen Dank!

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 24 Feb 2014 13:01:23 UT
* Message-ID: <65482626-69936766$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

energie in kWh.
ich ab die daten von hier: http://openv.wikispaces.com/share/view/65878710, siehe die datei config.zip.
aus der ecnDataPointType.xml such dir die geräteID aus und die entsprechende parameterbezeichnung. dann kannst du die parameter-addressen aus ecnEventType.xml hernehmen.
vermutlich hast du auch vitocal+vitotronic200, oder? dazu passen die parametern "WPR_..." und/oder "WPR300_...". obwohl nicht alle parametern sind auslesbar und andere haben falsche byte-länge.
steve

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: maxmax20 <userid-51793276@openv.wikispaces>
* Date: Sun,  9 Nov 2014 22:06:33 UT
* Message-ID: <65482626-76619222$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Hallo.
Ich versuche auch gerade aus meiner Vitocal 300 die Energiebilanz auszulesen.
die JAZ werte bekomme ich korrekt retour.
Aber die Energiewerte leider nicht.
Ich habe dies so gemacht in der vito.xml - zb für die Energiebilanz der Heizwärme:

<command name='getEBHW' protocmd='getaddr'>
      <description>Energiebilanz Heizwaerme</description>
      <addr>1640</addr>
      <len>4</len>
      <unit>CO</unit>
</command>

Ich bekomme zwar eine Antwort zurück hier zb: 27071.000000 sind das schon die kWH ? oder kann das nicht stimmen.
Hoffentlich hat jemand eine Antwort für mich.
Vielen Dank,
Mark

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Mon, 10 Nov 2014 06:02:41 UT
* Message-ID: <65482626-76635022$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Moin,
Du musst den Wert nochmal durch 10 Teilen. Ich habe das damals mit einer eigenen Unit in der vcontrol.xml gelöst:
 <unit name='Counter/10'>
           <abbrev>CO10</abbrev>
           <calc get='V/10' set='V*10'/>
           <type>int</type>
           <entity></entity>
          </unit>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: maxmax20 <userid-51793276@openv.wikispaces>
* Date: Mon, 10 Nov 2014 10:58:06 UT
* Message-ID: <65482626-76639740$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Hi,
Danke für den Hinweis. So etwas habe ich mir dann auch gedacht. 
Ich habe auch die eigene Unit für einen anderen Parameter schon definiert.
Nur nochmals der Vollständigkeit - welche Werte sind was:

WPR_Heizwaerme01~0x1640                                        <-- Heizwärme in kWH
WPR_WO1H_Energie_Verdichter1~0x1650                  <-- gebrauchte Energie um die Heizwärme zu erzeugen in kWH
WPR_WO1H_Energie_Verdichter1_1660~0x1660        <-- gebrauchte Energie um das WW zu erzeugen in kWH
WPR_WO1H_ElektroenergieWW_Verdichter1~0x1670 <-- Heizwärme WW in kWH

stimmt das so?
Grüße,
Mark

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Mon, 10 Nov 2014 11:46:54 UT
* Message-ID: <65482626-76640718$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Hi Mark,
die Werte stimmen so. Ich möchte aber Deine Erwartung bzgl. einer dauerhaften Auswertung etwas dämpfen.
Beim mir das das Auslesen 8 Monate funktioniert, dann fingen die Summenzähler an Rückwärts zu laufen :-)
Am Panel der Vitrotronic ist die Energie-Bilanz ok, aber eben die Zähler 1640-1670 machen seit 9 Wochen was sie wollen (==> siehe Wochenstatistik KW34/2014ff http://warmup.mypump.de/Live-Daten/Live-Daten.html)

Gruß

Simon

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: cchiro <userid-46021696@openv.wikispaces>
* Date: Sat,  3 Jan 2015 17:55:18 UT
* Message-ID: <65482626-78191220$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Die Energie-Bilanz des laufenden Kalendar Woche, (kWh  Strom Heizung u.Warmwasser, kWh Wärme Heizung u. Warmwasser) seht man bei mir am Vitrotronic Panel unter Energie-Bilanz. Die werden quasi im Echtzeit aktualisiert. Die Zähler Adressen 1640-1670 werden einmal die Woche aktualisiert.
Die 4 Parametern bekomme ich unter folgende Adressen:

16D0 kWh Strom Warmwasser
16D1 kWh Strom Heizung
16D2 kWh Wärme Warmwasser
16D3 kWh Wärme Heizung

Das sind alle 2 Byte Params.
Seit ein paar Tage vergleiche ich sie mit der Vitrotronic Energie-Bilanz Panel und bis jetzt stimmen sie. Vielleicht hilft es jemand.
PS: Meine Anlage hat DeviceID 2048  vitocal + vitotronic200

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: cchiro <userid-46021696@openv.wikispaces>
* Date: Sat,  3 Jan 2015 18:12:43 UT
* Message-ID: <65482626-78191342$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Bei mir, scheinen die Zähler Adressen 1640-1670 ein Art Jahressumme zu sein. Wie Simon oben geschrieben hat, haben meine Zählern, nachdem 52. Betriebswoche, “Rückwarts” gelaufen. Das wäre zu erklären wenn die Zähler eine Summe der letzten 52 Wochen sei. Die älteste Woche wird abgezogen und die neue Woche kommt hinzu. Hierbei kann ein kleinere Summe raus kommen wenn die neueste Woche wenigen Energie verbraucht als die älteste. Ich hoffe ich habe das verständlich geschrieben.

Bei mir kommt es genau hin wenn ich meine Zählern mit die Vitrotronic Panel Energie-Bilanz vergleiche.
Über eine Rückmeldung wurde ich mich sehr freuen falls jemand das Bestätigen oder widerlegen könnte.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Sat,  3 Jan 2015 18:20:25 UT
* Message-ID: <65482626-78191436$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Hi - das Thema ist mittlerweile geklärt .... hat vergessen hier was zu posten - es verhält sich wie DU gerade geschrieben hast:
Die Vitocal arbeitet mit einem rollierenden Speicher, der nach 52 Wochen immer den ältesten Wert überschreibt. Somit kann man nach einem Jahr Betrieb nicht mehr einfach die Alte von der neuen Woche abziehen.
Jetzt habe ich mal die neue Formel KW[Aktuell]-KW[Alt]+KW[Alt des letzten Jahres] eingebaut und die Werte korrigiert.

Mit den neuen Werten für die aktuelle Energie-Bilanz gehört das aber zur Vergangenheit und ich kann endlich die Auswertung umbauen und zum Stichtag am Monatsende eine Auswertung fahren :-)))

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitotronic V200 WO1B - Energiebilanz
* Wiki-Page: home
* From: cchiro <userid-46021696@openv.wikispaces>
* Date: Sun,  4 Jan 2015 14:05:51 UT
* Message-ID: <65482626-78197452$openv.wikispaces>
* In-Reply-To: <65482626-68453028$openv.wikispaces>

Ich wurde gefragt wie ich die Parametern entdeckt habe.
Mir hat geholfen größere Speicher Blöcke mit Optolink auszulesen. Ich kann bis zu 56 Byte Blöcke abfragen.
Zum Beispiel ‘read size 56 addr 0x16D0’
Da habe ich größere Speicher Blöcke ausgelesen und im Hex Dump Werte von Vitrotronic Panel gesucht und die entsprechend Adressen geortet und beobachtet.
Ich habe leider keine weiter Info über andere Adressen. Ich habe leider auch keine Dokumentationen dies bezüglich.

s0170071, 2018-04-11:
wenn bei 
16D0 kWh Strom Warmwasser
16D1 kWh Strom Heizung
16D2 kWh Wärme Warmwasser
16D3 kWh Wärme Heizung
die Werte der aktuellen Woche liegen, wo finde ich dann die Werte der vergangenen 51 Wochen ? 
Oder kann man von jeder dieser Adressen mehrere Werte (also einen Block) abfragen ?

WojtusJ, 2020-04-16:
@WikiSpaxe, how did you manage to read larger data block? Is it possible with OPENV? I want to find Gas Consumption in my heater (Vitodens 200-W), and I am looking for a way to do this. The addresses you mention return an error in my case, so it has to be somewhere else unfortunately. :-/