openv/openv #174: Vitocal Warmwasser einmalig nachladen

WikiSpaxe, 2017-10-26:
* Subject: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: PGMonster <userid-18659820@openv.wikispaces>
* Date: Wed,  1 Jan 2014 14:55:31 UT
* Message-ID: <65536402-68559992$openv.wikispaces>

Hallo,

hat noch jemand Adressen für die Vitocal (mit WRP300) herausgefunden? Ich suche schon seit einiger Zeit nach der Adresse für "Warmwasser einmalig nachladen". Kann mir jemand einen Tip geben, ob das auch über eine Betriebsartenumschaltung bzw. Partybetrieb realisierbar ist. Ich bin für jede noch nicht in der Adressenliste dokumentierte Adresse dankbar.

Tobi

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Fri, 21 Feb 2014 14:12:39 UT
* Message-ID: <65536402-69877606$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

hi Tobi,
schau mal hier: http://openv.wikispaces.com/share/view/65878710
die datei ecnEventType.xml bringt dich weiter. mit Excel öffnen, dann die erste spalte alphabetisch sortieren. die einträge mit "WPR3_..." sind glaub ich was du suchst.
lg
steve

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: PGMonster <userid-18659820@openv.wikispaces>
* Date: Fri, 21 Feb 2014 22:11:48 UT
* Message-ID: <65536402-69893660$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

Hallo Steve,

danke für den Link. Ich kannte den Beitrag schon und habe die Adresse aus dieser Datei auch schon probiert (0xB020). Leider kann ich die Adressen 0xB... an meiner Regelung weder lesen noch schreiben. Hast Du eine Vitocal? Kannst Du die Adressen lesen/schreiben?

Tobi

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Sat, 22 Feb 2014 14:50:06 UT
* Message-ID: <65536402-69902320$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

ich hab Vitocal-300A mit Vitotronic 200. geräteerkennung: 2049.
ich hab die gewollte addressen in eine XLS datei kopiert, dazu entwickle gerade ein VBA program damit man die daten direkt aus Excel lesen kann. eine erste rudimentäre version lauft bereits.
ich hatte auch schwierigkeiten am anfang mit dem optolink adapter, das solltest du auch prüfen ob zumindest die poll-bytes 05 empfangen kannst.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: PGMonster <userid-18659820@openv.wikispaces>
* Date: Sat, 22 Feb 2014 22:17:18 UT
* Message-ID: <65536402-69907576$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

Hallo Steve,

kleines Missverständnis. Ich habe seit über drei Jahren einen Eigenbau-Optolink an einer Wago-Steuerung hängen und lese damit meine Vitocal 300G aus (mit WRP300). Ich suche nur immernoch die Adresse für "WW einmalig nachladen". Die Adressen der Vitotronic und des WPR300 unterscheiden sich leider ein wenig (die Vitotronic ist kommunikativer als der alte WPR). 
So long! 
Tobi

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Sun, 23 Feb 2014 12:28:01 UT
* Message-ID: <65536402-69914080$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

ja, ich hab auch einige unstimmigkeiten, bei einiger addressen stimmt die parameter länge nicht...
ja, die adresse mit B... kommt komisch vor. ich würde statdessen bei WW_Betriebsart suchen, vielleicht ist unter einen eingabewert verborgen. hast du schon mal dort probiert?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: smbunn <userid-40480070@openv.wikispaces>
* Date: Mon, 24 Feb 2014 03:45:09 UT
* Message-ID: <65536402-69929324$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

I found a web page from Viessmann that lists most of their devices and the addresses. Look at my GitHub to see the documents.  https://github.com/smbunn/Viessmann-Control

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 24 Feb 2014 09:34:32 UT
* Message-ID: <65536402-69934386$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

hi smbunn,
thx for the link, i would be interested in VT200WO1AB. on your site there is only a subset of the addresses. do you have the complete list of all parameters as well? the I could compare it with the one listed under the link see my first answer.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 24 Feb 2014 11:32:42 UT
* Message-ID: <65536402-69935770$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

@pgmonster,
hab gerad geschaut, die adresse B020 für 1xWW ist nur für VT200WO1AB spezifiziert, aber nicht für WRP300. ausserdem für Vitocal 300G ist der datensatz entsprechend zu VT200WO1B anzuwenden, und nicht WPR300. (siehe seite 3, http://www.viessmann.de/content/dam/internet-global/pdf_documents/toptechnik/tt-vitotronic.pdf)
das bedeutet, du solltest eigentlich die daten ändern können...teoretisch.
nur praktisch, wie gesagt, bei mir funktioniert nicht mal das auslesen einiger parametern, und wieder andere parametern haben anderen byte-länge wie spezifiziert...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal Warmwasser einmalig nachladen 
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Sun, 21 Jun 2015 18:32:45 UT
* Message-ID: <65536402-82741104$openv.wikispaces>
* In-Reply-To: <65536402-68559992$openv.wikispaces>

Hi PGmonster,
jetzt bin ich auch auf dieses problem gestürtzt, wollte "1xWW-Bereitung" setzen, geht leider nicht, stattdessen wird über Adresse B020 "Manueller Betrieb" aktiviert.
Hast Du schon inzwischen die richtige Adresse für 1xWW Bereitung (WPR300) herausgefunden?
VG,steve

dumpfheimer, 2021-09-18:
Servus!

es ist zwar schon alt, aber ich hatte gerade dasselbe Problem. Die Lösung ist, dass B020 kein Status Register ist, sondern ein Modus.

0 dürfte Normalbetrieb sein
1 Manueller Betrieb
2 Warmwasserbereitung auf Temp2

ob es mehr gibt weiß ich nicht.

LG