openv/openv #185: Fehler beim Setzen der Zeit

WikiSpaxe, 2017-10-26:
* Subject: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: Morum <userid-47052016@openv.wikispaces>
* Date: Sun, 16 Feb 2014 14:02:34 UT
* Message-ID: <66158564-69730148$openv.wikispaces>

Hallo,
ich bekomme folgenden Fehler (Protokoll P300):
pi@raspberrypi ~ $ vclient -h 127.0.0.1:3002  -c "debug on",setSystemTime
debug on:

setSystemTime:
DEBUG:Sun Feb 16 13:59:45 2014 : Befehl: setSystemTime
DEBUG:Sun Feb 16 13:59:45 2014 : Process 7048 tries to aquire lock
DEBUG:Sun Feb 16 13:59:45 2014 : Process 7048 got lock
DEBUG:Sun Feb 16 13:59:45 2014 : >FRAMER: open device /dev/ttyAMA0 ProtocolID 41
DEBUG:Sun Feb 16 13:59:45 2014 : konfiguriere serielle Schnittstelle /dev/ttyAMA0
DEBUG:Sun Feb 16 13:59:45 2014 : >SEND: 04
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: len=1 05 (1150.0 ms)
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: received 05
DEBUG:Sun Feb 16 13:59:46 2014 : >FRAMER: closed
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 16
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 00
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 00
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: len=1 06 (20.0 ms)
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: received 06
DEBUG:Sun Feb 16 13:59:46 2014 : >FRAMER: opened
DEBUG:Sun Feb 16 13:59:46 2014 : aktuelle Sys.Zeit 20 14 02 16 00 13 59 46
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 41
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 05
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 01
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: F4
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 08
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 8E
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 08
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 98
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: len=1 06 (30.0 ms)
DEBUG:Sun Feb 16 13:59:46 2014 : <RECV: received 06
DEBUG:Sun Feb 16 13:59:46 2014 : >FRAMER: addr was still active FE06
DEBUG:Sun Feb 16 13:59:46 2014 : >FRAMER: Command send
DEBUG:Sun Feb 16 13:59:46 2014 : >SEND: 20
DEBUG:Sun Feb 16 13:59:46 2014 : >FRAMER: no preset result
DEBUG:Sun Feb 16 13:59:47 2014 : <RECV: len=1 15 (260.0 ms)
DEBUG:Sun Feb 16 13:59:47 2014 : <RECV: received 15
DEBUG:Sun Feb 16 13:59:52 2014 : <RECV: read timeout
DEBUG:Sun Feb 16 13:59:52 2014 : <RECV: received
DEBUG:Sun Feb 16 13:59:52 2014 : >FRAMER: read failure
DEBUG:Sun Feb 16 13:59:52 2014 : Fehler recv, Abbruch
DEBUG:Sun Feb 16 13:59:52 2014 : Fehler beim ausfuehren von setSystemTime
ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
<RECV: read timeout
>FRAMER: read failure
Fehler recv, Abbruch
Fehler beim ausfuehren von setSystemTime

wie kann ich die Zeit der Anlage setzen?
Gruß,
Morum

WikiSpaxe, 2017-10-26:
* Subject: Re: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: Morum <userid-47052016@openv.wikispaces>
* Date: Sun, 16 Feb 2014 17:16:47 UT
* Message-ID: <66158564-69731778$openv.wikispaces>
* In-Reply-To: <66158564-69730148$openv.wikispaces>

auch wenn ich auf KW2 Protokoll umstelle klappt es nicht:
pi@raspberrypi ~ $ vclient -h 127.0.0.1:3002  -c "debug on",setSystemTime
debug on:

setSystemTime:
DEBUG:Sun Feb 16 17:12:52 2014 : Befehl: setSystemTime
DEBUG:Sun Feb 16 17:12:52 2014 : Process 7141 tries to aquire lock
DEBUG:Sun Feb 16 17:12:52 2014 : Process 7141 got lock
DEBUG:Sun Feb 16 17:12:52 2014 : >FRAMER: open device /dev/ttyAMA0 ProtocolID 00
DEBUG:Sun Feb 16 17:12:52 2014 : konfiguriere serielle Schnittstelle /dev/ttyAMA0
DEBUG:Sun Feb 16 17:12:52 2014 : aktuelle Sys.Zeit 20 14 02 16 00 17 12 52
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 04
DEBUG:Sun Feb 16 17:12:52 2014 : >FRAMER: no preset result
DEBUG:Sun Feb 16 17:12:52 2014 : Warte auf 05
DEBUG:Sun Feb 16 17:12:52 2014 : <RECV: 05 (60.0 ms)
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 01
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: F4
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 08
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 8E
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 08
DEBUG:Sun Feb 16 17:12:52 2014 : >SEND: 20
DEBUG:Sun Feb 16 17:12:52 2014 : >FRAMER: no preset result
DEBUG:Sun Feb 16 17:12:55 2014 : <RECV: len=1 05 (2260.0 ms)
DEBUG:Sun Feb 16 17:12:55 2014 : <RECV: received 05
DEBUG:Sun Feb 16 17:12:55 2014 : 05 -> SYNC (NOT OK)
DEBUG:Sun Feb 16 17:12:55 2014 : SYNC (NOT OK)
SYNC (NOT OK)
pi@raspberrypi ~ $ vclient -h 127.0.0.1:3002  -c getSystemTime
getSystemTime:
So,16.02.2014 17:09:22
---------
die Uhr der Anlage geht  weiterhin 2,5 min nach.

WikiSpaxe, 2017-10-26:
* Subject: Re: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: luftdieb <userid-40691500@openv.wikispaces>
* Date: Fri,  9 Jan 2015 18:38:20 UT
* Message-ID: <66158564-78334294$openv.wikispaces>
* In-Reply-To: <66158564-69730148$openv.wikispaces>

Der Post ist zwar schon ~ 1 Jahr alt, aber ich bin auf der Suche nach einer Lösung für das gleiche Problem hier gelandet. Da ich die Lösung herausgefunden habe, ist sie vielleicht für andere nützlich. Ich habe eine Vitodens 200-W WB2C und das setzen der Zeit in der Heizung habe ich über ein Shellscript realisiert mit folgendem Inhalt: 
export Startzeit=`date "+20 %g %m %d 0%u %H %M %S"`
export file=/var/log/vitoSetTime
echo getSystemTime >$file
echo unit off >>$file
echo setSystemTime $Startzeit >>$file
echo unit on >>$file
#echo setSystemTime >>$file
#tail $file

/usr/local/bin/vclient -h localhost:3002 -f $file
/usr/local/bin/vclient -h localhost:3002 'getSystemTime'

Der Fehler im Post liegt an der fehlenden Übergabe der Systemzeit. Vorher muss die Einheit deaktiviert werden (unit off) und die Systemzeit im 2er Häppchen übertragen werden (wie es im Post bei getSystemTime zu sehen ist). Zwischen Datum und Uhrzeit ist der Wochentag als Zahl. Diese wird beim setzen der Zeit nicht berücksichtigt, da die Heizung das vermutlich selbst umrechnet. Zum Schluss habe ich noch eine Kontrollausgabe eingebaut. Die Datei kann über einen CronJob z.B. 1x am Tag ausgeführt werden.

Gruß
luftdieb

WikiSpaxe, 2017-10-26:
* Subject: Re: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Wed, 22 Feb 2017 19:32:33 UT
* Message-ID: <66158564-93108571$openv.wikispaces>
* In-Reply-To: <66158564-69730148$openv.wikispaces>

Ich versuche gerade unter dem P300-Protokoll den Beginn und das Ende der Ferien zu setzen. Im Grunde genommen ist der Befehl dazu ja vergleichbar mit setSystemTime. Doch leider bekomme ich immer eine Fehlermeldung

vctrld>getVitoFerienBeginn
DEBUG:Wed Feb 22 19:28:48 2017 : Befehl: getVitoFerienBeginn
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 41
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 05
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 00
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 01
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 23
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 09
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 08
DEBUG:Wed Feb 22 19:28:48 2017 : >SEND: 3A
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 06 (60.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: received 06
DEBUG:Wed Feb 22 19:28:48 2017 : ▒▒~▒
DEBUG:Wed Feb 22 19:28:48 2017 : >FRAMER: Command send
DEBUG:Wed Feb 22 19:28:48 2017 : >FRAMER: no preset result
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 41 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 0D (10.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 01 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 01 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 23 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 09 (10.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 08 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 19 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: received 41 0D 01 01 23 09 08 19
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 70 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 01 (10.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 01 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 04 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 00 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 00 (10.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 00 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: len=1 D2 (0.0 ms)
DEBUG:Wed Feb 22 19:28:48 2017 : <RECV: received 70 01 01 04 00 00 00 D2
19 70 01 01 04 00 00 00
DEBUG:Wed Feb 22 19:28:48 2017 : Empfangen: 19 70 01 01 04 00 00 00
vctrld>setVitoFerienBeginn 20 17 04 01 00 00 00 00
DEBUG:Wed Feb 22 19:29:04 2017 : Befehl: setVitoFerienBeginn 20 17 04 01 00 00 00 00
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 41
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 06
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 00
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 02
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 23
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 09
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 08
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 00
DEBUG:Wed Feb 22 19:29:04 2017 : >SEND: 3C
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 06 (40.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: received 06
DEBUG:Wed Feb 22 19:29:04 2017 :
DEBUG:Wed Feb 22 19:29:04 2017 : >FRAMER: Command send
DEBUG:Wed Feb 22 19:29:04 2017 : >FRAMER: no preset result
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 41 (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: received 41
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 05 (10.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: received 05
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 01 (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 02 (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 23 (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 09 (10.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 08 (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: len=1 3C (0.0 ms)
DEBUG:Wed Feb 22 19:29:04 2017 : <RECV: received 01 02 23 09 08 3C
DEBUG:Wed Feb 22 19:29:04 2017 : >FRAMER: unexpected length 8 08
DEBUG:Wed Feb 22 19:29:04 2017 : Fehler recv, Abbruch
DEBUG:Wed Feb 22 19:29:04 2017 : Fehler beim ausfuehren von setVitoFerienBeginn 20 17 04 01 00 00 00 00
ERR:
>FRAMER: Command send
>FRAMER: unexpected length 8 08
Fehler recv, Abbruch
Fehler beim ausfuehren von setVitoFerienBeginn 20 17 04 01 00 00 00 00

Habt ihr eine Idee?

WikiSpaxe, 2017-10-26:
* Subject: Re: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: heat7 <userid-38967508@openv.wikispaces>
* Date: Fri,  1 Sep 2017 18:08:35 UT
* Message-ID: <66158564-94955857$openv.wikispaces>
* In-Reply-To: <66158564-69730148$openv.wikispaces>

Hallo thopeter,

Läuft bei dir das P300-Protokoll mit Senden und Lesen? Ich verwende VScotHO1 ID=20CB Protokoll:P300: Lesen funktioniert, jedoch Senden bereitet Probleme. 

Besten Dank

WikiSpaxe, 2017-10-26:
* Subject: Re: Fehler beim Setzen der Zeit
* Wiki-Page: vcontrold
* From: hsellsch <userid-58957011@openv.wikispaces>
* Date: Mon,  9 Oct 2017 07:29:16 UT
* Message-ID: <66158564-95372891$openv.wikispaces>
* In-Reply-To: <66158564-69730148$openv.wikispaces>

Hallo heat7,

bei mir ging senden mit P300 und iD 20cb auch nicht. 
Ich habe in der vcontrold.xml folgende Änderung vorgenommen

....
  <protocols>
      <protocol name='P300'>
	  <pid>41</pid>
	  <macros>
		<macro name='GETADDR'>
		    <command>SEND 00 01</command>
		</macro>
		<macro name='SETADDR'>
		
		    <command>SEND 00 02</command>
		</macro>
	  </macros>

Mit dieser Änderung konnte ich bisher alle Werte Senden.

uelpenich, 2022-07-24:
setSystemTime funktioniert für mich nicht. 

Meine Heizung: Viessmann Vitodenz 300W Typ B3HA, das Greät meldet sich mit dem DevType 20CB, Baujahr 2015.
Ich möchte / muss die Systemzeit setzen, das bei einem schon relativ kurzem Stromausfall die Systemzeit verloren geht, und das Gerät erst weiter heizt, wenn die Uhrzeit eingegeben wurde (vergl. die ellenlangen Diskussionen zum Thema "Gangreserve" und gealterte Stützakkus). 
Ich habe alle gefundenen Tipps zu "setSystemTime" durchprobiert und hänge bei einer Fehlermeldung, die der von **thopeter** zum Setzen des Ferien Beginns sehr ähnlich ist. 

Nun die Details:
_Die Systemzeit kann ich richtig lesen._

> pi@ioBroker:~ $ vclient -h localhost:3002 'debug on','getSystemTime','debug off'
> debug on:
> 
> getSystemTime:
> DEBUG:Sun Jul 24 17:51:28 2022 : Befehl: getSystemTime
> DEBUG:Sun Jul 24 17:51:28 2022 : Process 5435 tries to aquire lock
> DEBUG:Sun Jul 24 17:51:28 2022 : Process 5435 got lock
> DEBUG:Sun Jul 24 17:51:28 2022 : >FRAMER: open device 192.168.39.12:8234 ProtocolID 41
> DEBUG:Sun Jul 24 17:51:28 2022 : ClI Net: verbunden 192.168.39.12:8234 (FD:1)
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 04
> DEBUG:Sun Jul 24 17:51:28 2022 : <RECV: len=1 05 (0.0 ms)
> DEBUG:Sun Jul 24 17:51:28 2022 : <RECV: received 05
> DEBUG:Sun Jul 24 17:51:28 2022 : >FRAMER: closed
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 16
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:28 2022 : <RECV: len=1 05 (0.0 ms)
> DEBUG:Sun Jul 24 17:51:28 2022 : <RECV: received 05
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 16
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:28 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: len=1 05 (2000.0 ms)
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: received 05
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 16
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: len=1 06 (20.0 ms)
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: received 06
> DEBUG:Sun Jul 24 17:51:30 2022 : >FRAMER: opened
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 41
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 05
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 00
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 01
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 08
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 8E
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: 08
> DEBUG:Sun Jul 24 17:51:30 2022 : >SEND: A4
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: len=1 06 (100.0 ms)
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: received 06
> DEBUG:Sun Jul 24 17:51:30 2022 : >FRAMER: addr was still active FE06
> DEBUG:Sun Jul 24 17:51:30 2022 : >FRAMER: Command send
> DEBUG:Sun Jul 24 17:51:30 2022 : >FRAMER: no preset result
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: len=8 41 (0.0 ms)
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: received 41 0D 01 01 08 8E 08 20
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: len=8 22 (0.0 ms)
> DEBUG:Sun Jul 24 17:51:30 2022 : <RECV: received **22 07 24 07 17 51 24 AD**
> DEBUG:Sun Jul 24 17:51:30 2022 : So,24.07.2022 17:51:24
> So,24.07.2022 17:51:24
> debug off:
> DEBUG:Sun Jul 24 17:51:30 2022 : Befehl: debug off
> pi@ioBroker:~ $

Hier fällt auf, dass die gelesenen 8 Bytes eine andere Reihenfolge bzw. Bedeutung haben im Vergleich mit der beschriebenen setSystemTime Lösung oben im Thread. Zu Beginn fehlt das Jahrhundert, umd am Ende finde ich ein Byte AD, dessen Sinn ich nicht verstehe. 

> pi@ioBroker:~ $ date
> So 24. Jul 17:59:45 CEST 2022
> pi@ioBroker:~ $ date "+20 %g %m %d 0%u %H %M %S"
> 20 22 07 24 07 17 59 45

**Frage:** kann ich von dem Datenformat der Abfrage **getSystemTime** auf das Datenformat des Setzbefehls **setSystemTime** schliessen? 

[... 106 lines cut ...]
>       <pid>41</pid>
>       <macros>
>         <macro name="GETADDR">
>           <command>SEND 00 01</command>
>         </macro>
>         <macro name="SETADDR">
>           <command>SEND 00 02</command>
>         </macro>
>       </macros>
>       <commands>
>         <command name="getaddr">
>           <send>GETADDR $addr $hexlen;RECV $len $unit</send>
>         </command>
>         <command name="setaddr">
>           <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
>         </command>
>         <command name="gettestaddr">
>           <send/>
>         </command>
>       </commands>
>     </protocol>

**Frage**: ist da ein Problem mit dem Parameter **RECV 1** im send Befehlim setaddr command **<send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>** ? 

**Frage**: welche Parameter müssen in welcher Schreibweise im setSystemTime angegeben werden?