openv/openv #190: Vitoladens 300C + Vitotronic 200, Lesen geht prima, Schreiben nicht so.

WikiSpaxe, 2017-10-26:
* Subject: Vitoladens 300C + Vitotronic 200, Lesen geht prima, Schreiben nicht so.
* Wiki-Page: home
* From: heizknecht <userid-47847190@openv.wikispaces>
* Date: Wed,  5 Mar 2014 20:03:29 UT
* Message-ID: <66420352-70245032$openv.wikispaces>

Hallo 

bis jetzt läuft alles eigentlich super. Ich habe mir die vorcompilierten vcontrold/vclient binaries auf meinen raspberry gepackt daran den Viessmann Optokoppler gehangen und ich kann ohne Probleme alle 5 min. Daten auslesen und mit rrdb aufbereiten. Soweit so klasse und ich bin damit auch so schon sehr zufrieden, wäre aber noch einen Tacken glücklicher, wenn ich die Heizung auch steuern könnte. 

Das ist mir aber bisher nicht geglückt. Ich kriege dann immer einen Fehler wie diesen, wenn ich z.B. den Neigungswinkel verändern will:

vctrld>setNeigungM1 1.4
DEBUG:Wed Mar  5 19:45:43 2014 : Befehl: setNeigungM1 1.4
DEBUG:Wed Mar  5 19:45:43 2014 : Process 16340 tries to aquire lock
DEBUG:Wed Mar  5 19:45:43 2014 : Process 16340 got lock
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: open device /dev/vitoir0 ProtocolID 41
DEBUG:Wed Mar  5 19:45:43 2014 : konfiguriere serielle Schnittstelle /dev/vitoir0
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 04
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: len=1 05 (80.0 ms)
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: received 05
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: closed
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 16
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 00
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 00
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: len=1 06 (10.0 ms)
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: received 06
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: opened
DEBUG:Wed Mar  5 19:45:43 2014 : Send Exp:V*10 [V=1.400000]
DEBUG:Wed Mar  5 19:45:43 2014 : Typ: short (Bytes: 0E )  
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 41
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 05
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 01
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: F4
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 27
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: D3
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 01
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: F5
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: len=1 06 (10.0 ms)
DEBUG:Wed Mar  5 19:45:43 2014 : <RECV: received 06
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: addr was still active FE06
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: Command send
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 0E
DEBUG:Wed Mar  5 19:45:43 2014 : >SEND: 00
DEBUG:Wed Mar  5 19:45:43 2014 : >FRAMER: no preset result
DEBUG:Wed Mar  5 19:45:44 2014 : <RECV: len=1 15 (250.0 ms)
DEBUG:Wed Mar  5 19:45:44 2014 : <RECV: received 15
DEBUG:Wed Mar  5 19:45:49 2014 : <RECV: read timeout
DEBUG:Wed Mar  5 19:45:49 2014 : <RECV: received
DEBUG:Wed Mar  5 19:45:49 2014 : >FRAMER: read failure
DEBUG:Wed Mar  5 19:45:49 2014 : Fehler recv, Abbruch
DEBUG:Wed Mar  5 19:45:49 2014 : Fehler beim ausfuehren von setNeigungM1 1.4
ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
<RECV: read timeout
>FRAMER: read failure
Fehler recv, Abbruch
Fehler beim ausfuehren von setNeigungM1 1.4
vctrld>

Ich benutze das p300 Protokoll und hab mir die xml Dateien aus dem svn gezogen.

Muss man an der Anlage erst noch was konfigurieren, damit das geht ?

Gruß
   heizknecht

P.S.  Ich versteh das Protokoll nur so halb, der erste Teil der Daten wird mit Prüfsumme und header geschickt, aber der zweite Teil (der short wert) 0xE 0x0  dann nicht ?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoladens 300C + Vitotronic 200, Lesen geht prima, Schreiben nicht so.
* Wiki-Page: home
* From: fnobis <userid-15534813@openv.wikispaces>
* Date: Thu,  6 Mar 2014 08:50:15 UT
* Message-ID: <66420352-70264736$openv.wikispaces>
* In-Reply-To: <66420352-70245032$openv.wikispaces>

Moin,

das sieht für mich stark danach aus, als ob der P300 Frame ohne die Argumente bereits an die Anlage geschickt wird. Habe ich bei anderen Befehlen auch schon gesehen.

Ich müsste mal etwas Zeit finden, um das genauer im Source zu verfolgen. Das P300 Protokoll ist noch nicht so lange im Source.

Gruß
Frank

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoladens 300C + Vitotronic 200, Lesen geht prima, Schreiben nicht so.
* Wiki-Page: home
* From: heizknecht <userid-47847190@openv.wikispaces>
* Date: Wed, 12 Mar 2014 00:44:45 UT
* Message-ID: <66420352-70430730$openv.wikispaces>
* In-Reply-To: <66420352-70245032$openv.wikispaces>

Also wenn ich das richtige verstehe, wird bei sowas im XML

		<command name="setaddr">
		     <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
		</command>

nach jedem semicolon einmal framer_send aufgerufen, was aber dann zu früh (vor SEND BYTES) die CRC sendet. 
Man müsste framer_send  in  framer_send_begin,  framer_send_partial und framer_send_end aufspalten und den context halt in eine eigene struct packen.

Oder aber die SEND compilePtr vorher konsolidieren...

Gruß
   heizknecht

abrits56, 2024-10-30:
Hallo,
ist zwar wohl schon etwas her....

Ich habe die gleiche Heizung/Steuerung bin aber erst jetzt zu vcontrold gekommen. 

Einige Werte kann kann ich auslesen, aber ansonsten fehlen mir einfach die Infos für die xml Eintragungen.

Deine vcontrold.xml und vito.xml wären natürlich für mich von höchstem Interesse.

Grüße Johannes