openv/openv #200: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)

WikiSpaxe, 2017-10-26:
* Subject: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Sat, 17 May 2014 21:13:44 UT
* Message-ID: <67582694-72467292$openv.wikispaces>

Habe nun endlich meinen , nach eurer Anleitung, selbstgebastelten Optolink-Adapter am laufen. Vielen Dank für diese Anleitung!
Nun bekomme ich aber leider die Geraete-ID meiner Heizung nicht heraus gekitzelt. Leider steht diese ebenfalls nicht in der Geraetelist. Und die Software voIdent spuckt auch nix aus. Bleibt mir jetzt nur das Ausprobien aller möglichen, noch unbekannten ID-Hexwerte? Kann jemand helfen? Hat jemand eine Loseung oder bin ich in Deutschland der einzige mit dieser Anlagen-Kombination? :-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Sat, 17 May 2014 22:06:26 UT
* Message-ID: <67582694-72468218$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Koennte ne 2048 oder 2049 sein. Bei der getTempA-Abfrage im vcontrold kommt aber leider ein falscher Wert. 2.500000 Grad. Irgendetwas stimmt da wohl nicht. Ich stocher mal weiter im Dunkel. Vielleicht wird ja Licht :-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 19 May 2014 09:22:19 UT
* Message-ID: <67582694-72499974$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

hi, ich hab ein Vitocal 300-A mit Vitotronic 200 (WO1B).
welche HW/SW verwendest du fürs auslesen?
Vcontrold.xml hat eine macke für den 300 protokol, siehe: http://openv.wikispaces.com/share/view/66289078

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Mon, 19 May 2014 11:54:16 UT
* Message-ID: <67582694-72502424$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Ich habe den Optolink nach Bauanleitung Nr.3 gebaut. Also mit RS232-Schnittstelle.
Ich erhalte im Terminal-Programm die 05-Meldung von der Heizung.
Mit vcontrol kann ich mich ebenfalls am System anmelden. Aber leider gibt mir die Heizung "falsche" Werte zurück.
Ich vermute, das die vito.xml nicht auf mein System passt.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 19 May 2014 22:18:42 UT
* Message-ID: <67582694-72523118$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

was ist es denn einen "falschen wert"?
kannst bitte ein antwort-log hier posten?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Mon, 19 May 2014 22:37:22 UT
* Message-ID: <67582694-72523768$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

welchen vito.xml hast du probiert?
es kann durchaus sein, dass die addressen nicht mit deinem system zusammenpassen.
hast du schon dein gerät-id auslesen können?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Tue, 20 May 2014 07:40:30 UT
* Message-ID: <67582694-72534954$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Habe die vcontrold-Software aus Subversion augecheckt und verwende sie auf einem RaspberryPi.
Die vito.xml und die vcontrold.xml habe ich aus dem Subversion-Ordner ...xml-32/xml/... . Also KW2-Protokoll. Dort habe ich die "vermutete" Geraete-ID (2046, 2047, 2048 oder 2049) eingetragen und bei "getTempA" die Adressen aus der Adress-Tabelle dieses Wiki.

Soweit ich das sehe liegt die Geraete-ID bei allen Geraeten auf der selben Adresse. Also sollte das doch mit der passenden Geraete-ID ohne weitere (Adress-)Aenderung funktionieren. Oder sehe ich das falsch?

Mit "falschem Wert" meine ich, das die Antwort bei der Abfrage "getTempA" mit allen getesteten ID's immer "2.50000 Grad C" ist. Da stimmt was nicht.

Ich habe zur Abfrage der Geraete-ID jetzt nicht den Fehlercode griffbereit. Ich versuche es gleich nochmal und melde mich dann wieder mir der Info.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Tue, 20 May 2014 07:53:36 UT
* Message-ID: <67582694-72535132$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Ich glaube, ich sehe das etwas falsch ist. Die Anlage antwortet ja immer nur mit 05...?
Bin ich vielleicht doch noch nicht richtig verbunden? Hardware-Fehler auf der selbstgebauten Platine?

Hier "getTempA":

vctrld>getTempA
DEBUG:Tue May 20 05:51:09 2014 : Befehl: getTempA
DEBUG:Tue May 20 05:51:09 2014 : >SEND: 04
DEBUG:Tue May 20 05:51:09 2014 : >FRAMER: no preset result
DEBUG:Tue May 20 05:51:09 2014 : Warte auf 05
DEBUG:Tue May 20 05:51:09 2014 : <RECV: 05 (530.0 ms)
DEBUG:Tue May 20 05:51:09 2014 : >SEND: 01
DEBUG:Tue May 20 05:51:09 2014 : >SEND: F7
DEBUG:Tue May 20 05:51:09 2014 : >SEND: 6F
DEBUG:Tue May 20 05:51:09 2014 : >SEND: 01
DEBUG:Tue May 20 05:51:09 2014 : >FRAMER: no preset result
DEBUG:Tue May 20 05:51:11 2014 : <RECV: len=1 05 (1980.0 ms)
DEBUG:Tue May 20 05:51:11 2014 : <RECV: received 05
DEBUG:Tue May 20 05:51:11 2014 : Typ: char (in float: 5.000000)
DEBUG:Tue May 20 05:51:11 2014 : (FLOAT) Exp:V/2 [B0:05 B1:00 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
DEBUG:Tue May 20 05:51:11 2014 : 2.500000 Grad Celsius
2.500000 Grad Celsius
vctrld>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Tue, 20 May 2014 11:21:14 UT
* Message-ID: <67582694-72538334$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Ich hab's gefunden. Die Platine ist defekt. Habe jetzt mit Drahtbrücken gearbeitet, da einige Kupferverbindungen sich gelöst haben. Wahrscheinlich aufgrund mechanischer Belastung beim Einbau ins Gehäuse.

Jetzt kann ich zumindest senden. Und bekomme Werte zurück. Habe übrigends die xml-Dateien aufs P300 Protokoll geändert. Ggf. gehe ich zurück auf KW2.

vctrld>getDevType
DEBUG:Tue May 20 11:17:30 2014 : Befehl: getDevType
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 41
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 05
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 00
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 01
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 00
DEBUG:Tue May 20 11:17:30 2014 : >SEND: F8
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 08
DEBUG:Tue May 20 11:17:30 2014 : >SEND: 06
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 06 (10.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: received 06
DEBUG:Tue May 20 11:17:30 2014 : 
DEBUG:Tue May 20 11:17:30 2014 : >FRAMER: Command send
DEBUG:Tue May 20 11:17:30 2014 : >FRAMER: no preset result
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=5 41 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 F8 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 08 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 20 (10.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: received 41 0D 01 01 00 F8 08 20
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=3 48 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 00 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 00 (10.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 03 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 46 (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: len=1 CB (0.0 ms)
DEBUG:Tue May 20 11:17:30 2014 : <RECV: received 48 02 09 00 00 03 46 CB
DEBUG:Tue May 20 11:17:30 2014 : 20 48 02 09 00 00 03 46 -> UNKNOWN
DEBUG:Tue May 20 11:17:30 2014 : UNKNOWN
UNKNOWN

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Tue, 20 May 2014 12:37:30 UT
* Message-ID: <67582694-72539918$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

So, das heisst dann wohl, wenn ich die Daten oben richtig lese, das ich eine ID 2048 habe, korrekt?
Das heisst nun muss ich die richtigen Adressen finden fuer die Werte. Ich fange mal mit der Aussentemperatur an. Das geht bisher nicht. Es kommt jetzt 0.0000 Grad Celsius zurück. Das heisst wahrscheinlich lese ich die falsche Adresse (01 01) aus.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: deathrattle <userid-49716466@openv.wikispaces>
* Date: Tue, 20 May 2014 15:08:05 UT
* Message-ID: <67582694-72542562$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

Jetzt klappt es...

vctrld>getTempA
DEBUG:Tue May 20 15:07:25 2014 : Befehl: getTempA
DEBUG:Tue May 20 15:07:25 2014 : Process 2516 tries to aquire lock
DEBUG:Tue May 20 15:07:25 2014 : Process 2516 got lock
DEBUG:Tue May 20 15:07:25 2014 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
DEBUG:Tue May 20 15:07:25 2014 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
DEBUG:Tue May 20 15:07:25 2014 : >SEND: 04
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 05 (1990.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: received 05
DEBUG:Tue May 20 15:07:27 2014 : >FRAMER: closed
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 16
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 00
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 00
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 06 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: received 06
DEBUG:Tue May 20 15:07:27 2014 : >FRAMER: opened
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 41
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 05
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 00
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 01
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 01
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 01
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 02
DEBUG:Tue May 20 15:07:27 2014 : >SEND: 0A
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 06 (10.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: received 06
DEBUG:Tue May 20 15:07:27 2014 : >FRAMER: addr was still active FE06
DEBUG:Tue May 20 15:07:27 2014 : >FRAMER: Command send
DEBUG:Tue May 20 15:07:27 2014 : >FRAMER: no preset result
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=2 41 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: received 41 07
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=4 01 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 02 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 02 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 01 (10.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: len=1 10 (0.0 ms)
DEBUG:Tue May 20 15:07:27 2014 : <RECV: received 01 01 01 01 02 02 01 10
DEBUG:Tue May 20 15:07:27 2014 : Typ: short (in float: 258.000000)
DEBUG:Tue May 20 15:07:27 2014 : (FLOAT) Exp:V/10 [B0:02 B1:01 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
DEBUG:Tue May 20 15:07:27 2014 : 25.799999 Grad Celsius
25.799999 Grad Celsius

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Thu, 22 May 2014 10:18:31 UT
* Message-ID: <67582694-72605584$openv.wikispaces>
* In-Reply-To: <67582694-72467292$openv.wikispaces>

hi, nur als info, mit P300 musst du nicht unbedingt ein 04 schicken, ich sende immer nur 16 00 00, erwarte 06 antwort, und dann gehts weiter los. so geht es ein bischen schneller...

squashball, 2020-02-14:
Guten Abend, ich hänge mich hier mal an diesen alten Thread an.
Ich habe auch eine WO1B. Müsste also id 2049 sein. Gibt es eine vito.xml schon irgendwie zu finden oder kämpft sich da jeder selbst durch? Wäre für Tipps dankbar!
Grüße, der Squashball