openv/openv #201: Vitocal 200-G, Anfängerprobleme

WikiSpaxe, 2017-10-26:
* Subject: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Fri, 11 Jul 2014 00:03:05 UT
* Message-ID: <68716492-73989366$openv.wikispaces>

Hallo zusammen,

ich steige gerade erst in die Materie ein und habe einen Raspberry mit vcontrold installiert. Läuft noch nicht also Dienst, ich starte noch manuell mit vcontrol -n und teste so. 

Habe mir eine kleine vito.xml und vcontrold.xml gebaut auf Basis der hier verlinkten Master. Leider bin ich mir nicht mal sicher welche Device-ID ich verwenden sollte. Also habe ich einfach das KW2-Protokoll angegeben und versucht den Device Type auszulesen - die Adresse scheint ja überall die selbe zu sein. 

Leider bekomme ich zu JEDER Adresse nur 05 als Antwort (je nach erwarteter Länge 1x oder 2x 05). Ich nutze den USB-Adapter von bytelink. Es gibt keine Timeouts, die LEDs am Adapter leuchten wenn ich ein Kommando absetze und die 05 sollte ja auch eine "echte" Antwort der Wärmepumpensteuerung sein (wie gesagt keine Timeouts).

Wie kann ich das nun weiter eingrenzen?

Danke,
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Fri, 11 Jul 2014 05:30:10 UT
* Message-ID: <68716492-73992276$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Also für mich hört sich das erstmal an, als würde der Adapter nix senden - um genauere Infos/Eingrenzungen durchzuführen solltest du in der vcontrold.xml den Debug den Debug-Level auf "Y" setzen und mal das Log nach einer Abfrage posten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Fri, 11 Jul 2014 19:29:16 UT
* Message-ID: <68716492-73997690$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Danke für die Antwort, hier mal ein Auszug des Logs für ein Kommando (Aussentemperatur):

[2768] Fri Jul 11 19:25:00 2014 : Befehl: getOutsideTemp
[2768] Fri Jul 11 19:25:00 2014 : >SEND: 04
[2768] Fri Jul 11 19:25:00 2014 : >FRAMER: no preset result
[2768] Fri Jul 11 19:25:00 2014 : Warte auf 05
[2768] Fri Jul 11 19:25:01 2014 : <RECV: 05 (980.0 ms)
[2768] Fri Jul 11 19:25:01 2014 : >SEND: 01
[2768] Fri Jul 11 19:25:01 2014 : >SEND: F7
[2768] Fri Jul 11 19:25:01 2014 : >SEND: 01
[2768] Fri Jul 11 19:25:01 2014 : >SEND: 01
[2768] Fri Jul 11 19:25:01 2014 : >SEND: 02
[2768] Fri Jul 11 19:25:01 2014 : >FRAMER: no preset result
[2768] Fri Jul 11 19:25:02 2014 : <RECV: len=1 05 (1020.0 ms)
[2768] Fri Jul 11 19:25:03 2014 : <RECV: len=1 05 (1030.0 ms)
[2768] Fri Jul 11 19:25:03 2014 : <RECV: received 05 05
[2768] Fri Jul 11 19:25:03 2014 : Typ: short (in float: 1285.000000)
[2768] Fri Jul 11 19:25:03 2014 : (FLOAT) Exp:V/10 [B0:05 B1:05 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
[2768] Fri Jul 11 19:25:03 2014 : 128.500000 Degrees Celsius

Naiv betrachtet sieht das für mich so aus als würde etwas gesendet werden, und es kommen 05 zurück. Seltsamerweise aber durchgehend 05 für alle Kommandos die ich absetze.

Kann man aus dem Protokoll irgendetwas ablesen, das mir bei der Fehlersuche helfen kann?

Danke!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Fri, 11 Jul 2014 19:49:16 UT
* Message-ID: <68716492-73997818$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi Jens,
also die Anlage gibt auf Deine Kommandos gar nix zurück. Sie sendet einfach die zyklischen 05 weiter (deshalb auch die 1 Sekunde Wartezeit bis zur 05)
So richtig weiß ich aber auch nicht voran es liegen könnte. Sicher, dass die Sendediode richtig sitzt?
Was mich noch wundert ist das erste SENDEN von 04. Sollte IMO nix ausmachen, aber meines Wissens wird bei KW-Protokoll auf 05 gewartet und danach zum lesen 01 F7 und die abzufragende Adresse gesendet ... wozu also die 04 am Anfang?

Gruß
Simon

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Fri, 11 Jul 2014 19:58:01 UT
* Message-ID: <68716492-73997894$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Danke Dir Simon. 

Ich habe jetzt im Protokoll das Macro SYNC angepasst und die 04 entfernt - jetzt sieht das Protokoll wie folgt aus:

[2813] Fri Jul 11 19:55:51 2014 : Befehl: getOutsideTemp
[2813] Fri Jul 11 19:55:51 2014 : Process 2813 tries to aquire lock
[2813] Fri Jul 11 19:55:51 2014 : Process 2813 got lock
[2813] Fri Jul 11 19:55:51 2014 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 00
[2813] Fri Jul 11 19:55:51 2014 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[2813] Fri Jul 11 19:55:51 2014 : >FRAMER: no preset result
[2813] Fri Jul 11 19:55:51 2014 : Warte auf 05
[2813] Fri Jul 11 19:55:52 2014 : <RECV: 05 (660.0 ms)
[2813] Fri Jul 11 19:55:52 2014 : >SEND: 01
[2813] Fri Jul 11 19:55:52 2014 : >SEND: F7
[2813] Fri Jul 11 19:55:52 2014 : >SEND: 01
[2813] Fri Jul 11 19:55:52 2014 : >SEND: 01
[2813] Fri Jul 11 19:55:52 2014 : >SEND: 02
[2813] Fri Jul 11 19:55:52 2014 : >FRAMER: no preset result
[2813] Fri Jul 11 19:55:53 2014 : <RECV: len=1 05 (1030.0 ms)
[2813] Fri Jul 11 19:55:54 2014 : <RECV: len=1 05 (1020.0 ms)
[2813] Fri Jul 11 19:55:54 2014 : <RECV: received 05 05
[2813] Fri Jul 11 19:55:54 2014 : Typ: short (in float: 1285.000000)
[2813] Fri Jul 11 19:55:54 2014 : (FLOAT) Exp:V/10 [B0:05 B1:05 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
[2813] Fri Jul 11 19:55:54 2014 : 128.500000 Degrees Celsius

Die 04 ist jetzt weggefallen, aber der Effekt bleibt der selbe. Kann ich kontrollieren ob die Sendediode funktioniert? Ist das Infrarot oder müsste ich da was blinken sehen?

Danke und Gruß,
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Fri, 11 Jul 2014 20:12:02 UT
* Message-ID: <68716492-73997978$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Also für mich sieht das aus, als würde der Adapter nicht senden - wie man das kontrollieren kann - Keine Ahnung

Kann jemand anders aus dem Forum weiterhelfen?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Tue, 15 Jul 2014 13:50:42 UT
* Message-ID: <68716492-74035318$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Anscheinend kommt nix bei der WP an.
Bitte Überprüfe den Adapter!
Möglichkeiten:
- D1 polarität prüfen/wechseln;
- R2 Wert senken (330 -> 220 Ohm).

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Mon, 21 Jul 2014 19:17:59 UT
* Message-ID: <68716492-74106096$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi Jens,
hmm, ich bin da etwas ratlos - irgendwas scheint da noch nicht so zu laufen. Die Abfrage der Geräte-ID darf ja nicht 2x was anderes ergeben. Bei der Außentemperatur scheint es auch so, als würde mal wieder nix zurückkommen (2x 05 nach fast einer Sekunde).
Kann es sein, dass es vielleicht ein optischer Kurzschluss besteht. Die Dioden müssen vor und nach der Platine (also durch die V-Aussparung) mit bspw. einem schwarzen Papier getrennt sein.
Andere Idee - was für eine Steuerung hast du denn dran .... sicher, dass die AT auf 0101 und nicht auf 0800 liegt? - Würde aber auch nicht die 2 unterschiedlichen Dev-IDs erklären

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Tue, 22 Jul 2014 13:24:22 UT
* Message-ID: <68716492-74115464$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Ich würde die Sache in diesem Fall ein bischen deterministischer angehen.
Also, Empfang funktioniert, das haben wir gesehen (REC: 05).
Jetzt muss nur noch die Senderichtung getestet werden.
Dazu würde ich den Optolink Adapter einfach über USB mit dem PC verbinden, dadurch bekommt mann eine serielle Schnittstelle zum Gerät.
Dann Putty, oder RealTerm, oder Hterm installieren (falls nicht geschehen) aufmachen, und mit  der (4800bps,8E2) Einstellung eine Verbindung zum Gerät aufbauen. Die Anzeige auf HEX umstellen.
Wenn alles richtig gelaufen ist, dann sollte man in 2 Sek. takt immer wieder 0x05 bekommen.
Dann die Senderichtung testen durch das Senden von [0x16 0x00 0x00]. Wenn alles gut lauft, die Antwort sollte ein 0x06 sein.
Test OK ==> der Adapter ist OK.
Test NOK ==> der Adapter ist nicht OK.
Bitte lass uns das Ergebnis wissen.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Wed, 23 Jul 2014 23:40:48 UT
* Message-ID: <68716492-74136956$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

stevestrong, danke für die Hinweise, das werde ich am Wochenende per Putty auf Windows (statt vcontrold auf Raspberry) durchexerzieren.

warmup, die Sache mit dem optischen Kurzschluss hatte ich bereits an anderer Stelle gelesen und verstehe ich Dich richtig, daß die Dioden VOR und HINTER der Platine optisch zu trennen sind? Ich habe

Hinter der Platine - Platine - Vor der Platine (Zwischenraum zwischen Adapter und Optolinkanschluss an Wärmepumpe) - Wärmepumpensteuerung 

Ich habe in dem Zwischenraum zwischen Adapterplatine und Wärmepumpe einen optischen Kurzschluss mit einem Stück schwarzen Schaumstoff ausgeschlossen aber benötige ich auch eine zusätzliche Trennung HINTER der Platine (also dort wo die ganzen Bauteile aufgelötet sind)?

Danke und Grüße,
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Thu, 24 Jul 2014 06:31:12 UT
* Message-ID: <68716492-74141044$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Moin Jens,
da hast ja den Adapter hier aus dem Forum? -  dazu gibt es den Hinweise
WICHTIG! Da die beiden Opto Bauelemente sich auf Höhe der Platinenrückseite befinden, spiegeln sich die Infrarotstrahlen an der Gehäuseinnenwand. Es muss mit einer Trennwand im Gehäuse, z.B. schwarze Pappe, dafür gesorgt werden, daß kein optischer Kurzschluß zwischen den Bauteilen entsteht. Sonst gibt es Empfangsprobleme.
Also ich habe seinerzeit um das auszuschließen ein Stück Pappe in die V-Aussparung reingesteckt und diese noch einige cm hinter der Platine stehen lassen. Später habe ich dann einen Kasten aus milchigem Plexiglas drumgebaut (war noch über :-)
http://warmup.mypump.de/Information/Vito-Fotos.html

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:39:36 UT
* Message-ID: <68716492-74148448$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, nun habe ich nochmals getestet. Ich habe nun den Adapter nicht nur zwischen Platine und Dioden an der Viessmannsteuerung sondern zusätzlich HINTER der Platine mit Schaumstoff "optisch isoliert". Jetzt bekomme ich konsistent einen Device Type 20 00. Auf 08 00 bekomme ich sinnfällige und konsistente Werte für Aussentemperatur. 

Wie weiter...
- wenn ich nacheinander via telnet verschiedene Kommandos absetze, scheint etwas durcheinander  zu kommen, ich bekomme dann öfters mal 05 als einen Rückgabewert, teilweise auch für alle (je nach Länge des erwarteten Ergebnisses). Nutze ich vclient und setze die Kommandos einzeln ab, scheint es zu funktionieren.
- Für meinen Device Type 20 00 finde ich hier im Wiki keine brauchbaren Adressen. Ein Durchprobieren der gängigen Adressen führt nicht zum Erfolg, für die meisten Werte bekomme ich nur 05-Folgen oder Werte die von der Größe her nicht plausibel sind. Gibt es irgendwo eine Dokumentation / ein Beispiel für den Device Type?

Danke für Eure bisherige Unterstützung!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:39:47 UT
* Message-ID: <68716492-74148456$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, nun habe ich nochmals getestet. Ich habe nun den Adapter nicht nur zwischen Platine und Dioden an der Viessmannsteuerung sondern zusätzlich HINTER der Platine mit Schaumstoff "optisch isoliert". Jetzt bekomme ich konsistent einen Device Type 20 00. Auf 08 00 bekomme ich sinnfällige und konsistente Werte für Aussentemperatur. 

Wie weiter...
- wenn ich nacheinander via telnet verschiedene Kommandos absetze, scheint etwas durcheinander  zu kommen, ich bekomme dann öfters mal 05 als einen Rückgabewert, teilweise auch für alle (je nach Länge des erwarteten Ergebnisses). Nutze ich vclient und setze die Kommandos einzeln ab, scheint es zu funktionieren.
- Für meinen Device Type 20 00 finde ich hier im Wiki keine brauchbaren Adressen. Ein Durchprobieren der gängigen Adressen führt nicht zum Erfolg, für die meisten Werte bekomme ich nur 05-Folgen oder Werte die von der Größe her nicht plausibel sind. Gibt es irgendwo eine Dokumentation / ein Beispiel für den Device Type?

Danke für Eure bisherige Unterstützung!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:39:49 UT
* Message-ID: <68716492-74148458$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, nun habe ich nochmals getestet. Ich habe nun den Adapter nicht nur zwischen Platine und Dioden an der Viessmannsteuerung sondern zusätzlich HINTER der Platine mit Schaumstoff "optisch isoliert". Jetzt bekomme ich konsistent einen Device Type 20 00. Auf 08 00 bekomme ich sinnfällige und konsistente Werte für Aussentemperatur. 

Wie weiter...
- wenn ich nacheinander via telnet verschiedene Kommandos absetze, scheint etwas durcheinander  zu kommen, ich bekomme dann öfters mal 05 als einen Rückgabewert, teilweise auch für alle (je nach Länge des erwarteten Ergebnisses). Nutze ich vclient und setze die Kommandos einzeln ab, scheint es zu funktionieren.
- Für meinen Device Type 20 00 finde ich hier im Wiki keine brauchbaren Adressen. Ein Durchprobieren der gängigen Adressen führt nicht zum Erfolg, für die meisten Werte bekomme ich nur 05-Folgen oder Werte die von der Größe her nicht plausibel sind. Gibt es irgendwo eine Dokumentation / ein Beispiel für den Device Type?

Danke für Eure bisherige Unterstützung!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:39:50 UT
* Message-ID: <68716492-74148460$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, nun habe ich nochmals getestet. Ich habe nun den Adapter nicht nur zwischen Platine und Dioden an der Viessmannsteuerung sondern zusätzlich HINTER der Platine mit Schaumstoff "optisch isoliert". Jetzt bekomme ich konsistent einen Device Type 20 00. Auf 08 00 bekomme ich sinnfällige und konsistente Werte für Aussentemperatur. 

Wie weiter...
- wenn ich nacheinander via telnet verschiedene Kommandos absetze, scheint etwas durcheinander  zu kommen, ich bekomme dann öfters mal 05 als einen Rückgabewert, teilweise auch für alle (je nach Länge des erwarteten Ergebnisses). Nutze ich vclient und setze die Kommandos einzeln ab, scheint es zu funktionieren.
- Für meinen Device Type 20 00 finde ich hier im Wiki keine brauchbaren Adressen. Ein Durchprobieren der gängigen Adressen führt nicht zum Erfolg, für die meisten Werte bekomme ich nur 05-Folgen oder Werte die von der Größe her nicht plausibel sind. Gibt es irgendwo eine Dokumentation / ein Beispiel für den Device Type?

Danke für Eure bisherige Unterstützung!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:39:51 UT
* Message-ID: <68716492-74148464$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, nun habe ich nochmals getestet. Ich habe nun den Adapter nicht nur zwischen Platine und Dioden an der Viessmannsteuerung sondern zusätzlich HINTER der Platine mit Schaumstoff "optisch isoliert". Jetzt bekomme ich konsistent einen Device Type 20 00. Auf 08 00 bekomme ich sinnfällige und konsistente Werte für Aussentemperatur. 

Wie weiter...
- wenn ich nacheinander via telnet verschiedene Kommandos absetze, scheint etwas durcheinander  zu kommen, ich bekomme dann öfters mal 05 als einen Rückgabewert, teilweise auch für alle (je nach Länge des erwarteten Ergebnisses). Nutze ich vclient und setze die Kommandos einzeln ab, scheint es zu funktionieren.
- Für meinen Device Type 20 00 finde ich hier im Wiki keine brauchbaren Adressen. Ein Durchprobieren der gängigen Adressen führt nicht zum Erfolg, für die meisten Werte bekomme ich nur 05-Folgen oder Werte die von der Größe her nicht plausibel sind. Gibt es irgendwo eine Dokumentation / ein Beispiel für den Device Type?

Danke für Eure bisherige Unterstützung!
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu, 24 Jul 2014 21:44:53 UT
* Message-ID: <68716492-74148520$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Entschuldigt die Mehrfachpostings, kann sie leider nicht löschen...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Fri, 25 Jul 2014 00:33:46 UT
* Message-ID: <68716492-74149660$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Mit dem Terminal, du wartest warscheinlich auf den nächsten 05 nicht immer. Der vclient schon.
Ehrlich gesagt, DevID "20 00" finde ich ein bischen komisch, das letzte byte sollte nicht 00 sein.
Geräte Übersicht findest du hier:
http://openv.wikispaces.com/Ger%C3%A4te
Hast du auch ein Vitotronic daran? Welches Model?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Mon, 28 Jul 2014 22:27:01 UT
* Message-ID: <68716492-74182856$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

So, wieder ein kleines Stück weiter. 

Ich hatte mich im Vorfeld hier im Wiki informiert und fand unter "Geräte" auch "Vitocal 2xx-G / 3xx-G mit Vitotronic 200 (WO1)" und bin davon ausgegangen dass ich mit meiner Vitocal 200-G Erfolg haben könnte. Ich hab nun ein wenig weiter recherchiert und auch mit meinem Viessmann-"Betreuer" gesprochen, es scheint so als hätte ich "nur" eine Steuerung CD70.

Daraufhin bin ich auf folgendes interessanten Thread gestossen:
http://openv.wikispaces.com/share/view/33000304
Dort hat wohl jemand eine CD70 zumindest auslesen können, leider erreiche ich den Threadersteller nicht.

Ich habe nun - ähnlich wie er - versucht die nächsten zwei Bytes nach der Adresse für den Device Type auzulesen, und bekomme zusammengesetzt nun konsistent
20 00 45 00

[2752] Mon Jul 28 22:16:40 2014 : Befehl: getDevType
[2752] Mon Jul 28 22:16:40 2014 : Process 2752 tries to aquire lock
[2752] Mon Jul 28 22:16:40 2014 : Process 2752 got lock
[2752] Mon Jul 28 22:16:40 2014 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 00
[2752] Mon Jul 28 22:16:40 2014 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[2752] Mon Jul 28 22:16:40 2014 : >FRAMER: no preset result
[2752] Mon Jul 28 22:16:40 2014 : Warte auf 05
[2752] Mon Jul 28 22:16:40 2014 : <RECV: 05 (300.0 ms)
[2752] Mon Jul 28 22:16:40 2014 : >SEND: 01
[2752] Mon Jul 28 22:16:40 2014 : >SEND: F7
[2752] Mon Jul 28 22:16:40 2014 : >SEND: 00
[2752] Mon Jul 28 22:16:40 2014 : >SEND: F8
[2752] Mon Jul 28 22:16:40 2014 : >SEND: 02
[2752] Mon Jul 28 22:16:40 2014 : >FRAMER: no preset result
[2752] Mon Jul 28 22:16:40 2014 : <RECV: len=1 20 (20.0 ms)
[2752] Mon Jul 28 22:16:40 2014 : <RECV: len=1 00 (0.0 ms)
[2752] Mon Jul 28 22:16:40 2014 : <RECV: received 20 00
[2752] Mon Jul 28 22:16:40 2014 : 20 00 -> UNKNOWN
[2752] Mon Jul 28 22:16:40 2014 : UNKNOWN
[2752] Mon Jul 28 22:16:40 2014 : Befehl: quit
[2752] Mon Jul 28 22:16:40 2014 : Process 2752 released lock
[2752] Mon Jul 28 22:16:40 2014 : Verbindung beendet (fd:5)
[2752] Mon Jul 28 22:16:47 2014 : Client verbunden 127.0.0.1:52959 (FD:5)

[2752] Mon Jul 28 22:16:47 2014 : Befehl: getDevType2
[2752] Mon Jul 28 22:16:47 2014 : Process 2752 tries to aquire lock
[2752] Mon Jul 28 22:16:47 2014 : Process 2752 got lock
[2752] Mon Jul 28 22:16:47 2014 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 00
[2752] Mon Jul 28 22:16:47 2014 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[2752] Mon Jul 28 22:16:47 2014 : >FRAMER: no preset result
[2752] Mon Jul 28 22:16:47 2014 : Warte auf 05
[2752] Mon Jul 28 22:16:47 2014 : <RECV: 05 (30.0 ms)
[2752] Mon Jul 28 22:16:47 2014 : >SEND: 01
[2752] Mon Jul 28 22:16:47 2014 : >SEND: F7
[2752] Mon Jul 28 22:16:47 2014 : >SEND: 00
[2752] Mon Jul 28 22:16:47 2014 : >SEND: F9
[2752] Mon Jul 28 22:16:47 2014 : >SEND: 02
[2752] Mon Jul 28 22:16:47 2014 : >FRAMER: no preset result
[2752] Mon Jul 28 22:16:47 2014 : <RECV: len=1 45 (10.0 ms)
[2752] Mon Jul 28 22:16:47 2014 : <RECV: len=1 00 (10.0 ms)
[2752] Mon Jul 28 22:16:47 2014 : <RECV: received 45 00
[2752] Mon Jul 28 22:16:47 2014 : 45 00 -> UNKNOWN
[2752] Mon Jul 28 22:16:47 2014 : UNKNOWN

Nun bin ich etwas zwiegespalten: Ich würde total gerne weitermachen, einige wenige plausible Werte sind ja auch "auffindbar" - Aussentemperatur bpsw. Allerdings ist das "Suchen" der Adressen sehr mühsam, auf den üblichen der Vitotronic'en scheint da leider nix zu sein. 

Habt Ihr eine Idee wie ich vorgehen könnte?

Danke und Gruß,
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: stevestrong <userid-45994172@openv.wikispaces>
* Date: Tue, 29 Jul 2014 08:47:52 UT
* Message-ID: <68716492-74189140$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Du könntest mal tesweise versuchen auch mit dem P300 Protokoll die Daten auszulesen (erstmal mit dem PC, weil ich glaub vclient kommt damit nicht ganz zurecht), Vitotronic 200 (WO1) sollte P300 unterstützen, meins (WO1B) tut es.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Thu, 11 Sep 2014 20:54:28 UT
* Message-ID: <68716492-74887396$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi j-dimension,
Bist du noch hier und liest mit? Ich hab die gleiche Heizung wie du .... und die gleichen probe .. melde dich mal ^^

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Sat, 13 Sep 2014 22:26:47 UT
* Message-ID: <68716492-74936086$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo MartinFrankP,

Ja ich bin noch dran an der Sache, es gestaltet sich allerdings ziemlich mühsam.

Die gute Nachricht ist dass man die CD70 mit vclient auslesen kann. Allerdings gibt es keine Adressen, die gängigen funktionieren nicht.

Momentan suche ich mühsam den gesamten Adressbereich ab. Gefunden habe ich

Außentemperatur
Vorlauftemperaturen primär und sekundär
Temperaturen zweier angeschlossener Pufferspeicher
Status Zirkulationspumpe

Könnte da auch Unterstützung gebrauchen :-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Sun, 14 Sep 2014 07:19:26 UT
* Message-ID: <68716492-74941626$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi j-Dimension
meine Anlage sendet mir auch 20004500 .. also wird es die gleiche sein, wie deine. Wird heute mal ein paar Dumps via Viss Data v1 ziehen, mal sehen, was ich erkenne (in meinem Post von Kehrenberger vorgeschlagen). Verrätst du mir die Adressen, die Du rausgefunden hast? Vielleicht trägt kehrenberger sie ja unter Vitocall 200 G  (2045) in die Liste ein

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Sun, 14 Sep 2014 20:55:16 UT
* Message-ID: <68716492-74949322$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Gerne, hier die Adressen:

0800 Aussentemperatur (die hab ich auch noch an anderen Stellen gefunden, scheint redundant zu sein)
080C Speicher1 Temperatur oben (ich habe wie gesagt zwei Pufferspeicher dran, mit jeweils zwei Sensoren unten und oben)
080F Speicher2 Temperatur oben
0808 Temperatur Vorlauf Sekundär
0816 Temperatur Vorlauf Primär
0846 Status Zirkulationspumpe

Ich bin wie gesagt hoch interessiert an weiteren Adressen dieser Steuerung / Anlage! Insbesondere Status Verdichter etc. suche ich noch, da ich auf dieser Basis einige Funktionen der Hausautomatisierung steuern möchte.

Grüße,
j-dimension

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Sun, 14 Sep 2014 20:56:38 UT
* Message-ID: <68716492-74949342$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

PS: diese Dump-Funktionalität kenne ich nicht, bin auch in einer reinen Linux-Umgebung unterwegs. Es klingt jedenfalls vielversprechend / weniger mühsam. Momentan probiere ich (halbwegs geskripted) alles per Hand durch und schaue ob irgendetwas plausibles dabei ist.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Mon, 15 Sep 2014 07:26:02 UT
* Message-ID: <68716492-74965432$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

moin j
die Anlage bzw Steuerung ist echt ziemlich störrig. Der Dum funktioniert nicht.
Also, wenn ich das richtig sehe, gibt die Steuerung aud jeden Fall 2 Byte pro Adresse raus. Es lässt sich aber nicht jede Adresse oder auch jede länge abfragen. In dieser XML Datei, die hier mal irgendwo gepostet wurde (diese original von viessmann) ist unsere Steuerung natürlich auch nicht aufgelistet.
So, wie nun weiter? Die Anlage sendet jede Sekunde eine 05, dann kann man genau einen Befehl absetzten. Entweder kommt 05 zurück, oder ein Wertepaar. Das Würde bei einem Adressbereich von 0 - 6000 (hex) nur knappe 400 Stunden dauern :(. Dadurch hätte man aber alle Adressen, auf die man lesend zugreifen kann. Na mal sehn, ich melde mich, wenn ich weiter bin

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Mon, 15 Sep 2014 07:32:53 UT
* Message-ID: <68716492-74965544$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi,
warum versucht Ihr nicht mal die Adressen, die unter V333MW1 (http://openv.wikispaces.com/Adressen) stehen. Ich kann mir nicht vorstellen, dass eure Steuerung völlig anders aufgebaut ist und interessanterweise  sind 0800 und 0846 schonmal gleich belegt - vielleicht hast du Doich beim Probieren der anderen Adressen ein wenig in der Zuordnung vertan?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Mon, 15 Sep 2014 08:08:49 UT
* Message-ID: <68716492-74966042$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Gute Idee, nur scannen muss ich dennoch, da viele Werte nicht dort zu finden sind, das es sich um eine Sole WP handelt.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Mon, 15 Sep 2014 20:50:00 UT
* Message-ID: <68716492-74981722$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Dass die Anlage "störrisch" ist kann ich bestätigen. Bei etlichen Requests bekommt man immer mal wieder invalide Werte. Würde zu MartinFrankP's Aussage passen "Entweder kommt 05 zurück, oder ein Wertepaar.".
Ich habe mir da erstmal etwas drumherumgestrickt: ich kann konfigurieren welchen Wert ich abfragen möchte und welche Validierung dazu genommen wird (bspw. > -50 und <55 für Aussentemperatur). Wird ein invalider Wert von vclient zurückgegeben, so wird nochmal angefragt. Nach 3 Fehlschlägen ist schluss. Es gibt aber relativ wenige invalide Werterückgaben. Dieses Prozedere läuft bei mir für die weiter oben genannte Menge an Adressen seit Wochen stabil. Ich frage die Werte alle 30min ab.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Tue, 16 Sep 2014 07:24:33 UT
* Message-ID: <68716492-74998396$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

so, ich schon wieder.
also wie gestern gesagt, scan ich im Moment meine Heizung.
hier die ersten Bereiche, die tatsächlich Daten zurück geben.
00F8 - 00FD
0800 - 08FF
2700 - 2722
im Moment scan ich bis 6000. Mal sehen, welche Bereiche ich dann noch bekomme.
Wenn ich erstmal alle Bereiche habe, dann wird ich die Bereiche bei immer anderen zuständen scannen und hoffentlich so Einblick in die Anlage bekommen.
Beispielausschnitt vom 1. Scan
08 D2 --> 00 20 14 09 16 02 05 27 04 
08 E0 --> 20 14 09 16 02 05 27 15 
08 F0 --> 37 32 39 36 38 31 39 
08 F7 --> 94 9A 
08 FA --> 04 16 
08 FB --> 16 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 
08 FD --> 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 
08 FE --> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
08 FF --> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 00 00 00 00

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Tue, 16 Sep 2014 07:24:34 UT
* Message-ID: <68716492-74998398$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

so, ich schon wieder.
also wie gestern gesagt, scan ich im Moment meine Heizung.
hier die ersten Bereiche, die tatsächlich Daten zurück geben.
00F8 - 00FD
0800 - 08FF
2700 - 2722
im Moment scan ich bis 6000. Mal sehen, welche Bereiche ich dann noch bekomme.
Wenn ich erstmal alle Bereiche habe, dann wird ich die Bereiche bei immer anderen zuständen scannen und hoffentlich so Einblick in die Anlage bekommen.
Beispielausschnitt vom 1. Scan
08 D2 --> 00 20 14 09 16 02 05 27 04 
08 E0 --> 20 14 09 16 02 05 27 15 
08 F0 --> 37 32 39 36 38 31 39 
08 F7 --> 94 9A 
08 FA --> 04 16 
08 FB --> 16 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 
08 FD --> 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 00 C8 00 C8 00 00 00 00 00 00 00 00 00 00 00 
08 FE --> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
08 FF --> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 00 00 00 00

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Fri, 19 Sep 2014 07:48:47 UT
* Message-ID: <68716492-75113304$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Moin j-dimension 
selbst unsere Anlagen scheinen sich zu unterscheiden. 
Die 0846 Status Zirkulationspumpe kann ich nicht bestätigen. Evtl. weil ich sie über ein Zeitprogramm steuer?!? Auf der 0816 kommt bei mir immer 00 00. Dafür scheint bei mir die 0801 die PriVorlauftemp zu sein.
Und nun eine neue Adresse für dich ... 0807 ist (zumindest bei mir) die SekRücklauftemp.
08E0 ist die Systemzeit. Wobei das Byte zwischen Datum und Uhrzeit den Wochentag angibt. 01 = Montag, 02=Dienstag .. usw. Sonntag muss ich noch gucken .. 00 oder 07.
Im Moment ist es ehr schwierig gute Daten zu bekommen, da die Heizung nur 30 Min für WW am Tag läuft.

Einige Adressbereiche, welche Daten zurück geben, hab ich oben schon genannt. 3700, 4700, 5700, 6700 und 7700 geben wie die 2700 auch an einigen stellen Daten zurück. Die x7xx Bereiche sehen ein wenig wie Zeitprogramme aus. Mich interessiert jetzt für den Anfang eigentlich nur noch die Zirkulationspumpe und vor allen Dingen die PriRücklauftemp. Wobei ich mir nichtmal sicher bin, ob die gespeichert wird, da sie im Display auch nicht angezeigt wird.
Hast Du noch was Raus gefunden?

Grüsse Martin

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Sun, 21 Sep 2014 23:45:22 UT
* Message-ID: <68716492-75164722$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Moin Martin,

danke erstmal für die Arbeit - wir scheinen uns einer Lösung zu nähern :-)

Ich bin eben erst von einer Dienstreise rein, konnte noch nicht viel probieren, kann aber bestätigen dass diese von dir genannten Bereiche auch bei mir Daten zurückgeben:
00F8 - 00FD
0800 - 08FF
2700 - 2722

Frage: woher weisst Du wieviele Bytes zu lesen sind? Ich habe mich bei meinen Versuchen immer nur auf 2 Bytes konzentriert, um Temperaturen zu finden. 

Ich zitiere:
"Beispielausschnitt vom 1. Scan
08 D2 --> 00 20 14 09 16 02 05 27 04
08 E0 --> 20 14 09 16 02 05 27 15 "

Ich mache bei meinen Tests immer bspw. 
vclient -h 127.0.0.1:3002 -c 'get 08 D2'

Aber das schnappt sich lt. Konfiguration vito.xml ja immer nur 2 Bytes. 

Gruß,
Jens

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Mon, 22 Sep 2014 07:25:22 UT
* Message-ID: <68716492-75175590$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Moin Jens
hatte mir nen kleines Windowstool selbst gebastelt, um möglichst flexibel zu sein.
Hab also zB 01 F7 08 E0 02 gesendet und alles gelesen, was bis zur nächsten 05 rein kam .. und siehe da, es kam 20 14 09 16 02 05 27 15.
Werde jetzt erstmal meine DB erweitern und die gefundenen Temperaturen loggen. Mal sehn, ob ich irgendwann noch was gutes finde. Eigentlich müsste man nen Heizungsmensch finden, der unsere Steuerung mit der Viessmann Software auslesen kann, und bei nem guten Kaffee die Config-Files der Software abschwatzen. Dann hätte man alles. Vielleicht klappt es ja auch irgendwann, da ich mir eine sparsame Umwälzpumpe einbauen lassen will. Meine läuft mit 100 W  :(

Gruss
Martin

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Mon, 22 Sep 2014 17:57:43 UT
* Message-ID: <68716492-75186804$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Sorry, so ganz hab ich noch nicht verstanden, was ihr da macht... Es gibt doch schon eine ganze Reihe Adressen für Wärmepumpen in der Adressliste. Sucht ihr andere?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Mon, 22 Sep 2014 22:57:06 UT
* Message-ID: <68716492-75194772$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Martin,

die 0801 und 0807 kann ich jetzt auch noch bestätigen. Temperaturtechnisch habe ich nun eigentlich alles was ich brauche. 

Primär Rücklauf zeigt die Anlage m.W. nur in dieser ominösen "Fachbetriebsebene" an. 

Die Status-Flags (Verdichter, Pumpe, ...) muss ich suchen wenn die Pumpe was zu tun hat. Momentan reicht die Solarthermie noch aus, und ohne sich bewegende Daten ist es schwer etwas zu finden. 

Melde mich wieder, sobald ich etwas habe.

Grüße!

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Mon, 22 Sep 2014 23:06:42 UT
* Message-ID: <68716492-75194948$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo kehrenberger,

es geht um eine nicht mehr ganz so "frische" Sole-WP Vitocal 200G, die eine CD70 (keine Vitotronic) - Steuerung hat. Device ID ist wohl die 2045. Die Adressen haben mit den hier im Wiki verfügbaren praktisch nichts zu tun, bis auf die AT.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Mon, 22 Sep 2014 23:26:27 UT
* Message-ID: <68716492-75195472$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Aha, na dann viel Glück....

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Tue, 23 Sep 2014 07:40:50 UT
* Message-ID: <68716492-75205472$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo kehrenberger, 
genauso ist es leider. Die Device ID fangen hier ich glaub bei 2046 oder 47 an. Die Adressen sind aber leider komplett andere, als bei unserer Steuerung. 

Hallo Jens
wo soll die Anlage die Primär Rücklauftemperatur in der Fachbetriebsebene anzeigen .. Wenn ich den Momentanwert ablesen kann, find ich wahrscheinlich auch die Adresse ^^

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Tue, 23 Sep 2014 15:06:21 UT
* Message-ID: <68716492-75211152$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo j-Dimension,
passt zwar nicht 100% auf dein Post, aber ich habe eine Vitocal 222G Sole-WaePu. Gerätecode 2044, somit auch so ein "Exot" wo keine allg. verfügbaren Registerlisten passen. Habe mir daher auch alles mit der Hand am Arm selber ermitteln müssen. Bei mir finden sich die Temperaturen auch in 08xx

Komponentenstati habe ich folgende bei mir identifiziert:
Wärmepumpe ein/aus 088A (ein Byte, 0 oder 1, auch alle nachfolgenden)
Primärpumpe an/aus  08BF
Warmwassererwärmung ein/aus   08B0
Umwälzpumpe Warmwasser	08B3
Warmwassererwärmung (?) ein/aus (läuft kurz nach) 08D0
Solltemperatur Heizkreis    0898  (2 Bytes)
(Dummerweise finde ich den Status des Heizstabes (ein/aus) nicht...)
Zahlreiche Einstellungen (meist auch änderbar) finden sich ab folgenden Adressen (gerne mehr Details, wenn interessiert):
WäPu-Einstellungen ab 5700
Warmwasser-Paramter ab 6700
Anlagendef/E-Heizung/interne Hydraulik ab 7700
Zeitstempel ab 08E0
Seriennummer ab 08F0

Ich dachte mall, ich schreib Dir das, den 2045 ist nicht soweit von 2044 entfernt, ggf. deckt sich ja was..... ;-)

Gruß
mimax

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Tue, 23 Sep 2014 21:55:32 UT
* Message-ID: <68716492-75225450$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Also, anscheinend handelt es sich ja um eine Steuerung, die doch sehr viel mit den aktuellen Steuerungen gemein hat, und Leute wie mimax haben ja schon eine ganze Menge von Adressen gefunden, warum seid ihr Euch dann nicht mal bei der DeviceID sicher? Kommt nicht immer die gleiche Antwort, wenn ihr 01 F7 00 F8 02  nach einer empfangenen 05 sendet? Also, wenn bei Euch das KW Protokoll grundsätzlich funktioniert, dann würde ich Viessdata oder ein selbst programmiertes Tool benutzen, die Daten-Adressliste mit den bekannten Adressen füllen und erst mal diese verifizieren. Das kann mit der Anzeige an der Steuerung passieren (Temperaturen, Zeit, Zeitprogramme, Schaltzustände), oder durch Messungen/Beobachtungen (Geräusche, Messgeräte)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Tue, 23 Sep 2014 22:31:39 UT
* Message-ID: <68716492-75226670$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo mimax,

1000 Dank für die Infos! 

Ist "Wärmepumpe ein/aus 088A" der Status des Verdichters? Den suche ich verzweifelt...

Die gute Nachricht: für alle von Dir genannten Statusadressen bekomme ich valide Werte (0, da meine Heizung wie gesagt momentan noch nichts zu tun hat). Ich lasse das jetzt mal automatisch wegschreiben und schaue ob es passt. Bin sehr zuversichtlich!

Die anderen von Dir genannten Punkte interessieren mich brennend! Was ist da noch zu finden, und welche Werte hast Du schon erfolgreich auch SETZEN können?

Und an welche Adresse kann ich einen Kasten Bier schicken? :-)

Grüße,
j-dimension

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Tue, 23 Sep 2014 22:34:28 UT
* Message-ID: <68716492-75226714$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo Martin,

entgegen der Aussage meines Heizungs-Spezis finde ich die Primär-Rücklauftemperatur in der Fachbetriebsebene NICHT. Ob ich die letzte Software-Version für die Steuerung dieser Anlage habe, kann ich jetzt nicht sagen. Ich wüsste auch nicht wie das selbst gemacht werden kann. Er hatte die Vermutung dass diese Anzeige erst später hinzugekommen ist. :-(

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Tue, 23 Sep 2014 22:41:17 UT
* Message-ID: <68716492-75226874$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo kehrenberger,

ich sehe das nicht ganz so pessimistisch :-)

Martin und ich haben das selbe Modell (Vitocal 200-G mit CD70-Steuerung) und wir bekommen beide konsistent die ID 2045. 

mimax hat eine Vitocal 222-G, wahrscheinlich auch mit CD70 und ID 2044. 

Auf den ersten Blick scheinen für beide IDs
- die Adressen übereinzustimmen
- die üblichen Wärmepumpen-Adressen bei Steuerung Vitotronic durchgehend NICHT zu passen

Insofern könnten wir hier über die nächsten Tage noch etwas weiter verifizieren und dann wäre das m.E. eine neue Spalte in der Adressliste wert. Aber wie gesagt - ich bin ja Neuling hier und lasse mich gern eines Besseren belehren!

Grüße,
j-dimension

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Tue, 23 Sep 2014 23:00:20 UT
* Message-ID: <68716492-75227338$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Ja, dann stimme ich Dir zu, aber oben schreibt ihr, "glaube ich" und "ist wohl", da scheint ihr wohl noch nicht konsistente Antworten bekommen zu haben. Ich habe mir mal die Serviceanleitung Eurer Anlage angesehen, die Steuerung sieht ja besser als die Vitotronic aus, keine Codierungen, alles Klartext, super! Sowas hätte ich auch gerne ;^) Nun zu den Adressen zurück...funktionieren nun Dumps, also z.B. das Lesen von Adressbereichen? Damit könnte man natürlich schneller interessante Daten lokalisieren. So habe ich das gemacht: Entweder Dump auslesen und nach einem bestimmten Datum per Hexedit im Dump suchen. Oder auch Dump auslesen, Änderung an der Steuerung eingeben, weiteren Dump lesen und sehen was sich geändert hat.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Wed, 24 Sep 2014 11:55:06 UT
* Message-ID: <68716492-75242406$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hi zusammen
mimax, erstmal vielen Dank für deine Infos. Die Adressen scheinen ja wirklich gleich zu sein (wie auch die Adressbereiche).
Jens, nicht schlimm mit der Rücklauftemperatur, bekommen wir bestimmt noch raus ^^
kehrenberger, Dumps funzen im 'klassischen' Sinne nicht. Du musst glück haben, und eine Adresse finden, die dir lesen erlaubt. dann kannst du teilweise bis zu 0x3f Daten abholen, auf über Adressen, die dir direkt angesprochen, das lesen verbieten. Also hatte ich einen Scan über einen großen Bereich gemacht, um so herauszufinden, welche Adressen ich lesen darf. Nicht schön, aber selten. ZZ les ich die mir bekannten Temperaturadressen aus und speicher sie. Irgendwann wird's dann vielleicht erweitert.
noch eine Frage. Wie werden negative Zahlen dargestellt?  0xFFFF = -1, 0xFEFF = -2 ... ?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Wed, 24 Sep 2014 15:03:13 UT
* Message-ID: <68716492-75246064$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

so, hab noch was gefunden, nur was es im einzelnen heißt, weiß ich noch nicht.
ich hatte den unter anderem die Adressen 0x08B0, 0x08C0 und 0x08D0 mit einer Länge von 0x16 minütlich gelogged.
Bei der Warmwasserbereitung hab ich folgende Änderungen entdeckt
Hab mal nur die Interessanten Dinge kopiert
0x8B0 - 0x8BF                                           8D0-1  8E0
00000000000000000000000000000000  0000  2014092006161129 
00000000000000000000000000000000  0000  2014092006161229 
00000000000000000000010000000000  2121  2014092006161330 
00000001000000000000010000000001  6464  2014092006161429 
00000001000000000000010000000001  6464  2014092006161529
01010000010000010101010000000001  6464  2014092006161629
01010000010000010101010000000001  6464  2014092006161730
01010000010000010101010000000001  6464  2014092006161829
01010000010000010101010000000001  6464  2014092006161929
01010000010000010101010000000001  6464  2014092006162029
01010000010000010101010000000001  6464  2014092006162130
01010000010000010101010000000001  6464  2014092006162229
01010000010000010101010000000001  6464  2014092006162329
01010000010000010101010000000001  6464  2014092006162430
01010000010000010101010000000001  6464  2014092006162530
01010000010000010101010000000001  6464  2014092006162629
01010000010000010101010000000001  6464  2014092006162729
01010000010000010101010000000001  6464  2014092006162830
01010000010000010101010000000001  6464  2014092006162930
01010000010000010101010000000001  6464  2014092006163029
01010000010000010101010000000001  6464  2014092006163129
01010000010000010101010000000001  6464  2014092006163230
00000000000000000000010000000000  0000  2014092006163330
00000000000000000000010000000000  0000  2014092006163429
00000000000000000000000000000000  0000  2014092006163529
00000000000000000000000000000000  0000  2014092006163630

8B4 Scheint die Zirkulationspumpe zu sein. Kommt alle 30 min 5 min lang, so wie ich es eingestellt hab. Wird aber hier von der WW Bereitung frühzeitig abgestellt.
Was nun die anderen Bytes bedeuten, weiss ich noch nicht, da ich nicht was, was in welcher Reihenfolge eingeschaltet wird.
Die PrimärEinTemp fällt an dem Zeitpunkt, wo die Zirku aus geht. Da wird dann also der Kompressor gestartet worden sein .. viel Spass beim tüfteln ^^

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Wed, 24 Sep 2014 15:03:54 UT
* Message-ID: <68716492-75246086$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

ups .. 8b3 = Zirku ...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Wed, 24 Sep 2014 17:25:10 UT
* Message-ID: <68716492-75251038$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

@kehrenberger: zu deiner Frage "warum seid ihr Euch dann nicht mal bei der DeviceID sicher?"... ganz einfach, als ich vor 2 1/2 Jahren mich in das Thema vertieft habe, war ich scheinbar der einzige, der einen so exotischen Gerätecode bei sich hatte. Alle anderen beschäftigten sich mit den "normalen" Vitocal-Adressen, für die 2044 hatte ich keinen validen Existenzbeweis von anderen. Daher die grundsätzlichen Zweifel...war mir aber dann auch wurscht, denn ich hatte ja meinen Adressen, die mich interessierten. Mittlerweile tauchen aber noch WaePu-Besitzer auf, die so "exotische" Typen haben, scheinbar bin ich doch normal veranlagt... ;-)
Und daher auch sicher, dass die 2044 stimmt. Die ID wird ja auch stabil geliefert.

@:MartinFrankP und j-dimension: Ich wäre noch sehr an der Adresse für den Heizstabstatus (ein/aus) interessiert. Dadurch dass der bei mir so gut wie nie angeht, ist es mir noch nie gelungen, die Adresse per Datenabgleich zu ermitteln. Da der aber laut Betriebsstundenzähler ganz, ganz selten doch wohl zu heizen scheint, würde ich ihn gern auch in meinen Charts sehen wollen. Also wenn einer von Euch ne schwach dimensionierte Anlage hat, wo das Ding schon bei unter +10 Grad zu glühen anfängt... ich wär interessiert  ;-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Wed, 24 Sep 2014 17:27:47 UT
* Message-ID: <68716492-75251150$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Ach ja, die Betriebsstundenzähler für die Einzelkomponenten - das wär auch noch so ein Wunschthema. Leider habe ich die noch nicht gefunden....

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: MartinFrankP <userid-50903050@openv.wikispaces>
* Date: Thu, 25 Sep 2014 11:42:04 UT
* Message-ID: <68716492-75272550$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

sorry mimax, ich habe (zum Glück) kein Heizstab

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: ChristianK78 <userid-54719575@openv.wikispaces>
* Date: Sun, 11 Oct 2015 14:33:08 UT
* Message-ID: <68716492-84630693$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo, will das Thema mal wieder aufwärmen... :-)
Gibt es mittlerweile "dokumentierte" Adressen der CD70?
Meine Eltern haben auch so eine, und ich würde diese gerne per FHEM abfragen...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Mon, 19 Oct 2015 23:59:45 UT
* Message-ID: <68716492-84880219$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Würde mich auch für die Adressen interessieren! 
Hab ne Luft/Wasser Wärmepumpe Vitocal 350 mit CD70 Regelung
ID = 20 00 E4 00

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Mon, 26 Oct 2015 21:13:34 UT
* Message-ID: <68716492-85092043$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Ich habe dann irgendwann aufgehört weiter zu suchen - man müsste zu zwei verschiedenen Zeitpunkten einen kompletten Dump machen und Wertänderungen an der CD70 mit Wertänderungen im Dump korrelieren - war mir letztlich zu viel Aufwand. Ich benutze jetzt seit Monaten zumindest folgende Werte und vearbeite / charte sie in OpenHAB:

 <command name='getTempAussen' protocmd='getaddr'>
      
      <addr>0800</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Determine the outside temperature in degrees C</description>
  </command>

 <command name='getTestTemp' protocmd='getaddr'>
      
      <addr>089B</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>tbd</description>
  </command>
 <command name='get080A' protocmd='getaddr'>
      <addr>080A</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>unbekannte aber sinnhaftige Temperatur</description>
  </command>
 <command name='getTempPufferBrauchwasser' protocmd='getaddr'>
      <addr>080C</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Temperatur Pufferspeicher rechts Warmwasser oben</description>
  </command>
 <command name='get0802' protocmd='getaddr'>
      <addr>0802</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>scheint nochmal Aussentemperatur zu sein</description>
  </command>
 <command name='getTempSekVorlauf' protocmd='getaddr'>
      <addr>0808</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Temperatur Sekundär Vorlauf</description>
  </command>
 <command name='getTempHeizVorlauf' protocmd='getaddr'>
      <addr>0816</addr>
      <len>2</len>   
      <unit>UT</unit>
      <description>Temperatur Heizkreis Vorlauf</description>
  </command>
 <command name='getTempPufferHeizwasser' protocmd='getaddr'>
      <addr>080F</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Temperatur Pufferspeicher links Heizen oben</description>
  </command>

  <command name='getHeatPumpStatus' protocmd='getaddr'>
             <addr>088A</addr>
             <len>1</len>
             <unit>RT</unit>
             <description>Status Waermepumpe/Verdichter</description>
                
  </command>
  <command name='getPrimaryPumpStatus' protocmd='getaddr'>
             <addr>08BF</addr>
             <len>1</len>
             <unit>RT</unit>
             <description>Status Primaerpumpe</description>
  </command>
  <command name='getCircPumpStatus' protocmd='getaddr'>
             <addr>08B3</addr>
             <len>1</len>   
             <unit>RT</unit>
             <description>Status Zirkulationspumpe</description>
  </command>
  <command name='getWarmwaterHeatingStatus' protocmd='getaddr'>
             <addr>08B0</addr>
             <len>1</len>   
             <unit>RT</unit>
             <description>Status Warmwasserbereitung</description>
  </command>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Wed,  4 Nov 2015 12:15:59 UT
* Message-ID: <68716492-85371799$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo j-dimension,

könntest du mal deine vcontrold.xml posten? Mich würd es interessieren wie du die Sync bei der CD70 gemacht hast!

Mfg
Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Wed,  4 Nov 2015 18:56:15 UT
* Message-ID: <68716492-85386675$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

<?xml version="1.0"?>
<V-Control xmlns:vcontrol="http://www.openv.de/vcontrol">
  <unix>
     <config>
	<serial>
		<tty>/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A702J94K-if00-port0</tty>
	</serial>
	<net>
		<port>3002</port>
		<allow ip='127.0.0.1'/>
		<allow ip='192.168.1.0/24'/>
	</net>
	<logging>
		<file>/dev/null</file>
		<syslog>n</syslog>
		<debug>y</debug>
	</logging>
	<device ID="2098"/>
     </config>
  </unix>
  <units>
	<unit name='Temperatur'>
	   <abbrev>UT</abbrev>
	   <calc get='V/10' set='V*10'/>
	   <type>short</type>
	   <entity>Degrees Celsius</entity>
	</unit>
	<unit name='Neigung'>
	   <abbrev>UN</abbrev>
	   <calc get='V/10' set='V*10'/>
	   <type>short</type>
	   <entity></entity>
	</unit>	
	<unit name='Temperatur 1Byte'>
          <abbrev>UT1</abbrev>
          <calc get='V/2' set='V*2'/>
          <type>char</type>
          <entity>Degrees Celsius</entity>
        </unit>
	<unit name='Temperatur 1Byte unsigned'>
          <abbrev>UT1U</abbrev>
          <calc get='V/2' set='V*2'/>
          <type>uchar</type>
          <entity>Degrees Celsius</entity>
        </unit>
	<unit name='Temperatur 1Byte ganzzahlig'>
          <abbrev>UTI</abbrev>
          <calc get='V' set='V'/>
          <type>uchar</type>
          <entity>Degrees Celsius</entity>
        </unit>
	<unit name='Status'>
	   <abbrev>ST</abbrev>
	   <calc get='V' set='V'/>
	   <type>char</type>
	   <entity></entity>
	</unit>
	<unit name='Counter'>
	   <abbrev>CO</abbrev>
	   <calc get='V' set='V'/>
	   <type>int</type>
	   <entity></entity>
	</unit>
[... 303 lines cut ...]
	   	     <send>SYNC;GETXADDR;SEND BYTES;SEND 01 04;RECV 1</send>
		</command>
		 <command name="getkmaddr">
                     <send>SYNC;GETKMDDR $addr $hexlen 04;RECV $len $unit</send>
		     <retry>3</retry>
		     <recvTimeout>150</recvTimeout>
                </command>
		<command name="getkmtestaddr">
		     <retry>3</retry>
		     <recvTimeout>150</recvTimeout>
	   	     <send>SYNC;GETKMADDR;SEND BYTES;SEND 01 04;RECV 1</send>
		</command>
                <command name="init">
                     <send>SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100</send>
                </command>
		<command name="setaddr">
		     <send>SYNC;RECV 1</send>
		</command>
	  </commands>
      </protocol>
  </protocols>
  <extern xmlns:xi="http://www.w3.org/2003/XInclude">
     <xi:include href="vito.xml" parse="xml"/>
  </extern>
</V-Control>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Wed,  4 Nov 2015 18:56:33 UT
* Message-ID: <68716492-85386685$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

PS: ich verwende Protokoll KW2

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Thu,  5 Nov 2015 12:36:41 UT
* Message-ID: <68716492-85414129$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo, 

ich mache das so:

<macro name='SYNC'>

<command>SEND 01;WAIT 05</command>
</macro>
<macro name='GETADDR'>
<command>SEND F7</command>
</macro>

In allen anderen Variationen hatte ich extreme Lesefehler!

Mfg
Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Thu,  5 Nov 2015 12:46:36 UT
* Message-ID: <68716492-85414271$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo j-dimension,

Wie kann ich den das Vcontrold mit Befehle erweitern? du hast z.B ein "getTestTemp" Befehl.
Den ich garnicht habe. Ist dieser einfach in der Vito.xml einzutragen oder muss da was im Source geändert werden?

Mfg
Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Thu,  5 Nov 2015 16:03:30 UT
* Message-ID: <68716492-85420725$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

Hallo j-dimension,

Hättest du eine Vorlage für einen Dump? ich habe die die Adressen mühevoll mit Realterm ausgelesen von HEX auf DEZ umgerechnet und die Werte an der Heizung verglichen. Am liebsten wäre es mir auf Windows Ebene.

Mfg
Kevin

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 200-G, Anfängerprobleme
* Wiki-Page: home
* From: j-dimension <userid-50279738@openv.wikispaces>
* Date: Thu,  5 Nov 2015 20:36:28 UT
* Message-ID: <68716492-85432159$openv.wikispaces>
* In-Reply-To: <68716492-73989366$openv.wikispaces>

(1) Ja, ich habe auch ab und zu Lesefehler. Ich habe im Abfrageclient einen Retry für max. 3x, wenn die Werte inhaltlich keinen Sinn ergeben. Funktioniert soweit robust.
(2) Ja, nur die Konfigurationsfiles anpassen, Source wird nicht angefasst.
(3) Nein, kann ich leider nicht bieten :-(   Ich hatte automatisch ein paar Adressen abgefragt, aber den gesamten Adressraum zu ziehen dauert einfach zu lange bei den Antwortzeiten.

[14 more replies]