openv/openv #223: Werte Setzen > 255

WikiSpaxe, 2017-10-26:
* Subject: Werte Setzen > 255
* Wiki-Page: home
* From: domi81 <userid-52158814@openv.wikispaces>
* Date: Mon, 15 Dec 2014 23:38:39 UT
* Message-ID: <71050896-77914032$openv.wikispaces>

Hallo Community,

ich habe folgendes Problem. Ich habe eine Vitocal 300A mit einer Vitotronic 200 (Typ WO1B) Steuerung. Auslesen etc. klappt wunderbar. Auch die erste Werte setzen hat super funktioniert. Jetzt versuche ich aber dezimal Werte über 255 zu setzen, also 3.stellige HEX Werte. Z.B. beim Soll Wert des Warmwassers muss man bei 47 Grad den Dez Wert 470 setzen, also Hex = 1D6 setzen. Bis 25,5 Grad, also Dez 255 = HEX FF geht alles wunderbar. aber ab Dez 256 geht es nicht mehr. habe schon mit verschiedenen Byte-Längen, führenden Nullen etc. alles versucht. Hat jemand dies bereits gemacht ???

Hier mal mein funktionierende Konfiguration bis 255:

vito.xml:

 <device ID="2049" ExtID="020B" name="V200WO1B" protocol="P300"/>

<command name="setWWSoll47" protocmd="setaddr470">
            <addr>6000</addr>
            <len>2</len>
            <unit>CO</unit>
            <description>Setze WW SOLL TEMP1 auf 47</description>
</command>
 
vcontrold.xml:

<device ID="2049"/>

<command name="setaddr470">
	     <send>SEND 00 02 $addr $hexlen 1E 00;RECV 1 SR</send></command>

Danke im Voraus für Eure Lösungen....

meschmesch, 2021-03-21:
Hallo, ich habe dasselbe Problem. Gibt es hier inzwischen eine Lösung. Ich weiß, der Thread ist schon etwas älter. Aber das ist meine Anlage auch (204D). Werte bis Hex FF funktionieren, aber alles darüber geht nicht. Danke!

speters, 2021-03-21:
Ich kenne bei mir nur die an Adresse 0x6300 einzustellende Warmwassertemperatur.
Diese umfasst ein Byte und erlaubt ganzzahlige Werte zwischen 10 und 60, bzw. 10 und 90 (je nach Kodierung) als Warmwasser-Solltemperatur in °C.

meschmesch, 2021-03-21:
ok, aber wo mache ich das? Die Adresse kenne ich, ist 6000 bei mir. In der vito.xml habe ich z.B.
`                <command name="SetTempWW20" protocmd="setaddr20">
                        <addr>6000</addr>
                        <len>2</len>
                        <unit>UT</unit>
                        <description>Warmwasserzieltemperatur setzen 20 Grad</description>
`
und zugehörig in der vcontrold.xml:
` <units>
        <unit name='Temperatur'>
           <abbrev>UT</abbrev>
           <calc get='V/10' set='V*10'/>
           <type>short</type>
           <entity>Degrees Celsius</entity>
        </unit>
`
sowie
`                <command name="setaddr20">
                     <send>SEND 00 02 $addr $hexlen C8 00;RECV 2 SR</send>
                </command>
`

Danke!

speters, 2021-03-21:
Entschuldige bitte, ich war mit deiner WW-Solltemperatur auf dem Holzweg.

`0x6000` scheint in der Tat richtig zu sein, das ist ein Zwei-Byte Wert, der als Double gespeichert wird. Zur Umrechnung in °C dann 1/10. Wertebereich von 10,0 bis 70,0.
Die Definition in deiner XML sieht somit m.E. gut aus.

Zum Setzen der Temperatur übergibst du bei beim obigen Setup die einzustellende Temperatur als dimensionslose Größe.
Also z.B. 45, nicht den Rohwert 4500 (0x1194), denn der vcontrold rechnet den Wert ja gemäß der set-Definition im XML selber um.

Kannst du bitte auch nochmal sicherstellen, dass du die _aktuelle_ Version von [vcontrold](https://github.com/openv/vcontrold) verwendest.

meschmesch, 2021-03-21:
Tja, daran könnte es liegen. Alles läuft bei mir auf einem RaspberryPi 4 und ich habe mich an die Anleitung von https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi gehalten. Extra heute Mittag nochmals in /usr/local/bin bwz sbin die bisherigen Dateien vcontrold, vclient und vsim gelöscht und dann nochmals die Anleitung durchgegangen. Wenn ich dann aber ins Terminal von vctrld gehe (telnet localhost 3002) wird mir als Version 0.98.2_IPv6 angezeigt. Laut https://github.com/openv/vcontrold/releases ist die aktuelle Version aber 0.98.10 ?

speters, 2021-03-22:
Dann ersetze zunächst einmal deinen alten vcontrold/vclient durch aktuelle Versionen, um auszuschliessen, dass ein bereits behobener Fehler in vcontrold zu deinem Problem führt.

Du findest neue Versionen entweder vorkompiliert als Binärpakete unter https://github.com/openv/vcontrold/releases (für RPi4 wäre das dann für ARMHF) oder kannst sie selber aus den Sourcen kompilieren.

Es reicht, wenn du die ausführbaren Dateien `vcontrold` und `vclient` erneuerst..

speters, 2021-03-22:
Dein Kommando sendet 0x2d00 (11520) als rohen Temperaturwert. Das versteht die Heizung als 115,2°C, was deutlich oberhalb des akzeptierten Temperaturbereiches ist.

Zum Setzen von 45°C müsstest du 0x01c2 (450) senden (für 45,5°C 0x01c7, usw.).

<hr />

Für die Adresse 0x6300 findet sich in den XML-Dateien der offiziellen VitoSoft-Demo noch folgender Hinweis:

> Warmwassertemperatur einstellen
Hinweis: Die Wärmepumpe allein kann keine Warmwassertemperatur über 50°C (55°C 
bei VC350) bereitstellen (abhängig von Speicher und Anlagenauslegung). Diese 
Temperaturen sind im allgemeinen nur über Heizwasserdurchlauferhitzer oder 
Elektroheizstäbe im Speicher erreichbar.

In der Vitosoft-Demo ist für die Adresse ein Wertebereich von 10°C bis 70°C zulässig.

Öffentlich verfügbare Angaben zu dem Datenpunkt an Adresse 0x6000 habe ich lediglich in [vt200wo1ab_wpr300en.pdf](https://connectivity.viessmann.com/content/dam/vi-micro/CONNECTIVITY/Vitogate/Vitogate-200/7452573/EN/vt200wo1ab_wpr300en.pdf/_jcr_content/renditions/original.media_file.download_attachment.file/vt200wo1ab_wpr300en.pdf) finden können, das deckt sich aber mit dem, was auch aus den Konfigurationsdateien der VitoSoft-Demo zu ziehen ist.

Aber Achtung:
Ich kann in der VitoSoft-Demo kein Gerät mit der von Dir genannten Device-ID 204D finden, also könnte es da ggf. doch irgendwelche Fallstricke geben.

meschmesch, 2021-03-22:
Naja, ich habe testweise mal das Ursprüngliche mit 20°C probiert:

_vctrld>SetTempWW20
NOT OK
vctrld>GetTempWW
20.000000 Degrees Celsius_

Also einerseits die Meldung NOT OK, funktioniert aber dennoch. Die Solltemperatur vom Warmwasser hat in diesem Fall von (von Hand) eingestellten 48Grad auf 20Grad umgeschaltet, habe ich auch am Gerät selbst verifiziert. Die Adresse 0x6000 stimmt also.

Zugehörge Kommandos:
```
                <command name="setaddr20">
                     <send>SEND 00 02 $addr $hexlen C8 00;RECV 2 SR</send>

xxxxxxxxxxxxxxxxx

                <command name="SetTempWW20" protocmd="setaddr20">
                        <addr>6000</addr>
                        <len>2</len>
                        <unit>UT</unit>
                        <description>Warmwasserzieltemperatur setzen 20 Grad</description>
                </command>

```
By th way, 0x01c2 kann ich nicht senden, dann gibt es eine Fehlermeldung

_vctrld>SetTempWW45
ERR: >FRAMER: ERROR address 6000 code 33
Error in recv, terminating
Error executing SetTempWW45_

```
                <command name="setaddr45">
                     <send>SEND 00 02 $addr $hexlen 01 C2;RECV 2 SR</send>
                </command>
```

meschmesch, 2021-03-22:
Ich habe das Problem gelöst. Das Geheimnis liegt darin, den Hex-Wert der Zieltemperatur von hinten geschrieben anzugeben. Also 45 Grad ist als 450 zu schreiben: HEX wäre das 02 26. Das Kommando muss dann umgekehrt lauten 26 02. Also konkret

```
                <command name="setaddr45">
                     <send>SEND 00 02 $addr $hexlen 26 02;RECV 2 SR</send>
                </command>
```

speters, 2021-03-22:
Super, dass es nun funktioniert und danke für die Rückmeldung.

closed 2021-03-22 by speters