openv/openv #229: Adresse Betriebsmodus gesucht

WikiSpaxe, 2017-10-26:
* Subject: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: ingo0042 <userid-52297486@openv.wikispaces>
* Date: Sat,  3 Jan 2015 21:36:40 UT
* Message-ID: <71257176-78193496$openv.wikispaces>

Hallo,

ich habe eine VScotHO1 (20CB) und nach Tests (und lesen diverser Beiträge hier) habe ich festgestellt, dass ich den Partymodus nur im Nicht-Reduzierten Betrieb setzen kann.
Hat jemand von Euch schon eine Adresse heraus gefunden, über die ich den Betriebsmodus (Normalbetrieb, Reduzierter Betrieb (Tag-Nacht-Symbol auf der Vitotronic)) herausfinden kann?
Ich habe es mit der 23 00 und der 25 00 versucht, aber die ändern sich nicht beim Wechsel des Betriebsmodus.

Danke im Voraus
Ingo

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: ingo0042 <userid-52297486@openv.wikispaces>
* Date: Sat,  3 Jan 2015 21:40:09 UT
* Message-ID: <71257176-78193524$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

wah, ich meinte 2500 und 3500 wie ich in dieser Doku gelesen hatte:
http://www.viessmann.de/content/dam/internet-global/software-daten/vitogate_200_eib-daten2/vt200ho1a_en.pdf

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sun,  4 Jan 2015 00:03:14 UT
* Message-ID: <71257176-78194422$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

Ich habe die gleiche Therme. Akt. Betriebsmodus in 0x2323. Lade Dir Viessdata 2.0, darin ist eine Liste mit  Adressen, die bei meiner Therme funktionieren.

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: ingo0042 <userid-52297486@openv.wikispaces>
* Date: Sun,  4 Jan 2015 15:05:18 UT
* Message-ID: <71257176-78197834$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

Hallo,

unabhängig ob die Therme gerade im Normalbetrieb (Timer) oder im Reduzierten Betrieb (Timer) ist, ich lese aus
0x2323
0x3323
0x2500 und
0x3500 immer nur eine 02 heraus
Der Wert in 0x2323 bzw. 0x3323 ändert sich nur, wenn ich das Betriebsprogramm von Heizen+WW auf WW ändere.
Ich suche aber halt eine Adresse über die ich herauslesen kann, ob die Anlage im Reduzierten oder im Normalen Betrieb ist.

MfG Ingo

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sun,  4 Jan 2015 15:54:49 UT
* Message-ID: <71257176-78198178$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

Ah, verstehe, Du meinst, ob ein Heizkreis gerade laut Heizzeittabelle im normalen oder im reduzierten Modus unterwegs ist. Sorry, danach habe ich noch nicht gesucht. Könntest Du natürlich direkt mit der Heizzeittabelle auswerten. Ich bin aber fast sicher, Du findest die Info in Adresse 2500 oder eine der 21 folgenden. Ich habe da auch die aktuell aktive RT-Soll gefunden, sie steht an Adresse 0x250C und 0x250D. Diese Adresse könnte Dir auch nützen, allerdings musst Du aufpassen, da einige Codierungen Einfluss auf die aktuelle RTS haben, z.B. F8/F9, Party-/Sparbetrieb, etc.

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: ingo0042 <userid-52297486@openv.wikispaces>
* Date: Sun,  4 Jan 2015 16:23:25 UT
* Message-ID: <71257176-78198446$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

Genau, ich wollte es halt nu nicht unbedingt aus den Schaltzeiten berechnen sondern gerne abfragen.
Ich hab einfach mal 0x2500 - 0x251F bzw. da es bei mir um M2 geht 0x3500 - 0x351F abgefragt (jeweils 1 Byte Länge) aber ich habe keine Vernünftigen Antworten bekommen ...

WikiSpaxe, 2017-10-26:
* Subject: Re: Adresse Betriebsmodus gesucht
* Wiki-Page: home
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sun,  4 Jan 2015 17:40:00 UT
* Message-ID: <71257176-78199170$openv.wikispaces>
* In-Reply-To: <71257176-78193496$openv.wikispaces>

Hab mal etwas recherchiert.....ich glaube 0x2501 ist eine vielversprechende Adresse. Sie gibt bei Betriebsmodus "Heizen und WW" innerhalb der Heizzeit 02 und innerhalb des reduzierten Betriebs 00 aus. Nicht verifiziert, aber vielleicht ein erster Ansatzpunkt. Wichtig: Beim Lesen von 0x2500 und folgenden Adressen per Protokoll 300 sollten immer 22 Bytes gelesen werden und dann die interessanten Bytes/Bits aus diesen Daten gefiltert werden.

passuff, 2018-11-23:
Ich habe das selbe Problem. Lese derzeit 2500/3500 und erhalte immer Betriebsart Normal. Was muss ich genau tun? Ich habe schon die Adresse 2501/3501 versucht, leider ohne Erfolg. Wie soll ich 22 bytes lesen?