openv/openv #243: 209F Vitotronic 200 (KO2B) V200KO2B - Neulingfrage

WikiSpaxe, 2017-10-26:
* Subject: 209F Vitotronic 200 (KO2B) V200KO2B - Neulingfrage
* Wiki-Page: home
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat, 11 Apr 2015 23:08:22 UT
* Message-ID: <73137642-81067556$openv.wikispaces>

Liebes Forum, ich hoffe ich nerve nicht mit meiner Neulingfrage.

In der Geräteliste hier:  http://openv.wikispaces.com/Ger%C3%A4te  gibt es schon einen Eintrag zu meiner Steuerung, die ich mit meinem neu erstandenen Optolink auslesen möchte.

ID           209F
ExtID        0002
Protocol     300
Description  Vitotronic 200 KO2B: Witterungsgeführte Kessel- u. Heizkreisregelung für gleitend abgesenkte Kesselwassertemperatur mit Regelung für 2 Heizkreise mit Mischer

Jetzt suche ich die Adressliste und dachte ich finde sie hier:
http://openv.wikispaces.com/Adressen

Leider ist diese Steuerung noch nicht gelistet. Die Suche über Google und im Forum hat ergeben, dass es einige wenige User gibt, die diese Steuerung haben.

In der Hoffnung eine Antwort mit idealerweise einem Tip zur Adressliste zu erhalten ... poste ich diese Nachricht.

Viele Grüsse
Michael

WikiSpaxe, 2017-10-26:
* Subject: Re: 209F Vitotronic 200 (KO2B) V200KO2B - Neulingfrage
* Wiki-Page: home
* From: sbsUlm <userid-53864431@openv.wikispaces>
* Date: Wed,  1 Jul 2015 19:26:53 UT
* Message-ID: <73137642-82849293$openv.wikispaces>
* In-Reply-To: <73137642-81067556$openv.wikispaces>

Hallo, 

schau mal hier: http://openv.wikispaces.com/share/view/65878710

Gruß
Sebastian

Mario275, 2019-11-17:
Hallo,
bin das erste mal hier bei Github, der letzte Beitrag ist schon etwas her, aber ich habe seit kurzem auch so eine Viessmann Steuerung und habe mir den Optolink Adapter besorgt. Die Links zu Wikispaces gehen ja nicht mehr. Nun sehe ich hier nirgends eine aktuelle .xml oder Adressliste mit der ID 209F wo alle möglichen Parameter enthalten sind. Ich habe die Windows Software Viess-Data 2.0 und Vito-Test probiert sowie RS232 Test, bekomme aber nur vereinzelte Ist-Werte raus und dann hängt sich die Software teilweise auf. Möchte gerne das ganze später über vcontrold auslesen und mit ioBroker kommunizieren um die Werte mit meiner selbst programmierten Haussteuerung zu verknüpfen. 
Wäre nett wenn mir hier jemand behilflich sein könnte.

Gruß Mario

philverh, 2019-11-18:
Hello, sorry for writing in English, but my German is too poor to write.
The adresses in WikiSpaces have moved. For example, the address in the first append in this "issue" is now available at https://github.com/openv/openv/wiki/Ger%C3%A4te and the next one at https://github.com/openv/openv/wiki/Adressen where you will find some info about your device.
The actual content of the .xml files has always been left as an exercise to the reader :-(

GolterU, 2019-11-18:
Hallo Mario,
lies mal die [Beiträge unter #432](https://github.com/openv/openv/issues/432). Dort geht es um ähnliche Probleme und deren Lösungen. Wenn unter Windows gearbeitet wird, ist IMHO [Vies-sion](https://github.com/openv/openv/wiki/Vies-sion) immer noch das best geeignete (wenn auch veraltete) Programm.

Gruß Golter

Mario275, 2019-11-19:
Hallo Uwe, danke habe mir da alles mögliche durchgelesen. Ich denke da ich ja mittlerweilen Daten aus der Steuerung mit div. Programmen hier auf Github bekomme (USB Port passt also) liegt es an mein grundsätzliches Verständnis von Cygwin. Der Server Cygwin läuft ja und alle Anwendungen und xml Dateien liegen im richtigen Ordner. Also vcontrold.exe , vclient.exe , vito.xml und vcontrold.xml.Ich starte cygwin und bin im Commandfenster von Cygwin. So nun starte ich wie, oder besser gesagt woher weis ich das vcontrold läuft? Wenn ich vclient öffne z.b mit vclient -h 127.0.0.1:3003 -c 'getDevType' oder sonstige Befehle, bekomme ich entweder gar keine Antwort oder -bash: (jeweiliger Aufruf): Kommando nicht gefunden. Komm einfach nicht weiter. Kann sein das meine .xml Dateien total falsch eingestellt sind??Zumindest erkennt ioBroker mit dem Viessmann Adapter die vcontrold.xml als successfull. Nur die Vito.xml wird nicht ordnungsgemäß angenommen.Gruss MarioVon meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/openv/openv/issues/243?email_source=notifications\u0026email_token=ANZPN5UK4FOQEFUY5BULQ63QUKNZNA5CNFSM4EA5TMM2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEEKQVUA#issuecomment-555027152",

"url": "https://github.com/openv/openv/issues/243?email_source=notifications\u0026email_token=ANZPN5UK4FOQEFUY5BULQ63QUKNZNA5CNFSM4EA5TMM2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEEKQVUA#issuecomment-555027152",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

GolterU, 2019-11-20:
Hallo Mario,
ich hatte Dich so verstanden, dass es nur native Windows-Programme sind, die zum Laufen gebracht werden sollen. Wenn es um Linux-artige Dinge geht, die per Cygwin laufen sollen, bin ich leider nicht der richtige Ansprechpartner.
Aber es gibt ja genügend diesbezügliche Experten hier...
Gruß
Uwe