openv/openv #246: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only

WikiSpaxe, 2017-10-26:
* Subject: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat,  2 May 2015 21:16:32 UT
* Message-ID: <73543608-81693438$openv.wikispaces>

Hallo,

ich habe mal hier http://openv.wikispaces.com/Adressen die von mir verifizierten Adressen für die V200KO1B in einer neuen Spalte eingetragen.

Es ist total schade, aber es scheint, dass die Betriebsart nur read-only ist. Ich würde so gerne über mein openhab oder script die Betriebsart ändern können. Ich kann über setBetriebArtM1 die Betriebsart über vcontrold / vclient auf WW oder H+WW FS oder H+WW setzen ... es kommt OK zurück. An der Steuerung ändert sich jedoch die Einstellung nicht. Es bleibt einfach so, wie es per Hand am Display eingestellt wird. Bei Auslesen kommt dann z.B. ... egal was man setzt immer "WW" zurück, obwohl ich vorher über den Setzbefehl die Betriebsart eigentlich geändert hatte.

Es ist eine sehr grosse Einschränkung. hoffentlich hat jemand eine gute Idee :-) Würde mir total gefallen die Betriebsart über Handy / Openhab umschalten zu können ohne in den Keller laufen zu müssen.

vcontrold.xml
        <unit name='BetriebsArt'>
           <abbrev>BA</abbrev>
           <type>enum</type>
           <enum bytes='00' text='WW'/>
           <enum bytes='01' text='RED'/>
           <enum bytes='02' text='NORM'/>
           <enum bytes='04' text='H+WW FS'/>
           <enum bytes='03' text='H+WW'/>
           <enum bytes='05' text='ABSCHALT'/>
           <enum text='UNKNOWN'/>
        </unit>

Mein System:
Cubietruck / Linux System (Raspberry compatibel, nur schneller) mit SSD und USB Optolink über 5m aktives USB Verlängerungskabel angeschlossen
Software: Openhab, Cronjob mit Script zur Datenerfassung über vcontrold und über http-request/curl speicherung der Daten in Openhab, Influxdb zur Speicherung der Daten und Grafana zur optischen Darstellung.

WikiSpaxe, 2017-10-26:
* Subject: Re: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat,  2 May 2015 21:18:52 UT
* Message-ID: <73543608-81693462$openv.wikispaces>
* In-Reply-To: <73543608-81693438$openv.wikispaces>

Ich greife auf die Betriebsart über Adresse 0x2301 zu. Erfolgreich auslesen ist möglich aber eben kein umschalten.

WikiSpaxe, 2017-10-26:
* Subject: Re: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat,  2 May 2015 21:24:03 UT
* Message-ID: <73543608-81693512$openv.wikispaces>
* In-Reply-To: <73543608-81693438$openv.wikispaces>

Ich verwende das KW2 Protokoll
xxx@cubietruck:/etc/vcontrold$ less vito.xml

        <device ID="209F" name="V200KW2" protocol="KW2"/>
        <command name='getBetriebArtM1' protocmd='getaddr'>
                <addr>2301</addr>
                <len>1</len>
                <unit>BA</unit>
                <description>Betriebsart M1</description>
                <device ID="2053">
                    <addr>51</addr>
                    <len>1</len>
                    <unit>BA</unit>
                </device>
        </command>
        <command name='setBetriebArtM1' protocmd='setaddr'>
                <addr>2301</addr>
                <len>1</len>
                <unit>BA</unit>
                <description>Setze Betriebsart M1</description>
                <device ID="2053">
                    <addr>51</addr>
                    <len>1</len>
                    <unit>BA</unit>
                </device>
        </command>

WikiSpaxe, 2017-10-26:
* Subject: Re: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat,  2 May 2015 21:49:28 UT
* Message-ID: <73543608-81693658$openv.wikispaces>
* In-Reply-To: <73543608-81693438$openv.wikispaces>

Habe jetzt nach Internetrecherche noch die Adresse 2323 gefunden ... UNGLAUBLICH ... damit geht es. Sorry für den Post.

vito.xml ergänzt um:

        <command name='getBetriebArtM1Test' protocmd='getaddr'>
                <addr>2323</addr>
                <len>1</len>
                <unit>BA</unit>
                <description>Betriebsart M1</description>
        </command>
         <command name='setBetriebArtM1Test' protocmd='setaddr'>
                <addr>2323</addr>
                <len>1</len>
                <unit>BA</unit>
                <description>Setze Betriebsart M1</description>
        </command>

... und ausprobiert (nach reload der config)

xxx@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1'
getBetriebArtM1:
H+WW
xxx@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1Test'
getBetriebArtM1Test:
NORM

Jetzt auf WW am Display der Steuerung per Hand umgestellt

linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1Test'
getBetriebArtM1Test:
RED
linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1'
getBetriebArtM1:
WW

linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'setBetriebArtM1Test NORM'
setBetriebArtM1Test NORM:
OK
linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1'
getBetriebArtM1:
H+WW
linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'setBetriebArtM1Test RED'
setBetriebArtM1Test RED:
OK
linaro@cubietruck:/$ sudo vclient -h 127.0.0.1:3002 -c 'getBetriebArtM1'
getBetriebArtM1:
WW
linaro@cubietruck:/$

.... Juhuuu jetzt geht es :-) .... also über Adresse 2323

WikiSpaxe, 2017-10-26:
* Subject: Re: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: genius30mc <userid-52189194@openv.wikispaces>
* Date: Sun,  3 May 2015 08:44:18 UT
* Message-ID: <73543608-81697640$openv.wikispaces>
* In-Reply-To: <73543608-81693438$openv.wikispaces>

Unglaublich ???  in ViTalk wird die Afresse 2323 seit Jahren erfolgreich versendet !! (vito_parameter.c)

WikiSpaxe, 2017-10-26:
* Subject: Re: Habe V200KO1B erfolgreich angeschlossen - Betriebsart aber nur read-only
* Wiki-Page: Adressen
* From: henk.oosterlinck <userid-56576367@openv.wikispaces>
* Date: Sun, 13 Nov 2016 20:00:00 UT
* Message-ID: <73543608-91895220$openv.wikispaces>
* In-Reply-To: <73543608-81693438$openv.wikispaces>

Hello,

I know this thread is relatively old, .

When using what is described here I am a bit confused.
The meaning of betriebart is an enum as follows:
'00' ='WW'
'01' ='RED'
'02' ='NORM'
'03' ='H+WW'
'04' ='H+WW FS'
'05' ='ABSCHALT'

When you read the current operating mode with "getBetriebsartM1" you read address 2323.
When you write the current operating mode with "setBetriebsartM1" you write address 2323.
I would expect the same value, but not.

As you see in the above you write "NORM-02" to adress 2323, subsequently you read it and you get back "H+WW-03".
I would have expected "norm-02" which is the value you wrote in the subsequent read.

Afterwards you write a "RED-01", and you get with the subsequent read a "WW only -00".
The value you write is not the same as the value you read afterwards.

I made a correspondence table between what you write and what you read afterwards:
1-> '00' ='WW'
3 -> '01' ='RED'
4 -> '02' ='NORM'
2 -> '03' ='H+WW'
5 -> '03' ='H+WW'
0-> '05' ='ABSCHALT' 
 
Note that if you have 2 heating circuits/heisskreize you can use the adress 3323 for getbetriebsartM2 (for floor-heating or so).
Same problem with that address.

Can anyone explain?

What does "NORM" and "RED" correspond to on the display of viessman, in the menu of the heater I only have "ww", "ww+h", "off" (3 options not 5).

I guess "H + ww FS" means H+ww frostschutz, but i cannot really find this on the viessman itself.
It seems impossible to write that value.

Can any of you clarify this a bit more?

Henk

mactoolz, 2018-11-04:
Hi,

darf ich fragen in weit ihr die KO1B XML Beschreibung zur Verfügung stellen könnt.

Das wäre sehr nett ...

Danke

Gruß
MacToolz

mactoolz, 2018-11-06:
Hallo,

falls es jemanden interessiert ich habe Informationen was die KO1B angeht.

Gruß

Mario275, 2019-12-11:
Hallo,

suche verzweifelt eine Datenpunktliste oder fertige Vito.xml für meine KO2B Type "209F".
Ist doch die gleiche wie die KO1B. 

Gruß Mario275

king-rulez, 2020-11-23:
Hallo,

ich suche ebenfalls eine fertige Vito.xml und Vcontrold.xml für die KO1B Steuerung.

Julez318, 2024-01-23:
Hallo :)
gibt es zu der KO1B/KO2B Steuerung eine fertige Vito.xml / Vcontrold.xml oder andere nützlich Daten?

Gruß Julez

Juchar, 2024-10-01:
Hi @mactoolz , falls du noch aktiv bist - könntest du die Infos zur KO1B evtl. zur Verfügung stellen? Werde gerade wahnsinnig bei der Suche nach Adressen. Vielen Dank schon mal :-)

abrits56, 2024-10-30:
@Juchar 
Bist Du in Sachen KO1B/2B weitergekommen?
Grüße Johannes

Juchar, 2024-10-30:
@abrits56  leider nein, keine Rückmeldung bekommen.

Vinshi01, 2025-05-04:
Leider gleiches problem. Bei 2323 liest er mir keine BA mehr aus und bei 2301 schon aber setzen geht dann nicht. 

Es ist eine 20CB.