openv/openv #257: Vitoligno 300-P (Pellets)

WikiSpaxe, 2017-10-26:
* Subject: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: sthones <userid-54730189@openv.wikispaces>
* Date: Sat, 10 Oct 2015 20:12:25 UT
* Message-ID: <75598823-84621679$openv.wikispaces>

Hallo,

gibt es Nutzer mit dem Vitoligno 300-P, die openv aktuell nutzen? ich habe damit vor wenigen Wochen angefangen, und wuerde gerne ein paar Erfahrungen austauschen. Der letzte Thread, den ich hier gefunden hatte, war von 2013 ( http://openv.wikispaces.com/share/view/60727718 ), daher bin ich unsicher, ob es tatsaechlich aktive Nutzer gibt.

Gruss
  sthones

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: -Merkur- <userid-52601474@openv.wikispaces>
* Date: Tue, 27 Oct 2015 08:57:51 UT
* Message-ID: <75598823-85109359$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Hallo sthones,

ich habe einen dieser Kessel und auch schon einen Adapter für die Anbindung. Mir fehlte bisher nur der Stein des Anstoßes das mal in Betrieb zu nehmen, daher bin ich 'noch' kein aktiver Nutzer. Aber wo die Heizperiode wieder langsam Fahrt aufnimmt vermute ich, das mein Spieltrieb bald wieder die Oberhand gewinnt....

Welche Soft- & Hardware hast Du im Einsatz ?

Gruß,

Merkur

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: sthones <userid-54730189@openv.wikispaces>
* Date: Tue, 27 Oct 2015 09:28:24 UT
* Message-ID: <75598823-85109813$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Hallo Merkur,

ich habe einen Raspi mit dem selbstgebauten Adapter hier von der Seite (Bauanleitung Raspberry Pi) - wobei der Adapter selber eigentlich keinen Unterschied machen sollte, das sind alles bloss Adapter auf die serielle Schnittstelle, mit denen man dem vcontrold den Zugriff auf den Kessel erlaubt.

Als Software nutze ich eine Loesung, die auf der 'Auswertung mit RRDB' basiert und per cron alle paar Minuten die Werte liest und ueber den Webserver eben die Graphen erzeugt.

Ich habe von den Adressen (http://openv.wikispaces.com/share/view/62042248) einige bestaetigen koennen, und manche verwerfen muessen. Und auf einige Werte, die mich interessieren koennten, komme ich bislang gar nicht.

Gruss
  sthones

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: -Merkur- <userid-52601474@openv.wikispaces>
* Date: Tue, 27 Oct 2015 14:40:31 UT
* Message-ID: <75598823-85115487$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Hallo sthones,

jetzt kommt schon das Problem was ich bekommen werde - meine Kenntnisse in Linux sind bestenfalls rudimentär, trotz der Tatsache das ich (auch?) mein Geld mit IT verdiene. Einen Raspi habe ich zwar auch, aber nur für Kodi. D.h. ich muss mich wohl an unter Windows lauffähige Software halten. Aber ich vermute Du bist daran interessiert Adressen auszutauschen, und da spielt die verwendete Software ja keine Rolle. Womöglich finde ich die Woche Zeit das mal aufzubauen und zu testen. Dann lasse ich von mir hören.

Gruß,

Merkur

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: -Merkur- <userid-52601474@openv.wikispaces>
* Date: Wed, 28 Oct 2015 07:51:28 UT
* Message-ID: <75598823-85148231$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Moin,

gestern Abend habe ich mich doch tatsächlich mal hingesetzt und das aufgebaut und getestet - und es ging wider erwarten auch sofort. Die ersten Daten wie Brennerstarts und Betriebsstunden etc.pp konnte ich auslesen. Jetzt muss ich nur noch schauen welche Software sich für den Dauerbetrieb eignen würde. Kam schon auf die Idee mit AutoIT selber was zu schreiben - hab mir schonmal ein paar Funktionen zum Ansprechen der seriellen Schnittstelle aus dem Netz gesucht, aber ob ich da jemals richtig Lust zu haben werde.... ^^

Gruß,

Merkur

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Wed, 28 Oct 2015 09:07:21 UT
* Message-ID: <75598823-85149299$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Moin Merkur,
ich war vor 4 Jahren auch noch blutiger Unix-Laie (mit IT-Hintergrund). Angefangen mit WindowsXP und einem alten Notebook habe ich dann recht schnell auf Unix gesetzt.
Ich kann dir nur raten dich mal mit dem Raspberry zu beschäftigen. Die Investition ist überschaubar und die Probleme zumindest nach meiner Erfahrung klein. Du wirst mit Sicherheit mehr Zeit mit AutoIT zubringen, als mit dem "Unix-Lernen"! Zusätzlich hast Du ein sparsames Logging (5W).

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitoligno 300-P (Pellets)
* Wiki-Page: home
* From: sthones <userid-54730189@openv.wikispaces>
* Date: Wed, 28 Oct 2015 14:10:14 UT
* Message-ID: <75598823-85154529$openv.wikispaces>
* In-Reply-To: <75598823-84621679$openv.wikispaces>

Hallo Merkur,

na das war ja einfach :)
warmup's hinweis kann ich eigentlich nur bestaetigen... Linux ist fuer sowas klar zu empfehlen... ich kann aber auch die Einstiegshuerde nachvollziehen.
Und fuer die Adresses ist das Betriebssystem tatsaechlich nicht relevant.

Hier sind die Adresses, die ich lese (addr / len / unit - aus dem vito.xml)
7507 / 9 / ES / Fehler 1
5525 / 2 / TD / getTempAussen
0B12 / 2 / TD / getTempKessel
0810 / 2 / TD / getTempPufferOben
0812 / 2 / TD / getTempPufferUnten
0814 / 2 / TD / getTempWarmwasser
6500 / 2 / TD / getWarmwasserSoll
0B14 / 2 / TD / getTempFlamme
A305 / 1 / PC1 / getKesselLeistung (aber nur 0 oder 100)
08A7 / 4 / HS / getBrennerStunden
088A / 4 / CO / getBrennerStarts
08B0 / 4 / BKG / getBrennerVerbrauch
0846 / 1 / E_ST / getStatusZirkulation

ab hier wird's unsicher:

2323 / 1 / E_BA4 / getBetriebsmodusHK1 (eigentlich muesste der konstant auf 'Heizen und WW stehen', aber der schwankt gelegentlich mal.

Werte zu setzen habe ich mich bislang nicht getraut.

Gruss
  sthones

sam-ams, 2018-02-14:
Hallo Zusammen,
ich bin vor einigen Tage auf diesen Issue gestoßen und etwas Feuer gefangen, sprich ich habe mir ein Optolink Adapter besorgt und den Raspi aufgesetzt. Soweit so gut, ich bekomme Werte aus der Heizung ausgelesen, habe aber Schwierigkeiten Eure Adressen in den xml's zu konfigurieren und bin mir auch nicht sicher, ob ich das alles richtig verstehe, wie die Software wirklich arbeitet... Habt Ihr einen Tipp, welche Kapitel hier für das Verständnis notwendig sind? 
Gibt es pro Heizungstyp eine fertige xml? 
Sorry, ich stehe komplett auf dem Schlauch und Ihr könnt mich hier etwas unterstützen...

Viele Grüße,
Arne