openv/openv #259: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung)

WikiSpaxe, 2017-10-26:
* Subject: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Sat, 17 Oct 2015 18:52:43 UT
* Message-ID: <75707315-84816203$openv.wikispaces>

Hallo,

wie im Betreff schon beschrieben besitze ich eine Luft/Wasser Wärmepumpe Vitocal 350 Ausführung Außenaufstellung.
Ich habe den Optolinkadapter nach der Raspberry Pi Bauanleitung nachgebaut und mit einem USB/RS232 Wandler an den USB meines PC´angeschlossen.
Die Sende und Empfangsled habe ich Optisch Kurzgeschlossen und mit Hyperterminal (4800 8-E-2) getestet. Die Befehle die ich über Hyperterminal sende Empfängt mein PC auch wieder. Somit scheint der Adapter in Ordnung zu sein.

- Mit dem Programm voIdent bekomme ich von der CD Steuerung die "ID 3F20" "ExtID 3F4E" Protocol "KW"
- Mit dem Programm Vitotest bekomme ich von der CD Steuerung die "ID 3F20" aber auch den Fehler: An internal Error occurred.
- Mit der Software v-control 1.3.0M (Windows) wird bei der ist die Verbindung mal Okay, manchmal bekomme ich einen Init Error.
- Mit dem vcontrol (Linux) erhalte ich bei getDevType als Antwort mal ne "05 05" manchmal eine "20 05".

Optischer Kurzschluss schließe ich aus, da ich auf der V-Arretierung meiner gebauten Platine mit Isoband eine Optische Trennung habe.

Habe auch schon verschiedene Widerstände an der IR Sende LED Probiert. 100 Ohm, 180 Ohm 330 Ohm.

Meine Vermutung ist dass evtl. ein die Initialisierung auf dieser Steuerung ewas anders Funktioniert, oder das Timing der gesendeten Befehle nicht stimmt.

Meine Fragen an euch:

Hat jemand eine Idee Was das das Problem sein könnte?
Ist die ID von voIdent korrekt? Bedeutet das meine ID ist 204E?
Hat jemand eine CD70 + Vitocal 350 am Laufen?

Würde gerne das Projekt zum laufen bekommen und auch hier meine Ergebnisse / Masterdateien präsentieren!

Mfg
Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Sun, 18 Oct 2015 23:42:30 UT
* Message-ID: <75707315-84840731$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

Ich habe heute mit RealTerm gearbeitet. 
Die periodische 0x05 wurde empfangen. 
Beim senden des befehls 0x01 0xF7 0x00 0xF8 0x04 (ID Kennung) mit der einstellung 4800 8E2 NONE gab Realterm den Fehler "UART receiver framing error" es wurde plötzlich immer "3F" und alle 2sec. "05" geschrieben. 

Wenn ich den usb/rs232 wandler im pc gesteckt lasse und nur den optolink abziehe wird die "3F" weitergeschrieben die "05" nichtmehr. 
Somit glaube ich grad eher an einen fehler des usb/rs232 adapters. 

Laut i-net wird diese fehler ausgeben, wenn start/stopbit einen anderen zustand haben als der adapter erwartet. 

Mit der einstellung 4800 8E1 None kommt kein fehler und es lassen sich daten aus der heizung lesen.

Kann jemand ein adapterproblem bestätigen?
Oder gibt irgendwo eine einstellung, die ich übersehen habe?

Mfg 

Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Sun, 18 Oct 2015 23:42:31 UT
* Message-ID: <75707315-84840733$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

Ich habe heute mit RealTerm gearbeitet. 
Die periodische 0x05 wurde empfangen. 
Beim senden des befehls 0x01 0xF7 0x00 0xF8 0x04 (ID Kennung) mit der einstellung 4800 8E2 NONE gab Realterm den Fehler "UART receiver framing error" es wurde plötzlich immer "3F" und alle 2sec. "05" geschrieben. 

Wenn ich den usb/rs232 wandler im pc gesteckt lasse und nur den optolink abziehe wird die "3F" weitergeschrieben die "05" nichtmehr. 
Somit glaube ich grad eher an einen fehler des usb/rs232 adapters. 

Laut i-net wird diese fehler ausgeben, wenn start/stopbit einen anderen zustand haben als der adapter erwartet. 

Mit der einstellung 4800 8E1 None kommt kein fehler und es lassen sich daten aus der heizung lesen.

Kann jemand ein adapterproblem bestätigen?
Oder gibt irgendwo eine einstellung, die ich übersehen habe?

Mfg 

Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: TerminatorIII <userid-12797127@openv.wikispaces>
* Date: Mon, 19 Oct 2015 19:36:19 UT
* Message-ID: <75707315-84868873$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

also ich sende immer 0x01 0xF7 0x00 0xF8 0x02 , da ich nur 2 Bytes erwarte.
01 : antwort nach 05
F7 Befehl lese
00F8 Kennung 
02 Anzahl der erwarteten Bytes
Gruß

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Tue, 20 Oct 2015 23:21:39 UT
* Message-ID: <75707315-84916643$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

Hallo,

wenn ich  0x01 0xF7 0x00 0xF8 0x02 dende erhalte ich 204E

Also muss die ID 204E sein!

Aussentemperatur bekomme ich auch zurück:

vctrld>getTempA
DEBUG:Tue Oct 20 23:00:42 2015 : Befehl: getTempA
DEBUG:Tue Oct 20 23:00:42 2015 : >SEND: 01
DEBUG:Tue Oct 20 23:00:42 2015 : >FRAMER: no preset result
DEBUG:Tue Oct 20 23:00:42 2015 : Warte auf 05
DEBUG:Tue Oct 20 23:00:42 2015 : <RECV: 05 (520.0 ms)
DEBUG:Tue Oct 20 23:00:42 2015 : >SEND: F7
DEBUG:Tue Oct 20 23:00:42 2015 : >SEND: 08
DEBUG:Tue Oct 20 23:00:42 2015 : >SEND: 00
DEBUG:Tue Oct 20 23:00:42 2015 : >SEND: 01
DEBUG:Tue Oct 20 23:00:42 2015 : >FRAMER: no preset result
DEBUG:Tue Oct 20 23:00:42 2015 : <RECV: len=1 53 (20.0 ms)
DEBUG:Tue Oct 20 23:00:42 2015 : <RECV: received 53
DEBUG:Tue Oct 20 23:00:42 2015 : Typ: short (in float: 83.000000)
DEBUG:Tue Oct 20 23:00:42 2015 : (FLOAT) Exp:V/10 [B0:53 B1:00 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
DEBUG:Tue Oct 20 23:00:42 2015 : 8.300000 Grad Celsius
8.300000 Grad Celsius

Der Wert passt, den Aussenfühler habe ich erwärmt und der ausgelesene Wert gleicht der Regelung.

Ich benötige für die CD70 Regelung aber weitere Adressen.
Hat jemand irgendwelche Adressen für die CD70 Regelung? vielleicht sogar mit der ID 204E?
Gibt es eine Software für einen "Adressenscan"?

Oder gibt es eine Möglichkeit an die Adressen ranzukommen, ohne mühseliges selbstheruasfinden.
Gut kontrolliert muss es trozdem werden, aber dass man mal eine Basic an Ardessen hat.

Habe auch Viessmann bzl. der HEX Adressen angschrieben

Mfg
Kevin 
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Sun, 25 Oct 2015 15:12:36 UT
* Message-ID: <75707315-85052989$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

Habe folgende Temperaturen für die ID 20E4 herausgefunden:

0800 = Außentemp.
0801 = Primär Ein
0802 = Primär Aus
0804 = Sek.Vorlauf
0805 = Sek.Rücklauf
0806 = Pufferspeicher unten
0811 = Verdampfer
080A = Heizkreis2 Vorlauf
080B = Pufferspeicher oben
080E = Warmwasserspeicher

Mfg 
Kevin
(Alleinzuhaus)

WikiSpaxe, 2017-10-26:
* Subject: Re: Verbindungsprobleme mit Witterungsgeführte Wärmenpumpenregelung CD70 + Luft/Wasser Wärmepumpe Vitocal 350 (Außenaufstellung) 
* Wiki-Page: home
* From: Alleinzuhaus <userid-54742509@openv.wikispaces>
* Date: Sun,  8 Nov 2015 18:04:43 UT
* Message-ID: <75707315-85510479$openv.wikispaces>
* In-Reply-To: <75707315-84816203$openv.wikispaces>

0812 = Heißgas

Mfg
Kevin
(Alleinzuhaus)

ha-ge, 2020-12-31:
Guten Tag zusammen!

Ich habe eine Vitocal 350-A mit einer CD 70 Wärmepumpenregelung. Ich danke an alle die hier Vorarbeit geleistet haben. Ich kann zur Zeit folgende Werte auslesen:

Adresse, Größe, Multiplikator, Beschreibung
0800, 2 Byte mit Vorzeichen, 0.1, Außentemperatur
0801, 2 Byte mit Vorzeichen, 0.1, Primär Ein
0802, 2 Byte mit Vorzeichen, 0.1, Primär Aus
0803, 2 Byte mit Vorzeichen, 0.1, ? 
0804, 2 Byte mit Vorzeichen, 0.1, Sekundär Vorlauf
0805, 2 Byte mit Vorzeichen, 0.1, Sekundär Rücklauf
0806, 2 Byte mit Vorzeichen, 0.1, Pufferspeicher unten
0807, 2 Byte mit Vorzeichen, 0.1, ? Sekundär Rücklauf
0809, 2 Byte mit Vorzeichen, 0.1, ?
080A, 2 Byte mit Vorzeichen, 0.1, Heizkreis 2 Vorlauf
080B, 2 Byte mit Vorzeichen, 0.1, Pufferspeicher oben
080C, 2 Byte mit Vorzeichen, 0.1, Heizkreis 2 Raum
080E, 2 Byte mit Vorzeichen, 0.1, WW Speicher Oben
080F, 2 Byte mit Vorzeichen, 0.1, ?
0810, 2 Byte mit Vorzeichen, 0.1, ?
0811, 2 Byte mit Vorzeichen, 0.1, Verdampfer
0812, 2 Byte mit Vorzeichen, 0.1, Heißgas
0813, 2 Byte mit Vorzeichen, 0.1, Fernbedienung 2

08A0, 1 Byte, 1, Verdichter
08A2, 1 Byte, 1, Ventilator
08A3, 1 Byte, 1, Sekundär Pumpe
08AA, 1 Byte, 1, Warmwasser Ladepumpe

Auch auf den restlichen Adressen zwischen 08A0 und 08CF bekomme ich ein Byte Antworten. Zuordnen kann ich diese Daten aber mit meinen eigenen Beobachtungen nicht.

Ebenso kann ich bestätigen dass es vier Byte Antworten von den Adressen 27xx, 37xx, 47xx, 57xx, 67xx, und 77xx in den unteren Bereichen gibt mit xx kleiner 30, manchmal auch noch etwas über 30.

Mich würden noch die Einschaltzähler für Verdichter, Ventilator und Pumpen interessieren. Fall es hier Hinweise gibt wo diese zu finden sind, bin ich sehr dankbar.

Gruß,
ha-ge

christophkreutzer, 2025-04-28:
Hallo zusammen, hallo @ha-ge

Weiss nicht ob's noch interessant ist, aber hier meine Erfahrungen (Vitocal 350 AWI 110, CD70 Steuerung, Bj. 2008, Innenaufstellung, Anlagenschema 4 - Warmwasser + 1 Mischer-Heizkreis mit Puffer).

Erstmal vielen vielen Dank für eure Versuche/Tests/Zusammenstellungen hier. Vieles davon konnte ich wiederverwenden :)

Im Laufe des vergangenen Jahres habe ich immer wieder etwas herumprobiert und folgendes herausgefunden:

### Protokoll

Die CD70 kann mit dem KW2-Protokoll angesprochen werden und unterstützt auch das Senden/Empfangen mehrerer Befehle zwischen den Syncs. Ich verwende folgende Protokolldefinition:
```xml
    <protocol name="KW2">
      <macros>
        <macro name="SYNC">
          <command>SEND 04;WAIT 05;SEND 01</command>
        </macro>
        <macro name="GETADDR">
          <command>SEND F7</command>
        </macro>
        <macro name="SETADDR">
          <command>SEND 04;WAIT 05;SEND 01 F4</command>
        </macro>
      </macros>
      <commands>
        <command name="getaddr">
          <retry>3</retry>
          <recvTimeout>300</recvTimeout>
          <send>GETADDR $addr $hexlen;RECV $len $unit</send>
        </command>
        <command name="getsyncaddr">
          <retry>3</retry>
          <recvTimeout>4000</recvTimeout>
          <send>SYNC;GETADDR $addr $hexlen;RECV $len $unit</send>
        </command>
        <command name="setaddr">
          <retry>3</retry>
          <recvTimeout>300</recvTimeout>
          <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 01 SR</send>
        </command>
        <command name="gettestaddr">
          <retry>3</retry>
          <recvTimeout>4000</recvTimeout>
          <send>SEND 04;WAIT 05;SEND 01 F7;SEND BYTES;SEND $hexlen;RECV $len</send>
        </command>
        <!--
        <command name="sync">
          <retry>0</retry>
          <recvTimeout>0</recvTimeout>
          <send>SYNC</send>
        </command>
        -->
      </commands>
    </protocol>
```

Dann schicke ich einfach den ersten und alle paar Commands dazwischen mit `getsyncaddr`, damit kann man gut 20 Werte in <10s abfragen (für `getsyncaddr` verwende ich `getDevType` und `getDevType2`, welche ich nachher ohnehin als "Verfügbarkeitscheck" benutze).

Die Steuerung hat auch manchmal Schluckauf, das Konfigurieren von 3 Retries hat aber Wunder geholfen, habe nun quasi nie Lesefehler, die auf die Applikationsebene durchschlagen.

Das Auslesen grösserer Adressbereiche habe ich mit vcontrold/vclient gemacht. Startadresse angeben, dann Länge immer grösser am Stück, bis die CD70 offensichtlich nur noch Gulasch ausgibt. Das ganze dann strukturiert in einen Editor und dann Hex-Patterns bekannter Werte im Editor gesucht und durch Umsetzen/Auslesen validiert.

Beispiel-Command:
```xml
    <command name="getTest" protocmd="getsyncaddr">
      <description>Testabfrage</description>
      <addr>3719</addr>
      <len>24</len>
    </command>
```
[... 64 lines cut ...]
371D = Heizkreis 2 Solltemperatur Raum reduziert in Grad C
3703 = Heizkreis 2 Heizkurve Niveau in K
3704 = Heizkreis 2 Heizkurve Neigung

5790 = Warmwasser Solltemperatur in Grad C
5770 = Warmwasser 2. Solltemperatur soll in Grad C

Vermutungen:

3707 = Heizkreis 2 Vorlauf min in Grad C
3708 = Heizkreis 2 Vorlauf max in Grad C
3719 = Hysterese Abschaltung? 2K
371A = Hysterese Einschaltung? 4K

Im Anhang noch meine aktuellen vcontrold.xml/vito.xml:
[vitocal350_2025-04-28.zip](https://github.com/user-attachments/files/19947247/vitocal350_2025-04-28.zip)

Was ich immer noch suche/vermisse (für eine "SGready"-Emulation zum Speicher laden):
* "Warmwasser max." oder "einmalige Warmwasserbereitung", siehe auch #201 
* Pufferspeicher-Festwert

Dazu habe ich weder bei 08xx, 47xx oder 57xx etwas sinnvolles gefunden.

Viele Grüsse
Christoph

ha-ge, 2025-05-10:
Guten Abend @christophkreutzer,

Danke für das ausführliche Update. Das hilft auch mir weiter. Bei den Recherchen sind nicht zufälligerweise auch Zählerstände zu Tage gekommen, wie z.B. die Zähler der Einschaltungen vom Verdichter.

Vielen Dank und Gruß,
ha-ge