openv/openv #262: WO1B: aktuelle Leistungsaufnahme des Verdichters.

WikiSpaxe, 2017-10-26:
* Subject: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Mon, 16 Nov 2015 20:38:34 UT
* Message-ID: <76296327-85767573$openv.wikispaces>

Hallo zusammen,

warmup hat in seinen rrd-Diagrammen eine Datenreihe, die in der Legende mit "Verbrauch" betitelt ist, ich denke mal das ist die aktuelle Leistungsaufnahme in kW. Hat mir vielleicht jemand (warmup? :-) die Adresse hierfür? Das wäre klasse!

Vielen Dank vorab,
Bernd

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Tue, 17 Nov 2015 05:28:40 UT
* Message-ID: <76296327-85780043$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Moin,
also die Werte im Chart stammen direkt aus dem Smart-Meter und werden mit einem Perl-Programm ausgelesen.
Die Vito bietet aber in der Tat für Wärmemenge und Verbrauch die berechneten Werte an.
 <command name='getWMZH' protocmd='getaddr'>
      <description>HeizwÃ¤rme (Heizen) Verdichter 1</description>
      <addr>1640</addr>
      <len>4</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
    <command name='getKWHH' protocmd='getaddr'>
      <description>Elektroenergie (Heizen) Verdichter 1</description>
      <addr>1660</addr>
      <len>4</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
    <command name='getWMZWW' protocmd='getaddr'>
      <description>HeizwÃ¤rme (Warmwasser) Verdichter 1</description>
      <addr>1650</addr>
      <len>4</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
    <command name='getKWHWW' protocmd='getaddr'>
      <description>Elektroenergie (Warmwasser) Verdichter 1</description>
      <addr>1670</addr>
      <len>4</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
Gruß

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Tue, 17 Nov 2015 07:58:07 UT
* Message-ID: <76296327-85781723$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Ah... Ich habe alle Adressquellen rauf und runter gesucht, aber nix gefunden. Dass die Daten aus dem Smartmeter kommen, erklärt natürlich, warum. Die Adressen für die wöchentliche Energiebilanz kannte ich bereits. Allerdings brauche ich die  Werte für die entsprechende Vorjahreswoche, um die sinnvoll interpretieren zu können, oder? Und die habe ich leider noch nicht, da ich erst seit zwei Wochen logge :-( Aber eigentlich müssten die ja auch irgendwo hinterlegt sein. Abgesehen von dieser Problematik, wäre es natürlich schön, man hätte einen Wert, der öfter aktualisiert wird... Vielleicht weiß ja jemand noch was dazu...

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Tue, 17 Nov 2015 17:47:10 UT
* Message-ID: <76296327-85793165$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hi Bernd,
nein Du brauchst dafür die Vorjahreswerte nicht. Die Codes kummulieren über eine Woche den Verbrauch und die WMZ. Sonntags um 24:00 (zum Montag) werden die Zähler in die Statistik übernommen und auf 0 gesetzt. Somit brauchst du in Deinem Skript immer nur den Vortag zu "merken" und die Differenz zu nehmen.

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Wed, 18 Nov 2015 00:38:59 UT
* Message-ID: <76296327-85806191$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hmm, ich fürchte, ich stehe auf dem Schlauch. Die Werte an diesen Adressen ändern sich bei mir nur einmal pro Woche - von Sonntag auf Montag. Und sie werden von Woche zu Woche geringer. Beispiel:

Aufgenommene Energie für die Warmwasserbereitung:
Vorwoche 4123,0 
Vor-vorwoche 4229,8
Vor-vor-vorwoche 4269,5

Abgegebene Energie für die Warmwasserbereitung:
Vorwoche 2746,0 
Vor-vorwoche 2801,8
Vor-vor-vorwoche 2830,2

Irgendwie schaff ich's nicht, diese Werte sinnvoll zu interpretieren. Als Datentyp in der Unit-Definition habe ich int und dividiere beim Abruf durch 10.

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Wed, 18 Nov 2015 06:34:39 UT
* Message-ID: <76296327-85814237$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Moin Bernd,
entschuldige bitte ich hatte an die falsche Stelle in meiner Vito.xml geschaut. Du hast recht - die Werte die das ausgelesen werden sind immer die für die letzten 12 Monate kummulierten.
Die korrekten Adressen der IST-Werte der aktuellen Woche (immer durch 10 geteilt) lauten
<command name='getEWMZH' protocmd='getaddr'>
      <description>WMZ Energie-Bilanz aktuelle Woche (Heizen) Verdichter 1</description>
      <addr>16D3</addr>
      <len>2</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
    <command name='getEKWHH' protocmd='getaddr'>
      <description>KWH Energie-Bilanz aktuelle Woche (Heizen) Verdichter 1</description>
      <addr>16D1</addr>
      <len>2</len>
      <error>05 05</error>
      <unit>CO10</unit>
      </command>
    <command name='getEWMZWW' protocmd='getaddr'>
      <description>WMZ Energie-Bilanz aktuelle Woche (WW) Verdichter 1</description>
      <addr>16D2</addr>
      <len>2</len>
      <error>05 05</error>
      <unit>CO10</unit>
    </command>
    <command name='getEKWHWW' protocmd='getaddr'>
      <description>KWH Energie-Bilanz aktuelle Woche (WW) Verdichter 1</description>
      <addr>16D0</addr>
      <len>2</len>
      <error>05 05</error>
      <unit>CO10</unit>
      </command>
Sorry nochmal war keine Absicht!
Gruß

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Wed, 18 Nov 2015 11:59:34 UT
* Message-ID: <76296327-85817963$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hallo warmup,

Danke für die Info, werde die neuen Adressen gleich heute Abend testen.

Vg,
Bernd

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Thu, 19 Nov 2015 23:37:47 UT
* Message-ID: <76296327-85876769$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hallo,

ich nochmal. Leider gehen diese Adressen bei mir nicht. Du hast doch auch die WO1B, oder? Hier mal der Debug-Log:

DEBUG:Thu Nov 19 23:21:06 2015 : Befehl: getElektrEnergieHz2
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 41
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 05
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 00
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 01
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 16
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: D1
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: 02
DEBUG:Thu Nov 19 23:21:06 2015 : >SEND: EF
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 06 (20.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: received 06
DEBUG:Thu Nov 19 23:21:06 2015 :  ERROR address 16D1 code 1
DEBUG:Thu Nov 19 23:21:06 2015 : >FRAMER: Command send
DEBUG:Thu Nov 19 23:21:06 2015 : >FRAMER: no preset result
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=2 41 (0.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: received 41 06
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=2 03 (0.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 16 (10.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 D1 (0.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 01 (0.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 01 (0.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: len=1 F3 (10.0 ms)
DEBUG:Thu Nov 19 23:21:06 2015 : <RECV: received 03 01 16 D1 01 01 F3
DEBUG:Thu Nov 19 23:21:06 2015 : >FRAMER: ERROR address 16D1 code 1
DEBUG:Thu Nov 19 23:21:06 2015 : Fehler recv, Abbruch
DEBUG:Thu Nov 19 23:21:06 2015 : Fehler beim ausfuehren von getElektrEnergieHz2
ERR:  ERROR address 16D1 code 1

Hat jemand eine Idee dazu?

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Fri, 20 Nov 2015 18:07:03 UT
* Message-ID: <76296327-85899863$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

N´abend
ja ich habe auch eine WO1B - verstehe also gerade nicht, warum das bei dir nicht funktioniert - schau mal, dass im am WE auch ein Log produziere, vielleicht kann man da was erkennen

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Fri, 20 Nov 2015 22:02:01 UT
* Message-ID: <76296327-85906381$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

So hier mal mein Debug-Log
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 41
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 05
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 00
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 01
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 16
[16822] Fri Nov 20 21:57:31 2015 : >SEND: D3
[16822] Fri Nov 20 21:57:31 2015 : >SEND: 02
[16822] Fri Nov 20 21:57:31 2015 : >SEND: F1
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 06 (30.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: received 06
[16822] Fri Nov 20 21:57:31 2015 : >FRAMER: addr was still active FE06
[16822] Fri Nov 20 21:57:31 2015 : >FRAMER: Command send
[16822] Fri Nov 20 21:57:31 2015 : >FRAMER: no preset result
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 41 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 07 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: received 41 07
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 01 (10.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 01 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 16 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 D3 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 02 (10.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 68 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 07 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: len=1 63 (0.0 ms)
[16822] Fri Nov 20 21:57:31 2015 : <RECV: received 01 01 16 D3 02 68 07 63
[16822] Fri Nov 20 21:57:31 2015 : Typ: int (in float: 1896.000000)
[16822] Fri Nov 20 21:57:31 2015 : (FLOAT) Exp:V/10 [B0:68 B1:07 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
[16822] Fri Nov 20 21:57:31 2015 : 189.600006

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Fri, 20 Nov 2015 22:34:45 UT
* Message-ID: <76296327-85907035$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hallo warmup,

Danke für die Mühe. Ich stelle fest, bei Dir führt die gleiche Anfrage zu einem ganz anderen Ergebnis als bei mir. Hmm, merkwürdig... Kannst Du mir mal verraten, was die Kurzabfrage bei Dir ausspuckt? Insbesondere die Software-Versionen wären interessant. Hier meine Daten:

01: 09 32 03
02: 04 10 00
03: 07 00 00
04: 00 00 49
05: 00 00 00
06: 00 00 00
07: 00 00 00
08: 01 0D 00
09: 00 00 00
10: 03 46 6E

Grüße,
Bernd

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Sat, 21 Nov 2015 11:20:50 UT
* Message-ID: <76296327-85917267$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Moin, die Werte schicke ich dir heute Abend - es gibt natürlich einen unterschied bei unserem Send!
Nach der Längenangabe sendest Du ein EF und ich ein F1

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Sat, 21 Nov 2015 11:35:33 UT
* Message-ID: <76296327-85917445$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Das ist die Prüfsumme, die wegen der anderen Adresse abweicht - ich habe 16D1 abgefragt, du 16D3. Ist aber unerheblich, da ich den gleichen Fehler bei allen 4 Adressen bekomme.

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Sat, 21 Nov 2015 13:38:44 UT
* Message-ID: <76296327-85918975$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Anbei meine Kurzabfrage
01: 0B 32 03
02: 02 10 00
03: 07 00 00
04: 00 00 49
05: 00 00 00
06: 00 00 00
07: 00 00 00
08: 01 0F 00
09: 00 00 00
10: 03 52 6E

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Sat, 21 Nov 2015 14:55:54 UT
* Message-ID: <76296327-85919861$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Vielen Dank :-) Ich vesuche mal bei Viessmann rauszubekommen, wie sich die beiden Software-Stände unterscheiden.

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: FhemAlan <userid-55005541@openv.wikispaces>
* Date: Tue, 24 Nov 2015 20:56:24 UT
* Message-ID: <76296327-86023149$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Was ist eine Kurzabfrage?

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: zelmani <userid-54884615@openv.wikispaces>
* Date: Tue, 24 Nov 2015 23:12:43 UT
* Message-ID: <76296327-86027493$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Die Kurzabfrage ist eine Funktion der Viotronic 200, die wesentliche Informationen zum Soft- und Hardwarestand sowie der Anlagenkonfiguration in 10 Zeilen zu je drei Byte darstellt. Detaillierte INfos gibt's im Service-Handbuch auf Seite 102: http://www.viessmann.com/vires/product_documents/5681896VSA00001_1.PDF

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: FhemAlan <userid-55005541@openv.wikispaces>
* Date: Fri, 27 Nov 2015 06:45:52 UT
* Message-ID: <76296327-86092887$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Ah, wunderbar. Das hat meine Heizung definitiv nicht. Dafür ist sie wohl zu alt.

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: maxmax20 <userid-51793276@openv.wikispaces>
* Date: Wed, 13 Jan 2016 21:08:30 UT
* Message-ID: <76296327-86828549$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hallo zusammen,

ich habe dies nun auch bei meiner Vitocal probiert, komme mit den Werten aber nicht zusammen.

Addr 16D3 Wert:9020.900391 und abgelesen auf der Energiebilanz der Steuerung direkt: 840
Addr 16D2 Wert:2041.500000 und abgelesen auf der Energiebilanz der Steuerung direkt: 58
Addr 16D1 Wert:2139.300049 und abgelesen auf der Energiebilanz der Steuerung direkt: 181
Addr 16D0 Wert:4086.000000 und abgelesen auf der Energiebilanz der Steuerung direkt: 13

sind immer len 2 und unit CO10.
Ich habe die gleiche Vito wie warmup, nur kein Splitgerät.

Vielleicht kann mir jemand weiterhelfen.
Vielen Dank.!
Mark

WikiSpaxe, 2017-10-26:
* Subject: Re: WO1B: aktuelle Leistungsaufnahme des Verdichters. 
* Wiki-Page: home
* From: maxmax20 <userid-51793276@openv.wikispaces>
* Date: Sun, 17 Jan 2016 18:26:16 UT
* Message-ID: <76296327-86884063$openv.wikispaces>
* In-Reply-To: <76296327-85767573$openv.wikispaces>

Hallo,

hat etwas gedauert bei mir, aber der Fehler lag bei mir.
Ich habe die vito.xml im falschen Verzeichnis bearbeitet.... :-(
Jetzt klappt alles.
Grüße,
Mark

MrIcemanLE, 2018-02-16:
> Addr 16D3 Wert:9020.900391 und abgelesen auf der Energiebilanz der Steuerung direkt: 840
Addr 16D2 Wert:2041.500000 und abgelesen auf der Energiebilanz der Steuerung direkt: 58
Addr 16D1 Wert:2139.300049 und abgelesen auf der Energiebilanz der Steuerung direkt: 181
Addr 16D0 Wert:4086.000000 und abgelesen auf der Energiebilanz der Steuerung direkt: 13

Hat jemand diese Adressen erfolgreich für die WO1C getestet? Das sind genau die Informationen nach denen ich schon länger suche. Leider bekomme ich keine Werte bzw. 0-Werte auf den Adressen. Oder weiß jemand in welchem Adressbereich ich schauen muss?