openv/openv #267: Vies-sion Fragen

WikiSpaxe, 2017-10-26:
* Subject: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Sat, 19 Dec 2015 11:02:07 UT
* Message-ID: <76742725-86551539$openv.wikispaces>

Fragen zur Applikation...

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Tue, 22 Dec 2015 14:00:43 UT
* Message-ID: <76742725-86578323$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Erst mal Danke für die Info!!

Hier noch ein paar Daten

a: Baujahr 08.2012

b: Anlagenausführung 1
   Ein Heizkreis ohne Mischer A1 (mit/ohne Warmwasserbereitung)

c: HK gibt es nur einen, auf Radiatoren

d: Sollte dann auf HK1 sein

Ich habe bei mir "CB" eingestellt, ist doch richtig, oder ?

Lese gerade alles aus, dauert noch etwas.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Tue, 22 Dec 2015 16:46:21 UT
* Message-ID: <76742725-86579643$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Habe gerade aufgrund deiner Fragen das Help File angepasst (Erste Schritte - Applikation einstellen, Fehler - Daten, Hauptbildschirm - Steuerung). Kann jetzt heruntergeladen werden.
a. Deine Anlage ist 2 Jahre neuer als meine. Ich vermute die kann um einiges mehr (von dem ich nur träumen kann!)
b. Ich gehe davon aus, dass die Anlage richtig installiert und eingestellt wurde. Mit verschiedenen Anlagen kenne ich mich zu wenig aus. Daher ist es wichtig, dass du bevor du mit Vies-sion etwas änderst, sicher bist, dass auch die richtige Adresse verwendet wird. Ich hab diesen Test so gemacht, dass ich (ungefährliche) Parameter geändert habe und in Vies-sion überprüft habe ob das richtig angezeigt wurde. 
c. ok
d. ok, damit werden bei dir andere Adressen verwendet, statt 3xxx sind es 2xxx Adressen. Wenn du unter 'Timer' nachsiehst, wird deine Einstellung ganz links (unter A1) zu sehen sein.

CB stimmt!

Als erstes kann ich dir folgende Tipps geben! Danach sehen wir weiter.
1. Stelle unter Einstellungen die 'Steuerung' ab  - damit nichts auf die Anlage geschrieben werden kann!
2. Stelle alle Parameter für die 'Heizung', im XML unter category 'H' von 37xx auf 27xx um
3. Stelle die 'Sensor' Parameter von 3xxx auf 2xxx, Namen und Beschreibung kannst du auf A1 anpassen (2900 gibt es schon, macht aber nichts)
4. Stelle die 'Bedien'-ungs Parameter von 3xxx auf 2xxx um
5. Stelle die Steuerungsadressen auf die neuen Adressen um

Volstrom und Abgastemp kannst du von Viess-data erfahren. Die kannst du zu den Sensoren addieren - hinten anhängen!

Dann solltest du deine Anlage wiedererkennen :-)
Achtung: Wenn du alle Werte mit dem Panel verglichen hast und sie in Ordnung sind, dann kannst du Versuche mit der Steuerung machen

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Tue, 22 Dec 2015 17:41:22 UT
* Message-ID: <76742725-86580133$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Hab mal gerade ein paar Backups gemacht.... :-)

1. Hatte ich noch nie angestellt, wollte erst die normale Funktion testen.

2. war ich schon bei, ist jetzt erledigt

3. erledigt

4. wie wo, in welchem file ?

5. wie wo, in welchem file ?

"Volstrom und Abgastemp kannst du von Viess-data erfahren. Die kannst du zu den Sensoren addieren - hinten anhängen!"

Das kommt erst später dran wenn alles sauber läuft.

Danke schon mal für deine Hilfe !!

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Tue, 22 Dec 2015 18:59:09 UT
* Message-ID: <76742725-86580899$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Parameter sind im Parameter File (XML) anzupassen. Das ist das einzige File das du ändern musst. Die Namen der Kategorien findest du im Help File (zB: Hauptbildschirm - Allgemein). Die Seite hat als Überschrift "XML Categorie Code: A". Im XML findest du unter category diesen code. Am Leichtesten kannst du das File mit Excel editieren, aber Achtung beim Abspeichern - als .xml speichern

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Tue, 22 Dec 2015 18:59:13 UT
* Message-ID: <76742725-86580901$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Parameter sind im Parameter File (XML) anzupassen. Das ist das einzige File das du ändern musst. Die Namen der Kategorien findest du im Help File (zB: Hauptbildschirm - Allgemein). Die Seite hat als Überschrift "XML Categorie Code: A". Im XML findest du unter category diesen code. Am Leichtesten kannst du das File mit Excel editieren, aber Achtung beim Abspeichern - als .xml speichern

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Tue, 22 Dec 2015 19:02:48 UT
* Message-ID: <76742725-86580943$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

5. ist unter 'Options-Steuerung'

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Tue, 22 Dec 2015 21:40:36 UT
* Message-ID: <76742725-86582441$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Ich habe das ganze jetzt mehrmals mit einem sauberen "Viess-ion_CB.xml" nach deiner Anleitung durchkonfiguriert,
mit ein wenig Erflog.

Unter Warmwasser wird alles korrekt angezeigt.
Unter Heizung wird alles korrekt angezeigt.
Unter Timer (A1) & WW1 wird alles korrekt angezeigt.
Unter Bedienung wird alles korrekt angezeigt.
Unter Sensor wird alles korrekt angezeigt, bis wieder mal RT ist, dort steht wieder 20 Grad.
Unter Steuerung steht auch wieder 20 Grad.

Ich weiß z.Zt. nicht mehr so wirklich weiter warum er Adresse 0898 nicht wirklich ausliest,
unter "Adressen" hab ich den Wert auch eingetragen.

<address addr="0898" access="R" shortcut="RT ist" description="Raum Temperatur Ist-Wert" category="Sensor">
			<len>2</len>
			<div>10</div>
			<check>1</check>
			<min>10</min>
			<max>35</max>
			<unit>°C</unit>

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Wed, 23 Dec 2015 00:26:44 UT
* Message-ID: <76742725-86583343$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Ich vermute bei dir liegt 'RT is't auf Adresse "0896", da deine FB auf A1 eingestellt ist. Wenn's das war, dann gratuliere, jetzt kannst du dir aus dem .csv file Graphiken erstellen und die Anlage an das Verhalten der Familie anpassen. Bei mir läuft die Anlage nur mehr auf Minimallast (16,5%), das reicht sowohl für WW als auch Heizung bis -5°C. Würd mich interessieren wie viel du an Ersparnissen raus holen kannst. Bei Radiatoren ist das Potential normalerweise geringer. Wenn du nach Ende der Heizperiode deine Erfahrung postest, würd ich mich freuen.
Viel Spaß

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Wed, 23 Dec 2015 12:11:08 UT
* Message-ID: <76742725-86585617$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Aus Viessdata
-------------------

<datapoint name="Rücklauftemp" ID="Rücklauftemp">
      <addr>0C20</addr>
      <len>2</len>
      <precision>0.01</precision>
      <unit>Grad</unit>
      <description>Rücklauftemp</description>
      <enable>1</enable>
      <speichern>1</speichern>
    </datapoint>

---------------------

Für Viss-ion
---------------------

<address addr="0C20" access="R" shortcut="Rücklauftemp" description="Rücklauftemp (Sensor)" category="Sensor">
			<len>2</len>
			<div>10</div>
			<check>0</check>
			<min>5</min>
			<max>70</max>
			<unit>°C</unit>
		</address>
----------------------------------------------

Aus Viessdata
-------------------

<datapoint name="Volumenstrom" ID="Volumenstrom">
      <addr>0C24</addr>
      <len>2</len>
      <precision>0.1</precision>
      <unit>l/h</unit>
      <description>Volumenstrom</description>
      <enable>1</enable>
      <speichern>1</speichern>
    </datapoint>

-------------------------------------

Für Viss-ion
---------------------

<address addr="0C24" access="R" shortcut="Volumenstrom" description="Volumenstrom (Sensor)" category="Sensor">
			<len>2</len>
			<div>10</div>
			<check>0</check>
			<min>0</min>
			<max>1370</max>
			<unit>l/h</unit>
		</address>

-----------------------------------------------

Aus Viessdata
-------------------

<datapoint name="AbgasTmp" ID="AD Wandler AGT-S">
      <addr>0808</addr>
      <len>2</len>
      <precision>0.1</precision>
      <unit>Grad</unit>
      <description>Temperatur Sensor 15 </description>
      <enable>1</enable>
      <speichern>1</speichern>
    </datapoint>

----------------------------------------------

Für Viss-ion
---------------------

<address addr="0C24" access="R" shortcut="AbgasTmp" description="AD Wandler AGT-S (Sensor)" category="Sensor">
			<len>2</len>
			<div>10</div>
			<check>0</check>
			<min>10</min>
			<max>80</max>
			<unit>°C</unit>
		</address>

------------------------------------

Aus Viessdata
-------------------

<datapoint name="Förderhöhe mbar" ID="?">
      <addr>7688</addr>
      <len>2</len>
      <precision>1</precision>
      <unit>mbar</unit>
      <description>?</description>
      <enable>1</enable>
      <speichern>0</speichern>
    </datapoint>

----------------------------------------

Für Viss-ion
---------------------

<address addr="7688" access="R" shortcut="Förderhöhe mbar" description="Förderhöhe mbar (Sensor)" category="Sensor">
			<len>2</len>
			<div>10</div>
			<check>0</check>
			<min>5</min>
			<max>300</max>
			<unit>mbar</unit>
		</address>

--------------------------------------

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Wed, 23 Dec 2015 12:12:23 UT
* Message-ID: <76742725-86585619$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Sorry, das sollte eigentlich oben stehen :-((
Hatte ich gestern Abend nicht mehr geschrieben, hab das XML auf "0896" umgestellt, jetzt wird die "Raum Ist Temp" auch richtig 

angezeigt :-) Danke dir!

Das mit dem anpassen der Heizung habe ich die letzten zwei Jahre schon gemacht, mit Viessdata, hat ganz gut funktioniert.
Aber mal schauen wenn evtl. mal bei uns der Winter kommt :-) und ich die Heizung wieder auf Heizbetrieb stelle was ich mit deinem Programm 

noch so rausholen kann. Seit Anfang März macht sie bei uns nur WW und den Rest erledigt unser Jydepejsen Kaminofen.

Ich denke mal viel einsparen können wir bei uns nicht mehr, 185 m2, voll beheizt, 
Verbrauch in den letzten Jahren mit WW bei drei / vier Personen ca. 11000 kWh.

Aber ein Frage habe ich noch wegen weiterer Parameter im XML. Ich hätte da gerne den Volumenstrom, Abgastemp, Rücklauftemp und 

Differenzdruck in der Übersicht "Sensor" mit drin, habe ich das so richtig zusammen kopiert, und nicht darüber lachen in bin kein 

Programmierer ;-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Wed, 23 Dec 2015 13:16:28 UT
* Message-ID: <76742725-86586029$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Schaut alles gut aus!!!, Wenn du nicht ganz sicher bist, ob ein Wert passt, dann kannst du ihn unter 'Ändern' einmal auslesen. Dazu muss er nicht einmal im XML eingetragen sein.  Du brauchst dazu nur die Adresse, die Länge, min, max und den Divisor anpassen. Auch die Einheit ist bei der Umrechnung wichtig, da kannst du 'text' od. 'hex' Darstellung sehen

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: cisco0815 <userid-55293209@openv.wikispaces>
* Date: Wed, 23 Dec 2015 14:33:37 UT
* Message-ID: <76742725-86586397$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Vielen vielen Dank für deine Hilfe & deine Zeit !!!
Hab es erst mit "Ändern" getestet & anschließend in das XML eingetragen, Funktioniert jetzt alles super.

Eine Frage habe ich noch.
Da meine Freundin Ihre Notebook wieder haben möchte, welches z.Zt. im Keller neben der Heizung steht........
Wie sieht es mit einem mini-pc aus, brauch das Betreibssystem (z.b. Windows 10) 32 bit oder 64 bit damit die Anwendung läuft.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: whwh <userid-55284815@openv.wikispaces>
* Date: Wed, 23 Dec 2015 15:17:18 UT
* Message-ID: <76742725-86586575$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

du kannst alles ab Windows 7 aufwärts verwenden 32 oder 64 bit ist egal. .net 4.5.2 ist notwendig. Der Resourcenverbrauch ist minimal, also kannst du auch einen schwachen Atom verwenden. 
Ich selbst hab ein Dell Venue 7 od. 8 Tablet im Einsatz. War um 99 EUR ein Schäppchen, wichtig ist bei Tablets das sie USB OTG haben und gleichzeitig geladen werden können wenn der Optolink dranhängt.

WikiSpaxe, 2017-10-26:
* Subject: Re: Vies-sion Fragen
* Wiki-Page: Vies-sion
* From: helmut.lehmeyer <userid-57587718@openv.wikispaces>
* Date: Mon, 16 Jan 2017 17:44:45 UT
* Message-ID: <76742725-92645178$openv.wikispaces>
* In-Reply-To: <76742725-86551539$openv.wikispaces>

Hallo,

erst mal Danke für die tollen Programme Viessdata und Viss-ion.

Ich habe eine Steuerung mit KW-Protokoll, '<device ID="2098" name="V200 KW2" protocol="KW" />'
Kann man dieses auch auswählen/konfigurieren?

Leider habe ich keine Möglichkeit selber zu komlilieren.

Bernardo1971, 2017-11-10:
Hallo zusammen,
ich habe ebenfalls eine V200KW2 Anlage, dann wohl mit der Kennung 2098. Hat jemand eine solche Anlage schon einmal mit Viess-ion zum sprechen bekommen ?
LOG kommt mit dieser Meldung zurück - mit VIESS DATA lässt sich die Anlage jedoch ansprechen.
10.11.2017 20:25:45 --,  ---- Viess-ion Monitor V1.2.0.5 started ---- 
10.11.2017 20:25:45 MONSTRT-I-LOGGING, Start recording data
10.11.2017 20:25:45 SYNCCOM-E-SYNC, timeout for SYN request, retrying in one minute
10.11.2017 20:25:45 MONSTOP-I-LOGGING, communication stopped and reset, port closed
Dadurch, dass die Anlage nicht angesprochen wird, weil scheinbar auch kein File mit der 98 als Kennung angelegt und die Applikation greift weiter auf Default als XML zu. 
Bevor ich hier weitere Stunden verbrenne frage ich einfach mal rum.
Gruss
B71