openv/openv #280: Ferienbetrieb

WikiSpaxe, 2017-10-26:
* Subject: Ferienbetrieb
* Wiki-Page: Adressen
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Mon, 11 Jan 2016 08:45:53 UT
* Message-ID: <76868983-86769733$openv.wikispaces>

Hallo, hat schon mal jemand den Ferienbetrieb bei seiner Anlage per vcontrold aktiviert? Für meine VScotHO1 sind ja dafür die Adressen 2535 (Ferienbetrieb aktiv/inaktiv), 2309 (Abreisetag) und 2310 (Rückreisetag) zuständig. Bisher habe ich aber noch nicht herausgefunden, ob die 2535 lediglich als Anzeige für den Status dient und der Ferienbetrieb dann nur über ein entsprechend gesetztes Datum in der 2309 und der 2310 aktiviert bzw. deaktiviert werden kann.

WikiSpaxe, 2017-10-26:
* Subject: Re: Ferienbetrieb
* Wiki-Page: Adressen
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Mon, 11 Jan 2016 15:12:20 UT
* Message-ID: <76868983-86774177$openv.wikispaces>
* In-Reply-To: <76868983-86769733$openv.wikispaces>

Hab gerade herausgefunden, dass der Rückreisetag die Adresse 2311 hat. 2310 ist falsch.

WikiSpaxe, 2017-10-26:
* Subject: Re: Ferienbetrieb
* Wiki-Page: Adressen
* From: andies1 <userid-58839765@openv.wikispaces>
* Date: Sat, 30 Sep 2017 22:17:29 UT
* Message-ID: <76868983-95257195$openv.wikispaces>
* In-Reply-To: <76868983-86769733$openv.wikispaces>

Bei mir liefert die 2535 nur beständig 18 zurück. Egal, ob das Ferienprogram aktiv ist oder nicht. Wenn es nicht aktiv ist, werden allerdings die Termine gelöscht.

squashball, 2020-02-22:
Würde mich hier gerne anhängen. War schon irgendwer mit irgendeiner Anlage erfolgreich, was den Ferienbetrieb/Ferienprogramm angeht?