openv/openv #307: VScotHO1 Adresse Befüllfunktion ?

WikiSpaxe, 2017-10-26:
* Subject: VScotHO1 Adresse Befüllfunktion ?
* Wiki-Page: Adressen
* From: heinzrudolf <userid-30171658@openv.wikispaces>
* Date: Sat,  9 Jul 2016 13:21:16 UT
* Message-ID: <79052371-90331793$openv.wikispaces>

Kennt jemand zufällig die Adresse für die Befüllfunktion ?
Ich würde diese gerne nutzen, um meinen SOlar-Boiler bei zu viel Sonne zu kühlen bzw. zur kleinen Heizungsunterstützung.
Es funktioniert von Hand, ist dann allerdings nur eine halbe Stunde aktiv, wäre schön wenn ich das irgendwie über die Schnittstelle schicken könnte

schmaller, 2022-06-04:
Hallo heinzrudolf,
ich bin [hier](http://vitoopen.tripod.com/files/VScotHO1.zip) fündig geworden.
Anschließend habe ich eine neue "UNIT" in der vcontrold.xml und die get- und set-Operation in der vito.xml erstellt.
Das funktionierte bei mir auf Anhieb.

**vcontrold.xml**
```
   <unit name="EntlueftBefuell">
      <abbrev>EBP</abbrev>
      <type>enum</type>
      <enum bytes="00" text="NA"/>
      <enum bytes="01" text="Entlueftung"/>
      <enum bytes="02" text="Befuellung"/>
    </unit>
```
**vito.xml**
```
    <command name="getEntlueftBefuell" protocmd="getaddr">
      <description>Ermittle den Status Entlueftung/Befuellung</description>
      <device ID="20CB">
        <addr>572F</addr>
        <len>1</len>
        <unit>EBP</unit>
      </device>
    </command>
    <command name="setEntlueftBefuell" protocmd="setaddr">
      <description>Setze Entlueftung oder Befuellung</description>
      <device ID="20CB">
        <addr>572F</addr>
        <len>1</len>
        <unit>EBP</unit>
      </device>
    </command>
```

schmaller, 2022-06-30:
Hallo, 
die Frage hier ist zwar schon 5 Jahre alt, aber das Thema dank Gasknappheit wieder brandaktuell:
**Heizungsunterstützung der Gasbrennwerttherme durch solarthermisch beheizten Warmwassertank**

Seitdem meine Kollektoren bereits im April und bis in den Oktober hinein den Wasserspeicher viel mehr erwärmen, als wir es benötigen,
frage ich mich, ob man die überschüssige Wärme nicht für die Fussbodenheizung nutzen kann.
Ja, ich weiss, es gibt Lösungen, bei denen die Vitosol mit der Vitodens verbunden sind, und da klappt das Zusammenspiel wahrscheinlich "out of the box".
Bei uns sind Therme und Solarkollektoren aber nur über den Wassertank verbunden. Die Gastherme heizt den Wassertank auf, wenn die solare Wärmelieferung dazu nicht ausreicht.

Meine gefundene Lösung ist die, bereits vor Jahren von heinzrudolf erwähnte Befüllungsfunktion der Vitodens 300W,
welche das Umschaltventil in Mittelstellung bringt und die Zirkulationspumpe einschaltet.
Dadurch wird das warme Wasser aus dem Heizkreislauf des WW-Speichers in den Heizkessel der Gastherme befördert
und kann anschließend von dort über den Mischer der FBH zugeführt werden.

Zur Steuerung des ganzen habe ich ein kleines Python-Skript geschrieben, welches auf einem Raspberry Pi Zero läuft,
der mit einem USB-OptoLink-Adapter an der Vitodens angeschlossen ist.
https://github.com/schmaller/vcontrold/tree/master/tools/vito_watcher
Vorraussetzung dafür ist ein funktionierender vcontrold und die Erweiterungen in der vito.xml und vcontrol.xml für die Befüllungsfunktion.
https://github.com/schmaller/vcontrold/tree/master/xml/300

Fragen, Anregungen und Kommentar gerne über das Github-Repo: https://github.com/schmaller/vcontrold