openv/openv #314: Rückgabewerte von getBrennerStatus?

WikiSpaxe, 2017-10-26:
* Subject: Rückgabewerte von getBrennerStatus?
* Wiki-Page: Adressen
* From: l3u <userid-55623545@openv.wikispaces>
* Date: Sat,  8 Oct 2016 15:11:24 UT
* Message-ID: <79676142-91251436$openv.wikispaces>

Hallo :-)

Ich habe die Regelung Vitotronic 200 H01B.

Mit dem Befehl "getBrennerStatus", der hier Adresse 55D3 ausliest, bekomme ich nicht nur "00" oder "01" zurück, sondern einige verschiedene Werte. Bisher hatte ich außerdem "1A", "1D", "24", "3A" und "64".

Bei jedem dieser Werte scheint der Brenner zu laufen, zumindest war auf dem Display das Flammensymbol zu sehen, und man hat gehört, dass das Gerät etwas tut.

Kann mir jemand sagen, was diese Rückgabewerte bedeuten? Bzw. gibt es irgendwo eine Datenpunktliste für diese Regelung?

Vielen Grüße

WikiSpaxe, 2017-10-26:
* Subject: Re: Rückgabewerte von getBrennerStatus?
* Wiki-Page: Adressen
* From: l3u <userid-55623545@openv.wikispaces>
* Date: Sun,  9 Oct 2016 09:45:48 UT
* Message-ID: <79676142-91257248$openv.wikispaces>
* In-Reply-To: <79676142-91251436$openv.wikispaces>

Scheinbar sind die Rückgabewerte Prozente! Folgende Werte hatte ich:

hex: dec
1A: 26
1C: 28
23: 35
24: 36
26: 38
34: 52
3A: 58
43: 67
46: 70
59: 89
64: 100

Mehr als 100 war es noch nicht, und je länger ich die Werte aufzeichne, desto mehr füllen sich die "Lücken".

Gibt es irgendwo was "offizielles" zu der Adresse 55D3? So wie es aussieht ist das nicht "Läuft der Brenner?", sondern "Mit wie viel Prozent Leistung läuft der Brenner?" ...

WikiSpaxe, 2017-10-26:
* Subject: Re: Rückgabewerte von getBrennerStatus?
* Wiki-Page: Adressen
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Sat,  3 Dec 2016 10:30:14 UT
* Message-ID: <79676142-92238046$openv.wikispaces>
* In-Reply-To: <79676142-91251436$openv.wikispaces>

Dass unter der Adresse 55D3 die Prozentangaben für die Brennerflamme angezeigt werden, könnte durchaus sein. Ich habe nach einigem Herumprobieren dann aber die Adresse 55DE für den Flammenstatus genommen:

<addr>55DE</addr>
<len>1</len>
<unit>FLS</unit>
<description>Flammenstatus</description>

In der vcontrold.xml habe ich mir noch eine entsprechende unit definiert:

<unit name='Flammenstatus'>
  <abbrev>FLS</abbrev>
  <type>enum</type>
  <enum bytes='00' text='aus'/>
  <enum bytes='0B' text='an'/>
  <enum text='?'/>
</unit>

Das liefert bei meiner VScotHO1 einen zuverlässigen Wert, ob der Brenner an oder aus ist.

vito-user, 2021-03-15:
Hallo,
die Diskussion ist zwar schon etwas her, trotzdem hänge ich mich mal hier dran.
Ich nutze die o.g. Adresse 55DE für den Flammenstatus. vclient..detail.. liefert hier:
```
Bytes:00 Text:aus
Bytes:0B Text:an
```
Wenn ich über vclient die Werte abrufe, erhalte ich auch den richtigen Status. In der command.txt habe ich vor dem Status auch noch "unit off" eingefügt, so dass 00 bzw. 0B geliefert wird.
Irgendwie bekomme ich den Wert nicht in der rrd-Datenbank gespeichert. Da es mit den anderen Statusmeldungen, die nur 00 und 01 liefern problemlos klappt, vermute ich, dass es vielleicht an der etwas anderen Hex-Ausgabe liegen könnte.
In der rrd-Datenbank habe ich definiert:
`DS:StatusFlamme:GAUGE:300:0:11`

Die Erklärung [hier](https://github.com/openv/openv/wiki/Datenauswertung-mit-RRDB) mit der $R4-Template-Variable (Return Wert wie zurückgeliefert) konnte ich nicht nachvollziehen und bringt Fehler, so dass ich nur die klassischen $1, $2, usw. Variablen verwende. Wie sollte man Statuswerte in der rrd speichern, die Buchstaben (also wie beim Hex-Code) enthalten?
Viele Grüße

vito-user, 2021-03-16:
Hier noch zur Ergänzung die aktuelle Konfiguration:
**commands.txt**

```
getTempAussen
getErtragTagSolar
unit off
getStatusFlamme
```

**vito.tmpl**
```
if  [ "x$E1" != xOK ]; then
        echo -n `date`  >>/tmp/vc-err.txt
        echo "es ist ein Fehler aufgetreten: $C1:$E1" >>/tmp/vc-err.txt
        exit 1;
fi
if  [ "x$E2" != xOK ]; then
        echo -n `date`  >>/tmp/vc-err.txt
        echo "es ist ein Fehler aufgetreten: $C2:$E2" >>/tmp/vc-err.txt
        exit 1;
fi
if  [ "x$E3" != xOK ]; then
        echo -n `date`  >>/tmp/vc-err.txt
        echo "es ist ein Fehler aufgetreten: $C3:$E3" >>/tmp/vc-err.txt
        exit 1;
fi
if  [ "x$E4" != xOK ]; then
        echo -n `date`  >>/tmp/vc-err.txt
        echo "es ist ein Fehler aufgetreten: $C4:$E4" >>/tmp/vc-err.txt
        exit 1;
fi

rrdtool update /home/pi/vito.rrd N:$1:$2:$R4
```
vclient -h localhost:3002 -f commands.txt
bringt folgendes Ergebnis:
```
getTempAussen:
4.200000 °C
getErtragTagSolar:
0.000000 
unit off:

getStatusFlamme:
0B 
```
vclient -h localhost:3002 -f /home/pi/commands.txt -t /home/pi/vito.tmpl -x /home/pi/vito.sh
bewirkt, dass in vito.sh folgendes unten steht:
`rrdtool update /home/pi/vito.rrd N:4.200000:0.000000:0B`
und dieser Fehler kommt:
`ERROR: /home/pi/vito.rrd: Function update_pdp_prep, case DST_GAUGE - Converted '0B' to 0,000000, but cannot convert 'B'`

Sobald ich aber in der Template-Datei vito.tmpl
`rrdtool update /home/pi/vito.rrd N:$1:$2:$R4`
eintrage, steht dann in der vito.sh nur noch
`rrdtool update /home/pi/vito.rrd N:4.200000:0.000000:0.000000`

speters, 2021-03-16:
Nach meinen Informationen sieht die Struktur des Datensatzes an Adresse 0x55d3 etwas komplexer aus:
Ab Adresse 0x55d3 kann ein 9-Byte Datensatz gelesen werden. Bit 42 (also in Byte Position 5 ab 0 gezählt) soll dann den Brennerstatus enthalten. Ggf. kannst du die entsprechende Offsetadresse direkt nehmen, also 0x55d3+5 = 0x55d8 als einzelnes Byte abfragen und mit entsprechender Bitmanipulation das interessierende Bit anschauen.

Mein [vogod](https://github.com/speters/vogod) (nicht `vcontrold`) sieht das so:
```json
    "GWG_Flamme": {
        "id": "GWG_Flamme",
        "address": "0x55D3",
        "fcread": 1,
        "fcwrite": 2,
        "block_length": 9,
        "byte_position": 5,
        "byte_length": 1,
        "bit_position": 42,
        "bit_length": 1,
        "value_list": "0=AUS;1=EIN",
        "codec": "valueListCodec"
    }
```
Ich verwende allerdings bei meinem Gerät folgende Adresse:
```json
{
    "id": "nvoPWRState_CFDM_state",
    "address": "0xA38F",
    "description": "nvoCFDMPweState",
    "fcread": 1,
    "fcwrite": 2,
    "block_length": 2,
    "byte_position": 1,
    "byte_length": 1,
    "value_list": "0=AUS;1=EIN",
    "codec": "valueListCodec",
    "value": 1
}
```

Mit der Info müsstest du die XML-Datei für den `vcontrold` eigentlich anpassen können.
Viel Erfolg!

vito-user, 2021-03-18:
Vielen Dank für den Hinweis zur Datensatz. Ich muss ehrlicherweise gestehen, dass ich das mit dem Auslesen und Interpretieren der Bits noch nicht ganz verstanden habe. Nichts desto trotz habe ich mit einer unkonventionellen Lösung auch das richtige Ergebnis bekommen.
Ich bin bei der Adresse 0x55DE geblieben und habe den Uniteintrag in der vcontrold.xml wie folgt angepasst:
```
		<unit name='Flammenstatus'>
			<abbrev>FLS</abbrev>
			<type>enum</type>
			<enum bytes='00' text='0'/>
			<enum bytes='0B' text='1'/>
			<enum text='?'/>
		</unit>
```
Entsprechend habe ich in der command.txt das **unit off** gestrichen. Ich gebe zu, es ist nicht ganz so elegant, aber damit kann ich leben, dass jetzt "1"  "an" bedeutet. Auf jeden Fall versteht die rrd-Datenbank jetzt die Ausgabe als richtigen Zahlenwert.
Viele Grüße