openv/openv #320: vMon Alternative

WikiSpaxe, 2017-10-26:
* Subject: vMon Alternative
* Wiki-Page: openHab Integration
* From: hmueller01 <userid-53715464@openv.wikispaces>
* Date: Thu, 27 Oct 2016 14:47:49 UT
* Message-ID: <79869884-91571190$openv.wikispaces>

Ich verwende vclient und ein Python template Skript, um Daten in die openhab REST Schnittstelle zu füttern ...
S.a. "Auswertung mit RRDB". In der rest_config.py muss man nur noch host, port, username und password definieren.

#!/usr/bin/env python2
  1. rest_client.py.tmpl
import requests
import base64
import rest_config

debug = True

class rest_client():
	def _init_(self, host, port, user, passwd):
		self.openhab_host = host
		self.openhab_port = port
		self.username = user
		self.password = passwd

	def basic_header(self):
		""" Header for OpenHAB REST request - standard """
		self.auth = base64.encodestring('%s:%s'
                       %(self.username, self.password)
                       ).replace('\n', '')
		return {
            "Authorization" : "Basic %s" %self.auth,
            "Content-type": "text/plain"}

	def put_status(self, key, value):
		""" Put a status update to OpenHAB key is item, value is state """
		url = 'http://%s:%s/rest/items/%s/state'%(self.openhab_host,
                                self.openhab_port, key)
 		if debug: print url
		if debug: print value
		req = requests.put(url, data=value, headers=self.basic_header())
		if req.status_code != requests.codes.ok:
			req.raise_for_status()

ohc = rest_client(rest_config.host, rest_config.port, rest_config.username, rest_config.password)
ohc.put_status("$C1", sr($1))
ohc.put_status("$C2", sr($2))
...

WikiSpaxe, 2017-10-26:
* Subject: Re: vMon Alternative
* Wiki-Page: openHab Integration
* From: hsellsch <userid-58957011@openv.wikispaces>
* Date: Mon,  9 Oct 2017 08:23:53 UT
* Message-ID: <79869884-95373071$openv.wikispaces>
* In-Reply-To: <79869884-91571190$openv.wikispaces>

Hallo zusammen,

ich habe am Wochenende meine Heizung mit openhab-2 verbunden und habe keinen der hier vorgeschlagenen Wege verwendet, da man damit  keine Werten der Heizung setzen kann. Das war mir aber wichtig.

Anbei mein Vorgehen, das hoffentlich so beschrieben ist, dass man es nachbauen kann.

Ich nutze openhab in einem docker-container auf einem qnap-server. Meine Vitodens 222-W (ID=20CB) wird mit einem Raspberry Pi angesteuert.
Ich brauche für die Openhab-Integration weder VMon noch irgend ein kompliziertes Python-Script.
Der vclient reicht vollkommen aus

Das Lesen und Setzen von Werten in openhab erfolgt bei mir über die exec-binding, mit der man shell-scripte aufrufen kann. Dazu habe ich den vclient 
auf meinem openhab server installiert. vcontrold läuft auf meinem Raspi.

1. Werte der Heizung setzen (z.B: setBetriebPartyM1)
	
	a) Openhab-Thing (name z.B.: Heizung_Set_Party_Betrieb) anlegen mit dem Kommando "/openhab/conf/openv/HZSetBetriebPartyM1.bash %2$s"
	b) auf dem openhab server unter /openhab/conf/openv das shell-script HZSetBetriebPartyM1.bash ablegen
	c) Der Inhalt des scripts:

		#!/bin/bash
		cd /openhab/conf/openv
		/sbin/vclient -h 192.168.2.21:3002 -c "setBetriebPartyM1 $1"

	Erläuterung: Wenn man einen Wert (z.B: 1) in den input-channel des Things "Heizung_Set_Party_Betrieb" schreibt, dann wird das Kommando ausgeführt,
	also
		/openhab/conf/openv/HZSetBetriebPartyM1.bash 1
	Das script macht daraus 
		/sbin/vclient -h 192.168.2.21:3002 -c "setBetriebPartyM1 1"
	Es wird also das Kommando "setBetriebPartyM1 1" an vcontrold auf meinem Raspi (192.168.2.21) geschickt und ausgeführt.
	
2. Werte der Heizung auslesen
	a) Openhab-Thing (name z.B.: Heizung_Get_Items) anlegen mit dem Kommando "/openhab/conf/openv/HZGetItems.bash"
	b) auf dem openhab server unter /openhab/conf/openv das shell-script HZGetItems.bash ablegen
	c) Der Inhalt des scripts:

		#!/bin/bash
		cd /openhab/conf/openv

		/sbin/vclient -h 192.168.2.21:3002 -c getTempA -t ./GetItem.tmpl -x ./GetItem.sh
		/sbin/vclient -h 192.168.2.21:3002 -c getTempWWist -t ./GetItem.tmpl -x ./GetItem.sh
		/sbin/vclient -h 192.168.2.21:3002 -c getTempWWsoll -t ./GetItem.tmpl -x ./GetItem.sh
		...
	
	d) auf dem openhab server unter /openhab/conf/openv das Template GetItem.tmpl ablegen
	e) Der Inhalt des Templates:
	
		/sbin/curl --header "Content-Type: text/plain" --request PUT --data "$1" http://192.168.2.35:8080/rest/items/$C1/state

	Erläuterung: das openhab Things "Heizung_Get_Items" ist so konfiguriert, dass das Kommando alle 60 Sekunden ausgeführt wird,
	also
		/openhab/conf/openv/HZGetItems.bash
	Das script führt jedes get-Kommando (z.B. getTempA) einzelnd aus, und schreibt das Ergebnis via Template in die Datei GetItem.sh 
	GetItem.sh sieht dann z.B.: so aus 
		/sbin/curl --header "Content-Type: text/plain" --request PUT --data "10.50000" http://192.168.2.35:8080/rest/items/getTempA/state
	Anschliessend wird über die -x Option dieses Skript ausgeführt. Das curl-Kommando schreibt via Rest-Schnittstelle den Wert 10.5000 in das
	openhab Item getTempA, welches natürlich in openhab angelegt sein sollte.
	
vclient in Zusammenspiel mit der exec-binding von openhab reichen also vollkommen aus, um alle Werte der Heizung zu setzen und auszulesen

luismauriz1, 2018-03-02:
Hallo, ich hätte dazu eine Frage, sie schreiben, dass sie vclient auf ihrem openhab Server installiert haben. Wie genau muss ich mir das vorstellen? Wie funktioniert das? 
LG