openv/openv #322: Viessmann Vitodens 100

WikiSpaxe, 2017-10-26:
* Subject: Viessmann Vitodens 100
* Wiki-Page: home
* From: Cyanoazimin <userid-57325940@openv.wikispaces>
* Date: Sun,  6 Nov 2016 10:28:40 UT
* Message-ID: <79965248-91735460$openv.wikispaces>

Hallo liebe openv Mitglieder, Danke für die Erläuterungen und Pionierarbeit! 
I have successfully assembled/soldered the raspberry version (had to double-check orientation of LED and PNP-transistor). My raspberry listens but does not speak/send to the Vitodens 100 (see below). I can turn on/off the LED via GPIO 14 (successfully monitored via video camera). Thus:
1.	Do you know whether my Vitodens 100 from 2000/01 is capable of providing data at all? Apparently this Wiki is about Vitodens 200 and 300 mainly. (Optolink interface is mentioned in the Vitodens 100 service manual, though)
2.	Is there any further way to check the sending of data to the Vitodens 100?
3.	Do I have to adjust settings like speed / baud rate to connect to the Vitodens 100?
Thanks a lot for any help

vctrld>getDevType
DEBUG:Sat Nov  5 20:31:08 2016 : Befehl: getDevType
DEBUG:Sat Nov  5 20:31:08 2016 : >SEND: 04
DEBUG:Sat Nov  5 20:31:08 2016 : >FRAMER: no preset result
DEBUG:Sat Nov  5 20:31:08 2016 : Warte auf 05
DEBUG:Sat Nov  5 20:31:09 2016 : <RECV: 05 (820.0 ms)
DEBUG:Sat Nov  5 20:31:09 2016 : >SEND: 01
DEBUG:Sat Nov  5 20:31:09 2016 : >SEND: CB
DEBUG:Sat Nov  5 20:31:09 2016 : >SEND: F8
DEBUG:Sat Nov  5 20:31:09 2016 : >SEND: 03
DEBUG:Sat Nov  5 20:31:09 2016 : >FRAMER: no preset result
DEBUG:Sat Nov  5 20:31:10 2016 : <RECV: len=1 05 (1420.0 ms)
DEBUG:Sat Nov  5 20:31:12 2016 : <RECV: len=1 05 (1490.0 ms)
DEBUG:Sat Nov  5 20:31:13 2016 : <RECV: len=1 05 (1040.0 ms)
DEBUG:Sat Nov  5 20:31:13 2016 : <RECV: received 05 05 05
DEBUG:Sat Nov  5 20:31:13 2016 : 05 05 05 -> UNKNOWN
DEBUG:Sat Nov  5 20:31:13 2016 : UNKNOWN
UNKNOWN

WikiSpaxe, 2017-10-26:
* Subject: Re: Viessmann Vitodens 100
* Wiki-Page: home
* From: Cyanoazimin <userid-57325940@openv.wikispaces>
* Date: Tue, 27 Dec 2016 21:26:58 UT
* Message-ID: <79965248-92511492$openv.wikispaces>
* In-Reply-To: <79965248-91735460$openv.wikispaces>

Thank you for your feedback! In the meantime I have found out that I own a “GWG_VBES” and for such a GWG my dialog above apparently does make sense, though: 
Telegrammdetails:
01 Telegramm Start Byte 
CB Type „PHYSICAL READ“
8F Adresse „System Identifikation“
03 Anzahl der angefragten Bytes 
04 Telegramm Ende Byte
Probably it is rather a hardware flaw in my sender. I will double-check.

WikiSpaxe, 2017-10-26:
* Subject: Re: Viessmann Vitodens 100
* Wiki-Page: home
* From: kunksmuhme <userid-57546650@openv.wikispaces>
* Date: Tue,  3 Jan 2017 14:41:38 UT
* Message-ID: <79965248-92534914$openv.wikispaces>
* In-Reply-To: <79965248-91735460$openv.wikispaces>

If you are using a Raspberry Pi 3: Disable bluetooth and reenable the hardware-based serial port, i recently updated the instructions: http://openv.wikispaces.com/Bauanleitung+RaspberryPi
I had exactly the same problem, emitter LED blinking but no response to commands except the periodical "0x05". Problem was the new default "mini uart" of the Raspberry Pi 3 which is somewhat limited, e.g. it does not allow parities (I think they are required to communicate with Vitotronics). Now it works perfectly.

WikiSpaxe, 2017-10-26:
* Subject: Re: Viessmann Vitodens 100
* Wiki-Page: home
* From: Cyanoazimin <userid-57325940@openv.wikispaces>
* Date: Sun,  8 Jan 2017 14:37:37 UT
* Message-ID: <79965248-92572932$openv.wikispaces>
* In-Reply-To: <79965248-91735460$openv.wikispaces>

It works perfectly now! I have replaced the SF487 LED and the 2N3906 transistor. Turning on the LED (e.g. by powering up/down GPIO14 directly) you are able to see the “dark red” end of the IR, i.e. in a dark room you should be able to see the LED glowing faintly red (and shining fairly bright white when watched via a video camera).
I am using a Raspberry Pi 2

Arty68, 2019-09-16:
Hi I am also very interrested by your work.
I have a vitodens 100 wb1 from 2002. 
Will it work also ?
What can I control with the optolink interface.
Thanks for your answer.

Arty68, 2019-12-27:
Hi I also have a Vitodens 100 wb1. I succesfully established an communication with the esp8266 interface ( with esp-link) 
When I send : 01 CB 8F 03 04 
I receive the answer :
73 99 C4 81 01 CB 0E 2B FD 3E EE 10 05
For the moment I have tried with python script that communicate the telnet.
Perhaps I am missing something.