openv/openv #323: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?

WikiSpaxe, 2017-10-26:
* Subject: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: dripphone <userid-50743620@openv.wikispaces>
* Date: Thu, 10 Nov 2016 13:13:01 UT
* Message-ID: <80023378-91848728$openv.wikispaces>

Hallo,

mich würde interessieren, ob jemand weiss, was das für ein Kabel ist, welches bei dem sog. "Vitoconnect 100, Typ OPTO1 (WLAN)" dabei ist.

Es hat einen USB-Stecker auf der einen Seite und einen V-Optolink auf der anderen. Das Teil scheint es gerade vermehrt in der Bucht zu geben, wahrscheinlich weil Viessmann es mittlerweile kostenlos zum Kessel mitgibt(?)

Frage: Könnte man das Kabel alleine als eine Art USB/serial wie das original-Optolink-Kabel nutzen?

Oder wäre es sogar möglich, irgendwie an das Gerät an sich (welches ja wohl mit Viessmann-Servern kommuniziert) zu connecten und Werte abzufragen/setzen etc?

Ich habe nämlich derzeit wegen Server-Umbau Probleme, das alte RS232-Kabel wieder zum laufen zu bringen bzw. der USB/RS232 Adapter tut irgendwie nicht so wie er soll...

Vielen Dank und Gruss,
dripphone

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: SoenkePeters <1384994+speters@users.noreply.github.com>
* Date: Mon, 14 Nov 2016 13:01:32 UT
* Message-ID: <80023378-91904264$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

Ich habe in [Vitoconnect 100](http://openv.wikispaces.com/Vitoconnect+100) mal etwas zusammengetragen, was ich herausfinden konnte.
Ich vermute, dass das beiliegende Kabel einen "standard" seriell-USB Wandler drin hat, da weder Viessmann, noch QSC, noch Q-Loud eine eigene USB-Vendor ID registriert haben und eine eigene Implementierung ausser zur Marktabschottung kaum Sinn machte.

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: dripphone <userid-50743620@openv.wikispaces>
* Date: Mon, 14 Nov 2016 13:31:04 UT
* Message-ID: <80023378-91904470$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

WOW, das nenne ich Service! Vielen Dank für die Profi-Infos soweit. 
Ich bin kurz davor, mir einen zu ergattern. Werde dann darüber berichten und bin gerne bereit, mit Nachforschungen auszuhelfen, soweit ich kann.

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Mon, 21 Nov 2016 13:42:44 UT
* Message-ID: <80023378-92013806$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

Servus, also ich hab das Kabel vom Vitoconnect an meinem Raspberry wunderbar am Laufen...

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: dripphone <userid-50743620@openv.wikispaces>
* Date: Tue,  6 Dec 2016 18:05:16 UT
* Message-ID: <80023378-92283784$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

Update:
unter CentOS7 meldet sich bei mir folgendes USB Gerät und die Kommunikation über vcontrold scheint (für eine Weile) zu funktionieren:

Bus 001 Device 002: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light

Nach einigen Stunden scheint aber wiederholt (der Treiber oder das Gerät?) abzustürzen, was zu folgenden Fehlermeldungen im log führt:

kernel: cp210x ttyUSB0: cp210x_open - Unable to enable UART

Nach einem USB reset Befehl per software scheints wieder für eine Weile zu funktionieren.
Leonaldo, läuft es bei dir auf dem Raspberry 7x24h ohne Probleme? Vielleicht ist es ein CentOS/kernel problem?

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Wed,  7 Dec 2016 09:49:23 UT
* Message-ID: <80023378-92295968$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

Servus,  
ja bei mir soweit alles im Lot... lese seit Wochen alle 5 Minuten mehrere Werte aus und schreibe Sie in eine mysql-DB. 
Habe keine Probleme feststellen können. Bei mir erkennt er folgendes Device.:

ls -ltr /dev/serial
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

Grüße
Leo

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: dripphone <userid-50743620@openv.wikispaces>
* Date: Thu,  8 Dec 2016 12:56:37 UT
* Message-ID: <80023378-92315252$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

welches OS/version genau hast du denn auf dem Raspberry am laufen wenn ich fragen darf, und welche kernel-version?
Sprichst du auch direkt per vcontrold/vclient oder hast du was dazwischen, wie ser2net o.ä.

WikiSpaxe, 2017-10-26:
* Subject: Re: Frage zu "Vitoconnect 100  Typ OPTO1" /WLAN: USB-Kabel nutzbar?
* Wiki-Page: Die Optolink Schnittstelle
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Thu,  8 Dec 2016 13:54:30 UT
* Message-ID: <80023378-92316034$openv.wikispaces>
* In-Reply-To: <80023378-91848728$openv.wikispaces>

Servus, klar darfst du fragen:
Linux raspberrypi 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux

Spreche direkt mit vcontrold/vclient. Da ist nix dazwischengeschaltet.

kuebx, 2017-11-06:
Hallo,
habe das Vitoconnect Kabel (OPTO1), möchte es seriell an einer Wago Steuerung verwenden.
Kann man den USB Stecker abzwicken und einen RS232 Stecker anbringen ? hat jeman die Kabelbelegung von dem Opto1 Kabel bzw. welche Adern auf welche Stecker Pins müssen ?

Jojo-A, 2017-11-06:
@kuebx auf keinen Fall. Lies dir mal bitte ein bißchen was zu dem Innenleben des Kabels durch. Das ist ja nicht einfach nur eine dumme Drahtverbindung ;) .
Da ist ein Wandler drin von USB auf UART-seriell. USB ist USB. UART-seriell ist UART-seriell. Und eine "echte" RS232-Schnittstelle hat auch noch ganz andere Spannungspegel.
Was ist denn das für eine WAGO-Steuerung?

Grüße

kuebx, 2017-11-07:
@Jojo-A ah ok, hatte naiv gedacht das ist im prinzip nen serielles Kabel halt mit seriell auf usb wandler dran. den wandler abzwicken und ma hat die serielle Schnittstelle. Ich hab nen PFC200 750-8202 mit serieller on board sowie die extra serielle Karte 750-652, die eig. voll parametrierbar ist.
Die einzigen Infos zum Innenleben von dem Kabel hatte ich hier gefunden [(https://github.com/openv/openv/wiki/Vitoconnect-100)] da steht eben nur seriell.
Ist dann der OptoPi Adapter von der wiki seite das richtige ? gibt es irgendwo die Möglichkeit sich einen zu kaufen ?(ich hab im Forum gesucht aber nichts dazu gefunden) Heizung ist eine Vitodens 200, da würde das Kabel gut passen. 

Danke schonmal

Jojo-A, 2017-11-08:
@kuebx Also ich möchte nicht belehrend rüberkommen, aber gerade zu diesem Kabel gibt es hier im Wiki ja nun wirklich _viel_ Infomaterial. Zum Beispiel hier: https://github.com/openv/openv/wiki/Bauanleitung-USB

Eigentlich ist das alles gar nicht sooo schwierig. Du mußt nur wissen, was deine SPS für eine Schnittstelle hat. Und damit meine ich jetzt nicht "nen Comport", sondern wie die Hardware-Spezifikationen sind. Ich VERMUTE einfach mal, daß deine SPS eine "echte" RS232-Schnittstelle hat. Also eine serielle Schnittstelle, mit den für RS232 vorgesehenen Spannungspegeln. Dann brauchst du im Prinzip nur das hier: https://github.com/openv/openv/wiki/Bauanleitung oder https://github.com/openv/openv/wiki/Adapter-Eigenbau und fertig ist die Laube. Steht alles im Wiki ;) .

msglueck, 2023-02-26:
Kleines Update zum original [Viessmann Anschlussleitung USB-Optolink - 7856059 Vitoconnect 100 OPTO1/ OPTO2] bzw. die Variante die es aktuell bei eBay zu erstehen gibt.

Die Variante die ich erstanden habe meldet sich in Linux als  `QinHeng Electronics HL-340 USB-Serial adapter`

`$ lsusb
$ lsusb
Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
$ usb-devices
T:  Bus=01 Lev=02 Prnt=02 Port=02 Cnt=02 Dev#=  5 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=ff(vend.) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=1a86 ProdID=7523 Rev=02.62
S:  Product=USB2.0-Serial
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=98mA
I:  If#=0x0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=01 Prot=02 Driver=ch341`

In der `vitocontrold.xml` funktioniert das Teil mit folgender Config wunderbar:

`
<serial>
    <tty>/dev/ttyUSB0</tty>
</serial>
`