openv/openv #324: Mehr Infos?

WikiSpaxe, 2017-10-26:
* Subject: Mehr Infos?
* Wiki-Page: Vitoconnect 100
* From: SoenkePeters <1384994+speters@users.noreply.github.com>
* Date: Mon, 14 Nov 2016 12:56:54 UT
* Message-ID: <80055046-91904244$openv.wikispaces>

Ich habe mal oben zusammengetragen, was ich so dazu auffinden konnte. Leider steht mir kein VConnect 100 zur Verfügung, um das genau unter die Lupe zu nehmen. Interessant wäre die Frage, welcher Seriell-USB-Konverter im beiligenden Kabel verbaut ist (da der Preis einer VConnect 100 mit Kabel günstiger als ein originales Optolink-Kabel zu sein scheint - für Bastelunwillige und gewerbliche Anwender interessant).
Ausserdem wären hochauflösende Platinenfotos/-scans toll, um die Bauteile genau identifizieren zu können (OPTO1 und auch OT1 Typ).
Wer sich traut, kann ja mal schauen, ob die Pins der UART-Schnittstelle des AR9311 zugänglich sind und ob man darüber ein U-Boot- oder Linux-Consolen-Prompt erreichen kann. Auch auf den JTAG-Port könnte man mal lunsen. Erforschenswert wäre ausserdem natürlich der Inhalt des SPI-Flash-ROMs um mal im Firmware-Image herumgucken zu können.

WikiSpaxe, 2017-10-26:
* Subject: Re: Mehr Infos?
* Wiki-Page: Vitoconnect 100
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Wed, 14 Dec 2016 15:55:56 UT
* Message-ID: <80055046-92399798$openv.wikispaces>
* In-Reply-To: <80055046-91904244$openv.wikispaces>

Servus, 
guter Ansatz... also bei mir ist das USB Kabel folgendes:

ls -ltr /dev/serial
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

Grüße
Leo

WikiSpaxe, 2017-10-26:
* Subject: Re: Mehr Infos?
* Wiki-Page: Vitoconnect 100
* From: Leonaldo <userid-57158498@openv.wikispaces>
* Date: Wed, 14 Dec 2016 15:55:58 UT
* Message-ID: <80055046-92399800$openv.wikispaces>
* In-Reply-To: <80055046-91904244$openv.wikispaces>

Servus, 
guter Ansatz... also bei mir ist das USB Kabel folgendes:

ls -ltr /dev/serial
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

Grüße
Leo

WikiSpaxe, 2017-10-26:
* Subject: Re: Mehr Infos?
* Wiki-Page: Vitoconnect 100
* From: SoenkePeters <1384994+speters@users.noreply.github.com>
* Date: Thu, 26 Jan 2017 12:54:21 UT
* Message-ID: <80055046-92752153$openv.wikispaces>
* In-Reply-To: <80055046-91904244$openv.wikispaces>

Die haben mal "heimlich" etwas nachgebessert, aber Downloadlink und Dateiname gleich gelassen:
```diff

--- oss_vc100_v1.2.2-20161205.txt	2017-01-26 12:51:01.433953495 +0100
+++ oss_vc100_v1.2.2-20161216.txt	2017-01-26 12:51:12.378229976 +0100
@:@ -1,4 +1,4 @:@
-Archive:  /home/soenke/Downloads/oss_vc100_v1.2.2.zip
+Archive:  /home/soenke/Downloads/oss_vc100_v1.2.2_001.zip
  Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
 --------  ------  ------- ---- ---------- ----- --------  ----
 22716802  Defl:N 21926917   4% 2016-12-05 11:31 fcaabee5  binutils-2.24.tar.bz2
@:@ -7,8 +7,9 @:@
  1232516  Defl:N  1229712   0% 2016-12-05 11:31 b7254ecb  libmicrohttpd-0.9.42.tar.gz
   896005  Defl:N   894327   0% 2016-12-05 11:32 44e9b016  libnl-3.2.26.tar.gz
   521496  Defl:N   521125   0% 2016-12-05 11:32 f8e7009f  libusb-1.0.19.tar.bz2
-77835264  Defl:N 77847778   0% 2016-12-05 11:34 42ee400d  linux-3.14.49.tar.xz
+78489116  Defl:X 78501126   0% 2016-12-16 11:59 3d447ef2  linux-3.14.49.tar.xz
  5266102  Defl:N  5244462   0% 2016-12-05 11:32 99991a6a  openssl-1.0.2g.tar.gz
+  250830  Defl:N   248299   1% 2016-12-05 14:28 250d4e5c  u-boot_v1.1.4-patched_v1.2.2.tar.bz2
  2384404  Defl:N  2384769   0% 2016-12-05 11:36 6c1ad292  uClibc-ng-1.0.5.tar.xz
 --------          -------  ---                            -------
-113577859         112741111   1%                            9 files
+114482541         113642758   1%                            10 files

```

WikiSpaxe, 2017-10-26:
* Subject: Re: Mehr Infos?
* Wiki-Page: Vitoconnect 100
* From: JanM5 <userid-57680973@openv.wikispaces>
* Date: Sat, 11 Feb 2017 23:37:43 UT
* Message-ID: <80055046-92969877$openv.wikispaces>
* In-Reply-To: <80055046-91904244$openv.wikispaces>

Das Optolink-Kabel, das bei der Vitoconnect dabei liegt, funktioniert einwandfrei mit einem Raspberry/vcontrold und sogar mit Vitosoft 300 unter Windows, sofern man den richtigen Treiber installiert hat. 

Leider (leider) gibt's bei Viessmann 5 Jahre Garantie, wenn man die Anlage mittels Vitoconnect aufschaltet (und aufgeschaltet lässt), weshalb man vcontrold da leider nicht nutzen kann... Ein Mitschnitt des Netzwerkverkehrs der Vitoconnect ist relativ unergiebig gewesen; das Modul verbindet sich via SSL auf den Viessmann-Server und behält diese Verbindung aufrecht. Man kann also noch nicht einmal auf diesem Wege Daten der Anlage auslesen, da ja SSL-verschlüsselt... Warum hat Viessmann nicht zwei Optolink-Schnittstellen verbaut? :D

Eine Art Webinterface stellt die Vitoconnect schon zur Verfügung, allerdings nur, wenn diese noch nicht fertig konfiguriert ist und man mit dem VITOCONNECT-Wlan verbunden ist:
(Auszug aus der Serviceanleitung zur Vitoconnect):
"You can find the latest version of this docu-ment on your Vitoconnect 100 by following these steps:
Press the side button on the Vitoconnect 100 for 5 seconds until the lower LED starts flash-ing yellow in order to start the Vitoconnect in Access Point Mode. Then connect to the wifi “VITOCONNECT-<xxxx>” with your smartphone. You find the password for the wifi on the bar code at “WPA2 login”. (...) On your android device please open a browser and access http://vitoconnect.gateway or enter http://192.168.1.1 into the browser edit line. Then follow the link “Open Source Licenses”"
... habe ich noch nicht ausprobiert, einfach mal hier zur Dokumentation...

h0nIg, 2022-04-15:
@speters die Seite ist etwas undurchsichtig. Hast du damals die Firmware auslesen können bzw. den JTAG port gefunden? Bilder von der Vitoconnect wären das kleinste Problem.