openv/openv #325: Einfache Steuerung durch Viess-Data?

WikiSpaxe, 2017-10-26:
* Subject: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: rwhaus <userid-56597510@openv.wikispaces>
* Date: Mon, 21 Nov 2016 22:25:13 UT
* Message-ID: <80121858-92023004$openv.wikispaces>

Hallo,

ich habe bei mir eine Vitodens 200-W (19 kW, Bj 2014), die ich seit geraumer Zeit mit Viess-Data 2.0 auslese und manuell "steuere" (naja, eher ab und an ein paar Werte neu setzen zwecks Optimierung). 
Leider ist die Heizung auch ein "Kuhschwanzheizung", d. h. alle 4 Minuten ein Brennerstart, außer es sind draußen Minusgrade. Die Heizung ist einfach zu groß für einen neues, gut gedämmtes EFH bzw. kann nicht weit genug runter modulieren. Auch wenn die Heizleistung auf 5% reduziert wird und die HK-Pumpe bei >80% läuft, bekommt die Heizung die beim Brennerstart erzeugte Wärme nicht schnell genug weg. Man kann da jetzt mit Pufferspeicher, hydraulischer Weiche etc. rummachen, aber das ist alles noch ineffizienter.
Leider ist die Software der Heizung von Viessmann so grottenschlecht geschrieben, dass sie auf so eine Situation (niedriger Heizbedarf) nicht reagieren kann. Die Heizung taktet munter weiter und versucht Wärme zu produzieren, wo doch garkeine Wärme benötigt wird. Daher hatte ich folgende Idee: immer wenn in einer kurzen Zeitspanne eine bestimmte Anzahl von Brennerstarts erfolgt ist (z. B. 4 Brennerstarts in 17 min), ist das ein Zeichen, dass derzeit keine Wärme benötigt wird und die Heizung eigentlich inaktiv sein kann (z. B. RT_Soll auf von 21 °C auf 14°C setzen). Die RT_Soll wird also für z. B. 45 min auf 14°C gesetzt und die Heizung ist ruhig. Nach 45 min wird RT_Soll wieder auf den Ursprungswert gesetzt (hier 21 °C) und das Spiel beginnt von vorne. Entweder wird jetzt genügend Wärme benötigt und die Heizung läuft durch oder die Heizung probiert erfolglos 4 Brennerstarts und wird dann wieder mittels  RT_Soll = 14 °C aus dem Rennen genommen. Das Spiel beginnt von vorne.
Andere Viessmann-Kessel wie der Vitodens 300 sollen so eine Logik ansatzweise implementiert haben, aber das soll auch nicht richtig funktionieren.

OK, suchte ich nach einem Weg, die o. g. Logik umzusetzen. Da ich nicht so gut programmieren kann, dass ich Viess-Data erweitern kann (die Source sind ja verfügbar), dachte ich an eine Automatisierung mit einem externen Skript, das Viess-Data steuert. Hierfür kam https://www.autoitscript.com in Frage. Damit kann ich Tastendrücke und Mausklicks simulieren und so Viess-Data steuern. Nun, die Logik hatte ich schnell programmiert, aber die Ansteuerung von Viess-Data klappt leider nicht richtig. Werte, die das Skript in die Eingabemaske (Tab "Übersicht") eingibt, werden von Viess-Data nicht übernommen. Der "nochmal alle DP lesen"-Button lässt sich nicht per Skript klicken. Sonst hätte alles geklappt! Kurz vorm Ziel war also leider Schluß mit meiner Skript-Lösung :( 

Nun meine Fragen an kehrenberger, den Autor von Viess-Data:
1.) Wäre es möglich, den Tab "Übersicht" etwas robuster zu programmieren in dem Sinne, dass eingetragene Werte auch wirklich übernommen werden? Evtl. mit einem zusätzlichen OK-Button? Ich weiß, in der Anleitung steht, dass selbst bei manueller Bedienung dieser Tab etwa eigenartig reagieren kann.
2.) (sehr egoistische Frage) Wäre es möglich, die o.g. Logik zu implementieren? Ich könnte Code aus meinem Skript beisteuern. Die Foren sind voll mit Beschwerden über die ewige Takterei bei den Vitodensen. Das wäre also eine sinnvolle Erweiterung ;) Einstellparameter der Steuerung wären die abgesenkte RT_Soll, die Zeitspanne, bis wieder auf nominelle RT_Soll zurückgegangen werden soll und der Trigger (x Brennerstarts in y Minuten)

Ich wollte schon beides selber probieren und habe dafür die Source runtergeladen und in Visual Studio studiert, aber so richtig traue ich mich da nicht ran. Oder ist es doch ganz einfach? 

So, das war ein Feature Request. Mach was draus ;)

Grüße
rw

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Mon, 21 Nov 2016 23:28:13 UT
* Message-ID: <80121858-92023814$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Hallo, rw!
Freut mich, dass Viessdata 2.0 bei Dir grundsätzlich funktioniert. Leider bin ich seid der Programmierung auch etwas aus der Übung, daher zunächst einige Tips, die vielleicht schon weiterhelfen.
Eingaben in Übersicht können mit "Tab" oder "Return" abgeschlossen werden, so dass ein Übertragen der neuen Parameter an die Therme stattfindet. Allerdings müsste ein Script beachten, dass die Parameter nicht geschrieben werden, wenn gerade eine Abfrage aktiv ist.... Bei einer Veränderung eine Parameters werden nach dem Schreiben einmal alle Parameter gelesen.
Reduzierung der Heizleistung bringt nichts, in der Startphase wird immer mit ca. 60% der Maximalleistung gefahren, das kann man nicht reduzieren. Braucht die Therme um sicher zu starten.
Wenn die Thermensoftware dumm ist, muss der Anwender schlauer sein. Ein Weg ist die Heizzeiten (4 Stück pro Tag) auszunutzen, um nur 4 mal am Tag kurz zu heizen, z.B. 6-7, 9-10, 14-15, 19-20 Uhr. Dazwischen kühlt der Vorlauf so stark ab, dass die während des Startvorgangs erzeugte Wärme ohne Probleme abtransportiert wird, ohne die Abschalttemperatur zu erreichen.
Alternativ empfehle ich das Heizen bei niedrigem Wärmebedarf von der Außentemperatur abhängig zu machen, dann kann man sich die Skripterei, wie auch das Verändern der Heizzeiten, sparen. Ich habe mir eine AT ausbaldowert, bei der das Heizen tatsächlich nicht mehr notwendig ist und die Startabbrüche wahrscheinlich werden. Dazu dienen die beiden Sparschaltungen, die sowohl bei kurzfristigen Wärmeperioden, wie unter Tags im frühen Frühling oder im späten Herbst, als auch bei langen Wärmeperioden, wie im Sommer zuverlässig funktioniert. Für erstere Fälle nehme ich Sparschaltung Pumpe, die auf normalen AT basiert, und für letzteren die Sparschaltung Pumpe und Brenner, die auf der gedämpften AT basiert.
Aber auch eine Umstellung des Abschickens einer Parameteränderung mit einer Taste sollte nicht allzu schwer zu implementieren sein....vielleicht findet sich ja jemand, der dazu kommt....

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: rwhaus <userid-56597510@openv.wikispaces>
* Date: Sat, 26 Nov 2016 18:05:05 UT
* Message-ID: <80121858-92102492$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Hallo kehrenberger,

mein Skript sendet die Tastendrücke "Tab" oder "Return", aber aus mir noch unbekannten Gründen werden diese von Viess-data nicht immer angeommen. Ich muss da weiter testen. Ich habe eine Abfrage eingebaut, dass dies immer nur dann passiert, wenn der Fenstertitel den String "nicht verbunden" enthält.

Die 60% in der Startphase sind mir bekannt, aber die Reduktion bringt zumindest bei mir was, da das Gerät - wenn mal ein Brennerstart erfolgreich war - nicht mit > 5% Heizleistung losballert, sondern nur schwach heizt und somit für eine lange Zeit an ist.

4 Heizzeiten hatte ich auch schon eingestellt, aber dann taktet die dumme Viessmann-Software in 3 von 4 Heizzeiten. Zumal halte ich das für eine ziemlich uneleganten Workaround.

Das mit den niedrigen AT und der Sparschaltung könnte eine Möglichkeit sein, aber in meinem Fall nicht wirklich anwendbar. Der AT-Fühler hängt an der schattigen Nordwand, aber ich habe große Fensterflächen gen Süden. Schon leichter Sonnenschein wärmt die Bude gewollt auf (solarer Wärmegewinn), bei praller Wintersonne wird es schnell richtig schön warm. In diesem Fall schaltet der Raumtemperaturfühler den FB-Heitkreis aus. Der Wärmebedarf sinkt und jetzt sollte dann eben auch die Heizung das merken. Tut sie aber nicht, weil dumm :/ Der solare Wärmegewinn soll sich eben auch positiv auf den Gasverbrauch auswirken, aber das ist derzeit mit der Viessmann-Software nicht möglich. Ich werde trotzdem mal schauen, ob ich irgendwie die Sparschaltungen einsetzen kann. Danke für den Tipp!

Langfristig muss ich mich wohl wirklich in den Quelltest von Viess-data einarbeiten. In Form_Main stehen die ganzen Funktionen und Prozeduren drin. Wo soll ich denn mit lesen anfangen? Gibt es ein, zwei Funktionen, die besonders wichtig sein, weil sie den Programmablauf steuern? Ein paar Sätze zum grundsätzlichen Aufbau des Quelltextes würden mir schon helfen. Evtl. kannst Du dafür nochmal in den hinteren Hirnregionen kramen ;)

Danke und Grüße
rw

PS: sagte ich schon, dass die Viessmann-Software unheimlich dumm ist und das Viessmann-Forum jede Frage abschmettert und keinerlei Hilfe ist.

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sun, 27 Nov 2016 01:51:51 UT
* Message-ID: <80121858-92108272$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Hallo, rw,

Ja, nach einem Brennerstart wird die Leistungsbegrenzung aktiv, aber leider auch nicht auf 5%, die minimale Leistung liegt bei den Vitodensen zwischen ca. 16 un 30%, einfach mal im Graphen die rote Brennerleistung verfolgen.

Die Sparschaltungen machen natürlich nur bei AT-geführten Anlagen Sinn. Wenn Sie Raumtemperatur-geführt steuern, was meine Therme nicht tut, können evtl. andere Mechanismen aktiviert werden, die die Therme temporär bei geringem Wärmebedarf abschalten....

Wenn Sie tatsächlich in den Quelltext von Viessdata eingreifen wollen, nur soviel: Ich habe den Quelltext ebenfalls nur übernommen und erweitert. Es gibt einen Programmteil Screen_Update der die ausgelesenen Daten an jeder Adresse interpretiert, diese in Übersicht einträgt und Aktionen für bestimmte Adressen ausführt. Die für Sie interessante Adresse ist 0x088A, die die Anzahl von Brennerstarts enthält. Da die Abfrageschleife einmal pro Minute durchlaufen wird, könnte man beim Auslesen von 0x088A testen, ob seit dem letzten Auslesen die Anzahl von Brennerstarts zugenommen hat. Tut man dies in jeder Schleife und bildet einen laufenden Quotienten von Brennerstarts und Schleifendurchläufen, würde dieser Wert einen Anhaltspunkt von Brennerstarts pro Zeiteinheit geben. Mittelt man zum Beispiel laufend über 15 Minuten, und gibt es mehr als 2 Brennerstarts in diesen 15 Minuten, könnte man nach der Abfrageschleife den aktuellen Betriebsmodus aus 0x2323 speichern und auf nur WW Bereitung umschalten. Ist diese Anti-Kuhschwanz-Funktion aktiv, wird ca. 60 Schleifendurchläufe gewartet (1 Stunde) und danach der ursprüngliche Betriebsmodus wieder in 0x2323 geschrieben....und die Messung der Starts pro Zeiteinheit beginnt von Neuem...
So könnte man sicherstellen, dass das Starten für 1 Stunde verhindert wird und das Heizmedium bis zum nächsten Start abkühlt. 
Nur eine Idee....sicherlich gibt es viele andere Ansätze.  Problem bei dieser Lösung ist, wenn das Viessdata vor dem Zurückschalten beendet wird, bleibt der Heizbetrieb deaktiviert, bis er an der Therme wieder aktiviert wird.

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sun, 27 Nov 2016 01:59:32 UT
* Message-ID: <80121858-92108318$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Alternativ kann man natürlich die Raumtagsolltemperatur herabsetzen, wie von Ihnen vorgeschlagen, aber dann würde die Funktion nur während des normalen Heizbetriebs, nicht aber während der Nachtabsenkung, funktionieren....

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: rwhaus <userid-56597510@openv.wikispaces>
* Date: Sat, 25 Feb 2017 10:34:34 UT
* Message-ID: <80121858-93146435$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Hallo,

vor ein paar Wochen habe ich dann doch direkt im Quelltext von viessdata meine gewünschte Funktion eingebaut. Hat eine Weile gebraucht, bis ich den Quelltest kapiert habe und wusste, wo ich meinen Code hinzufügen musste. An dieser Stelle vielen Dank an kehrenberger, den Quelltest zu veröffentlichen!

Meine Idee war folgende (siehe auch der erste Post):
"immer wenn in einer kurzen Zeitspanne eine bestimmte Anzahl von Brennerstarts erfolgt ist (z. B. 4 Brennerstarts in 17 min), ist das ein Zeichen, dass derzeit keine Wärme benötigt wird und die Heizung eigentlich inaktiv sein kann (z. B. RT_Soll auf von 21 °C auf 14°C setzen). Die RT_Soll wird also für z. B. 45 min auf 14°C gesetzt und die Heizung ist ruhig. Nach 45 min wird RT_Soll wieder auf den Ursprungswert gesetzt (hier 21 °C) und das Spiel beginnt von vorne. Entweder wird jetzt genügend Wärme benötigt und die Heizung läuft durch oder die Heizung probiert erfolglos 4 Brennerstarts und wird dann wieder mittels RT_Soll = 14 °C aus dem Rennen genommen. Das Spiel beginnt von vorne."

Diese Idee habe ich im Programm umgesetzt und es läuft ohne Probleme. Hier ein Screenshot:
http://www.directupload.net/file/d/4643/oei5q3bc_png.htm

Erläuterung:
Um 4 Uhr morgens kommt die Heizung aus der Nachtabsenkung (nicht im Bild) und läuft dann bei 13% Brennermodulation (rote Kurve), dann um 6 Uhr WW-Bereitung. Danach geht die Heizung wieder zurück auf 13% Brennermodulation, bis sie wohl gegen 8:20 Uhr meint, es wäre genug geheizt und den Brenner ausschaltet. Bisher ist alles vollkommen OK. Dann versucht das Gerät ab ca. 9 Uhr mehrfach zu starten, was erfolglos ist. Gegen 9:45 Uhr wird der Trigger ausgelöst: 4 Brennerstarts in < 17 Minuten. Das wird von nun von meiner Programmlogik erkannt und RT_Soll für 45 min von 21 °C auf 14°C gesetzt, ebenso wird die Leistung der Heizkreispumpe verringert. Folge: das Gerät heizt nicht mehr.
Gegen 10:30 Uhr sind die 45 min "Zwangspause" um,  RT_Soll wird wieder auf 21 °C gesetzt und die Heizung meint, es gäbe wieder etwas zu tun. Jedoch folgen wieder 4 Brennerstarts in < 17 min, was eine erneute "Zwangspause" auslöst. So geht es dann die nächsten Stunden weiter.
Irgendwann gegen 20 Uhr (nicht im Bild) konnte der Brenner dann wieder so starten, dass er wieder bei 13% Brennermodulation durchlaufen konnte.

ToDo-Liste:
- Den neuen Tab "Steuerung" schöner machen und die Debug-Infos weglassen
- Trigger-Einstellungen vom Nutzer einstellbar (x Brennerstarts in y min, Werte für RT_Soll und min/max Leistung Heizkreispumpe)
- Unterbinden, dass die neue Funktion auch bei aktivierter Frostschutzfunktion aktiv wird.
- Programm und Quelltest veröffentlichen (gerne mit Hilfe von kehrenberger)

Viele Grüße
rw

WikiSpaxe, 2017-10-26:
* Subject: Re: Einfache Steuerung durch Viess-Data?
* Wiki-Page: Viess-Data 2.0
* From: kehrenberger <userid-45842210@openv.wikispaces>
* Date: Sat, 25 Feb 2017 19:34:43 UT
* Message-ID: <80121858-93148705$openv.wikispaces>
* In-Reply-To: <80121858-92023004$openv.wikispaces>

Super, sieht gut aus! Die Änderungen würde ich mir gern mal ansehen... Ich habe in der Zwischenzeit einige Bugs korrigiert, vielleicht können wir ja wieder zu einer Version kommen.

joxsi, 2021-11-20:
Hallo zusammen,
der oben beschriebene viess-data code zur Reduzierung der Viessmann Vitodens 200 Takterei würde mich auch interessieren. Viess-data 2.06 läuft bei mir einwandfrei. Ist zwar schon ne weile her wo das hier gepostet wurde aber über ne Rückmeldung würde ich mich sehr freuen.
Schöne Grüße J.S.