openv/openv #341: Ein Raspberry mit zwei Heizungssteuerungen?

WikiSpaxe, 2017-10-26:
* Subject: Ein Raspberry mit zwei Heizungssteuerungen?
* Wiki-Page: home
* From: gero112233 <userid-57755283@openv.wikispaces>
* Date: Thu, 16 Feb 2017 10:18:43 UT
* Message-ID: <80709643-93034903$openv.wikispaces>

Hallo,

ich habe in der Gastherme Vitodens 300 die Steuerung VScotHO1 verbaut und über LON verbunden zusätzlich in der Vitotronic 200-H die V050HK3W.

Muss/kann ich vcontrold zweimal starten, um mit zwei Steuerungen zu sprechen? Ich bin gerade dabei, ein zweites Optolink-Kable zu kaufen, aber kann vcontrold mit zwei Kabel umgehen?

Wo finde ich die besten XML-Dateien für VScotHO1 und V050HK3W?

Ich bitte um Unterstützung!

Beste Grüße,
Gero

WikiSpaxe, 2017-10-26:
* Subject: Re: Ein Raspberry mit zwei Heizungssteuerungen?
* Wiki-Page: home
* From: gero112233 <userid-57755283@openv.wikispaces>
* Date: Tue, 21 Feb 2017 08:26:24 UT
* Message-ID: <80709643-93087379$openv.wikispaces>
* In-Reply-To: <80709643-93034903$openv.wikispaces>

Aktueller Stand:
Hilfe gibt es leider momentan keine.
vcontrold kann mit –x das xml-file mitgegeben werden.
Default ist nicht /etc/vcontrold sondern /etc/vcontrold/vcontrold.xml
vcontrold kann somit theoretisch mit verschiedenen xml-Files gestartet werden.

Leider verhindert vcontrold mit Hilfe eines Pid-Files, dass es zweimal gestartet werden kann.
Denn Sinn dieser Sperre kenne ich nicht.

WikiSpaxe, 2017-10-26:
* Subject: Re: Ein Raspberry mit zwei Heizungssteuerungen?
* Wiki-Page: home
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Thu, 23 Feb 2017 13:21:28 UT
* Message-ID: <80709643-93119059$openv.wikispaces>
* In-Reply-To: <80709643-93034903$openv.wikispaces>

Ich habe mal meine vito.xml und vcontrold.xml für meine VScotHO1 hochgeladen. Allerdings verwende ich etwas andere commmands und zum Teil auch auch andere unit-Filter für die Adressen. Zudem verfügt unsere Anlage nur über einen einzigen Heizkreis, so dass ich alle diesbezüglich unnötigen Adressen gelöscht habe. Vielleicht hilft's ja trotzdem.

WikiSpaxe, 2017-10-26:
* Subject: Re: Ein Raspberry mit zwei Heizungssteuerungen?
* Wiki-Page: home
* From: gero112233 <userid-57755283@openv.wikispaces>
* Date: Sat,  4 Mar 2017 16:46:54 UT
* Message-ID: <80709643-93238653$openv.wikispaces>
* In-Reply-To: <80709643-93034903$openv.wikispaces>

Vielen Dank thopeter, aber wo finde ich die Datei?
Bei mir sind getTempKist, getTempKtp und getTempVListM1 identisch, da kann etwas nicht stimmen. Mir fehlt auch der Status der internen Pumpe.
Auf jeden Fall vielen Dank für die Hilfe!

WikiSpaxe, 2017-10-26:
* Subject: Re: Ein Raspberry mit zwei Heizungssteuerungen?
* Wiki-Page: home
* From: thopeter <userid-52868466@openv.wikispaces>
* Date: Wed,  5 Apr 2017 07:34:40 UT
* Message-ID: <80709643-93658583$openv.wikispaces>
* In-Reply-To: <80709643-93034903$openv.wikispaces>

Klick mal oben links auf "Pages and Files" und gib dann im Suchfeld "vito.xml" bzw. "vcontrold.xml" ein. Dann müssten Dir meine Dateien (user thopeter) angezeigt werden.

passuff, 2018-11-20:
> V050HK3W

Hey,

habe eine ähnliche Konfig (Vitodens 200 / HO1 und V050HK3W, ebenfalls über LON verbunden. Bist du mittlerweile weitergekommen? Ich finde leider weder Adressen noch ein Vito File für die V050HK3W. HO1 habe ich seit längerem erfolgreich laufen.  Vito.xml mit A1/M2 findest du im Anhang.

[vito.zip](https://github.com/openv/openv/files/2598625/vito.zip)

passuff, 2018-11-20:
Falls noch jamand Adressen für die HK3W sucht:

`Absenkzeit_gelerntA1M1~0x253D;
Absenkzeit_gelerntM2~0x353D;
Absenkzeit_gelerntM3~0x453D;
ADC_IstTemperaturwert_17A~0x080A;
ADC_IstTemperaturwert_17B~0x080C;
ADC_Isttemperaturwert_2_M1~0x0802;
ADC_IstTemperaturwert_ATS~0x0800;
ADC_Isttemperaturwert_KTS~0x0802;
ADC_IstTemperaturwert_STS1~0x0804;
ADC_IstTemperaturwert_STS2~0x0806;
Aufheizgradient_gelerntA1M1~0x253B;
Aufheizgradient_gelerntM2~0x353B;
Aufheizgradient_gelerntM3~0x453B;
Ausgang52_Mischerposition_M1~0x254C;
Ausgang52_Mischerposition_M2~0x354C;
Ausgang52_Mischerposition_M3~0x454C;
AutomaticFlag_WWA1M1~0x2300;
AutomaticFlag_WWM2~0x3300;
AutomaticFlag_WWM3~0x4300;
AutomaticFlag_ZPA1M1~0x2300;
AutomaticFlag_ZPM2~0x3300;
AutomaticFlag_ZPM3~0x4300;
Bedien_WW_Solltemperatur~0x6300;
BedienBetriebsartA1M1~0x2301;
BedienBetriebsartM2~0x3301;
BedienBetriebsartM3~0x4301;
BedienKT_Solltemperatur~0x5300;
BedienNeigungA1M1~0x2305;
BedienNeigungM2~0x3305;
BedienNeigungM3~0x4305;
BedienNiveauA1M1~0x2304;
BedienNiveauM2~0x3304;
BedienNiveauM3~0x4304;
BedienparameterA1M1FunktionReset~0xA051;
BedienparameterM2FunktionReset~0xA051;
BedienparameterM3FunktionReset~0xA051;
BedienPartybetriebA1M1~0x2303;
BedienPartybetriebM2~0x3303;
BedienPartybetriebM3~0x4303;
BedienRTSoll_PartyA1M1~0x2308;
BedienRTSoll_PartyM2~0x3308;
BedienRTSoll_PartyM3~0x4308;
BedienRTSolltemperaturA1M1~0x2306;
BedienRTSolltemperaturM2~0x3306;
BedienRTSolltemperaturM3~0x4306;
BedienRTSolltemperaturReduziertA1M1~0x2307;
BedienRTSolltemperaturReduziertM2~0x3307;
BedienRTSolltemperaturReduziertM3~0x4307;
BedienSparbetriebA1M1~0x2302;
BedienSparbetriebM2~0x3302;
BedienSparbetriebM3~0x4302;
Bedienteil_SW_Index~0x7330;
BedienteilBetriebsartA1M1~0x2320;
BedienteilBetriebsartM2~0x3320;
BedienteilBetriebsartM3~0x4320;
D_Ausgang20_M2~0x3906;
D_Ausgang20_M3~0x4906;
D2_Fehler_d6~0x757A;
D2_Fehler_d7~0x757A;
D2_Fehler_d8~0x757A;
DigitalAusgang_20~0x0843;
DigitalAusgang_21~0x0845;
DigitalAusgang_28~0x0846;
DigitalAusgang_50~0x0847;
DigitalAusgang_52Auf~0x084D;
DigitalAusgang_52Zu~0x084C;
DigitalAusgang_LED_Gruen~0x084A;
DigitalAusgang_LED_Rot~0x084B;
[... 432 lines cut ...]
TiefpassTemperaturwert_17B~0x081A;
TiefpassTemperaturwert_ATS~0x5525;
TiefpassTemperaturwert_RTS_M2~0x0898;
TiefpassTemperaturwert_RTS_M3~0x089A;
TiefpassTemperaturwert_STS1~0x0812;
TiefpassTemperaturwert_STS2~0x0814;
TProfilSchrittAktuellM1~0x2533;
Uhrzeit~0x088E;
UhrzeitJAHR~0x088E;
UhrzeitJH~0x088E;
UhrzeitMONAT~0x088E;
valmRPCClearErrorHistoryTable~0xA029;
vlogRPCClearMemberList~0xA009;
VorlauftemperaturM1~0x2900;
VorlauftemperaturM2~0x3900;
VorlauftemperaturM3~0x4900;
VT_SollKesselA1M1~0x2546;
VT_SollKesselM2~0x3546;
VT_SollKesselM3~0x4546;
VT_SolltemperaturA1M1~0x2544;
VT_SolltemperaturM2~0x3544;
VT_SolltemperaturM3~0x4544;
WW_SolltemperaturAktuell~0x6500;
WW_SolltemperaturKessel~0x6502;
WW_Status_NR2~0x650A;`