openv/openv #347: WO1C Adressen falsch

WikiSpaxe, 2017-10-26:
* Subject: WO1C Adressen falsch
* Wiki-Page: Vorschlag__Vito-Masterdateien
* From: Vitocal_dolor <userid-57904811@openv.wikispaces>
* Date: Mon,  6 Mar 2017 21:56:39 UT
* Message-ID: <80850643-93263399$openv.wikispaces>

Hallo,
auch mich hat es gepackt die Vitocal über die optische Schnittstelle auszulesen. Danke der Beschreibung hat dies auch super funktioniert.
Ich habe eine Vitocal 200 AWCI mit WO1C. Dafür habe ich die Beispieldateien von der_vitti genommen (ID:204D).
Viele Parameter konnte ich somit auch auslesen. Schreiben habe ich noch nicht probiert.
Allerdings haben ein paar commands Fehler zurückgebracht. z.B.
DEBUG:Mon Mar  6 20:03:18 2017 : Befehl: Außentemperatur
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 41
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 05
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 00
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 01
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 01
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: C1
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: 02
DEBUG:Mon Mar  6 20:03:18 2017 : >SEND: CA
DEBUG:Mon Mar  6 20:03:18 2017 : <RECV: len=1 05 (200.0 ms)
DEBUG:Mon Mar  6 20:03:18 2017 : <RECV: received 05
DEBUG:Mon Mar  6 20:03:18 2017 : >FRAMER: Error 05
DEBUG:Mon Mar  6 20:03:18 2017 : Fehler send, Abbruch
DEBUG:Mon Mar  6 20:03:18 2017 : Fehler beim ausfuehren von Außentemperatur
ERR: >FRAMER: Error 05
Fehler send, Abbruch
Fehler beim ausfuehren von Außentemperatur
Wenn ich die Adresse auf 0101 ändere funktioniert es. Das gleiche bei der Speichertemperatur oben. Obwohl die Adresse von der_vitti in der Vitogate200 Beschreibung stehen. Bisher hat die Änderung von C in 0 geholfen. Hat einer eine Erklärung warum dies so ist. @:der_vitti hast du mal die Außentemperatur mit deiner xml ausgelesen? 
Gruß, vitocal_dolor

Morg42, 2017-11-03:
Ich habe (eigentlich...?) die gleiche Steuerung (Vitotronic 200 an einer Vitocal 200-S, V200WO1C, id 204D), aber bei mir hat das in Masse nicht funktioniert. Ich habe daher eine Übersicht gemacht, die hänge ich hier mal an. [V200WO1C.xlsx](https://github.com/openv/openv/files/1442621/V200WO1C.xlsx)

Wenn noch jemandem dazu weitere Infos vorliegen, gerne aktualisieren.

ynghill, 2018-11-26:
Hallo
Ich spreche kein Deutsch, also habe ich Google als Übersetzer benutzt.
Ich protokolliere meine Daten über ein optolink rpi in eine mysql-Datenbank, von der ich schnell Daten zu meiner Android-App sowie zu PC zur Temperaturkurvenanalyse abrufen kann.
Ich habe eine Wärmepumpe VT 200 WO1C mit Systemtyp 204b.
Mit Adresse 0x088E kann ich Datum und Uhrzeit nicht abrufen.
Eine Antwort kann wie folgt aussehen: 018201a600210045.
Die Antwort ändert sich mit der nächsten Frage, also passiert etwas.
Hat jemand eine andere Adresse für Datum und Uhrzeit?
Sucht auch nach der Adresse des Umlaufpumpens und des Kompressordrucks.
Wenn Sie die Softwareversion lesen können, ist es sinnvoll, diese in Verbindung mit den Adressen verschiedener Geräte einzugeben.
Mit freundlichen Grüßen
Yngve