openv/openv #349: Lua script

WikiSpaxe, 2017-10-26:
* Subject: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: the.neon <userid-38712692@openv.wikispaces>
* Date: Thu, 16 Mar 2017 22:58:04 UT
* Message-ID: <80929187-93396241$openv.wikispaces>

Hi ,

hab ne Frage ist es auch möglich das Lua Script zu bekommen ?

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: Superbrudi <userid-13940967@openv.wikispaces>
* Date: Wed, 29 Mar 2017 16:49:27 UT
* Message-ID: <80929187-93571395$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Klar,

hier das Lua Script:

-- LUA script ESP8266
-- VITOTRON 200

staconfig={}
staconfig.ssid="your ssid here"
staconfig.pwd="your security here"

wifi.setmode(wifi.STATION)
wifi.sta.config(staconfig)
wifi.sta.connect()

countdown = 6					-- start to put values into sql after 60 secs
tmr.alarm(1, 10000, 1, function()	-- every 10 secs
  if countdown == 0 then
    countdown = 30				-- 30 * 10 secs = 5 mins
    if (wifi.sta.getip() ~= nil) and (t1 ~= "--") then
      getstr ="http://192.168.1.110/heating.php?t1=" .. t1 .. "&t2=" .. t2 .. "&t3=" .. t3 .. "&t4=" .. t4 
	  getstr = getstr .. "&t5=" .. t5 .. "&t6=" .. t6 .. "&t7=" .. t7
	  getstr = getstr .. "&s1=" .. s1 .. "&h1=" .. h1 .. "&sp=" .. sp .. "&pa=" .. pa .. "&br=" .. br
      http.get(getstr, nil, function(code, data)
        if (code < 0) then print("HTTP request failed")
                      else countdown = tonumber(string.match(data,'%d+')) end
		if (countdown < 1) or (countdown > 30) then countdown = 30 end
	    end)
    end
  end
  countdown = countdown - 1
end)

srv=net.createServer(net.TCP,30)
srv:listen(80,function(conn)
    conn:on("receive", function(client,request)
        buf='HTTP/1.1 200 OK\r\nContent-type: text/html\r\nConnection: close\r\n\r\n'
        buf=buf .. "ESP ID: " .. node.chipid() .."</br>"
        buf=buf .. "heap  : " .. node.heap() .."</br></br>"
        buf=buf .. "cnt   : " .. countdown .."</br></br>"
        buf=buf .. "Aussen1   : " .. t1 .. "</br>"
        buf=buf .. "Kessel1   : " .. t2 .. "</br>"
        buf=buf .. "Speicher1 : " .. t3 .. "</br>"
        buf=buf .. "Ruecklauf1: " .. t4 .. "</br>"
        buf=buf .. "Vorlauf1  : " .. t5 .. "</br>"
        buf=buf .. "Kessel2   : " .. t6 .. "</br>"
        buf=buf .. "Speicher2 : " .. t7 .. "</br>"
        buf=buf .. "Starts    : " .. s1 .. "</br>"
        buf=buf .. "Hours     : " .. h1 .. "</br>"
        buf=buf .. "Spar      : " .. sp .. "</br>"
        buf=buf .. "Party     : " .. pa .. "</br>"
        buf=buf .. "Brenner   : " .. br .. "</br>"
        buf=buf .. "</body></html>"
        conn:send(buf)
		conn:on("sent",function(conn) conn:close() end)
    end)
end)

t1 = 0
t2 = 0
t3 = 0
t4 = 0
t5 = 0
t6 = 0
t7 = 0
s1 = 0
h1 = 0
sp = 0
pa = 0
br = 0

uart.setup(0, 4800, 8, uart.PARITY_EVEN, uart.STOPBITS_2, 0)
stat = 0
cnt = 1
uart.on("data", 4,
  function(data)
  if stat == 0 then if data == string.char(0x05, 0x05, 0x05, 0x05)		-- wait for vitotronic bytes
    then 
	  stat = 1
	  uart.write(0,0x01,0xF7)								-- read command
	  if cnt ==  1 then uart.write(0,0x08, 0x00) end
	  if cnt ==  2 then uart.write(0,0x08, 0x04) end
	  if cnt ==  3 then uart.write(0,0x08, 0x0A) end
	  if cnt ==  4 then uart.write(0,0x08, 0x10) end
	  if cnt ==  5 then uart.write(0,0x08, 0x8A) end
	  if cnt ==  6 then uart.write(0,0x08, 0xA7) end
	  if cnt ==  7 then uart.write(0,0x23, 0x02) end
	  if cnt ==  8 then uart.write(0,0x55, 0x1E) end
	  uart.write(0, 0x04)									-- read 4 bytes
	  end
  elseif stat == 1
    then  
	  if cnt ==  1 then t1  = data:byte(1) + data:byte(2) * 256 
						t2  = data:byte(3) + data:byte(4) * 256 end
	  if cnt ==  2 then t3  = data:byte(1) + data:byte(2) * 256 end
	  if cnt ==  3 then t4  = data:byte(1) + data:byte(2) * 256
						t5  = data:byte(3) + data:byte(4) * 256 end
	  if cnt ==  4 then t6  = data:byte(1) + data:byte(2) * 256
						t7  = data:byte(3) + data:byte(4) * 256 end
	  if cnt ==  5 then s1  = data:byte(1) + data:byte(2) * 256 + data:byte(3) * 65536 + data:byte(4) * 16777216 end
	  if cnt ==  6 then h1  = data:byte(1) + data:byte(2) * 256 + data:byte(3) * 65536 + data:byte(4) * 16777216 end
	  if cnt ==  7 then sp  = data:byte(1) 
					    pa  = data:byte(2) end
	  if cnt ==  8 then br  = data:byte(1) end
	  cnt = cnt + 1
	  if cnt > 8 then cnt = 1 end
    stat = 0
	end
end, 0)

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: Superbrudi <userid-13940967@openv.wikispaces>
* Date: Wed, 29 Mar 2017 16:52:45 UT
* Message-ID: <80929187-93571421$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Dazu noch das PHP Script auf dem Server um die Datenbank zu füttern: "heating.php"

----------------------------------------------------------------------
<html>
<body>
<?php
  $time = microtime(true);
  $t1 = $_GET["t1"];
  $t2 = $_GET["t2"];
  $t3 = $_GET["t3"];
  $t4 = $_GET["t4"];
  $t5 = $_GET["t5"];
  $t6 = $_GET["t6"];
  $t7 = $_GET["t7"];
  $s1 = $_GET["s1"];
  $h1 = $_GET["h1"];
  $sp = $_GET["sp"];
  $pa = $_GET["pa"];
  $br = $_GET["br"];
  if ( ($t1 != "") && ($t2 != "") && ($t3 != "") && ($t4 != "") )
  {
     $con = mysqli_connect("","root","admin");
     mysqli_select_db($con,"heating");
     $ok = mysqli_query($con,"INSERT INTO `heating` (datetime, t1, t2, t3, t4, t5, t6, t7, s1, h1, sp, pa, br) VALUES ('$time', '$t1', '$t2', '$t3', '$t4', '$t5', '$t6', '$t7', '$s1', '$h1', '$sp', '$pa', '$br')");
     mysqli_close($con);
     $mmm = round((300 - $time % 300)/10);
  echo "$mmm";
  }
?>
</body>
</html>

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: the.neon <userid-38712692@openv.wikispaces>
* Date: Fri, 14 Apr 2017 17:02:40 UT
* Message-ID: <80929187-93781319$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Danke !

Eine frage hab aktuell die usb ttl version des adapters. Läuft diese auch mit dem esp8266 ? Wenn ja wie anschliesen

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: Superbrudi <userid-13940967@openv.wikispaces>
* Date: Wed, 26 Apr 2017 12:41:58 UT
* Message-ID: <80929187-93908581$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Das sollte damit genauso gehen. IN des Adapters an URXD, OUT and UTXD. Spannungsversorgung 3,3V. Der Spannungsregler des Adapters kann auch den ESP8266 versorgen.

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: the.neon <userid-38712692@openv.wikispaces>
* Date: Sat, 16 Sep 2017 09:21:41 UT
* Message-ID: <80929187-95086309$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Danke geht soweit. Hat jemand schon kw protokol am laufen mit dem esp ?

WikiSpaxe, 2017-10-26:
* Subject: Re: Lua script
* Wiki-Page: Bauanleitung ESP8266
* From: the.neon <userid-38712692@openv.wikispaces>
* Date: Sat, 16 Sep 2017 09:21:46 UT
* Message-ID: <80929187-95086311$openv.wikispaces>
* In-Reply-To: <80929187-93396241$openv.wikispaces>

Danke geht soweit. Hat jemand schon kw protokol am laufen mit dem esp ?

Hottabych-, 2017-11-01:
Hallo,
kann es sein das 'heating.php' beim Umzug verloren gegangen ist?

Danke

vitoopen, 2017-11-03:
wo sollte sich die Datei heating.php befunden haben?
wenn ich auf openv.wikispaces.com unter "Pages and Files" nach php suche, wird nur eine Seite "PHP Webfront" und eine Datei plotHeizung.php gefunden.

Hottabych-, 2017-11-10:
Sorry für die späte Antwort, aber ich war beruflich unterwegs. Im 3ten Post von oben 

> From: Superbrudi userid-13940967@openv.wikispaces
> Date: Wed, 29 Mar 2017 16:52:45 UT

wird darauf verwiesen.