openv/openv #353: 1xWW Bereitung WO1C setaddr P300

WikiSpaxe, 2017-10-26:
* Subject: 1xWW Bereitung WO1C setaddr P300
* Wiki-Page: home
* From: Vitocal_dolor <userid-57904811@openv.wikispaces>
* Date: Mon, 27 Mar 2017 00:20:13 UT
* Message-ID: <81000249-93525789$openv.wikispaces>

Hallo,
ich versuche eine einmalige Warmwasserbereitung anzutriggern. Leider gibt es Probleme dabei. Version hab ich vctrld>version 0.98.2_IPv6.
Als Adresse schreibe ich B020 mit dem wert 1. Habt ihr eine Idee? Log:
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 05
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: closed
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 16
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 00
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 00
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 06 (10.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 06
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: opened
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 41
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 07
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 00
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 02
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: B0
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 20
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 01
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 01
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: 00
DEBUG:Sun Mar 26 09:29:19 2017 : >SEND: DB
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 06 (30.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 06
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: addr was still active FE06
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: Command send
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: no preset result
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 41 (10.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 41
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 06 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 06
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 03 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 02 (10.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 B0 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 20 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 01 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 04 (0.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: len=1 E0 (10.0 ms)
DEBUG:Sun Mar 26 09:29:19 2017 : <RECV: received 03 02 B0 20 01 04 E0
DEBUG:Sun Mar 26 09:29:19 2017 : >FRAMER: ERROR address B020 code 4
DEBUG:Sun Mar 26 09:29:19 2017 : Fehler recv, Abbruch
DEBUG:Sun Mar 26 09:29:19 2017 : Fehler beim ausfuehren von set1xWW-Bereitung
ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
>FRAMER: ERROR address B020 code 4
Fehler recv, Abbruch
Fehler beim ausfuehren von set1xWW-Bereitung

WikiSpaxe, 2017-10-26:
* Subject: Re: 1xWW Bereitung WO1C setaddr P300
* Wiki-Page: home
* From: Vitocal_dolor <userid-57904811@openv.wikispaces>
* Date: Mon, 27 Mar 2017 13:13:29 UT
* Message-ID: <81000249-93532533$openv.wikispaces>
* In-Reply-To: <81000249-93525789$openv.wikispaces>

Fehler gefunden: Ees wurden zwei Byte übertragen. 41 07....es muss heißen 41 06. Leider wird keine einmalige WW Bereitung ausgeführt, sondern der Manuelle Betrieb. Lösung steht noch aus: http://www.viessmann-community.com/t5/Experten-fragen/1xWW-Bereitung-WO1C/qaq-p/54507

Morg42, 2017-12-28:
Das gleiche Problem habe ich auch. Manuelle Steuerung wird ausgelöst. Anlage ist eine WO1C, id 0x204D, Vitocal 200-S.

Interessanterweise kann ich bei "1xWW aus" den Datenpunkt auslesen ("0"); bei "1xWW an" erhalte ich "NOT OK".

FM-500, 2019-03-07:
> Das gleiche Problem habe ich auch. Manuelle Steuerung wird ausgelöst. Anlage ist eine WO1C, id 0x204D, Vitocal 200-S.
> 
> Interessanterweise kann ich bei "1xWW aus" den Datenpunkt auslesen ("0"); bei "1xWW an" erhalte ich "NOT OK".

Konntest du die einmalige Auslösung hinbekommen?
Laut Viessmann Community soll man ja 2 senden statt 1, dann sollte es gehen?

Ich bekomme es nur nicht hin, die B020 als Setzadresse in der vito.xml festzulegen für iobroker.
Auslesen kann iich den Status auch :-(

Morg42, 2020-04-17:
Besser spät als nie - mit 2 geht es! Ich weiß aber nicht, was du mit iobroker meinst...