openv/openv #356: Vito.xml für WO1B Wärmepumpe

WikiSpaxe, 2017-10-26:
* Subject: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Fri,  7 Apr 2017 21:00:12 UT
* Message-ID: <81103673-93699403$openv.wikispaces>

Hallo

Habe meinen Raspi mit der Viessmann Wärmepumpe verbunden, einige Werte kann ich schon auslesen allerdings hätte ich gern noch mehr Hex Adressen um meine vito.xml damit zu befüllen.
Gibt es irgendwo eine Liste allen Adressen für die Vitoconnect W01B ?

danke im Voraus

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Fri,  7 Apr 2017 21:23:15 UT
* Message-ID: <81103673-93699733$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

VITOCONECT W01B ist falsch von mir angegeben ........... sollte eigentlich Vitotronic 200 W01B heißen

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Fri,  7 Apr 2017 21:23:16 UT
* Message-ID: <81103673-93699735$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

VITOCONECT W01B ist falsch von mir angegeben ........... sollte eigentlich Vitotronic 200 W01B heißen

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: heid16 <userid-38990402@openv.wikispaces>
* Date: Fri,  7 Apr 2017 22:11:00 UT
* Message-ID: <81103673-93700345$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Hallo,

Welche Wärmepumpe hast du, die Adressen sind leicht unterschiedlich.

Lg

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Fri,  7 Apr 2017 22:59:57 UT
* Message-ID: <81103673-93700671$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

EINE Vitocal 300 mit einer Steuereinheit Vitotronic 200 W01B

lg

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Fri,  7 Apr 2017 23:10:54 UT
* Message-ID: <81103673-93700771$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Bekomme beim Auslesen der Auslesen folgenden Fehler  ( hier der Debug )
[30951] Fri Apr  7 21:07:23 2017 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[30951] Fri Apr  7 21:07:23 2017 : >SEND: 04
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 05 (2010.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: received 05
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: closed
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 16
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 00
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 00
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 06 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: received 06
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: opened
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 41
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 05
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 00
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 01
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 01
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 6E
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 02
[30951] Fri Apr  7 21:07:25 2017 : >SEND: 77
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 06 (20.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: received 06
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: addr was still active FE06
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: Command send
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: no preset result
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 41 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 06 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: received 41 06
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 03 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 01 (10.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 01 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 6E (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 01 (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 01 (10.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: len=1 7B (0.0 ms)
[30951] Fri Apr  7 21:07:25 2017 : <RECV: received 03 01 01 6E 01 01 7B
[30951] Fri Apr  7 21:07:25 2017 : >FRAMER: ERROR address 016E code 1
[30951] Fri Apr  7 21:07:25 2017 : Fehler recv, Abbruch
[30951] Fri Apr  7 21:07:25 2017 : Fehler beim ausfuehren von getTemp

Hier die vito.xml von getTemp Konfig: von getTemp
<command name='getTemp' protocmd='getaddr'>
                        <addr>016E</addr>
                        <len>2</len>
                        <unit>T1U</unit>
                        <description>Temperatur in °C</description>

Es sieht so aus als könnte ich mit dem Empfangenen Wert nichts anfangen
Danke

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: heid16 <userid-38990402@openv.wikispaces>
* Date: Sat,  8 Apr 2017 08:36:00 UT
* Message-ID: <81103673-93702881$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Hallo,

Die Außentemperatur steht auf Adresse 0101
Probier mal diese Adresse.

mfg

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: heid16 <userid-38990402@openv.wikispaces>
* Date: Sat,  8 Apr 2017 09:13:40 UT
* Message-ID: <81103673-93702981$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Noch einige weitere:

        <command name='getDevType' protocmd='getaddr'>
             <addr>00F8</addr>
             <len>2</len>
             <unit>DT</unit>
             <description>Device Type der Anlage</description>
        </command>
 

	<command name='getTempA' protocmd='getaddr'>
	     <addr>0101</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Aussentemperatur in Grad C</description>
	</command>	
	<command name='getTempA_solar' protocmd='getaddr'>
	     <addr>0111</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Aussentemperatur Solarfühler in Grad C</description>
	</command>	

	<command name='getTempR' protocmd='getaddr'>
	     <addr>0116</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur in Grad C</description>
	</command>	
	<command name='getTempR_2' protocmd='getaddr'>
	     <addr>0117</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur in Grad C</description>
	</command>	
	<command name='getTempR_3' protocmd='getaddr'>
	     <addr>0118</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur in Grad C</description>
	</command>	
	<command name='getTempRF_Soll' protocmd='getaddr'>
	     <addr>011B</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur Fernsteuerung Soll in Grad C</description>
	</command>	
	<command name='getTempRF_Soll_2' protocmd='getaddr'>
	     <addr>011C</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur Fernsteuerung Soll in Grad C</description>
	</command>	
	<command name='getTempRF_Soll_3' protocmd='getaddr'>
	     <addr>011D</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur Fernsteuerung Soll in Grad C</description>
	</command>	
	<command name='getTempR_Soll' protocmd='getaddr'>
	     <addr>1000</addr>
	     <len>2</len>
	     <unit>UT</unit>
[... 204 lines cut ...]
	     <description>Raumtemperatur Soll Tag Mittelstellung in Grad C</description>
	</command>
	<command name='getTempRaumSollNacht_M' protocmd='getaddr'>
	     <addr>2001</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Raumtemperatur Soll Nacht Mittelstellung in Grad C</description>
	</command>

	<command name='getKurveNiveau' protocmd='getaddr'>
	     <addr>2006</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Heizkennlinie Niveau</description>
	</command>
	<command name='getKurveSteigung' protocmd='getaddr'>
	     <addr>2007</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <description>Heizkennlinie Steigung</description>
	</command>	

Werte auf einigen Adressen sind nur vorhanden wenn z.B. Solaranlage eingebunden, Fernsteuerung mit Raumtemperaturmessung, Pufferspeicher vorhanden, etc.

mfg

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: tom2222as <userid-58060961@openv.wikispaces>
* Date: Sat,  8 Apr 2017 09:59:34 UT
* Message-ID: <81103673-93703173$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Super ! Danke - hab zwar 0101 auch schon gesetzt gehabt für die Außentemperatur dort hab ich den gleichen Fehler bekommen.
Deine vito-Einstellungen passen bei mir auch, scheint an der Unit gelegen zu sein, wenn ich die Unit auf UT ändere dann stimmt meine Abfrage auch. 
Der Wert 016E funkt bei beiden Units nicht :)

Auf jeden Fall danke für die schnelle Antwort !!!! Hast mir super weitergeholfen.

lg

WikiSpaxe, 2017-10-26:
* Subject: Re: Vito.xml für WO1B Wärmepumpe
* Wiki-Page: Protokoll 300
* From: heid16 <userid-38990402@openv.wikispaces>
* Date: Sat,  8 Apr 2017 10:33:12 UT
* Message-ID: <81103673-93703287$openv.wikispaces>
* In-Reply-To: <81103673-93699403$openv.wikispaces>

Freut mich.

Es lassen sich auch sämtliche Einstellparameter der Steuerung auslesen und setzen. Ein weites Betätigungsfeld...

mfg

s4ndst0rm, 2018-02-28:
Wie komme ich denn an die ErrorStates? Ich kann zwar abfragen ob es eine Störung gibt (Müsste Adresse 0A82 sein) aber ich hätte gerne den ErrorState.
Ist das die Adresse: 7507? 
Weiterhin hätte ich dann auch gerne noch den Detailfehler: In der 7507 steht dann ja "nur" ein "Fehler im Kältekreis" aber nicht was tatsächlich den Fehler ausgelöst hat. Beim nächsten Start der WP ist dieser erweiterte Fehler auch weg..

Danke!

s0170071, 2023-12-02:
Versuch mal 0x010d für die obere Wassertemperatur.