openv/openv #359: ESP8266/Arduino library

WikiSpaxe, 2017-10-26:
* Subject: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Schuppenzot <userid-58077647@openv.wikispaces>
* Date: Sun, 23 Apr 2017 16:07:07 UT
* Message-ID: <81200201-93867403$openv.wikispaces>

I took the opportunity to create a library to communication using the ESP8266 and the Arduino framework.
I've only tested with the ESP8266 (Wemos D1 mini) but I think it would work using any arduino compatible board.

You can check it out on https://github.com/bertmelis/VitoWifi

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Schuppenzot <userid-58077647@openv.wikispaces>
* Date: Sun, 23 Apr 2017 16:09:15 UT
* Message-ID: <81200201-93867411$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

Keep in mind there is still a lot of work to to before we can talk about a "final version", especially with the transformation functions to return the hex values into a human readable format.

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Mon, 24 Apr 2017 13:13:13 UT
* Message-ID: <81200201-93877629$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

Great!!! Does it also work with KW protocoll? I have a V150KB1

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Mon, 24 Apr 2017 13:13:24 UT
* Message-ID: <81200201-93877633$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

Great!!! Does it also work with KW protocoll? I have a V150KB1

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Schuppenzot <userid-58077647@openv.wikispaces>
* Date: Mon, 24 Apr 2017 14:36:51 UT
* Message-ID: <81200201-93878715$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

I've only implemented the P300 protocol. There's currently no plan to support the KW protocal as the current version is not final yet.

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Schuppenzot <userid-58077647@openv.wikispaces>
* Date: Tue, 25 Apr 2017 21:15:13 UT
* Message-ID: <81200201-93901301$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

A first working version is ready. If you can convince me I could implement the KW protocol. I've never looked into it, but it seems to be "easier".
But it's a matter of priorities. (all the different Datapoint-types are not yet implemented)

WikiSpaxe, 2017-10-26:
* Subject: Re: ESP8266/Arduino library
* Wiki-Page: Bauanleitung ESP8266
* From: Schuppenzot <userid-58077647@openv.wikispaces>
* Date: Mon,  9 Oct 2017 22:22:59 UT
* Message-ID: <81200201-95383557$openv.wikispaces>
* In-Reply-To: <81200201-93867403$openv.wikispaces>

A first version for the KW-protocol is published. Check out the develop-KW branch.

bertmelis, 2018-02-24:
can be closed