openv/openv #366: Probleme bei der Kommunikation zwischen Pi und Vcontrold.

WikiSpaxe, 2017-10-26:
* Subject: Probleme bei der Kommunikation zwischen Pi und Vcontrold.
* Wiki-Page: home
* From: luismauriz1 <userid-58794859@openv.wikispaces>
* Date: Thu, 12 Oct 2017 18:47:35 UT
* Message-ID: <82260325-95430391$openv.wikispaces>

Hallo, inzwischen habe ich soweit das vcontrold programm fertig. allerdings, wenn ich es testen möchte, über Putty oder von dem Pi selber über die shell und den telnet localhost 3002 befehl, dann kommt dauernd Connection refused. Also blockiert. Liegt es an den Ip Adressen die ich in dem Vcontrold xml drin habe oder an was liegt es? es geht weder vom Pc als auch vom Pi selber. Habe nur 2 Ip`s die 127.... und eine von meinem Computer wo ich über Putty drauf zugreifen möchte.Über eine schnelle Antwort würde ich mich freuen.
LG

hoki26, 2019-02-16:
Hallo, 
dein Thread ist zwar schon älter, aber ich habe derzeit das gleiche Problem. Kannst Du mir schildern, wie Du das Problem gelöst hast? Ich werde hier wahnsinnig ....

2Slow4, 2019-02-16:
Hatte sowas auch.... bist Du Dir sicher, dass vcontrol wirklich läuft?
Vermute es liegt am start script.
Hab es dann so gemacht: https://github.com/openv/openv/issues/392

hoki26, 2019-02-16:
So, bin jetzt etwas weiter ... es sieht so aus, dass die Kiste Probleme beim laden der XML-Dateien hat. Ich hatte eine Telnet-Session offen, hatte aber vergessen eine Anlagen-ID einzugeben. nach der Bearbeitung mit "nano" und einem durchstarten des Service bekomme ich folgenden Fehler, den ich vorher NICHT hatte:
```
root@Heizung:/etc/init.d# sh vcontrol status
● vcontrol.service - LSB: Initscript to start vcontrold deamon
   Loaded: loaded (/etc/init.d/vcontrol; generated; vendor preset: enabled)
   Active: active (exited) since Sat 2019-02-16 16:04:36 GMT; 6s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 833 ExecStop=/etc/init.d/vcontrol stop (code=exited, status=0/SUCCESS)
  Process: 872 ExecStart=/etc/init.d/vcontrol start (code=exited, status=0/SUCCESS)

Feb 16 16:04:36 Heizung systemd[1]: Starting LSB: Initscript to start vcontrold deamon...
Feb 16 16:04:36 Heizung vcontrol[872]: Fehler beim Laden von /etc/vcontrold/vcontrold.xml, terminiere!
Feb 16 16:04:36 Heizung systemd[1]: Started LSB: Initscript to start vcontrold deamon.

```
Da ich dachte, dass ich versehentlich die XML-Struktur zerdeppert habe, habe ich nochmal die originalen Files in /etc/vcontrold kopiert. Auch den "Include-Part" für die vito.xml auskommentiert um zu prüfen, ob es an Datei1 oder 2 liegt ... nix ... kanns nicht lesen. An was könnte es denn noch hängen?

hoki26, 2019-02-16:
Jetzt wird es schräg: auf der Suche nach der nicht ladbaren XML Dateien fand ich folgenden Effekt heraus. Ich öffne "vcontrold.xml" mit "nano", schreibe und lösche in Zeile 1 Position 1 ein Leerzeichen und speichere die Datei. Faktisch wurde nichts vom Inhalt geändert, aber danach kann der Daemon die XML Files nicht mehr lesen. Gleiches Phänomen kann man auch bei "vim" beobachten. Es scheint, dass die Dateien unlesbar werden, sobald sie gespeichert wurden.

Hat hier jemand einen Tip? Ich habe einen Raspi mit Debian Stretch laufen ...

DBa2016, 2019-02-16:
Versuch mal, nach dem Editieren unix2dos drüber laufen zu lassen.

hoki26, 2019-02-17:
> Versuch mal, nach dem Editieren unix2dos drüber laufen zu lassen.

Kein Effekt ... ich lass es mal für heute gut sein. Auch mit verschiedenen Editoren (Notepad++, XMLSpy, ...) und verschiedenen Codierungen kann vcontrold die Files nicht einlesen. 
Ich gehe jetzt mal eine Runde weinen...

hoki26, 2019-02-17:
[vcontrold.xml.txt](https://github.com/openv/openv/files/2872797/vcontrold.xml.txt)
[vito.xml.txt](https://github.com/openv/openv/files/2872798/vito.xml.txt)

2Slow4, 2019-02-17:
Die ID2049 und vitoir0 sind sicher richtig?

Probier mal diese abgespeckten files, dabei nur die ID2048 mit ID2049
ersetzen und ttyusb0 mit vitoir0 und V200WO1B mit Deiner ersetzen.

<?xml version="1.0"?>
<V-Control xmlns:vcontrol="http://www.openv.de/vcontrol">
  <unix>
     <config>
	<serial>
		<tty>/dev/ttyUSB0</tty>
	</serial>
	<net>
		<port>3002</port>
		<allow ip='127.0.0.1'/>
	</net>
	<logging>
		<file>vcontrold.log</file>
		<syslog>y</syslog>
		<debug>y</debug>
	</logging>
	<device ID="2048"/>
     </config>
  </unix>
  <units>
	<unit name='Temperatur'>
	   <abbrev>UT</abbrev>
	   <calc get='V/10' set='V*10'/>
	   <type>short</type>
	   <entity>Grad Celsius</entity>
	</unit>
	<unit name='Temperatur100'>
	   <abbrev>UTH</abbrev>
	   <calc get='V/100' set='V*100'/>
	   <type>short</type>
	   <entity>Grad Celsius</entity>
	</unit>
	<unit name='Neigung'>
	   <abbrev>UN</abbrev>
	   <calc get='V/10' set='V*10'/>
	   <type>short</type>
	   <entity></entity>
	</unit>	
	<unit name='Temperatur 1Byte'>
          <abbrev>UT1</abbrev>
          <calc get='V/2' set='V*2'/>
          <type>char</type>
          <entity>Grad Celsius</entity>
        </unit>
	<unit name='Temperatur 1Byte unsigned'>
          <abbrev>UT1U</abbrev>
          <calc get='V/2' set='V*2'/>
          <type>uchar</type>
          <entity>Grad Celsius</entity>
        </unit>
	<unit name='Temperatur 1Byte ganzzahlig'>
          <abbrev>UTI</abbrev>
          <calc get='V' set='V'/>
          <type>uchar</type>
          <entity>Grad Celsius</entity>
        </unit>
	<unit name='Status'>
	   <abbrev>ST</abbrev>
	   <calc get='V' set='V'/>
	   <type>char</type>
	   <entity></entity>
	</unit>
	<unit name='Counter'>
	   <abbrev>CO</abbrev>
	   <calc get='V' set='V'/>
[... 405 lines cut ...]
                        <description>Diagnose Anlagenuebersicht Wertebereich -20 - 95</description>
                </command>
                <command name="RLSole" protocmd="getaddr">
                        <addr>0104</addr>
                        <len>2</len>
                        <unit>UT</unit>
                        <description>Diagnose Anlagenuebersicht Wertebereich -20 - 95</description>
                </command>
                <command name="VLHK1" protocmd="getaddr">
                        <addr>0105</addr>
                        <len>2</len>
                        <unit>UT</unit>
                        <description>Diagnose Anlagenuebersicht Wertebereich 0 - 95</description>
                </command>
                <command name="RLHK1" protocmd="getaddr">
                        <addr>0106</addr>
                        <len>2</len>
                        <unit>UT</unit>
                        <description>Diagnose Anlagenuebersicht Wertebereich 0 - 95</description>
                </command>
                
                
                
        </commands>
</vito>

hoki26, 2019-02-19:
Überraschung und Lösung zugleich: das Geheimnis der "defekten" XML-Datei sind Zeilenumbrüche im Stile von "\n". Ich vermute, es gibt Probleme beim Einlesen der Datei (Zeilenweise statt am Stück bis EOF).

Die Lösung: den gesamten XML-Inhalt in eine Zeile packen oder -anders ausgedrückt- wir entfernen die Zeilenumbrüche aus den Dateien vcontrold.xml und vito.xml

```
tr -d '\n' < vcontrold.xml > ausgabedatei.xml
tr -d '\n' < vito.xml > ausgabedatei2.xml
```
Danach alle XML-Files löschen
`rm *.xml`

und die gesäuberten Files umbenennen.

```
cp ausgabedatei.xml vcontrold.xml
cp ausgabedatei2.xml vito.xml

```
Daemon starten und die Kiste läuft. Vielen lieben Dank an besonders an 2Slow4 für seine Mühe!