openv/openv #367: Nachricht senden funktioniert nicht

WikiSpaxe, 2017-10-26:
* Subject: Nachricht senden funktioniert nicht
* Wiki-Page: Bauanleitung ESP8266
* From: bjoern3003 <userid-59038465@openv.wikispaces>
* Date: Thu, 19 Oct 2017 11:04:11 UT
* Message-ID: <82323105-95521861$openv.wikispaces>

Moin, ich habe das Projekt hier nachgebaut, weil es genau meinen Anforderungen entspricht und schön simpel ist. 

Anfangs hatte auch alles wunderbar funktioniert, bis auf das nur Kauderwelch bei mir ankam, aber das könnte am Protokoll gelegen haben.

Nun wollte ich dem ein wenig auf den Grund gehen, indem ich über die Konsole den Datenverkehr mitlese, aber irgendwie bekomme ich es nicht mehr hin, dass die Heizung meine gesendeten Befehle empfängt. Ich sehe (mithilfe einer Kamera), dass die IR LED blinkt, sobald ich einen Befehl absetze, aber die Heizung antwortet durchgängig nur mit "05". Habe auch bereits eine andere SFH487-2 getestet (ohne Erfolg). WIe kann ich dem Problem nun am besten auf den Grund gehen? 

Es handelt sich um eine Viessmann Vitodens 300.
Heizung antwortet immer nur mit 05

WikiSpaxe, 2017-10-27:
* Subject: Re: Nachricht senden funktioniert nicht
* Wiki-Page: Bauanleitung ESP8266
* From: kunksmuhme <userid-57546650@openv.wikispaces>
* Date: Fri, 27 Oct 2017 00:12:35 UT
* Message-ID: <82323105-95636807$openv.wikispaces>
* In-Reply-To: <82323105-95521861$openv.wikispaces>

Wie sieht denn deine Hardware aus, welchen Adapter haste nachgebaut?

WikiSpaxe, 2017-10-27:
* Subject: Re: Nachricht senden funktioniert nicht
* Wiki-Page: Bauanleitung ESP8266
* From: bjoern3003 <userid-59038465@openv.wikispaces>
* Date: Fri, 27 Oct 2017 08:43:20 UT
* Message-ID: <82323105-95640167$openv.wikispaces>
* In-Reply-To: <82323105-95521861$openv.wikispaces>

Ich habe die ESP8266 Version nachgebaut. Mittlerweile hat sich aber rausgestellt, dass offensichtlich die Spannung immer zusammenbrach. Die Sendeseite habe ich nun noch um eine Transistorschaltung erweitert. Der erste Test lief erfolgsversprechend. Werde das über die Feiertage mal weiter ausgiebig testen.

sloesp, 2017-12-04:
Hallo zusammen.
Habe mich der gleichen Schaltung angenommen und erhalte das gleiche Ergebnis. Die Anlage sendet kontinuierlich 0x05. Den IR-Sender habe ich geprüft -> OK. Da ich vermutete, das die Sendeleistung zu gering ist, kam auch hier eine Transistorschaltung zum Einsatz. Mögliches Übersprechen von der Empfangs- zur Sendeseite ist unterdrückt.
Leider habe ich bislang keinerlei spürbaren Erfolg was das Absetzen einer Anlagenabfrage angeht. Gibt es nun schon etwas Neues was mehr Erfolg verspricht. Ach ja, meine Anlage ist eine Vitodens 222-F aus 2014.
Für Hinweise wäre ich dankbar.

gitzila, 2022-11-18:
@sloesp Habe leider vor dem Lesen Deiner Nachrichten die simple ESP8266-Schaltung nachgebaut. Die Infrarot LED leuchtet mit den 180 Ohm Vorwiderstand nicht mal auf, das ist maximal ein kurzes Aufglimmen, kaum wahrnehmbar. Es ist daher nicht verwunderlich, dass der ganze Kram an einem ESP8266 in der Form nicht funktioniert. Ich wage mal zu behaupten, dass das noch nirgendwo richtig funktioniert hat. Habe dann mit Transistoren experimentiert. Siehe da: Ein helles Leuchten der LED. Leider ist die LED aber immer auf voller Power, egal, ob der ESP8266 über TX "sendet" oder nicht. Mit einem Oszilloskop sehe ich auch keine Spannungsschwankungen (?!?). Alles sehr kryptisch. Hast Du damals Erfolg gehabt oder alles in die Ecke geschmissen?