openv/openv #368: Fehlermeldung vcontrold

WikiSpaxe, 2017-10-26:
* Subject: Fehlermeldung vcontrold
* Wiki-Page: Protokoll 300
* From: andies1 <userid-58839765@openv.wikispaces>
* Date: Tue, 24 Oct 2017 16:15:20 UT
* Message-ID: <82371143-95599203$openv.wikispaces>

Ich versuche, die Zeiten mit vcontrold zu setzen. Erst einmal musste ich mir aus dem Code heraussuchen, wie man das überhaupt macht - weil es nirgendwo steht. Leider kommt bei mir eine Fehlermeldung, von der ich vermute, dass sie falsch ist. Hier der genau debug:

vctrld>setTimerWWMo 07:40 10:10 12:00 12:30 15:30 16:00 19:00 20:30
DEBUG:Tue Oct 24 15:46:38 2017 : Befehl: setTimerWWMo 07:40 10:10 12:00 12:30 15:30 16:00 19:00 20:30
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 07:40 -> [3C]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 10:10 -> [51]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 12:00 -> [60]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 12:30 -> [63]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 15:30 -> [7B]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 16:00 -> [80]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 19:00 -> [98]
DEBUG:Tue Oct 24 15:46:38 2017 : Cycle Time: 20:30 -> [A3]
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 41
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 0D
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 00
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 02
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 21
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 00
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 08
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 3C
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 51
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 60
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 63
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 7B
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 80
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: 98
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: A3
DEBUG:Tue Oct 24 15:46:38 2017 : >SEND: BE
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: len=1 06 (90.0 ms)
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: received 06
DEBUG:Tue Oct 24 15:46:38 2017 : 
DEBUG:Tue Oct 24 15:46:38 2017 : >FRAMER: Command send
DEBUG:Tue Oct 24 15:46:38 2017 : >FRAMER: no preset result
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: len=1 41 (0.0 ms)
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: received 41
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: len=1 05 (0.0 ms)
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: received 05
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: len=6 01 (0.0 ms)
DEBUG:Tue Oct 24 15:46:38 2017 : <RECV: received 01 02 21 00 08 31
DEBUG:Tue Oct 24 15:46:38 2017 : >FRAMER: unexpected length 8 08
DEBUG:Tue Oct 24 15:46:38 2017 : Fehler recv, Abbruch
DEBUG:Tue Oct 24 15:46:38 2017 : Fehler beim ausfuehren von setTimerWWMo 07:40 10:10 12:00 12:30 15:30 16:00 19:00 20:30

Sind die Zeiten nun geschrieben oder nicht?

andreasloe, 2017-10-29:
Die Zeiten sind geschrieben, das habe ich nachprüfen können. Der Fehler muss bei der Rückmeldung erfolgen, anscheinend erwartet vcontrold etwas anderes als das, was dann kommt. Schaut man auf den Programmtext, versteckt sich der Fehler vermutlich in vcontrol.c in Zeile 465 f., denn dort steht

			/* falls ein Pre-Kommando definiert wurde, fuehren wir dies zuerst aus */
			if (cPtr->precmd &&(pcPtr=getCommandNode(cfgPtr->devPtr->cmdPtr,cPtr->precmd))) {
				logIT(LOG_INFO,"Fuehre Pre Kommando %s aus",cPtr->precmd);

				if (execByteCode(pcPtr->cmpPtr,fd,pRecvBuf,sizeof(pRecvBuf),sendBuf,sendLen,1,pcPtr->bit,pcPtr->retry,pRecvBuf,pcPtr->recvTimeout)==-1) {	
				  // todo semjfi vcontrol_semrelease();
				  logIT(LOG_ERR,"Fehler beim ausfuehren von %s",readBuf);
				  sendErrMsg(socketfd);
				  break;
				}
				else {
					bzero(buffer,sizeof(buffer));
					char2hex(buffer,pRecvBuf,pcPtr->len);
					logIT(LOG_INFO,"Ergebnis Pre-Kommand: %s",buffer);
				}
	
					
			}
				
			/* wir fuehren den Bytecode aus, 
			       -1 -> Fehler  
				0 -> Formaterierter String
				n -> Bytes in Rohform */

			count=execByteCode(cPtr->cmpPtr,fd,recvBuf,sizeof(recvBuf),sendBuf,sendLen,noUnit,cPtr->bit,cPtr->retry,pRecvBuf,cPtr->recvTimeout);
			// todo semjfi vcontrol_semrelease();

			if (count==-1) {	
				logIT(LOG_ERR,"Fehler beim ausfuehren von %s",readBuf);
				sendErrMsg(socketfd);
			}
und welche der beiden Zeilen das sind, kann ich aufgrund mangelnder Programmkenntnisse leider nicht beurteilen.

andreasloe, 2017-10-29:
Ich habe eine Lösung, die aber brute force ist. Sie funktioniert bei mir auch, aber mir wäre wohler, wenn die Autoren des Codes sich dem mal annehmen. In framer.c muss man in Zeile 442 die Abfrage ändern. Bei timer-Abfragen weist r_len eine Länge von 1 auf, während l_buf[P300_RESP_LEN_OFFSET]  nun acht ist (weil eben nicht ein Byte, sondern mehr zurückgeliefert wird - keine Ahnung wieso, im Quellcode steht was von "Error möglich"). Da ich das nur für ganz bestimmte Befehle nutze und der Fehler nur bei den Timern auftritt und das Kommando trotzdem ausgeführt wird, habe ich jetzt an der Stelle einfach geprüft, ob die zweite Zahl gleich 8 ist und dann lasse ich den Fehler schlichtweg nicht zu, also

`if ((r_len != l_buf[P300_RESP_LEN_OFFSET]) && (l_buf[P300_RESP_LEN_OFFSET] != 8))`

Bei mir geht das jetzt durch. Keine Fehlermeldung, die Werte werden gesetzt.

Dennoch wäre es schöner, wenn jemand, der den Code kapiert, sich dazu mal äußert.

hmueller01, 2017-11-07:
Hallo,

welche Version von vcontrold verwendest Du?
Hast Du den Patch (der auf der r106 basiert) eingespielt?
[https://sourceforge.net/p/vcontrold/patches/3/](https://sourceforge.net/p/vcontrold/patches/3/)

Gruß Holger

andreasloe, 2017-11-07:
Also ich scheine Rev 107 zu haben, denn ich kriege folgende Fehlermeldung

patch < P300_setaddr_patch_hmueller_20151124.diff 
patching file parser.c
Hunk #2 succeeded at 140 (offset 1 line).
Hunk #3 succeeded at 251 (offset 11 lines).
Hunk #4 succeeded at 270 (offset 11 lines).
Hunk #5 succeeded at 280 (offset 11 lines).
patching file framer.c
Hunk #19 FAILED at 528.
1 out of 19 hunks FAILED -- saving rejects to file framer.c.rej

hmueller01, 2017-11-07:
Wenn Du es schnell haben willst checke die r106 aus und patche dann. Ist eh keine wesentliche Änderung zwischen den beiden releases. Irgendwie hat der Entwickler Teile meines Patches schon übernommen, aber nicht alle. Deshalb geht der Patch an der Stelle schief. Du kannst aber auch warten, bis ich den Patch für die bisherige 107, die jetzt hier auf github übertragen wurde fertig habe. Muss es nur noch testen. Wird vmtl. diese Woche ...

andreasloe, 2017-11-07:
Ich habe einfach die framer.c und parser.c genommen, die auf der verlinkten seite als zip zu finden waren. Da ging alles durch. Teste ich jetzt noch, aber sieht gut aus. DANKE!

andreasloe, 2017-11-07:
Läuft.