openv/openv #76: Probleme bei der initialisierung

WikiSpaxe, 2017-10-26:
* Subject: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: andyyyy1 <None>
* Date: Mon, 19 Mar 2012 22:47:20 UT
* Message-ID: <51894274-51894274$openv.wikispaces>

Hallo Leute,

habe heute versucht vconrold unter Windows mit dieser Anleitung hier zu installieren, jedoch komme ich nicht über die initialisierung bzw. bekomme ihc vom vclient einen Fehler.
Folgende Ausgabe habe ich bei der initialisierung.

```

[3868] Mon Mar 19 22:36:20 2012 : Expandiere Kommandos fuer Device 2000
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 05 00 04;RECV 4 CO
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 05 00 04;RECV 4 CO
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 05 96 04;RECV 4 CS
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 05 96 04;RECV 4 CS
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 05 80 04;RECV 4 CS
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 05 80 04;RECV 4 CS
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 0A 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 0A 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 04 00 01;RECV 1 
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 04 00 01;RECV 1 
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 01 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 01 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 04 16 01;RECV 1 
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 04 16 01;RECV 1 
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 02 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 02 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 06 82 02;RECV 2 PS
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 06 82 02;RECV 2 PS
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 06 80 02;RECV 2 PS
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 06 80 02;RECV 2 PS
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 06 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 06 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 05 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 05 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 04 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 04 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 03 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 03 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 0D 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 0D 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 0B 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 0B 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 01 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 01 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR $addr $hexlen;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR 01 16 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 :    nach EXPAND:SEND 04;WAIT 05;SEND 01;SEND F7 01 16 02;RECV 2 UT
[3868] Mon Mar 19 22:36:20 2012 : protocmd Zeile: SYNC;GETADDR;SEND BYTES;SEND $len;RECV $len $unit
[3868] Mon Mar 19 22:36:20 2012 :   Nach Ersetzung: SYNC;GETADDR;SEND BYTES;SEND 10;RECV 10 
[... 148 lines cut ...]
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 3 Hexlen:1, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 2 Hexlen:2, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 0 Hexlen:0, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : BuildByteCode:SEND 04;WAIT 05;SEND 01
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 3 Hexlen:1, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 1 Hexlen:1, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 3 Hexlen:1, Unit: 
[3868] Mon Mar 19 22:36:21 2012 : 		Token: 0 Hexlen:0, Unit: 
[2496] Mon Mar 19 22:36:22 2012 : TCP socket 1234 geoeffnet

```

Wenn ich dann versuche etwas auszulesen bekomme ich.

```

C:\vcontrold>vclient -h 127.0.0.1:1234 -c getTempATS
[1192] Mon Mar 19 22:37:36 2012 : Fehler gethostbyname: 127.0.0.1▒":No address a
ssociated with name

```

Wo soll ich ansetzen ?? 

lg, Andy

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: warmup <None>
* Date: Tue, 20 Mar 2012 06:26:12 UT
* Message-ID: <51894274-51908876$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

Guten Morgen,
ich zwar diese Anleitung verbrochen-den Fehler aber noch nicht gehabt!
Funktioniert denn der Test unter Abschitt 3?
Wenn ja - könnte es sein, dass die Vito-Kennung in der vcontrold.xml nicht korrekt ist? Zusätzlich muss in der vcontrold.xml der richtige USB/COM-Port eingetragen werden (aber bei der Zählweise immer eins abziehen, da)
Hast Du denn auf Deinen System auch das Netzwerk eingerichtet?

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: andyyyy1 <userid-29233444@openv.wikispaces>
* Date: Thu, 22 Mar 2012 16:14:44 UT
* Message-ID: <51894274-52018334$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

Hallo warmup,

Hat ein wenig gedauert, aber ich habe nun alle "tests" gemacht.

Ja ich kann mit Vitotest Werte abfragen.
Kennung ist korrekt und Com Port ist egal welchen ich eingebe, der Fehler bleibt.
Mit [code] cat /dev/ttyS2 [code] bekomme ich die Welle als zyklische Antwort.
Netzwerk ist da.

Hab es nun schon auf 2 Verschiedenen Win 7 32bit und auf einem XP pro in einer VM probiert.

kann es nicht sein das mir beim cygwin noch irgendwo etwas fehlt ??

lg, Andy

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: BrainHunter <userid-9234109@openv.wikispaces>
* Date: Thu, 22 Mar 2012 16:48:41 UT
* Message-ID: <51894274-52020276$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

versuch doch mal bitte telnet. du gibst dazu in der console ein:
telnet localhost 1234

es sollte sich dann der vcontrold melden. dort kannst du dann die befehle testen

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: andyyyy1 <userid-29233444@openv.wikispaces>
* Date: Thu, 22 Mar 2012 16:54:25 UT
* Message-ID: <51894274-52020596$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

da meldet sich leider keiner

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: warmup <userid-27729442@openv.wikispaces>
* Date: Thu, 22 Mar 2012 19:29:35 UT
* Message-ID: <51894274-52030508$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

Hört sich merkwürdig an.
Ist der Cygwin-Dienst gestartet?
Ist vcontrol gestartet?
Wenn der Telnet nicht geht, dann stimmt was ander Basis nicht. 
Das Start-Log vom vcontrold zeigt, dass der Port geöffnet ist. Somit solltest Du Dich verbinden können
Bei mir gab es ja auch immer Probleme mit dem Adapter. 
Geh mal in vcontrold-Verzeichnis und versuche nachstehenden Init

-----------------------------------------
rem beenden der vcontrold
taskkill /F /IM vcontrold.exe
net stop cygserver

rem initialisieren der Schnittstelle (ttyS1=COM2)
stty -F /dev/ttyS1 4800

rem starten von vcontrold
net start cygserver
vcontrold.exe
---------------------------------------
So sollte 
telnet 127.0.0.1 1234
zur Verbindung führen ....

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: BrainHunter <userid-9234109@openv.wikispaces>
* Date: Thu, 22 Mar 2012 20:55:10 UT
* Message-ID: <51894274-52034390$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

ist evtl. eine firewall aktiv? wenn ja, dann mal testweise deaktivieren!

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: BrainHunter <userid-9234109@openv.wikispaces>
* Date: Thu, 22 Mar 2012 20:55:43 UT
* Message-ID: <51894274-52034414$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

...oder eben richtig einstellen ;-)

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: andyyyy1 <userid-29233444@openv.wikispaces>
* Date: Fri, 23 Mar 2012 15:44:30 UT
* Message-ID: <51894274-52068224$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

Hab nun alles deinstalliert neu geladen und installiert. nun funzt es. Danke

WikiSpaxe, 2017-10-26:
* Subject: Re: Probleme bei der initialisierung
* Wiki-Page: vcontrold-lite
* From: bertl85 <userid-47424366@openv.wikispaces>
* Date: Sun, 16 Feb 2014 23:23:47 UT
* Message-ID: <51894274-69737566$openv.wikispaces>
* In-Reply-To: <51894274-51894274$openv.wikispaces>

Hallo Zusammen

Ich habe irgendwie ein ähnliches Problem:
server laufen
Telnet funktioniert
bei einerm Aufruf mit vclient erhalte ich 127.0.0.1 unknown host.

Woran könnte das denn liegen?

Nutze Windows XP

Mario275, 2019-11-21:
Hallo, bin neu hier. Beiträge sind ja schon etwas her aber ich verzweifle langsam, bin schon seit über 1 Woche am probieren und habe schon so ziemlich alles hier bei github was cygwin, vcontrold, vclient und irgendwas mit Viessmann Heizungen zu tun hat durchsucht und probiert. Ich habe cygwin erfolgreich installiert. Im Ordner c:\vcontrold die vcontrold, vclient und dazugehörige .xml Dateien installiert. IP 127.0.0.1 Port 3002, ttyS6 (Com7) und meine Viessmann Type 209F angelegt bzw. angepasst. 
Leider scheitert es bei mir schon an der Vcontrold.exe zu starten. Telnet Aufruf bringt nur Verbindungsfehler. Ich arbeite auf Win10, kann das schon das Problem sein. Der Befehl stty unter Cmd geht auch nicht? Cygwin Server läuft aber unter Dienste!
Bin über jegliche Hilfe dankbar.

Gruß Mario

[image]

GolterU, 2019-11-22:
Erscheinen die Fehler auch, wenn statt der Powershell-Kommandozeile die normale (Admin)-Eingabeaufforderung benutzt wird?

Mario275, 2019-11-23:
Ja, leider gehts auch nicht. Wenn vcontrold laufen würde auch mit falschen xml Einstellungen dann müsste doch mindestens eine vcontrold.log erzeugt werden? Liegt es an Win10 64bit Betriebssystem? Läuft cygwin und vcontrold auch innerhalb einer Virtuellen Machine, dann könnte ich Win XP installieren?

Gruss Mario